bdring/FluidNC #419: How easy is it to switch between wifi and bluetooth?

IgorPack, 2022-05-12:
FluidNC 3.2.8

Can anyone explain how in FluidNC you can switch from wifi mode to bluetooth mode and back using the commands [esp...]? If I understand correctly, for this you need to run files on the PC either install-bt.bat or install-wifi.bat. Then only one thing will work. As for me, this is not very inconvenient and requires the use of a PC in the workshop. In the GRBL_ESP32 version, this is much easier and all this can be done from a smartphone or tablet. Or I don't know the correct commands for FluidNC. 

If so, can the developers tell how this can be done (if possible) from a smartphone or tablet, or simplify this procedure and make the switch between wifi and bluetooth and vice versa the way it is in GRBL_ESP32? 

Thanks.

bdring, 2022-05-12:
The Wifi and Bluetooth libraries are quite large. FluidNC has to compile all of the machine feature into the standard compile files, so both BT and WiFi will not fit for the standard ESP32 devices. 

If you want the option to hot swap you must either use Grbl_ESP32 or buy a larger memory ESP32 and compile yourself.

You need to change some things in platformio.ini

set **default_envs = wifibt**

and choose the **board_build.partitions** for your larger ESP32

MitchBradley, 2022-05-12:
The wifi+bt version still fits in the standard ESP32 devices, but there is very little room to spare.  You can still build it for now.  We chose to stop supporting it on the standard ESP32s because very few people use it - you are the only one we know of.  We might revise that decision in the future.

MitchBradley, 2022-05-12:
Sorry that is incorrect. It only fits when compiled using an experimental branch that I am working on. It does not fit with the mainline code

IgorPack, 2022-05-13:
Guys, thanks for your answers.

I make cnc machines for my clients and use ESP32 modules. I don’t know, maybe a separate Wi-Fi or Bluetooth connection is not bad for individual use by someone, but for my clients it’s more convenient to be able to simply switch between Wi-Fi and Bluetooth without using a PC. In GRBL_ESP32 you just enter the commands [esp110]bt / ap and [esp115]on and everything works, and you can assign buttons to these commands in WebUi, GRBL Controller, Candle or others. Isn't that great? And what opportunities does this open up for others!

I really like FluidNC, but due to the lack of such a switch in it, I have to make a choice in favor of GRBL_ESP32, which, it seems to me, slightly limits the excellent possibilities of FluidNC. Unfortunately, I myself am not strong in programming, and if someone ever manages to solve this problem, then I think that this will be the best solution on the CNC machine market. And then happiness will come to our house))). Please let me know when this happens.

Good luck and health to all.

closed 2022-05-13 by bdring

IgorPack, 2022-05-17:
Mitch, I will definitely do this as soon as my startup is fully operational and circumstances permit. 
I have it in my plans.

IgorPack, 2022-06-18:
Hi Bart and Mitch,
I want to say, FluidNC is a cool thing, everything is clear and easily a wonderful thing. True, I couldn’t test all the possibilities, I tried all versions from 3.2.8 - 3.4.7, I used several Espduino + cnc shield, with your firmware - I constantly get a Wi-Fi connection failure or a bad connection or a disconnection. I want to try using bluetooth, but I didn’t understand how to load the 3axis_v 3*.yaml file into the controller, can you tell me how to do it correctly? Thanks in advance.

MitchBradley, 2022-06-18:
You can connect with FluidTerm over either USB serial or Bluetooth serial, then type CTRL-U.  That will bring up a file selector on your host system where you can select the file you want to upload.  Then, in FluidTerm, type the name that you want for the file on the local FluidNC filesystem, or enter to use the same name that the file has on the host.

MitchBradley, 2022-06-18:
CTRL-U upload via Bluetooth only works on the todays release 3.4.8.  It works on USB serial since a long time ago.

IgorPack, 2022-06-18:
Thanks for the quick response and information, you are amazing, v3.4.8. already downloaded, will try tomorrow. I regret that I can't make friends with WebUi, I really like it. By the way, have you thought about making the same application as WebUi for bluetooth, such as the GRBL Controller? It would be great and I think a lot of people would be interested.

MitchBradley, 2022-06-18:
We did not make the WebUI application; it is a separate project by a separate author.  We do not have any spare time to make, and especially to support, a new app.  We barely have time to continue to develop and support FluidNC.

A lot of people are "interested" in taking advantage of free stuff.  Fewer people are willing to pay for it.

IgorPack, 2022-06-20:
Regarding v3.4.8. - when I try to load *.yaml, I get this window. In previous versions, this is not, but there are other problems.

[image]

1. In all versions I see such messages in the third line, I don’t know if this is normal, I don’t understand this well. I noticed the difference in the firmware in the second line.

[image]

[image]

2. When you try to simultaneously move the X and Y axes, the controllers (ESPduino by KOM3, ESP32 - 38PIN by Silicon Labs ...) freeze, the network adapter on the laptop turns off and you need to reconnect the general Wi-Fi connection. And that's without the connected motors.

[image]

I don't know, maybe I'm setting something up wrong?

MitchBradley, 2022-06-20:
> I can repro the 406 wrong data problem. I am working on it.
The escape sequences in the new configurable GRBL startup message are confusing the JSON parser in WebUI.  It is unhappy with seeing "\V" in the JSON encoding of the settings values.  Syntax always bites you somewhere.

IgorPack, 2022-06-20:
With the config.file, I do so, create a new file of type 3axes_v*.yaml with my parameters, then change the config.file in WebUi on my. I just wanted to understand what the message in the third line means. Thanks for the clarification.

The question remains unclear what to do with the board hanging while simultaneously turning on the X and Y axes.

MitchBradley, 2022-06-21:
PR #501 fixes the WebUI breakage in 3.4.8

MitchBradley, 2022-06-21:
> The question remains unclear what to do with the board hanging while simultaneously turning on the X and Y axes.

File a separate issue with all of the information about that problem in one place.  This issue has become too complicated because it addresses more than one topic.

bdring, 2022-06-21:
The message will go away when you upload a file and change to **$Config/Filename=\<your filename\>**

IgorPack, 2022-06-21:
I thought so, just wanted to make sure. And yes, I agree about creating a separate issue about the hanging of the board.