bdring/FluidNC #421: Soft reset boot error with TMC2209

TinkerBoi, 2022-05-14:
Hello!

I'm running an ESPDuino32 board with the CNC Shield V3. I swapped out the DVR8825 for TMC2209 drive, everything seems to run fine except when I hit the soft reset on the board I get a boot error. This only happened if I press the onboard reset (my other option is to cut power, unplug usb and reconnect which isn't always practical).

rst:0x10 (rtcwdt_rtc_reset),oot:0x22 (download_boot(uart0/art1/sdio_rei_feo_v2)) waiting for download

There are two things I'm not certain about - 

1. Are my TX/RX pins configured correctly for this setup? 
2. I think it's running in legacy mode as changing the steps in the config file didn't make any difference (but jumpers on the board did). 

If I leave the UART pins (NC, TX, RX) unconnected (for direct replacement of 8825), it throw this error on pressing the reset:

rst:0x10 (RTCWDT_RTC_RESET),boot:0x32 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371

X Motor Setting (I don't have any other axises declared as Im bench testing only with this motor)

axes:
  shared_stepper_disable_pin: gpio.12
  x:
    steps_per_mm: 160.000
    max_rate_mm_per_min: 10000.000
    acceleration_mm_per_sec2: 1000.000
    max_travel_mm: 600
    soft_limits: true
    homing:
      cycle: 1
      mpos_mm: 0
      pulloff_mm: 3
      allow_single_axis: true
      positive_direction: false
      settle_ms: 250.000
      seek_mm_per_min: 2000.000
      feed_mm_per_min: 1000.000
      seek_scaler: 1.1
      feed_scaler: 5.0
    motor0:
      limit_all_pin: gpio.13:high
      hard_limits: true    
      tmc_2209:
        uart:
          txd_pin: gpio.22
          rxd_pin: gpio.21
          baud: 115200
          mode: 8N1
        addr: 0
        r_sense_ohms: 0.110
        run_amps: 0.500
        hold_amps: 0.250
        microsteps: 32
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: StealthChop
        homing_mode: StealthChop
        use_enable: true
        step_pin: gpio.26
        direction_pin: gpio.16
        disable_pin: NO_PIN
    motor1:
      null_motor:

MitchBradley, 2022-05-14:
Please supply complete config files, not excerpts of the parts that you think are implicated, and also boot messages.  Boot errors are often caused by signal levels on certain gpios. Unless we can see everything, we cannot help you find the problem ones.

TinkerBoi, 2022-05-14:
> Please supply complete config files, not excerpts of the parts that you think are implicated, and also boot messages. Boot errors are often caused by signal levels on certain gpios. Unless we can see everything, we cannot help you find the problem ones.

Sorry about that Mitch! I update the original post with the full config file and more boot message details.

MitchBradley, 2022-05-14:
Sometimes GPIO 12 causes boot problems.  Perhaps the new stepsticks affect the signal level on that line during reset.

MitchBradley, 2022-05-14:
Did you remove the pullup resistor as described in the issue that you linked to?

MitchBradley, 2022-05-14:
I don't think it is possible to use TMC2209 in UART mode with a Protoneer CNC Shield V3.  According to [these schematics](https://blog.protoneer.co.nz/wp-content/uploads/2013/07/Arduino-CNC-Shield-Schematics.png), the UART pin which is on stepstick pins 13 and 14 not connected back to the Arduino shield pins, so the ESP32 would not be able to drive them.  Therefore, it is probably better to configure them as "stepstick" instead of TMC2209.

The boot problem could be caused by pulldown resistors on the stepstick modules or in the TMC2209 chips themselves.  The TMC2209 datasheet vaguely suggests that the chips might have internal pulldowns on the ENN pin, but does not call it out clearly.  There are two ways to fix that:

1. You could clip the enable pins on the stepsticks, thus disconnecting them from GPIO 12.  The motors would be always enabled then - although you could add a switch to pull up those pins to disable them manually.
2. It is possible to set an efuse in the ESP32 chip to make it ignore the state of GPIO 12.  There is some discussion of that somewhere, but I don't remember where it is.  We do not support that because changing efuses can brick your ESP32 unless you know exactly what you are doing.

TinkerBoi, 2022-05-14:
> Did you remove the pullup resistor as described in the issue that you linked to?

Yes I did! Everything worked great except for the soft reset.

> I don't think it is possible to use TMC2209 in UART mode with a Protoneer CNC Shield V3. According to [these [...]

Thank you so much for the suggestions! I'm fine to run it in stepstick  mode, I wasn't sure if my config was correct?

Let me try with the EN disconnected, I'll report back.

bdring, 2022-05-14:
We have that controller listed under the [Do Not Use](https://github.com/bdring/FluidNC/wiki/Hardware-that-Runs-FluidNC#do-not-use) section of the wiki.

Yes, some people have managed to make it work, but please don't expect us to answer support questions on literally the cheapest CNC controller in the world.

TinkerBoi, 2022-05-14:
> We have that controller listed under the [Do Not Use](https://github.com/bdring/FluidNC/wiki/Hardware-that-Runs-FluidNC#do-not-use) section of the wiki.
>
> Yes, some people have managed to make it [...]

I apologize as I didn't mean to come off wrong or have any expectation to get any help for this issue, and this issue I could easily live with by power cycle option as I mentioned above. I know its literally the cheapest but its the only option I have on hand to work with. I was actually looking to buy your V4 controller with 3 axis and I kept checking back but they're sold out so im left with this option. If you have any of those board on hand (even used or prototype), I'd love to get one to replace this problematic setup.

Appreciated Mitch and your help very much!

closed 2022-05-16 by TinkerBoi