bdring/FluidNC #424: SD not stable or initialized fail while User Outputs defined

gonebirds, 2022-05-15:
Please only submit bugs for the **latest** version of the code. You can check the version number in the startup messages.

**Do not add the bug label**. The developers of this repo will do that after confirming the issue.

Please provide the [FluidNC boot messages per this page](https://github.com/bdring/FluidNC/wiki/Requesting-Help#fluidnc-boot-messages).

Please answer the following questions.

What version of the firmware are you using?
3.4.4
Is the problem repeatable?
yes
Under what conditions does the bug occur?
1. make an yaml file to enable SD card and check SD functionality in OK status.
2. modified the yaml file to enable user outputs for gpio: 4, 25, 26, 27 (with no confilict to other assignment)
**Important** If you paste firmware code, please use [Markdown Code and Syntax Highlighting](https://github.com/adam-p/markdown-here/wiki/Markdown-Cheatsheet#code) with language C++. Use the three back tick method.

```fluidterm
Grbl 3.4 [FluidNC v3.4.4 (wifi) '$' for help]
$SD/List
No SD Card
error:60
$SD/List

[FILE:/test.txt|SIZE:229]
[FILE:/boot.log|SIZE:3246]
[FILE:/8ADC.LOG|SIZE:6066960]
[FILE:/ADC_Values_20210829.log|SIZE:119595]
[FILE:/YG8838v03.nc|SIZE:24528]
[FILE:/Date1.log|SIZE:216000]
[SD Free:950.85 MB Used:6.16 MB Total:957.02 MB]
ok
$SD/List
No SD Card
error:60
$SD/List
No SD Card
error:60
$SD/List
No SD Card
error:60
$SD/List
```

bdring, 2022-05-15:
What hardware are you using. Is it one you purchased or made yourself. If you made it yourself, can you send a picture of it?

gonebirds, 2022-05-17:
I made one with modifications based on 6pac v1.8. after diagnostics i've found the problems are in hardware. now a workaround is applied, and this issue is resolved and canbe closed. Thanks.

closed 2022-05-17 by gonebirds