bdring/FluidNC #429: Can not compile in platformio

be-engineer, 2022-05-16:
Please only submit bugs for the **latest** version of the code. You can check the version number in the startup messages.

**Do not add the bug label**. The developers of this repo will do that after confirming the issue.

Please provide the [FluidNC boot messages per this page](https://github.com/bdring/FluidNC/wiki/Requesting-Help#fluidnc-boot-messages).

Please answer the following questions.

What version of the firmware are you using?
V3.4.4
Is the problem repeatable?
When I compile in vscode with platformio,I got some error messages:
FluidNC/test/Pins/PinOptionParsing.cpp: In static member function 'static void Pins::PinOptionParsing_NoArgs_Test::runDetail()':
```
FluidNC/test/Pins/PinOptionParsing.cpp:19:19: warning: variable 'it' set but not used [-Wunused-but-set-variable]
         for (auto it : parser) {
                   ^
FluidNC/test/Pins/GPIO.cpp: In constructor 'Pins::GPIOISR::GPIOISR(int, int, int)':      
FluidNC/test/Pins/GPIO.cpp:304:43: error: expected primary-expression before ',' token   
             gpio16.attachInterrupt<GPIOISR, &GPIOISR::HandleISR>(this, mode);
                                           ^
FluidNC/test/Pins/GPIO.cpp:304:72: warning: left operand of comma operator has no effect 
[-Wunused-value]
             gpio16.attachInterrupt<GPIOISR, &GPIOISR::HandleISR>(this, mode);
                                                                        ^
FluidNC/test/Pins/GPIO.cpp:304:76: error: invalid operands of types 'void (Pins::GPIOISR::*)()' and 'int' to binary 'operator>'
             gpio16.attachInterrupt<GPIOISR, &GPIOISR::HandleISR>(this, mode);
                                                                            ^
Compiling .pio\build\wifi\lib882\TMCStepper\source\CHOPCONF.cpp.o
*** [.pio\build\wifi\src\test\Pins\GPIO.cpp.o] Error 1
============================== [FAILED] Took 24.99 seconds ==============================
Environment    Status    Duration
-------------  --------  ------------
wifi           FAILED    00:00:24.986
========================= 1 failed, 0 succeeded in 00:00:24.986 =========================
```
I just clone project from github, without any modification. What can I do?

Under what conditions does the bug occur?

**Important** If you paste firmware code, please use [Markdown Code and Syntax Highlighting](https://github.com/adam-p/markdown-here/wiki/Markdown-Cheatsheet#code) with language C++. Use the three back tick method.

```C++
  #define EASIER_TO_READ true
```

MitchBradley, 2022-05-16:
We need a lot more information, such as the way that you started the compilation, the version of platformio, and the complete output from the compilation.  Just showing the error messages does not give us the context that we need to understand what is different about your setup and ours.

bdring, 2022-05-16:
Show all the text from when you try to compile. The text at the beginning of the compile is very important to make sure are starting it correctly.

Be sure to answer the questions from MitchBradley as well as mine.

MitchBradley, 2022-05-16:
The problem seems to be caused by the new Platformio version 6.

be-engineer, 2022-05-22:
I changed a computer today and recompiled, and the compilation passed. A little weird

MitchBradley, 2022-05-22:
The problem is fixed by PR #431

closed 2022-05-22 by MitchBradley