bdring/FluidNC #436: Problem: SD Card Not Mounting After v3.4.5 Install

Lewery, 2022-05-20:
### Controller Board

6-pack

### Help From Board Vendor

- [ ] Yes
- [ ] No
- [X] Not Applicable

### Machine Description

'ESP-WROOM-32' (fluidterm: "chip is ESP32-D0WDQ6 (revision 1)", with 4MB flash), in a 6-Pack_CNC_Controller, 24V 10A Aclorol power supply, X & Y controlled by 'DRV8825' (presently no Z, maybe a servo later), running 'FluidNC v3.4.5 (wifi)'

### Configuration file

```yaml
board: 6 Pack
name: 6 Pack StepStick XY
meta: 
stepping:
  engine: I2S_stream
  idle_ms: 250
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0
  segments: 6

axes:
  shared_stepper_disable_pin: NO_PIN
  shared_stepper_reset_pin: NO_PIN
  x:
    steps_per_mm: 200.300
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 50.000
    max_travel_mm: 1300.000
    soft_limits: true
    homing:
      cycle: 2
      allow_single_axis: true
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 2500.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.16:pu
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 2.000
      stepstick:
        step_pin: I2SO.2
        direction_pin: I2SO.1
        disable_pin: I2SO.0
        ms1_pin: NO_PIN
        ms2_pin: NO_PIN
        ms3_pin: I2SO.3
        reset_pin: NO_PIN

  y:
    steps_per_mm: 200.050
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 50.000
    max_travel_mm: 800.000
    soft_limits: true
    homing:
      cycle: 1
      allow_single_axis: true
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 2500.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.26:pu
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 2.000
      stepstick:
        step_pin: I2SO.5
        direction_pin: I2SO.4
        disable_pin: I2SO.7
        ms1_pin: NO_PIN
        ms2_pin: NO_PIN
        ms3_pin: I2SO.6
        reset_pin: NO_PIN

    motor1:
      limit_neg_pin: gpio.27:pu
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 2.000
      stepstick:
        step_pin: I2SO.10
        direction_pin: I2SO.9
        disable_pin: I2SO.8
        ms1_pin: NO_PIN
        ms2_pin: NO_PIN
        ms3_pin: I2SO.11
        reset_pin: NO_PIN

  z:
    steps_per_mm: 100.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 5.000
    soft_limits: false
    homing:
      cycle: 0
      allow_single_axis: true
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 50.000
      seek_mm_per_min: 200.000
      settle_ms: 250
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 2.000
      rc_servo:
        output_pin: gpio.12
        pwm_hz: 50
        min_pulse_us: 1000
        max_pulse_us: 2000

kinematics:
  Cartesian:

i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5
  card_detect_pin: NO_PIN

control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

coolant:
  flood_pin: NO_PIN
  mist_pin: NO_PIN
  delay_ms: 0

probe:
  pin: NO_PIN
  check_mode_start: true

macros:
  startup_line0: 
  startup_line1: 
  macro0: 
  macro1: 
  macro2: 
  macro3: 

start:
  must_home: true
  deactivate_parking: false
  check_limits: false

user_outputs:
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 5000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000
  digital0_pin: NO_PIN
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: NO_PIN

arc_tolerance_mm: 0.002
junction_deviation_mm: 0.010
verbose_errors: true
report_inches: false
enable_parking_override_control: false
use_line_numbers: false
planner_blocks: 16
NoSpindle:
```

### Startup Messages

```text
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:12804
ho 0 tail 12 room 4
load:0x40080400,len:3032
entry 0x400805e4

[MSG:INFO: FluidNC v3.4.5]
[MSG:INFO: Compiled with ESP32 SDK:v4.4-beta1-189-ga79dc75f0a]
[MSG:INFO: Local filesystem type is SPIFFS]
[MSG:INFO: Configuration file:220519_1325.yaml]
[MSG:INFO: Machine 6 Pack StepStick XY]
[MSG:INFO: Board 6 Pack]
[MSG:INFO: I2SO BCK:gpio.22 WS:gpio.17 DATA:gpio.21]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN]
[MSG:INFO: Stepping:I2S_stream Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:250ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (0.000,1300.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.2 Dir:I2SO.1 Disable:I2SO.0]
[MSG:INFO:     Neg Limit gpio.16:pu]
[MSG:INFO: Axis Y (0.000,800.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.5 Dir:I2SO.4 Disable:I2SO.7]
[MSG:INFO:     Neg Limit gpio.26:pu]
[MSG:INFO:   Motor1]
[MSG:INFO:     stepstick Step:I2SO.10 Dir:I2SO.9 Disable:I2SO.8]
[MSG:INFO:     Neg Limit gpio.27:pu]
[MSG:INFO: Axis Z (-5.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     rc_servo Pin:gpio.12 Pulse Len(1000,2000)]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Using spindle NoSpindle]
[MSG:INFO: Connecting to STA SSID:maison]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 192.168.0.44]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl 3.4 [FluidNC v3.4.5 (wifi) '$' for help]
[MSG:INFO: '$H'|'$X' to unlock]
```

### User Interface Software

WebUI and Fluidterm

### What happened?

I updated from v3.4.4 to v3.4.5, no other changes to hardware or the config file were made, I ejected and reinserted the card when I realized it was missing. I restarted the controller many times and turned everything off for about an hour and had the same result after.

SD card has been working without issue, I expected the SD card to continue working.

WebUI reports "No SD card", here is output from Fluidterm:
```
--- Available ports:
---  1: /dev/cu.Bluetooth-Incoming-Port 'n/a'
---  2: /dev/cu.SLAB_USBtoUART 'CP2102 USB to UART Bridge Controller'
--- Enter port index or full name: 2
--- Fluidterm on /dev/cu.SLAB_USBtoUART  115200,8,N,1 ---
--- Quit: Ctrl+] or Ctrl+Q | Upload: Ctrl+U | Reset: Ctrl+R ---

ok
$x
[MSG:INFO: Caution: Unlocked]
ok
<Idle|MPos:0.000,0.000,0.000|FS:0,0|WCO:951.000,552.000,0.000>
$sd/list
No SD Card
error:Failed to mount device

--- exit ---
```

Lewery, 2022-05-20:
The card that was showing up, but is no longer is a SANDISK 4GB Micro SDHC Memory Card SDSDQM-004G

MitchBradley, 2022-05-20:
This might be related https://github.com/espressif/arduino-esp32/issues/6189

MitchBradley, 2022-05-20:
Here is the commit that fixes the problem.  That commit is in the master branch of the arduinoespressif32 framework, but not yet in the released version. https://github.com/espressif/arduino-esp32/commit/bb7df04446679c3c52015abaea133319e2de4178

Lewery, 2022-05-20:
The problem has been solved by fluidnc-SDFix-test-posix.

MitchBradley, 2022-05-20:
Okay, I deleted the test release and we will proceed to a proper release with the new framework version.

PR #443 (2022-05-21): Use platform espressif32@^4.3.0
Fixes #436
Thanks to @jiffyjeff
commit 50dd6b68 (2022-05-21): Use platform espressif32@^4.3.0 (#443)

Fixes #436
Thanks to @jiffyjeff