bdring/FluidNC #442: Feature: Quadrature output signal for servo motors

kakka666, 2022-05-21:
### Machine Context

I am building a setup using servo motors. These take up Quadrature signal as their signal.
I was thinking if it can produce the step direction signal, so it can also produce a quadrature signal.
This will add the possibility of having industrial servos for running axis.

### Feature Description

the quadrature configuration is of two type
**first** is where we require A-phase B-phase and we have to generate a two outputs. 
so lets assume we have a two binary digit number XY that can be initiated to 00 in beginning and lets assume that we need to add one pulse for positive axis direction and one for negative axis direction. 
these represent the states of the two outputs defined for this axis.
Now the addition rules will be as follows(addition means adding positive pulse)
00 + 1 = 01
01 + 1 = 11
11 + 1 = 10
10 + 1 = 00

similarly the subtraction rules will be as follows(subtraction means adding negative pulse)
00 - 1 = 10
10 - 1 = 11
11 - 1 = 01
01 - 1 = 00

so this can generate a A-Phase and B-Phase quadrature output

Second  type is A-phase, (-)A phase, B-phase, (-)B phase.
This is very similar to the earlier explained method. the difference is that it has 4 outputs. Two are same as A & B, other are A and B through not gate creating a differential signal.
[image]
[image]
 

### Other Approaches

I tried using step and direction in servo drives. these are available in some drives.
in some drives this feature is there. But they ask for signal in 24Vdc. i tried using an optocoupler to generate signal from 3.3v to 24v.
I could get the direction signal easily as when it changes, the state remains fixed but for pulse, i couldnt get as a 4 us value was not long enough to change the state and generate a signal. on other values(>4us) the pulses were not generated at all(i done know why)   

Other than this if i use a stepper or step-servo, The motors if not disabled will continuously draw current and keep on getting warmer. Whereas servos this is not the case, they will do error correction and will consume a minimal current.

### How I Can Help

I am a noob in programming, but i am having a mechanical setup to test this. Also i have some spare servos of different brands like mitsubishi/Siemens AB etc. i can check how these work on each and give a feedback
I can do basic arduino programming and have ran these motors with simple programs easily. so i can help some in developing the logic behind

MitchBradley, 2022-05-21:
Let's approach this another way.  In principle, the optocoupler techniques that you tried should have worked, so there is probably some simple change to get the rest of the way.

To help you, we will need the following:

* Details of the motor (or one of the motors) that you tried with optocouplers.  In particular we need the electrical specifications - voltages and timings - for the step/direction input.  The best way to provide that data is with a link to a datasheet.
* The config file that you used in trying to change the timing.  Attach it to a comment with the file extension changed to ".txt"
* The startup messages from FluidNC
* A schematic of the optocoupler circuit that you used to interface between the FluidNC system and the motor
* Details of the ESP32 controller hardware that you use.

Basically, that list above is the same stuff that you must enter in order to file a "Problem" ticket, so one way to provide it would be to open such a ticket and fill in the form, being sure that all the info above is provided somewhere.

kakka666, 2022-05-21:
> Let's approach this another way. In principle, the optocoupler techniques that you tried should have worked, so there is probably some simple change to get the rest of the way.
>
> To help you, we will [...]

Hello,
Initially i was using steppers with feedbacks. I got them working by directly connecting the output pins to  the terminals of the motor drives. The issue here is not been able to run on step and direction. 
I have recently bought used milling machine and it has power feed fed using servos and they are using jog wheels to jog or move the bed manually. These jog wheels are quadrature encoders.
i am a big fan of fluidnc and was hoping if i could use fluid nc and generate a same signal as these encoder, then this would be a big help to me.

MitchBradley, 2022-05-21:
We have an existing "Unipolar" motor type that sort of does what you want - but nobody uses it as far as I know.  That creates a support problem for us because whenever we want to make an improvement in that part of the system, we must ensure that the change does not break the Unipolar code - and due to the lack of interest, we do not keep a test rig for it readily at hand, so the existence of that code is just a nuisance.  One-user code like that creates a similar problem.  We can't test it ourselves, and it hampers our ability to make progress that benefits multiple users.

That is why I am more interested in solving the problem another way, as opposed to adding code for which there is not a lot of existing demand.

closed 2022-06-02 by MitchBradley