bdring/FluidNC #445: Problem: Multiple rapid movements after soft reset or job cancel

rysn351, 2022-05-22:
### Controller Board

6-pack

### Help From Board Vendor

- [ ] Yes
- [ ] No
- [ ] Not Applicable

### Machine Description

Sherline lathe.

### Configuration file

```yaml
board: 6 Pack
name: 6 Pack StepStick XYZ

stepping:
  engine: I2S_STREAM
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN
  
  x:
    steps_per_mm: 5039.370
    max_rate_mm_per_min: 800.000
    acceleration_mm_per_sec2: 60.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.33
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm:1.000
      stepstick:
        ms3_pin: i2so.19
        step_pin: I2SO.18
        direction_pin: I2SO.17
        disable_pin: I2SO.16

  y:
    steps_per_mm: 5039.370
    max_rate_mm_per_min: 800.000
    acceleration_mm_per_sec2: 60.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.32
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm:1.000
      stepstick:
        ms3_pin: i2so.3
        step_pin: I2SO.2
        direction_pin: I2SO.1
        disable_pin: I2SO.0
        
  z:
    steps_per_mm: 5039.370
    max_rate_mm_per_min: 800.000
    acceleration_mm_per_sec2: 60.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100
        
    motor0:
      limit_neg_pin: gpio.35
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm:1.000
      stepstick:
        ms3_pin: i2so.22
        step_pin: I2SO.21
        direction_pin: I2SO.20
        disable_pin: I2SO.23
        

i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  card_detect_pin: NO_PIN
  cs_pin: gpio.5
  
user_outputs:
  digital0_pin: I2SO.15:low
  digital1_pin: I2SO.8:low
  digital2_pin: I2SO.7:low
```

### Startup Messages

```text
[MSG:INFO: FluidNC v3.4.1]
[MSG:INFO: Compiled with ESP32 SDK:v3.3.5-1-g85c43024c]
[MSG:INFO: Configuration file:RS_L7.yaml]
[MSG:INFO: Machine 6 Pack StepStick XYZ]
[MSG:INFO: Board 6 Pack]
[MSG:INFO: I2SO BCK:gpio.22 WS:gpio.17 DATA:gpio.21]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 dectect:NO_PIN]
[MSG:INFO: Stepping:I2S_stream Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:255ms]
[MSG:INFO: User Digital Output:0 on Pin:I2SO.15:low]
[MSG:INFO: User Digital Output:1 on Pin:I2SO.8:low]
[MSG:INFO: User Digital Output:2 on Pin:I2SO.7:low]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (-300.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.18 Dir:I2SO.17 Disable:I2SO.16]
[MSG:INFO:     Neg Limit gpio.33]
[MSG:INFO: Axis Y (-300.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.2 Dir:I2SO.1 Disable:I2SO.0]
[MSG:INFO:     Neg Limit gpio.32]
[MSG:INFO: Axis Z (-300.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.21 Dir:I2SO.20 Disable:I2SO.23]
[MSG:INFO:     Neg Limit gpio.35]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Using spindle NoSpindle]
[MSG:INFO: Connecting to STA SSID:CenturyLink4006]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 192.168.0.99]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
ok
```

### User Interface Software

Universal Gcode Sender, bCNC

### What happened?

In the middle of a job, if I cancel the job my lathe makes many additional movements at maximum movement speed. This can cause the tool to crash into the part. The movements seem to be the next moves the lathe would have made, but the speed as defined in the gcode is not preserved. I have observed this behavior using both UGS and bCNC. I can also recreate the behavior by pausing the job and clicking soft-reset in UGS. The console output from UGS is "[Error] Error while processing response <ok>: An unexpected command was completed by the controller.". I believe this error message prints once per additional movement after the stop.

Screen capture of behavior:
[image]

Gcode:
[n10.txt](https://github.com/bdring/FluidNC/files/8749944/n10.txt)

### Other Information

I am using Universal Gcode Sender version 2.0.11. All default settings (tested on a fresh install).

Console output in UGS showing error messages after pressing stop:

>>> F300.00X-0.951
Skipping comment-only line: (Move fast Z axis)
ok
>>> F300.00Z0.000
Skipping comment-only line: (Plunge in fast)

**** Pausing file transfer. ****

**** Canceling file transfer. ****

ok
Grbl 3.4 [FluidNC v3.4.1 (wifi) '$' for help]
>>> $$
ok
>>> $G
ok
[Error] Error while processing response <ok>: An unexpected command was completed by the controller.
[Error] Error while processing response <ok>: An unexpected command was completed by the controller.
[Error] Error while processing response <ok>: An unexpected command was completed by the controller.
$10 = 1    (Status report options, mask)
[Error] Error while processing response <ok>: An unexpected command was completed by the controller.
[GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F300 S0]
[Error] Error while processing response <ok>: An unexpected command was completed by the controller.

Exception report from UGS:

com.willwinder.universalgcodesender.AbstractController$UnexpectedCommand: An unexpected command was completed by the controller.
	at com.willwinder.universalgcodesender.AbstractController.commandComplete(AbstractController.java:801)
[catch] at com.willwinder.universalgcodesender.GrblController.rawResponseHandler(GrblController.java:151)
	at com.willwinder.universalgcodesender.AbstractController.rawResponseListener(AbstractController.java:822)
	at com.willwinder.universalgcodesender.AbstractCommunicator.sendEventToListeners(AbstractCommunicator.java:174)
	at com.willwinder.universalgcodesender.AbstractCommunicator.lambda$new$0(AbstractCommunicator.java:196)
	at java.base/java.lang.Thread.run(Unknown Source)

MitchBradley, 2022-05-22:
Try switching to I2S_STATIC and see if that makes a difference.

rysn351, 2022-05-22:
Thanks, but same behavior when switching to I2S_STATIC.

commit 46f5e553 (2022-05-23): Fix #445 by add a flushRx() method to the Channel class

The Uart class, derived from Channel, has a queue to
capture data bytes while looking for a realtime characters.
That queue was holding data that had been sent ahead.
commit 8fdb9c76 (2022-05-23): Merge pull request #446 from bdring/Fix445

Fix #445 by add a flushRx() method to the Channel class
closed 2022-05-23 by bdring