bdring/FluidNC #453: Problem:

jongough, 2022-05-28:
### Controller Board

6-pack

### Help From Board Vendor

- [ ] Yes
- [ ] No
- [ ] Not Applicable

### Machine Description

Flatbed router with DM556 drivers, M091-FD09 [https://www.datasheets.com/en/part-details/m091-fd09-superior-electric-79098766](https://www.datasheets.com/en/part-details/m091-fd09-superior-electric-79098766)

### Configuration file

```yaml
name: 6 Pack External XYZ with relay
board: 6-pack
meta: b44

kinematics:
  Cartesian:

stepping:
  engine: I2S_STREAM
  idle_ms: 255
  dir_delay_us: 5
  pulse_us: 4
  disable_delay_us: 4

axes:
  x:
    steps_per_mm: 98.175
    max_rate_mm_per_min: 20000.000
    acceleration_mm_per_sec2: 400.000
    max_travel_mm: 2580.000
    soft_limits: false
    homing:
      cycle: 2
      allow_single_axis: true
      mpos_mm: 0.000
      positive_direction: false
      settle_ms: 250.000
      seek_mm_per_min: 5000.000
      feed_mm_per_min: 200.000
      seek_scaler: 1.100
      feed_scaler: 1.100
    motor0:
      limit_neg_pin: gpio.33:high
      limit_pos_pin: gpio.32:high
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 5.000
      standard_stepper:
        direction_pin: i2so.1:high
        step_pin: i2so.2:low
        disable_pin: i2so.0:low
  y:
    steps_per_mm: 98.195
    max_rate_mm_per_min: 20000.000
    acceleration_mm_per_sec2: 400.000
    max_travel_mm: 1270.000
    soft_limits: false
    homing:
      cycle: 2
      allow_single_axis: true
      mpos_mm: 0.000
      positive_direction: false
      settle_ms: 250.000
      seek_mm_per_min: 5000.000
      feed_mm_per_min: 200.000
      seek_scaler: 1.100
      feed_scaler: 1.100
    motor0:
      limit_neg_pin: gpio.35:high
      limit_pos_pin: gpio.34:high
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 5.000
      standard_stepper:
        direction_pin: i2so.4:high
        step_pin: i2so.5:low
        disable_pin: i2so.7:low
  z:
    steps_per_mm: 399.000
    max_rate_mm_per_min: 1000.000
    acceleration_mm_per_sec2: 200.000
    max_travel_mm: 70.000
    soft_limits: false
    homing:
      cycle: 1
      allow_single_axis: true
      mpos_mm: 0.0
      positive_direction: true
      settle_ms: 250.000
      seek_mm_per_min: 1000.000
      feed_mm_per_min: 200.000
      seek_scaler: 1.100
      feed_scaler: 1.1000
    motor0:
      limit_pos_pin: gpio.2:high
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 5.000
      standard_stepper:
        direction_pin: i2so.9:low
        step_pin: i2so.10:low
        disable_pin: i2so.8:low

i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5

control:
  feed_hold_pin: gpio.36
  cycle_start_pin: gpio.39

coolant:
  mist_pin: gpio.14
  delay_ms: 1000.000

probe:
  pin: gpio.25:high
  check_mode_start: true

macros:
  startup_line0: 
  startup_line1: 
  macro0: 
  macro1: 
  macro2: 
  macro3: 

start:
  must_home: true
  check_limits: true
  deactivate_parking: false

user_outputs:

relay:
  tool_num: 0
  speed_map: 0=0.0% 18000=100.0%
  spinup_ms: 2000
  spindown_ms: 10000
  output_pin: gpio.26
  disable_with_s0: false
  s0_with_disable: false
  spinup_ms: 0
  spindown_ms: 0
arc_tolerance_mm: 0.002
junction_deviation_mm: 0.010
verbose_errors: true
report_inches: false
enable_parking_override_control: false
use_line_numbers: true
```

### Startup Messages

```text
[MSG:INFO: FluidNC v3.4.4]
[MSG:INFO: Compiled with ESP32 SDK:v3.3.5-1-g85c43024c]
[MSG:INFO: Configuration file:config.yaml]
[MSG:INFO: Machine 6 Pack External XYZ with relay]
[MSG:INFO: Board 6-pack]
[MSG:INFO: I2SO BCK:gpio.22 WS:gpio.17 DATA:gpio.21]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN]
[MSG:INFO: Stepping:I2S_stream Pulse:4us Dsbl Delay:4us Dir Delay:5us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (0.000,2580.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.2:low Dir:I2SO.1 Disable:I2SO.0:low]
[MSG:INFO:     Neg Limit gpio.33]
[MSG:INFO:     Pos Limit gpio.32]
[MSG:INFO: Axis Y (0.000,1270.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.5:low Dir:I2SO.4 Disable:I2SO.7:low]
[MSG:INFO:     Neg Limit gpio.35]
[MSG:INFO:     Pos Limit gpio.34]
[MSG:INFO: Axis Z (-70.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.10:low Dir:I2SO.9:low Disable:I2SO.8:low]
[MSG:INFO:     Pos Limit gpio.2]
[MSG:INFO: FeedHold gpio.36]
[MSG:INFO: CycleStart gpio.39]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Relay Spindle Ena:NO_PIN Out:gpio.26 Dir:NO_PIN]
[MSG:INFO: Using spindle Relay]
[MSG:INFO: Mist coolant gpio.14]
[MSG:INFO: Probe Pin: gpio.25]
[MSG:INFO: Connecting to STA SSID:BA WIFI]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 192.168.1.2]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
```

### User Interface Software

WebUI, Fluidterm

### What happened?

Using FluidNC 3.4.4 all is OK. Installed 3.4.6 and the stepper motors start missing steps, no error messages but a grinding noise from the motors. Reinstall 3.4.4 and it works as expected. Tried slowing down the movement rate (using a pen in the router rather than a bit AND did the process above the job). With 3.4.6 (and 3.4.5) the issue showed up. Tried just doing a 1.5mx1.2m rectangle plus the diagonals as an air cut and the issue occurred. With 3.4.4 there is no issue with the steppers or locations. Tried putting resistance on the Y-axis beam as the job moved on the X-axis and it would pull me over, the same with the Y-Axis.

The failures seemed to occur irrespective of the complexity of the g-code. i.e. complex curves with lots of g-code did not affect the rate of failures. The speed of the movement seemed to increase the number of errors when they occurred. 

On slower jobs, you can hear the stepper motors stutter (very slight hesitation) and if watching the X-axis rails you can just see the slight hesitation. On faster jobs, the whole axis locks up.
[pattern - example.g.txt.zip](https://github.com/bdring/FluidNC/files/8790463/pattern.-.example.g.txt.zip)
 

### Other Information

Testing the new version of FluidNC on 6-Pack_CNC_Controller which was using Grbl-esp32 previously. Changeover was made to make it easier to maintain by owner due to text file configurations.

MitchBradley, 2022-06-01:
Try adding "segments: 20" to the stepping: section, like this:

```
stepping:
  engine: I2S_STREAM
  idle_ms: 255
  dir_delay_us: 5
  pulse_us: 4
  disable_delay_us: 4
  segments: 20
```
 I am not sure if that is the problem, but that helped with my machine.

jongough, 2022-06-01:
The dip switch settings are 1:Off, 2:Off, 3:Off, 4:Off,5:Off, 6:On, 7:On, 8:Off. This equates to software configure Peak Current/RMS Current of 0.5-5.6A and 10 micorstep with Steps/rev of 2000. We did try turning on SW6 & 7 to make it default software configured. Also tried turning off SW5 to give Microstep 5 and Steps/rev to 1000. 

We tried multiple combinations of these settings:
     engine: I2S_STREAM, I2S_STATIC
     dir_delay_us: 5, 6, 7
     pulse_us: 3, 4, 5, 6
     disable_delay_us: 3, 4, 5, 6
and other numbers like 10 to see if there was a timing issue. We did not find any noticeable difference in the behaviour. Some of the settings slowed down the movements but did not stop the X-axis lockups.

The table has 3 motors the same all driving through a cog to a chain. The X-axis has a rod between both runners and a chain on this to pull the carriage back and forward. The Y-axis has only one chain and the Z-axis has a step down chain to a ball screw. So we did quite a bit of tuning to get the distance per step correct starting at 100mm and going up to 1500mm and got the accuracy to about .1mm in a single direction with .9mm back lash when changing direction. Multiple back and forward movements ended up in the same place each time.

The X-axis demonstrated the lock up most frequently, with the Y-axis only doing so a couple of times. We tried doing air cuts of small and large boxes with movement in single axes then doing the diagonals. The faster the speed the sooner the lock up occurred. However, it normally occurred within a few minutes.

By loading the 3.4.4 software we did not have any lockups. Loading newer software, 3.4.5, SD-fix test and 3.4.6, the only change made, the lockups occurred. 

We moved the X-axis by pushing the Y-axis gantry with both power on and off. as well as unplugging the X-asis motor. The only difference is there seemed to be with the oower off and the motor plugged in to the driver was it generated enough power to turn on the light on the driver and it seemed a little 'lumpy' as if the feed back through the driver was fighting our movement. With the power on and 3.4.4 loaded we moved the X-axis at about 100mm/s and holding onto it it pull me along the floor. So there is no lack of power. We did surface some ironwood, cut down 60years ago and seasoned, with a 1" endmill at a depth of 11mm at 100mm/s and it cut fine. A couple of large chips jammed the X-axis at one point so the job had to be restarted after removing the chip. On the next job the drive chain had to be re setup as the force had broken the chain connector (link with a U-shaped clip, standard chain stuff).

It was like the pulses were being sent too quickly to the driver such that the microstep never quite completed and the motor moved back, i.e. oscillated between two positions making a 'grinding' type noise, sort of growling. We did check the motor temperature and the drivers and both were normal. Turned the power off and the locked axis moved smoothly from the issue location.

MitchBradley, 2022-06-01:
It increases the size of the segment buffer, thus preventing buffer underruns when running GCode.  We have recently discovered that 6 is too small in many cases.  The next release will have a larger default.

jongough, 2022-06-01:
OK. I will let you know the outcome as soon as I can try it out.

MitchBradley, 2022-06-02:
> If I put this in for 3.4.4 will it cause any issues?

I don't think so, but I could predict the result of all possible scenarios, then there would never be any issues.

jongough, 2022-06-02:
I am remote to the actual machine and am arranging a time with the owner to do the testing you asked for. I am a little worried about hard limits not working as this is one of the safety features that is enabled. For a quick test I can do that, but it cannot be left like that for long or it may damage the machine when it hits the hard stops. There are emergency buttons to hit, but....

I was wondering if there are 'known' issues with 3.4.4 and steppings of 20? If it causes the esp32 software issues I wanted to know before doing the test as the manual says to leave it at 6 and we are not fine tuning a special application.

MitchBradley, 2022-06-02:
The known issue is with stepping of 6.  We have not experienced problems with larger values

jongough, 2022-06-02:
We are mainly using the WebUI over WiFi using it as a hotspot as it will not connect as a client to the WiFi router due to an issue with 'special' characters in the password and the inability to reliably set then via the USB port or the WebUI. We use the WebUI to upload the gcode to the SD card and then pick the 'run' icon to run the file. That seems to be working fine. We do not have BlueTooth available on the PC used to run the system.

MitchBradley, 2022-06-02:
The startup messages show it connecting in STA mode

We do not recommend AP mode because the ESP32 core software has memory leaks in AP mode.

You can put special characters in the password using URI escaping.

```
        switch (c) {
            case '%':  // The escape character itself
                escaped += "%25";
                break;
            case '!':  // Cmd::FeedHold
                escaped += "%21";
                break;
            case '?':  // Cmd::StatusReport
                escaped += "%3F";
                break;
            case '~':  // Cmd::CycleStart
                escaped += "%7E";
                break;
```

bdring, 2022-06-02:
https://github.com/bdring/FluidNC/wiki/FluidNC-FAQ#special-character-issue

MitchBradley, 2022-06-02:
You have to use the escape sequences via serial.  In WebUI you should be able to enter them directly.

Perhaps you have a static IP configured for STA mode?

Another trick is to use a dedicated router.  Many people have several old ones lying around gathering dust.

jongough, 2022-06-03:
On my ESP32 I have the password working with 'special' characters and added steppings: 20 and it booted OK. When I can get onto the workshop machine I will give it a go with the suggestions above.

jongough, 2022-06-03:
I have now set the segments to 20 and tried it with both 3.4.4 and 3.4.6 and they both appear to work the same and provide the same force.
The connection issue was fixed and was due to two wifi routers with the same credentials using different, non-routable/non-bridgeable addresses (a wifi router configuration issue).
This issue is not solved, many thanks.

closed 2022-06-03 by jongough