bdring/FluidNC #460: Would you please add my controller card

Skywalkerf34, 2022-05-31:
### Machine Context

https://www.tindie.com/products/smartelectronic_technologies/6-pack-compact-cnc-controller-for-grbl-esp32/

### Feature Description

This board is compatible with your original 6 pack

### Other Approaches

Some of my customer already move to FLUID

### How I Can Help

Just add in the WIKI list

bdring, 2022-05-31:
I think you should also change the name. It would be confusing to have 2 different controllers on the list with the same name.

Skywalkerf34, 2022-06-12:
> 

Hi Bart,

My design for the board is just a "small enhancement" from your original and make it smaller for some customer's requirement
and because of this at that time PCB design is not open source.
So that is why I name it "6 PACK **compact**" for reference to the great job you have done with the original one, we also purposely change the PCB color (RED and now White) instead of the black to avoid confusion .
This design now also offers a Serial port to connect our pendant controller.

What do you mean by "contributing to project" ? bring some firmware dev? 
Yes I am on discord, we already exchange some time ago by id is the same that GitHub

Best regards

closed 2022-10-02 by bdring

Skywalkerf34, 2022-10-11:
> Please this provide information for the Wiki
>
> Is the design open source? If so is there a link to the source?
>
> Do you contribute to the project?
>
> Are you on our Discord Server? If so what is your username?

Hi Bart,

No, my design is not open source
What do you mean by "contribute to the project"?
my discord username is  skywalkerf34

Changing Name: This is slightly different from your board with "*COMPACT*"
which is the main characteristic/differentiation (except PCB colour that I
purposely choose to be NOT black)