bdring/FluidNC #461: Add hyperlinks to wiki for parameter labels in WebUI

daxliniere, 2022-05-31:
### Machine Context

N/A

### Feature Description

The FluidNC wiki is a wealth of knowledge. I often want to look up the specifics of parameters listed in the Config page.
Since it's just a webpage after all, it would be great if the parameters (eg /axes/z/homing/feed_scaler) linked to the relevant section of the wiki.

### Other Approaches

N/A

### How I Can Help

If you set up the first one, I'm more than happy to do all the grunt work on this one.

MitchBradley, 2022-05-31:
We will be switching over to WebUI version 3 in the near future so the change will have to be done in that context.  To save space, which is short in the FLASH filesystem, it is probably better to have a base URL that is the same for all parameters, then generate the final link by concatenating the parameter name, perhaps as a section reference.

daxliniere, 2022-06-04:
No worries, Mitch. If you remember it once the WebUI 3 transition has happened, let me know and I'll be happy to jump on this. :)

MitchBradley, 2022-06-04:
If it is done as a base url + name there will be no followon grunt work