bdring/FluidNC #462: Problem:  in v3.4.6 and v3.4.5 axes X, Y do not work

vadim7272, 2022-06-01:
### Controller Board

in v3.4.6 and v3.4.5 axes X, Y do not work and axes Z, A work
in v3.4.1..2..3 all axes work
I have coreXY , tried it in different ways and rolled back
please correct and how can I help you?

### Help From Board Vendor

- [ ] Yes
- [ ] No
- [ ] Not Applicable

### Machine Description

coreXY , esp32 , 4axis xyza , tangential knife 
[image]

### Configuration file

```yaml
name: "ESP32 V4 corexy"
board: "ESP32"
meta: "N_ver 3.4.1"

stepping:
  engine: RMT
  idle_ms: 254
  dir_delay_us: 2
  pulse_us: 1
  disable_delay_us: 0

kinematics:
  corexy:
  

axes:
  shared_stepper_disable_pin: gpio.25:low

  x:
    steps_per_mm: 80.000
    max_rate_mm_per_min: 6000.000
    acceleration_mm_per_sec2: 1000.000
    max_travel_mm: 300.000
    homing:
      cycle: 2
      mpos_mm: -3.000
      feed_mm_per_min: 200.000
      settle_ms: 200
      positive_direction: false
    
    motor0:
      limit_all_pin: gpio.36:low
      hard_limits: true
      pulloff_mm: 3.000
      stepstick:
        direction_pin: gpio.14
        step_pin: gpio.15

  y:
    steps_per_mm: 80.000
    max_rate_mm_per_min: 6000.000
    acceleration_mm_per_sec2: 1000.000
    max_travel_mm: 300.000
    homing:
      cycle: 2
      mpos_mm: -3.000
      feed_mm_per_min: 50.000
      settle_ms: 200
      positive_direction: false
    
    motor0:
      limit_all_pin: gpio.35:low
      hard_limits: true
      pulloff_mm: 3.000
      stepstick:
        direction_pin: gpio.27
        step_pin: gpio.13

  z:
    steps_per_mm: 426.000
    max_rate_mm_per_min: 500.000
    acceleration_mm_per_sec2: 300.000
    max_travel_mm: 50.000
    homing:
      cycle: 1
      mpos_mm: 3.000
      feed_mm_per_min: 100.000
      positive_direction: false
    
    motor0:
      limit_pos_pin: gpio.34:low
      hard_limits: false
      pulloff_mm: 1.000
      stepstick:
        direction_pin: gpio.26
        step_pin: gpio.12

  a:
    steps_per_mm: 4.444
    max_rate_mm_per_min: 30000.000
    acceleration_mm_per_sec2: 10000.000
    homing:
      cycle: 0
      positive_direction: false
      
    motor0:
      stepstick:
        direction_pin: gpio.32
        step_pin: gpio.33

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5
  card_detect_pin: NO_PIN

coolant:
  flood_pin: gpio.4

start:
  must_home: false
  deactivate_parking: true
  check_limits: false

Laser:
  pwm_hz: 5000
  output_pin: gpio.2
  enable_pin: gpio.17
  disable_with_s0: false
  s0_with_disable: true
  tool_num: 100
  speed_map: 0=0.000% 255=100.000%
```

### Startup Messages

```text
normal all ok
```

### User Interface Software

webui

### What happened?

in v3.4.1 everything worked, the configuration file did not change, then I installed v3.4.6 the xX, Y axes do not work, the log passes normally, the Z, A axes work
then returned to v3.4.1 everything works again
put v3.4.1,2,3 everything works
set v3.4.5,6 does not work XY

bdring, 2022-06-01:
I will try to reproduce the problem here. 

1. What happens when you send and X or Y move. Does the status suggest the axis is moving, but the motor do not?

2. Could you try removing CoreXY as an experiment to help narrow down the cause.

vadim7272, 2022-06-02:
@bdring 

when I send to the X or Y axis, it shows that the axis is moving and the engine is standing and so in different programs.

@MitchBradley

in order to exclude coreXY, I first switched to conventional kinematics, but the result did not change

I ordered a new board from aliexpress mks tynibee I will try it on it, but I think it will be the same

and yet, if after version 3.4.5 I install only install-fs.bat from v4.3.1, then everything works fine again

vadim7272, 2022-06-02:
I can make a video and put it on YouTube and you can see what is happening or we can talk on Skype, but I'm sorry I don't know English, I'm from Russia, I'm Russian, but I think we will understand each other

MitchBradley, 2022-06-02:
When the problem occurs, are the X and Y motors locked?  In other words, is it possible to turn them by hand.

MitchBradley, 2022-06-02:
Also try setting hard_limits to false to see if it could be related to limit switches.

MitchBradley, 2022-06-02:
Another thing to try is idle_ms: 255

vadim7272, 2022-06-03:
Thanks, I set hard_limits to false on the X,Y axis and the X,Y axes worked fine.
question again...
if I put limit switches , then I will need to set hard_limits to true , and I will get not to use the axis again, what should I do?
thanks again .
if you need my help as a tester of my machine, write, I will participate. I will receive a mks tyni bee board in a month, I will also test it. and I also have optical limit switches, but I have not been able to get them to work yet.

bdring, 2022-06-03:
This will help us debug your problem. You can turn them back on after we figure it out.

You can still use homing and soft limits while hard limits are off.

I do not see switches in your picture. 
 - Do you have them installed?
 - What is the actual controller you are using now?
 - Do you have a schematic for it? 
 - I noticed you did not have have the internal pull up resistor attribute (:pu) [Wiki](https://github.com/bdring/FluidNC/wiki/FluidNC-Setting-Up-I-O-Pins#input-pin-attributes). Does your circuit have pull ups?

vadim7272, 2022-06-03:
I tried to put optical limit switches and I have them built under the brackets, but I could not get them to work on 4.3.1. i set pullup to 5v and put attribute (:pu) , i tried many solutions but always gave alarm . ,.
But another friend put the usual trailers, too, with an attribute (: pu) pull-up, and everything works for him. it has a different scheme and is placed on separate drivers. on YouTube, he posted a video on setting up limit switches, he is also Russian.
I'll try on 3.4.7 I'll try to put optical limit switches and post the video on YouTube.

bdring, 2022-06-03:
What alarm! 

You never mentioned an alarm.

IgorPack, 2022-06-05:
Hello everyone, I am also interested in connecting optical switches. I will try it soon, I will write the results. Vadim, by the way, I'm also from Russia, if you want, we can talk.

vadim7272, 2022-06-20:
got results after many tests
3.4.1,2,3 work fine
3.4.4 works, but it happens that the buffer fills up and freezes, but it rarely happens.
4.4.5,6,7,8 at the first gcoode I get an error after 3-5 seconds of work and an alarm and a complete freeze.

3.4.8 the configuration web interface does not work, I rearranged the firmware 2 times, the same result.

bdring, 2022-06-20:
Did you solve the issue in the title of this?

vadim7272, 2022-06-21:
yes, I decided, it was hard_limits: true replaced with hard_limits: false , but a new problem appeared on 3.4.4,5,6,7,8 after starting gcoode, everything stops after 5-20 seconds and it is necessary to overload the controller.
the mks tiny bee board will come soon, I will test it, but for now I work on 3.4.3, it works fine

bdring, 2022-06-21:
Good.

Please start a new issue with your new problem. Many of the things you posted about your machine have probably changed. we need fresh information and a accurate title for the issue.

closed 2022-06-21 by bdring