bdring/FluidNC #468: retract_on_feedhold=TRUE

daxliniere, 2022-06-04:
### Machine Context

In a CNC mill setup where there is a spindle engaging the material (though, I can see the potential for laser cutters as well), feed hold is essential, but it can lead to burning of the material as the cutter stays down in the material.

### Feature Description

It would be great to get a new option **retract_on_feedhold=TRUE**.

When feedhold is triggered, current machine co-ords Z value would be stored in memory then the Z-axis would retract to machine Z0. When the program is resumed, Z-axis would return to the previously stored position and cutting would continue.

I suppose there could also be the added possibility of spinning down, as well. Perhaps best as a second option (spindown_on_feedhold) which can't be enabled if **retract_on_feedhold=TRUE** as it's not good to stop or start with material engaged.

### Other Approaches

I cannot think of any other way around this, currently.

### How I Can Help

Please let me know what you think I could assist with. Testing, obviously, but otherwise let me know.

bdring, 2022-06-05:
We support the standard Grbl feature of [parking](https://github.com/bdring/FluidNC/wiki/Parking-Feature). That will do what you want and more. Parking is typically associated with a door switch, but can also be activated with a command.

The parameters are compile time values in [config.h](https://github.com/bdring/FluidNC/blob/main/FluidNC/src/Config.h#L201) , but the defaults should work for most people.

Making changes to this would not be a priority for us at this time. If you wanted to sponsor the changes with a donation, we might consider it.

daxliniere, 2022-06-05:
Beautiful! Thanks Bart. I think this is exactly what I'm looking for.

closed 2022-06-05 by daxliniere

daxliniere, 2022-06-05:
Would wiring **cycle_start_pin** to the normally-open (opposite) side of my emergency stop (safety_door_pin) switch work to resume or would it be constantly sending a Cycle Start command?

NC - safety_door_pin
COMMON - GND
NO - cycle_start_pin