bdring/FluidNC #469: Feature:

forbin1ab, 2022-06-05:
### Machine Context

Wallplotter kinematics test machine.

### Feature Description

I was pretty excited to stumble across FluidCNC recently and specifically the WallPlotter kinematic module. I managed to get things kind of working on my machine but it seemed like there were inconsistencies with how the segmentation was working during the cartesian to plotter space conversion. I ended up studying the code in wallplotter.cpp that was created by pentacular and made some changes in order to get it working better. I am a bit new to git, github and discord but I have created a fork with the changes to the latest devt branch at forbin1ab/FLuidNC along with a sample .yaml file that I made. I have also created some documentation regarding the test machine used in markdown and put it here at https://github.com/forbin1ab/FluidNCWallplotter.

### Other Approaches

N/A

### How I Can Help

I can help with any additional testing for wallplotter kinematics and possibly other kinematic modules since I am a bit more familiar with these now.

MitchBradley, 2022-06-05:
The repo linked above contains no code, only documentation, so I cannot review your code changes.

forbin1ab, 2022-06-05:
Sorry my bad. I am new to this. First open source project for me and first use of github like this. I put the documentation in a separate repo since I was not sure where to put it. The code changes are in a forked repo of the devt branch at https://github.com/forbin1ab/FluidNC/tree/Devt

MitchBradley, 2022-06-05:
This eliminates multiplies and divides inside the loop
```c++
        float seg_x = position[X_AXIS];
        float seg_y = position[Y_AXIS];
        float seg_z = position[Z_AXIS];
        dx /= segment_distance;
        dy /= segment_distance;
        dz /= segment_distance;

        for (uint32_t segment = 0; segment < segment_count; segment++) {
            // next segment endpoint
            seg_x += dx;
            seg_y += dy;
            seg_z += dz;
```

MitchBradley, 2022-06-05:
Regarding:
```c++
    void WallPlotter::lengths_to_xy(float h1, float h2, float& x, float& y) {
        // calculate x y lengths from the two given hypotenuses
        // Assumes origin is to the right and below of left motor...
        // TODO Optimize equations to reduce floating point operations
        float m = abs(_left_anchor_x)+abs(_right_anchor_x); // motor separation
        x = (h1*h1-h2*h2+m*m) / (2*m); // solve for X1
        ...
```
The only algorithmic change here (not counting variable name substitution and subexpression factoring), that I can see, is the substitution of
```c++
        float m = abs(_left_anchor_x)+abs(_right_anchor_x); // motor separation
```
instead of
```c++
        float distance = _right_anchor_x - _left_anchor_x;
```
The new version seems incorrect.  If, say, left anchor is 1 and right anchor is 4, the old version gives the correct distance of 3, whereas the new version gives 5.  The only case where the new version gives the correct answer is when the anchors have a different sign.

The old version potentially gives the wrong distance when right_anchor < left_anchor, but arguably that is an error for other reasons.

MitchBradley, 2022-06-06:
If you put <hashtab><issue_number> in the PR commentary, the PR and the issue will be linked for easy navigation and tracking.  GitHub has some nice convenience features like that.

Now, care to tackle the feedrate conversion math?  My first guess would be scale the feedrate by the ratio of the vector lengths in cable space and cartesian space, but maybe I am missing something.

forbin1ab, 2022-06-07:
> If you put <issue_number> in the PR commentary, the PR and the issue will

I will checkout feedrate conversion and would also like to test whether
configuring z axis as pen with stepper and servo works.
I am out of town for next two weeks. I will get up to speed on github
features such as one you mentioned. Thanks.

CCERocks, 2023-07-11:
> I will checkout feedrate conversion and would also like to test whether configuring z axis as pen with stepper and servo works. I am out of town for next two weeks. I will get up to speed on github features such as one you mentioned. Thanks.
> […](#)

hi forbin1lab, do you have working demo video of this wall plotter using FluidNC ?

CCERocks, 2025-05-20:
@forbin1ab  its been a while.. i have MKS DLC 32 board With the help of ChatGPT i can make both motor move but i couldnt set machine configuration for 
    anchor_distance: 1000.0
    anchor_height:   500.0
    carriage_offset: [0.0, 0.0]
maybe added feedrate is also a great idea.. 
which pin for Servo ? GPIO 32 ? (LASER)

whould you share your working setting?

CCERocks, 2025-05-21:
> anchor_distance, anchor_height, and carriage_offset are not FluidNC parameters.  ChatGPT might have gotten confused or trained on something else.  The FluidNC parameters for WallPlotter are left_axis, left_anchor_x, left_anchor_y, right_axis, right_anchor_x, right_anchor_y, and segment_length.

Hi @MitchBradley, 😅 how you know I'm using ChatGPT? Hehehehehe hehehe 😂 any latest guide to make WallPlotter using FluidNC? I have board MKSDLC32 V2.1 with TMC2209. 

Hopefully can just upload gcode files or add into sdcard & run it with battery powered. 😅