bdring/FluidNC #470: Problem:

Dainius-G, 2022-06-05:
### Controller Board

TMC2209 4 Axis Controller

### Help From Board Vendor

- [ ] Yes
- [ ] No
- [ ] Not Applicable

### Machine Description

TMC2209 4 Axis Controller

### Configuration file

```yaml
name: TMC2209 4x Controller
board: unknown

kinematics:
  Cartesian:

stepping:
  engine: RMT
  idle_ms: 250
  dir_delay_us: 0
  pulse_us: 3
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: gpio.25
  x:
    steps_per_mm: 100.000
    max_rate_mm_per_min: 1000.000
    acceleration_mm_per_sec2: 200.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 2
      mpos_mm: 0.000
      positive_direction: true
      settle_ms: 250.000
      seek_mm_per_min: 2000.000
      feed_mm_per_min: 200.000
      seek_scaler: 1.100
      feed_scaler: 5.000
    motor0:
      limit_all_pin: gpio.35:low
      hard_limits: false
      tmc_2209:
        direction_pin: gpio.27
        step_pin: gpio.26
        r_sense_ohms: 0.110
        addr: 0
        run_amps: 0.250
        hold_amps: 0.125
        microsteps: 16
        stallguard: 16
        stallguard_debug: false
        run_mode: Stallguard
        homing_mode: Stallguard
        use_enable: false
        uart:
          txd_pin: gpio.22
          rxd_pin: gpio.21
          baud: 115200
          mode: 8N1
    motor1:
      null_motor:
  y:
    steps_per_mm: 100.000
    max_rate_mm_per_min: 1000.000
    acceleration_mm_per_sec2: 200.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 2
      mpos_mm: 0.000
      positive_direction: true
      settle_ms: 250.000
      seek_mm_per_min: 2000.000
      feed_mm_per_min: 200.000
      seek_scaler: 1.100
      feed_scaler: 5.000
    motor0:
      limit_all_pin: gpio.34:low
      hard_limits: false
      tmc_2209:
        direction_pin: gpio.32
        step_pin: gpio.33
        r_sense_ohms: 0.110
        addr: 1
        run_amps: 0.250
        hold_amps: 0.125
        microsteps: 16
        stallguard: 16
        stallguard_debug: false
        run_mode: Stallguard
        homing_mode: Stallguard
        use_enable: false
    motor1:
      null_motor:
  z:
    steps_per_mm: 100.000
    max_rate_mm_per_min: 1000.000
    acceleration_mm_per_sec2: 200.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 1
      mpos_mm: 0.000
      positive_direction: false
      settle_ms: 250.000
      seek_mm_per_min: 2000.000
      feed_mm_per_min: 200.000
      seek_scaler: 1.100
      feed_scaler: 5.000
    motor0:
      limit_all_pin: gpio.39:low
      hard_limits: false
      tmc_2209:
        direction_pin: gpio.14
        step_pin: gpio.2
        r_sense_ohms: 0.110
        addr: 2
        run_amps: 0.250
        hold_amps: 0.125
        microsteps: 16
        stallguard: 16
        stallguard_debug: false
        run_mode: Stallguard
        homing_mode: Stallguard
        use_enable: false
    motor1:
      null_motor:
  a:
    steps_per_mm: 100.000
    max_rate_mm_per_min: 1000.000
    acceleration_mm_per_sec2: 200.000
    max_travel_mm: 300.000
    soft_limits: false
    motor0:
      tmc_2209:
        direction_pin: gpio.15
        step_pin: gpio.16
        r_sense_ohms: 0.110
        addr: 3
        run_amps: 0.250
        hold_amps: 0.125
        microsteps: 16
        stallguard: 16
        stallguard_debug: false
        run_mode: Stallguard
        homing_mode: Stallguard
        use_enable: false
    motor1:
      null_motor:

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5

control:

coolant:
  flood_pin: gpio.12
  mist_pin: gpio.17
  delay_ms: 1000.000

probe:
  pin: gpio.36:low
  check_mode_start: false

macros:
  startup_line0: 
  startup_line1: 
  macro0: 
  macro1: 
  macro2: 
  macro3: 

start:
  must_home: true
  check_limits: true
  deactivate_parking: false

user_outputs:

PWM:
  tool_num: 0
  speed_map: 0=0.0% 1000=100.0%
  spinup_ms: 0
  spindown_ms: 0
  output_pin: gpio.4
  enable_pin: gpio.13
  disable_with_s0: false
  s0_with_disable: false
  pwm_hz: 5000
arc_tolerance_mm: 0.002
junction_deviation_mm: 0.010
verbose_errors: false
report_inches: false
enable_parking_override_control: false
use_line_numbers: false
```

### Startup Messages

```text
[MSG:INFO: FluidNC v3.4.7]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is SPIFFS]
[MSG:INFO: Configuration file:config.yaml]
[MSG:INFO: Machine TMC2209 4x Controller]
[MSG:INFO: Board unknown]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN]
[MSG:INFO: Stepping:RMT Pulse:3us Dsbl Delay:0us Dir Delay:0us Idle Delay:250ms]
[MSG:INFO: Axis count 4]
[MSG:INFO: Shared stepper disable gpio.25]
[MSG:INFO: Axis X (-300.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO: Trinamic Stepper Uart Tx:gpio.22 Rx:gpio.21 RTS:NO_PIN Baud:115200]
[MSG:INFO:     tmc_2209 Step:gpio.26 Dir:gpio.27 Disable:NO_PIN Addr:0 R:0.110]
[MSG:INFO:     All Limit gpio.35:low]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Y (-300.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2209 Step:gpio.33 Dir:gpio.32 Disable:NO_PIN Addr:1 R:0.110]
[MSG:INFO:     All Limit gpio.34:low]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Z (0.000,300.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2209 Step:gpio.2 Dir:gpio.14 Disable:NO_PIN Addr:2 R:0.110]
[MSG:INFO:     All Limit gpio.39:low]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis A (-300.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2209 Step:gpio.16 Dir:gpio.15 Disable:NO_PIN Addr:3 R:0.110]
[MSG:INFO:   Motor1]
[MSG:INFO: X Axis driver test passed]
[MSG:INFO: Y Axis driver test passed]
[MSG:INFO: Z Axis driver test passed]
[MSG:INFO: A Axis driver test passed]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: PWM Spindle Ena:gpio.13 Out:gpio.4 Dir:NO_PIN Freq:5000Hz Res:13bits]
[MSG:INFO: Using spindle PWM]
[MSG:INFO: Flood coolant gpio.12]
[MSG:INFO: Mist coolant gpio.17]
[MSG:INFO: Probe Pin: gpio.36:low]
[MSG:INFO: Connecting to STA SSID:VM7609012]
[MSG:INFO: Connecting.]
[MSG:INFO: Connected - IP is 192.168.0.30]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl 3.4 [FluidNC v3.4.7 (wifi) '$' for help]
[MSG:INFO: '$H'|'$X' to unlock]
```

### User Interface Software

WebUI

### What happened?

I can`t rotate any of the motors all I get is a little jitter on the motor shaft

MitchBradley, 2022-06-05:
If you set idle_ms to 255, do the motors lock?

MitchBradley, 2022-06-06:
Sometimes, when converting from Grbl_Esp32 to FluidNC, the problem can be caused by signal inversion differences.  In Grbl_Esp32, the inversion is handled by masks in NVS settings, whereas FluidNC handles it with :low pin attributes.  The translator program cannot see the values of your settings because they are not in the machine.h file that is the input to the translator, so it can't automatically add :low in all cases.

bdring, 2022-06-06:
I am testing and getting the same issue. It seems like the current is not working right. If you boost the current it wiggles a little more.

I am looking into the code.

bdring, 2022-06-06:
My TMC2209 4x works with the example configs, but not yours.

Trying to figure out what is the issue.

bdring, 2022-06-06:
It has something to do with StallGuard. StealthChop works fine.

bdring, 2022-06-06:
I think it might be due to the  tmc2209->TCOOLTHRS(...) value. 

Each chip implements a lot of things differently. I guess I'll have to spend another day on this.

You should be able to operate if you change the modes to StealthChop.

MitchBradley, 2022-06-06:
This might be a translator issue.  https://github.com/bdring/Grbl_Esp32/issues/1217 has this in the machine file:

```
#define TRINAMIC_UART_RUN_MODE       TrinamicUartMode :: StealthChop
#define TRINAMIC_UART_HOMING_MODE    TrinamicUartMode :: StallGuard
```
but this in the YAML:

```
        run_mode: Stallguard
        homing_mode: Stallguard
```

Dainius-G, 2022-06-06:
name: TMC2209 4x Controller
board: unknown

kinematics:
  Cartesian:

stepping:
  engine: RMT
  idle_ms: 250
  dir_delay_us: 0
  pulse_us: 3
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: gpio.25
  x:
    steps_per_mm: 100.000
    max_rate_mm_per_min: 1000.000
    acceleration_mm_per_sec2: 200.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 2
      mpos_mm: 0.000
      positive_direction: true
      settle_ms: 250.000
      seek_mm_per_min: 2000.000
      feed_mm_per_min: 200.000
      seek_scaler: 1.100
      feed_scaler: 5.000
    motor0:
      limit_all_pin: gpio.35:low
      hard_limits: false
      tmc_2209:
        direction_pin: gpio.27
        step_pin: gpio.26
        r_sense_ohms: 0.110
        addr: 0
        run_amps: 0.250
        hold_amps: 0.125
        microsteps: 16
        stallguard: 16
        stallguard_debug: false
        run_mode: StealthChop
        homing_mode: StealthChop
        use_enable: false
        uart:
          txd_pin: gpio.22
          rxd_pin: gpio.21
          baud: 115200
          mode: 8N1
    motor1:
      null_motor:
  y:
    steps_per_mm: 100.000
    max_rate_mm_per_min: 1000.000
    acceleration_mm_per_sec2: 200.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 2
      mpos_mm: 0.000
      positive_direction: true
      settle_ms: 250.000
      seek_mm_per_min: 2000.000
      feed_mm_per_min: 200.000
      seek_scaler: 1.100
      feed_scaler: 5.000
    motor0:
      limit_all_pin: gpio.34:low
      hard_limits: false
      tmc_2209:
        direction_pin: gpio.32
        step_pin: gpio.33
        r_sense_ohms: 0.110
        addr: 1
        run_amps: 0.250
        hold_amps: 0.125
        microsteps: 16
        stallguard: 16
        stallguard_debug: false
        run_mode: StealthChop
        homing_mode: StealthChop
        use_enable: false
    motor1:
      null_motor:
  z:
    steps_per_mm: 100.000
    max_rate_mm_per_min: 1000.000
    acceleration_mm_per_sec2: 200.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 1
      mpos_mm: 0.000
      positive_direction: false
      settle_ms: 250.000
      seek_mm_per_min: 2000.000
      feed_mm_per_min: 200.000
      seek_scaler: 1.100
      feed_scaler: 5.000
    motor0:
      limit_all_pin: gpio.39:low
      hard_limits: false
      tmc_2209:
        direction_pin: gpio.14
        step_pin: gpio.2
        r_sense_ohms: 0.110
        addr: 2
        run_amps: 0.250
        hold_amps: 0.125
        microsteps: 16
        stallguard: 16
        stallguard_debug: false
        run_mode: Stallguard
        homing_mode: Stallguard
        use_enable: false
    motor1:
      null_motor:
  a:
    steps_per_mm: 100.000
    max_rate_mm_per_min: 1000.000
    acceleration_mm_per_sec2: 200.000
    max_travel_mm: 300.000
    soft_limits: false
    motor0:
      tmc_2209:
        direction_pin: gpio.15
        step_pin: gpio.16
        r_sense_ohms: 0.110
        addr: 3
        run_amps: 0.250
        hold_amps: 0.125
        microsteps: 16
        stallguard: 16
        stallguard_debug: false
        run_mode: StealthChop
        homing_mode: StealthChop
        use_enable: false
    motor1:
      null_motor:

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5

control:

coolant:
  flood_pin: gpio.12
  mist_pin: gpio.17
  delay_ms: 1000.000

probe:
  pin: gpio.36:low
  check_mode_start: false

macros:
  startup_line0: 
  startup_line1: 
  macro0: 
  macro1: 
  macro2: 
  macro3: 

start:
  must_home: true
  check_limits: true
  deactivate_parking: false

user_outputs:

PWM:
  tool_num: 0
  speed_map: 0=0.0% 1000=100.0%
  spinup_ms: 0
  spindown_ms: 0
  output_pin: gpio.4
  enable_pin: gpio.13
  disable_with_s0: false
  s0_with_disable: false
  pwm_hz: 5000
arc_tolerance_mm: 0.002
junction_deviation_mm: 0.010
verbose_errors: false
report_inches: false
enable_parking_override_control: false
use_line_numbers: false 

This works fine, probably needs to be cleaned up though

MitchBradley, 2022-06-06:
Yes, there was a bug in the translator.  Quiz question - what is wrong with this code:
```c++
    const char* ret;
    switch (mode) {
        case TrinamicMode::None:
            ret = "ERROR";
        case TrinamicMode::CoolStep:
            ret = "CoolStep";
        case TrinamicMode::StealthChop:
            ret = "StealthChop";
        case TrinamicMode::StallGuard:
            ret = "Stallguard";
    }
    return ret;
```

MitchBradley, 2022-06-06:
See https://github.com/bdring/Grbl_Esp32/issues/1218 for the results using the fixed translator program.

closed 2022-06-06 by MitchBradley

Dainius-G, 2022-06-06:
Homing in "StallGuard" mode does not work. Enabling "stallguard_debug: true" shows noisy readings 
`[MSG:INFO: X Axis SG_Val: 14   Rate: 1000.000 mm/min SG_Setting:16]
[MSG:INFO: X Axis SG_Val: 74   Rate: 1000.000 mm/min SG_Setting:16]
[MSG:INFO: X Axis SG_Val: 20   Rate: 1000.000 mm/min SG_Setting:16]
[MSG:INFO: X Axis SG_Val: 74   Rate: 1000.000 mm/min SG_Setting:16]
[MSG:INFO: X Axis SG_Val: 54   Rate: 1000.000 mm/min SG_Setting:16]
[MSG:INFO: X Axis SG_Val: 38   Rate: 1000.000 mm/min SG_Setting:16]
[MSG:INFO: X Axis SG_Val: 20   Rate: 1000.000 mm/min SG_Setting:16]
[MSG:INFO: X Axis SG_Val: 38   Rate: 1000.000 mm/min SG_Setting:16]
[MSG:INFO: X Axis SG_Val: 54   Rate: 1000.000 mm/min SG_Setting:16]
[MSG:INFO: X Axis SG_Val: 20   Rate: 1000.000 mm/min SG_Setting:16]
[MSG:INFO: X Axis SG_Val: 54   Rate: 880.001 mm/min SG_Setting:16]
[MSG:INFO: X Axis SG_Val: 20   Rate: 520.001 mm/min SG_Setting:16]
[MSG:INFO: X Axis SG_Val: 2   Rate: 0.000 mm/min SG_Setting:16]
[MSG:INFO: X Axis SG_Val: 2   Rate: 1000.000 mm/min SG_Setting:16]
[MSG:INFO: X Axis SG_Val: 0   Rate: 1000.000 mm/min SG_Setting:16]
[MSG:INFO: X Axis SG_Val: 22   Rate: 1000.000 mm/min SG_Setting:16]
[MSG:INFO: X Axis SG_Val: 72   Rate: 1000.000 mm/min SG_Setting:16]
[MSG:INFO: X Axis SG_Val: 22   Rate: 1000.000 mm/min SG_Setting:16]
[MSG:INFO: X Axis SG_Val: 72   Rate: 1000.000 mm/min SG_Setting:16]
[MSG:INFO: X Axis SG_Val: 38   Rate: 1000.000 mm/min SG_Setting:16]
[MSG:INFO: X Axis SG_Val: 54   Rate: 1000.000 mm/min SG_Setting:16]
[MSG:INFO: X Axis SG_Val: 22   Rate: 1000.000 mm/min SG_Setting:16]
[MSG:INFO: X Axis SG_Val: 54   Rate: 1000.000 mm/min SG_Setting:16]
[MSG:INFO: X Axis SG_Val: 38   Rate: 40.001 mm/min SG_Setting:16]
[MSG:INFO: X Axis SG_Val: 20   Rate: 0.000 mm/min SG_Setting:16]
[MSG:INFO: X Axis SG_Val: 38   Rate: 0.000 mm/min SG_Setting:16]
[MSG:INFO: X Axis SG_Val: 40   Rate: 0.000 mm/min SG_Setting:16]
[MSG:INFO: X Axis SG_Val: 2   Rate: 1000.000 mm/min SG_Setting:16]
[MSG:INFO: X Axis SG_Val: 0   Rate: 1000.000 mm/min SG_Setting:16]
[MSG:INFO: X Axis SG_Val: 4   Rate: 1000.000 mm/min SG_Setting:16]
[MSG:INFO: X Axis SG_Val: 0   Rate: 1000.000 mm/min SG_Setting:16]
[MSG:INFO: X Axis SG_Val: 4   Rate: 1000.000 mm/min SG_Setting:16]
[MSG:INFO: X Axis SG_Val: 2   Rate: 1000.000 mm/min SG_Setting:16]
[MSG:INFO: X Axis SG_Val: 126   Rate: 1000.000 mm/min SG_Setting:16]
[MSG:INFO: X Axis SG_Val: 0   Rate: 1000.000 mm/min SG_Setting:16]
[MSG:INFO: X Axis SG_Val: 4   Rate: 1000.000 mm/min SG_Setting:16]
[MSG:INFO: X Axis SG_Val: 4   Rate: 1000.000 mm/min SG_Setting:16]
[MSG:INFO: X Axis SG_Val: 0   Rate: 1000.000 mm/min SG_Setting:16]
[MSG:INFO: X Axis SG_Val: 2   Rate: 1000.000 mm/min SG_Setting:16]
[MSG:INFO: X Axis SG_Val: 0   Rate: 1000.000 mm/min SG_Setting:16]
[MSG:INFO: X Axis SG_Val: 2   Rate: 1000.000 mm/min SG_Setting:16]
[MSG:INFO: X Axis SG_Val: 2   Rate: 1000.000 mm/min SG_Setting:16]
[MSG:INFO: X Axis SG_Val: 2   Rate: 1000.000 mm/min SG_Setting:16]
[MSG:INFO: X Axis SG_Val: 0   Rate: 1000.000 mm/min SG_Setting:16]
[MSG:INFO: X Axis SG_Val: 124   Rate: 1000.000 mm/min SG_Setting:16]
[MSG:INFO: X Axis SG_Val: 2   Rate: 1000.000 mm/min SG_Setting:16]
[MSG:INFO: X Axis SG_Val: 124   Rate: 1000.000 mm/min SG_Setting:16]
[MSG:INFO: X Axis SG_Val: 2   Rate: 1000.000 mm/min SG_Setting:16]
[MSG:INFO: X Axis SG_Val: 0   Rate: 1000.000 mm/min SG_Setting:16]
[MSG:INFO: X Axis SG_Val: 2   Rate: 1000.000 mm/min SG_Setting:16]
[MSG:INFO: X Axis SG_Val: 0   Rate: 1000.000 mm/min SG_Setting:16]
[MSG:INFO: X Axis SG_Val: 2   Rate: 1000.000 mm/min SG_Setting:16]
[MSG:INFO: X Axis SG_Val: 2   Rate: 1000.000 mm/min SG_Setting:16]
[MSG:INFO: X Axis SG_Val: 126   Rate: 1000.000 mm/min SG_Setting:16]
[MSG:INFO: X Axis SG_Val: 0   Rate: 1000.000 mm/min SG_Setting:16]
[MSG:INFO: X Axis SG_Val: 126   Rate: 1000.000 mm/min SG_Setting:16]
[MSG:INFO: X Axis SG_Val: 2   Rate: 1000.000 mm/min SG_Setting:16]
[MSG:INFO: X Axis SG_Val: 0   Rate: 1000.000 mm/min SG_Setting:16]
[MSG:INFO: X Axis SG_Val: 2   Rate: 1000.000 mm/min SG_Setting:16]
[MSG:INFO: X Axis SG_Val: 0   Rate: 1000.000 mm/min SG_Setting:16]
[MSG:INFO: X Axis SG_Val: 2   Rate: 1000.000 mm/min SG_Setting:16]
[MSG:INFO: X Axis SG_Val: 2   Rate: 1000.000 mm/min SG_Setting:16]
[MSG:INFO: X Axis SG_Val: 4   Rate: 1000.000 mm/min SG_Setting:16]
[MSG:INFO: X Axis SG_Val: 0   Rate: 1000.000 mm/min SG_Setting:16]
[MSG:INFO: X Axis SG_Val: 124   Rate: 1000.000 mm/min SG_Setting:16]
[MSG:INFO: X Axis SG_Val: 4   Rate: 1000.000 mm/min SG_Setting:16]
[MSG:INFO: X Axis SG_Val: 124   Rate: 1000.000 mm/min SG_Setting:16]
[MSG:INFO: X Axis SG_Val: 2   Rate: 1000.000 mm/min SG_Setting:16]
[MSG:INFO: X Axis SG_Val: 0   Rate: 1000.000 mm/min SG_Setting:16]
[MSG:INFO: X Axis SG_Val: 4   Rate: 1000.000 mm/min SG_Setting:16]
[MSG:INFO: X Axis SG_Val: 0   Rate: 1000.000 mm/min SG_Setting:16]
[MSG:INFO: X Axis SG_Val: 4   Rate: 1000.000 mm/min SG_Setting:16]
[MSG:INFO: X Axis SG_Val: 2   Rate: 1000.000 mm/min SG_Setting:16]
[MSG:INFO: X Axis SG_Val: 124   Rate: 1000.000 mm/min SG_Setting:16]
[MSG:INFO: X Axis SG_Val: 0   Rate: 1000.000 mm/min SG_Setting:16]
[MSG:INFO: X Axis SG_Val: 124   Rate: 1000.000 mm/min SG_Setting:16]
[MSG:INFO: X Axis SG_Val: 4   Rate: 1000.000 mm/min SG_Setting:16]
[MSG:INFO: X Axis SG_Val: 0   Rate: 1000.000 mm/min SG_Setting:16]
[MSG:INFO: X Axis SG_Val: 2   Rate: 1000.000 mm/min SG_Setting:16]
[MSG:INFO: X Axis SG_Val: 0   Rate: 1000.000 mm/min SG_Setting:16]
[... cut ...]
[MSG:INFO: X Axis SG_Val: 4   Rate: 1000.000 mm/min SG_Setting:16]
[MSG:INFO: X Axis SG_Val: 2   Rate: 1000.000 mm/min SG_Setting:16]
[MSG:INFO: X Axis SG_Val: 124   Rate: 1000.000 mm/min SG_Setting:16]
[MSG:INFO: X Axis SG_Val: 2   Rate: 1000.000 mm/min SG_Setting:16]
[MSG:INFO: X Axis SG_Val: 124   Rate: 1000.000 mm/min SG_Setting:16]
[MSG:INFO: X Axis SG_Val: 2   Rate: 1000.000 mm/min SG_Setting:16]
[MSG:INFO: X Axis SG_Val: 0   Rate: 1000.000 mm/min SG_Setting:16]
[MSG:INFO: X Axis SG_Val: 2   Rate: 1000.000 mm/min SG_Setting:16]`

MitchBradley, 2022-06-07:
What happens if you go much slower?

bdring, 2022-06-07:
I think some registers need to change. The method that worked in other drivers is having trouble here.