bdring/FluidNC #471: Soft limit alarm. G-code motion target exceeds machine travel.

daxliniere, 2022-06-05:
### Controller Board

DLC32

### Help From Board Vendor

- [ ] Yes
- [ ] No
- [X] Not Applicable

### Machine Description

N/A

### Configuration file

```yaml
board: MKS-DLC32 V2.1
name: SWOLE-CNC
meta: (23.04.2022) Dax Liniere, originally by Skorpi

kinematics:
  Cartesian:

stepping:
  engine: I2S_STATIC
  idle_ms: 0
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0
  
axes:
  shared_stepper_disable_pin: I2SO.0
  x:
    steps_per_mm: 802.341
    max_rate_mm_per_min: 1500.000
# was 18000.000
    acceleration_mm_per_sec2: 80.000
# was 1500.000
    max_travel_mm: 301.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 20.000
      seek_mm_per_min: 200.000
# was 100 in first SWOLE-CNC iteration running MKS firmware
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: gpio.36:high
      hard_limits: true
      pulloff_mm: 0.500
      stepstick:
        step_pin: I2SO.1
        direction_pin: I2SO.2:low
        disable_pin: NO_PIN
        ms1_pin: NO_PIN
        ms2_pin: NO_PIN
        ms3_pin: NO_PIN
        reset_pin: NO_PIN

  y:
    steps_per_mm: 802.322
    max_rate_mm_per_min: 1500.000
# was 18000.000
    acceleration_mm_per_sec2: 80.000
# was 300.000
    max_travel_mm: 187.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 20.000
# was 300.000
      seek_mm_per_min: 200.000
# was 5000.000 (was 100 with MKS firmware)
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: gpio.35:high
      hard_limits: true
      pulloff_mm: 0.500
      stepstick:
        step_pin: I2SO.5
        direction_pin: I2SO.6:low
        disable_pin: NO_PIN
        ms1_pin: NO_PIN
        ms2_pin: NO_PIN
        ms3_pin: NO_PIN
        reset_pin: NO_PIN
  z:
    steps_per_mm: 802.497
    max_rate_mm_per_min: 1000.000
# was 18000.000
    acceleration_mm_per_sec2: 50.000
# was 500.000
    max_travel_mm: 46.000
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 20.000
# was 300.000
      seek_mm_per_min: 100.000
# was 1000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.34:high
      hard_limits: true
      pulloff_mm: 0.500
      stepstick:
        step_pin: I2SO.3
        direction_pin: I2SO.4:low
        disable_pin: NO_PIN
        ms1_pin: NO_PIN
        ms2_pin: NO_PIN
        ms3_pin: NO_PIN
        reset_pin: NO_PIN

i2so:
  bck_pin: gpio.16
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.12
  mosi_pin: gpio.13
  sck_pin: gpio.14

sdcard:
  cs_pin: gpio.15
  card_detect_pin: NO_PIN
# This could be GPIO.39, but Card Detect has no supported functions in FluidNC

control:
  safety_door_pin: gpio.33:low:pu
  cycle_start_pin: NO_PIN
  feed_hold_pin: NO_PIN
  reset_pin: NO_PIN
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

macros:
  startup_line0:
  startup_line1:
  macro0: $SD/Run=lasertest.gcode
  macro1:
  macro2:
  macro3:

coolant:
  flood_pin: gpio.0
# continuous air
  mist_pin:  gpio.4
# pulsed air 
  delay_ms: 0

probe:
  pin: gpio.22:low
  check_mode_start: true
  

10V:
# Spindle
  direction_pin: NO_PIN
  forward_pin: gpio.5:low
  reverse_pin: NO_PIN
  output_pin: gpio.32
  enable_pin: NO_PIN
  pwm_hz: 5000
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 10000
  spindown_ms: 20000
  tool_num: 0
  speed_map: 0=0% 0=25% 5868=25% 23237=99% 24000=100%
#DLC32 Spindle TTL max output voltage is 4.89v, so it can never reach 24000rpm/400Hz

#pwm:
#  direction_pin: NO_PIN
#  output_pin: gpio.32
#  enable_pin: gpio.5:low
#  pwm_hz: 5000
#  disable_with_s0: false
#  s0_with_disable: true
#  spinup_ms: 10000
#  spindown_ms: 20000
#  tool_num: 0
#  speed_map: 0=0% 0=25% 5868=25% 23237=99% 24000=100%

user_outputs:
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 5000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000
  digital0_pin: NO_PIN
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: NO_PIN

start:
  must_home: false
# Pins that could be used: 0(SDA), 4(SCL), 5, 18, 19, 22, 23, 25, 26, 27, 32 (TTL spindle control), 33, 39, I2SO.7
```

### Startup Messages

```text
N/A
```

### User Interface Software

UGS 2.0.11+dev

### What happened?

I sstarted testing soft limits enabled. I issued a move command which was outside of the machine's boundaries. These are the messages I received:

**[MSG:INFO: Soft limit on Z target:21.100]

[Error] An error was detected while sending 'G91 g54 G0 Z21.1': (ALARM:2) Soft limit alarm. G-code motion target exceeds machine travel. Machine position retained. Alarm may be safely unlocked. Streaming has been paused.

[MSG:INFO: Reset to continue]**

The third line contradicts the second and is indeed correct, I did need to perform a reset to continue. Unlock didn't work.

brianredbeard, 2022-06-11:
One minor note, it appears that you're homing your Z axis by moving to the maximum positive value (I'm assuming this because of `$/axes/z/homing/positive_direction=true`) and you set `$/axes/z/max_travel_mm=46.000`, thus 46 would be your max height and then you would theoretically be able to move downwards 46mm _after_ homing all the way to 0.

The one question I have is around the value `$/axes/z/homing/mpos_mm=0`.  As per the documentation on [`mpos_mm`](https://github.com/bdring/FluidNC/wiki/FluidNC-Motor-Setup#user-content-mpos_mm):

-----
- <a id="mpos_mm">**mpos_mm:** 

  - Type: [Float](https://github.com/bdring/FluidNC/wiki/FluidNC-Config-File-Overview#user-content-float)
  - Range: -1000000.000 to 1000000.000
  - Default: 0.000
  - Details: Sets the machine position after homing and limit switch pulloff in millimeters. If you want the machine position to be zero at the limit switch, set this to zero. Keep in mind the homing direction you choose this number.
-----

So, I should _think_ that you would want that to be the value of:

`$/axes/z/max_travel_mm` - `$/axes/z/motor0/pulloff_mm`  In your case that would be 45.500 (46.000 -  0.500).

Then again, it's entirely possible that I'm off base here. :smile:

MitchBradley, 2022-06-11:
In the current way of doing things, the pulloff is considered to be outside the travel envelope.  With the provided config, the travel envelope is from -46 to 0.

IgorPack, 2022-06-18:
Can I ask the author a question, maybe off topic, why do you have steps_per_mm: on the axes X (802.341), Y (802.322) and Z (802.497) differ from each other? I assume you have T8 and HEMA23 so why such different settings? As for me, there should be 800 everywhere, or am I mistaken?

daxliniere, 2022-06-18:
@IgorPack, I have no idea, but that's what I got from multiple measurements. These were taken when I was using the original Arduino V4 GRBL board, but seem to have held up.
I have plans to go over every element of alignment/calibration with a fine-tooth comb, but have had more pressing things on lately.

daxliniere, 2022-06-20:
@MitchBradley @buildlog, any thoughts on the issue in the first post?

MitchBradley, 2022-06-20:
There is not enough information to make sense of it.  The first post did not say where the machine was when the offending move was issues, nor the work offset, nor what "didn't work" exactly means.

daxliniere, 2022-06-20:
@MitchBradley, sure, so just ask me for more. I'm always happy to provide that.

The issue is not regarding the absolute position, but the way FNC reacts to an out-of-bounds request with soft limits.
There is no movement of any axes, so no steps can be lost it, so it should not require the system to be reset. It should only require an unlock.

daxliniere, 2022-06-20:
The contradiction is this:
"Machine position retained. Alarm may be safely unlocked." and "Reset to continue"

bdring, 2022-06-20:
There are cases where position is lost with soft limits. If the machine is still executing a previous move and a new move is entered that fails soft limits, the machine will abruptly stop the current move with a soft limit alarm. This is probably true with most flavors of Grbl. That is why you get that notification.

Generally soft limits occur with a gcode file that exceeds the range of the machine. In that case the job is not recoverable because there are areas to machine outside the reach of the machine.

It can also occur when you are "jogging" the machine under manual control with regular gcode. Recently we modified the [$J style jogging](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Jogging) to prevent soft limit errors. The move will be truncated to the machine range.

We are considering modifying the way soft limits are handled. Rather than faulting immediately, the gcode would be rejected, motion still in the buffers would be completed, then the alarm would be issued. If you were running some long slow lines, this could take a while. Another alternative would be to issue a feedhold and give the user the option to complete the buffered motion.

We have added this to the wish list tracker. This is not likely to be a simple task and could take a while.

https://github.com/bdring/FluidNC/projects/1#column-17014978

daxliniere, 2022-06-20:
Okay, that's brilliant. Thanks for the update and the recent jogging changes.

MitchBradley, 2022-06-21:
The other reason to require a reset on a soft limit that happens in the execution of gcode is to make absolutely certain that a sender does not just keep on sending afterwards.  Most senders treat reset as having the additional semantics of stopping the job from the sending perspective.

daxliniere, 2022-06-21:
Not a bad point, @MitchBradley, but since FNC requires an unlock, to continue, is that not already solved?

daxliniere, 2022-06-22:
Or an option in the settings?

MitchBradley, 2022-06-22:
Every option doubles the test matrix.

bdring, 2022-06-22:
You can't add that option without making sure all soft limit alarms are recoverable. We have it on the list. In the mean time, try not to hit the soft limit so much.