bdring/FluidNC #473: Feed Hold to stop homing cycle

daxliniere, 2022-06-05:
### Machine Context

CNC mill. Engaging my feedhold button doesn't stop homing cycles, but it does stop jogging and any running program.

This is less-than-ideal and feedhold should be expanded to act on homing as well, since there's just as much chance of crashing during a homing cycle.

### Feature Description

Engaging feedhold should also stop homing cycles.

I'm yet to test how this works with the parking feature, but will report back.
(For context, I'm considering reconfiguring my feed_hold switch to safety_door.)

### Other Approaches

Be a wizard?

### How I Can Help

You tell me what I can do. :)

bdring, 2022-06-05:
Homing runs in a different loop than other motion modes. It does this so it can read the limit switches frequently and do stuff like stress free squaring. It ignores a lot of other things.

The normal method of stopping homing is a reset. You resume by running homing again.

Other things like parking in the middle of a homing sequence will not work either.

A lot of location stuff is not set until homing completes, so FNC cannot assume locations are valid. Things like single axis homing can be very dangerous for this reason. The alarm is cleared but other axes may not have been homed. This places the responsibility on the user. At this time I don't think I want to make that potentially worse by changing the context in which homing operates.

daxliniere, 2022-06-06:
I see your point, though I wonder if it's more risky for the user to have to think, in a split second, which of the 3 stop buttons they should push to avoid a homing crash. I like the idea of feed hold also applying to homing for that very reason.

Thanks for taking the time to explain some of the back end processes going on.

daxliniere, 2022-06-06:
Actually, I guess it's 4 stop methods: hard e-stop, feed hold, door switch and reset.

bdring, 2022-06-06:
It is possible we could have other methods stop the homing sequence, but there would be no resume. You would have to rehome.

E-Stop should always be 100% independent of firmware and kill the main power.

closed 2022-06-06 by bdring

daxliniere, 2022-06-06:
Yes, I agree, I don't think it would make sense to have resume on a homing. Restarting the homing cycles of the most logical.