bdring/FluidNC #479: Failed to communicate with FW!

T6Baldy, 2022-06-07:
### Controller Board

None, just the ESP32 currently

### Help From Board Vendor

- [ ] Yes
- [ ] No
- [X] Not Applicable

### Machine Description

New ESP32 on first install

### Configuration file

```yaml
N/A
```

### Startup Messages

```text
$ss
[MSG:INFO: FluidNC v3.4.7]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is SPIFFS]
[MSG:WARN: Cannot open configuration file:config.yaml]
[MSG:INFO: Using default configuration]
[MSG:INFO: Axes: using defaults]
[MSG:INFO: Machine Default (Test Drive)]
[MSG:INFO: Board None]
[MSG:INFO: SPI not defined]
[MSG:INFO: Stepping:RMT Pulse:4us Dsbl Delay:0us Dir Delay:0us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (-1000.000,0.000)]
[MSG:INFO: Axis Y (-1000.000,0.000)]
[MSG:INFO: Axis Z (-1000.000,0.000)]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Using spindle NoSpindle]
[MSG:INFO: Connecting to STA SSID:*****]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connecting...]
[MSG:INFO: Connecting....]
[MSG:INFO: Connected - IP is 192.168.0.70]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
```

### User Interface Software

WebUI

### What happened?

First install on new (blank) ESP32.

- Downloaded latest release (3.4.7) from releases link in wiki.
- Ran install-fs.bat.
- Ran install-wifi.bat
- Put Wifi details in through fluidterm using $Sta commands
- Rebooted (replugged) the ESP32
- Tried to connect to WebUI using IP address shown in startup messages

Every time I try and connect to the WebUI I'm presented with a message "Failed to communicate with FW!". If I mash F5 I sometimes get a working page and then have to repeatedly hit the refresh key on the settings page until I see the options.

If I connect through Telnet it works fine but the WebUI rarely works.

MitchBradley, 2022-07-03:
The requested information has not been provided so there is no more that we can do to help.  Closing as stale.

closed 2022-07-03 by MitchBradley

helLf1nGer, 2024-02-19:
Failed to communicate with FW!

I have this issue with WebUI in either AP or STA modes. instead of webui I get this:
[image]

MitchBradley, 2024-02-19:
Try running install-fs.bat again.  That message can happen if the filesystem is bad.  Also try a more recent version of the firmware.

helLf1nGer, 2024-02-20:
> Try running install-fs.bat again. That message can happen if the filesystem is bad. Also try a more recent version of the firmware.

Install-fs.bat fixed the issue. I was using the latest stable version of win64. Thank you