bdring/FluidNC #484: Problem: error:22 Feed rate has not yet been set or is undefined.

Althafzahra, 2022-06-10:
### Controller Board

board 4 axis external drive 

### Help From Board Vendor

- [ ] Yes
- [X] No
- [ ] Not Applicable

### Machine Description

laser engrave co2

### Configuration file

```yaml
name: "ESP32 Dev Controller V4"
board: "ESP32 Dev Controller V4"

stepping:
  engine: RMT
  idle_ms: 250
  dir_delay_us: 1
  pulse_us: 2
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: gpio.13:low
  
  x:
    steps_per_mm: 20.000
    max_rate_mm_per_min: 20000
    acceleration_mm_per_sec2: 3000
    max_travel_mm: 1170
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: false 
      mpos_mm: 0
      feed_mm_per_min: 200
      seek_mm_per_min: 2000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    
    motor0:
      limit_all_pin: gpio.34:low:pu
      limit_pos_pin: NO_PIN
      limit_neg_pin : NO_PIN
      hard_limits:false
      pulloff_mm: 1.000
      standard_stepper:
        direction_pin: gpio.2
        step_pin: gpio.0
    motor1:
      null_motor:

  y:
    steps_per_mm: 80.0302
    max_rate_mm_per_min: 10000
    acceleration_mm_per_sec2: 600
    max_travel_mm: 800
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 0
      feed_mm_per_min: 200
      seek_mm_per_min: 2000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_all_pin: gpio.35:low:pu
      limit_pos_pin: NO_PIN
      limit_neg_pin : NO_PIN
      hard_limits:false
      pulloff_mm: 1.000
      standard_stepper:
        direction_pin: gpio.15
        step_pin: gpio.26
        
    motor1:
      limit_all_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_neg_pin : NO_PIN
      hard_limits:false
      pulloff_mm: 1.000
      standard_stepper:
        direction_pin: gpio.33
        step_pin: gpio.27

  z:
    steps_per_mm: 800
    max_rate_mm_per_min: 2000
    acceleration_mm_per_sec2: 25
    max_travel_mm: 1000
    
    motor0:
      limit_all_pin: gpio.36:low:pu
      pulloff_mm: 1.000
      standard_stepper:
        direction_pin: gpio.14
        step_pin: gpio.12
    motor1:
      null_motor:

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5
  card_detect_pin: NO_PIN

  
control:
  safety_door_pin: NO_PIN
  reset_pin: gpio.32:low
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  
coolant:
  flood_pin: gpio.21
  mist_pin:  NO_PIN

        
probe:
  pin: NO_PIN

LASER:
  pwm_hz: 5000
  output_pin: gpio.25
  enable_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  tool_num: 0
  speed_map: 0=0% 255=100%
```

### Startup Messages

```text
Layer Cutting
Grbl 3.4 [FluidNC v3.4.2 (wifi) '$' for help]
error:22
Feed rate has not yet been set or is undefined.
On or near line 12122:
Job halted
Stream completed in 14:56
error:22
Feed rate has not yet been set or is undefined.
error:22
Feed rate has not yet been set or is undefined.
error:22
Feed rate has not yet been set or is undefined.
error:22
Feed rate has not yet been set or is undefined.
error:22
Feed rate has not yet been set or is undefined.
error:22
Feed rate has not yet been set or is undefined.
```

### What happened?

I hope this problem can be resolved so that it can work better

### Other Information

[image]

MitchBradley, 2022-06-10:
The error message mentions line 12122 of the GCode file. It would be nice to have a look at that file to see what is happening at and before that line, but you did not attach the file.

MitchBradley, 2022-06-22:
Unless the requested information is provided soon, I will close this ticket because we cannot solve it without more information.

You could try upgrading to the latest version of FluidNC.  I do not know of any recent bug fixes that could explain your problem, but it is always a good idea to try the latest code when reporting problems.

MitchBradley, 2022-07-03:
The requested extra information has not been provided so there is no more that we can do to help.  Closing.

closed 2022-07-03 by MitchBradley