bdring/FluidNC #485: Problem: After firmware upgrade one motor does not power up and rates are off

lagnat, 2022-06-11:
### Controller Board

CNC xPro V5

### Help From Board Vendor

- [ ] Yes
- [X] No
- [ ] Not Applicable

### Machine Description

Bulkman 3D Ultimate Bee, configured for dual X motors.

### Configuration file

```yaml
board: unknown
name: CNC_xPRO_V5_Machine_Template
meta: 
stepping:
  engine: RMT
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 0
  disable_delay_us: 0
  segments: 6
axes:
  shared_stepper_disable_pin: NO_PIN
  shared_stepper_reset_pin: NO_PIN
  x:
    steps_per_mm: 160.000
    max_rate_mm_per_min: 6000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 800.000
    soft_limits: false
    homing:
      cycle: 1
      allow_single_axis: true
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 200.000
      seek_mm_per_min: 3000.000
      settle_ms: 250
      seek_scaler: 1.000
      feed_scaler: 1.500
    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: gpio.34:low
      hard_limits: true
      pulloff_mm: 10.000
      tmc_5160:
        step_pin: gpio.27
        direction_pin: gpio.26
        disable_pin: NO_PIN
        r_sense_ohms: 0.050
        run_amps: 1.800
        hold_amps: 1.250
        microsteps: 8
        toff_disable: 0
        toff_stealthchop: 5
        use_enable: true
        cs_pin: gpio.17
        spi_index: 1
        run_mode: StallGuard
        homing_mode: StallGuard
        stallguard: 16
        stallguard_debug: false
        toff_coolstep: 3
        tpfd: 4
    motor1:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      tmc_5160:
        step_pin: gpio.33
        direction_pin: gpio.32
        disable_pin: NO_PIN
        r_sense_ohms: 0.050
        run_amps: 1.800
        hold_amps: 1.250
        microsteps: 8
        toff_disable: 0
        toff_stealthchop: 5
        use_enable: true
        cs_pin: NO_PIN
        spi_index: 2
        run_mode: StallGuard
        homing_mode: StallGuard
        stallguard: 16
        stallguard_debug: false
        toff_coolstep: 3
        tpfd: 4
  y:
    steps_per_mm: 200.000
    max_rate_mm_per_min: 6000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 550.000
    soft_limits: false
    homing:
      cycle: 1
      allow_single_axis: true
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 200.000
      seek_mm_per_min: 3000.000
      settle_ms: 250
      seek_scaler: 1.000
      feed_scaler: 1.500
    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: gpio.35:low
      hard_limits: true
      pulloff_mm: 5.000
      tmc_5160:
        step_pin: gpio.12
        direction_pin: gpio.14
        disable_pin: NO_PIN
        r_sense_ohms: 0.050
        run_amps: 1.800
        hold_amps: 1.250
        microsteps: 8
        toff_disable: 0
        toff_stealthchop: 5
        use_enable: true
        cs_pin: NO_PIN
        spi_index: 3
        run_mode: StallGuard
        homing_mode: StallGuard
        stallguard: 16
        stallguard_debug: false
        toff_coolstep: 3
        tpfd: 4
    motor1:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      null_motor:
  z:
    steps_per_mm: 400.000
    max_rate_mm_per_min: 2000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 1
      allow_single_axis: true
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 200.000
      seek_mm_per_min: 1000.000
      settle_ms: 250
      seek_scaler: 1.000
      feed_scaler: 1.500
    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: gpio.39:low
      hard_limits: true
      pulloff_mm: 5.000
      tmc_5160:
        step_pin: gpio.15
        direction_pin: gpio.2
        disable_pin: NO_PIN
        r_sense_ohms: 0.050
        run_amps: 1.800
        hold_amps: 1.250
        microsteps: 8
        toff_disable: 0
        toff_stealthchop: 5
        use_enable: true
        cs_pin: NO_PIN
        spi_index: 4
        run_mode: StallGuard
        homing_mode: StallGuard
        stallguard: 16
        stallguard_debug: false
        toff_coolstep: 3
        tpfd: 4
    motor1:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      null_motor:
kinematics:
  Cartesian:
spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18
sdcard:
  cs_pin: gpio.5
  card_detect_pin: NO_PIN
control:
  safety_door_pin: gpio.16:low:pu
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  macro0_pin: gpio.13:low:pu
  macro1_pin: gpio.0:low:pu
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN
coolant:
  flood_pin: NO_PIN
  mist_pin: gpio.21
  delay_ms: 1000
probe:
  pin: gpio.22:low:pu
  check_mode_start: false
macros:
  startup_line0: 
  startup_line1: 
  macro0: 
  macro1: 
  macro2: 
  macro3: 
start:
  must_home: true
  deactivate_parking: false
  check_limits: true
user_outputs:
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 5000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000
  digital0_pin: NO_PIN
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: NO_PIN
arc_tolerance_mm: 0.002
junction_deviation_mm: 0.010
[Error] An error was detected while sending '$cd': verbose_errors: false. Streaming has been paused.
report_inches: false
enable_parking_override_control: false
use_line_numbers: false
planner_blocks: 16
PWM:
  pwm_hz: 5000
  direction_pin: NO_PIN
  output_pin: gpio.25
  enable_pin: gpio.4
  disable_with_s0: false
  s0_with_disable: false
  spinup_ms: 0
  spindown_ms: 0
  tool_num: 0
  speed_map: 0=0.000% 12000=100.000%
```

### Startup Messages

```text
ets Jun  8 2016 00:22:57
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:12812
load:0x40080400,len:3032
entry 0x400805e4
**** Connected to /dev/ttyUSB0 @ 115200 baud ****
[MSG:INFO: FluidNC v3.4.7]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is SPIFFS]
[MSG:INFO: Configuration file:v5.yaml]
[MSG:INFO: Machine CNC_xPRO_V5_Machine_Template]
[MSG:INFO: Board unknown]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN]
[MSG:INFO: Stepping:RMT Pulse:4us Dsbl Delay:0us Dir Delay:0us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (0.000,800.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_5160 Step:gpio.27 Dir:gpio.26 CS:gpio.17 Disable:NO_PIN Index:1 R:0.050]
[MSG:INFO:     All Limit gpio.34:low]
[MSG:INFO:   Motor1]
[MSG:INFO:     tmc_5160 Step:gpio.33 Dir:gpio.32 CS:NO_PIN Disable:NO_PIN Index:2 R:0.050]
[MSG:INFO: Axis Y (0.000,550.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_5160 Step:gpio.12 Dir:gpio.14 CS:NO_PIN Disable:NO_PIN Index:3 R:0.050]
[MSG:INFO:     All Limit gpio.35:low]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Z (-300.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_5160 Step:gpio.15 Dir:gpio.2 CS:NO_PIN Disable:NO_PIN Index:4 R:0.050]
[MSG:INFO:     All Limit gpio.39:low]
[MSG:INFO:   Motor1]
[MSG:INFO: X Axis driver test passed]
[MSG:INFO: X2 Axis driver test passed]
[MSG:INFO: Y Axis driver test passed]
[MSG:INFO: Z Axis driver test passed]
[MSG:INFO: Door gpio.16:low:pu]
[MSG:INFO: Macro 0 gpio.13:low:pu]
[MSG:INFO: Macro 1 gpio.0:low:pu]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: PWM Spindle Ena:gpio.4 Out:gpio.25 Dir:NO_PIN Freq:5000Hz Res:13bits]
[MSG:INFO: Using spindle PWM]
[MSG:INFO: Mist coolant gpio.21]
[MSG:INFO: Probe Pin: gpio.22:low:pu]
[MSG:INFO: STA SSID is not set]
[MSG:INFO: AP SSID FluidNC IP 192.168.0.1 mask 255.255.255.0 channel 1]
[MSG:INFO: AP started]
[MSG:INFO: WiFi on]
[MSG:INFO: Captive Portal Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
```

### User Interface Software

UGS

### What happened?

I was a few revisions behind and did the upgrade to 3.4.7.  

After doing so, one of my X axis motors does not power up.  I can turn the stepper by hand, while the others I am not able to.  After $MD, I can turn them all.  

The feed rates and distances are also not correct.  G91 G0 Y100, for example, moves about 1/4 of the expected distance and not at the expected rate.  steps_per_mm looks right, but I wasn't smart enough to make a backup of the config before the upgrade so unless the upgrade process changed something, it should be right.

I haven't tried to revert the firmware (yet).

lagnat, 2022-06-11:
Note that Y steps_per_mm says 200, but that is because I was playing around with some settings and forgot to reset before $CD, but that is NOT what is saved in the file.  I can snag the file with $localfs if you need me to.

lagnat, 2022-06-11:
Update: Reverted firmware and then tried the next several releases.

| Version | Results |
| -- | -- |
| 3.4.3 | Works |
| 3.4.4 | Works |
| 3.4.5 | Doesn't work as described above |
| 3.4.6 | Doesn't work at all.  Threw a bunch of errors. |
| 3.4.7 | Doesn't work as described above |

bdring, 2022-06-11:
That controller is unique with daisy chained TMC5160 drivers and the SPI use_enable. It is not something I could reproduce here. The supplier said they were sending us a least one several months ago, but we never got one.

If the resolution is off, it is likely the microstepping value is not getting to the motor driver.

I can try some basic TMC5160 checks next week.

lagnat, 2022-06-11:
Anything I can do to help debug?  I could spin up the dev tools if needed.

lagnat, 2022-06-11:
`The supplier said they were sending us a least one several months ago, but we never got one.`

@spark-concepts  @Mboss32

MitchBradley, 2022-07-03:
Try [v3.5.0-pre5](https://github.com/bdring/FluidNC/releases/tag/v3.5.0-pre5).  It has a number of fixes to problems that resulted from the switchover to the new framework.  That switchover happened in v3.4.5 so your problem could be related.

lagnat, 2022-07-03:
I built main a couple days ago and that didn't help but I'll try pre5 as well and report back,

MitchBradley, 2022-07-03:
The only reason to build from main would be if you needed to change something in the compile-only configuration, such as adding OLED support.  Every time that we update main, we do an official release.  main is never ahead, and usually behind, our work in progress.

lagnat, 2022-07-05:
pre5 is slightly better.  Looks like the rate and distance is correct again but one of the motors is still not operating.

lagnat, 2022-07-28:
Might have been wrong about Pre5 being slightly better.  Is there anything I can do to help debug?  Do you suspect this is a framework bug?  I'm way out of my comfort zone with regard to hardware but I *think* all of the drivers are daisy-chained yet only the two that are ganged together are having a problem.  I was suspicious that cs_pin had to be set for both motors but the firmware specifically told me I was wrong.  If you can point me in the general direction I should be able to put in some debugging code and collect some data.

MitchBradley, 2022-07-28:
Debugging by remote control is too slow and too subject to misinterpretation.  If you can find some way to get an XPro into our hands we can solve the problem. Otherwise it will always be on the back burner because the process is too inefficient to displace all of the other things we are also doing.

lagnat, 2022-07-28:
Fair answer.  I understand that it's not practical if it's back and forth copy-paste.  I'd be more than happy to roll up my sleeves and give it a go but I'd need some bootstrapping in the form of "this is how it's supposed to work".  40+ years as a software engineer, coded c++ professionally for 15 of them however quite rusty these days.  I messaged the XPro people but that didn't prove to be effective.  I can't justify the expense but if there are enough other XPro people out there, I'd be happy to contribute.  It would be more cost effective for me to buy a 6-pack and had I known about FluidNC before I started, that's what I would have done.  The next best thing I can do is get you ssh access to my setup but I'm fairly sure I know what you'll think about that idea.  Either way, thanks for FluidNC.  It fantastic!

MitchBradley, 2022-07-28:
Okay this is how it is supposed to work.  I really don't have time to write this but...

We use the third-party TMCStepper library to access the TMC devices.  It has a complicated class structure that tries to capitalize on the similarities between different TMC device families while also handling their differences.  That is a tall order because, in my opinion, the Trinamic hardware engineers did not understand how to design their register interfaces to accommodate the needs of software.  TMCStepper is pulled into the PlatformIO compilation setup via a "lib_deps" line in platformio.ini

The TMCStepper library sits on top of the Arduino framework, which is a set of C++ classes that tries to make it possible to use Arduino coding principles and associated Arduino libraries with ESP32.  That framework includes an esp32-hal - layer that interfaces between the Arduino C++ classes and the C-style device drivers provided by Espressif's ESP-IDF SDK.  ESP-IDF has C device drivers that are themselves built on top of yet another HAL layer that tries to abstract away differences between the classic ESP32 chip and new variants like ESP32-S2 and even the version that is a RISC-V instruction set instead of Xtensa.  And that ESP-IDF HAL layer does not go directly to the hardware; instead it invokes yet another "ll" (low level) HAL-ish layer that is a bunch of always-inline functions with a layer of #define "functions" on top.

The ESP-IDF source code can be inspected if you know how to find the actual version that is included in the Arduino framework package.  That inclusion is in the form of precompiled ".a" library files, not as directly-compilable source.  It is possible to build ESP-IDF from source but that requires a different platformio.ini setup, which adds yet another layer of complexity to the compilation environment and risks getting versions that do not reflect the code you are trying to debug.  With the default precompiled setup, you cannot make code changes at the ESP-IDF level for debugging purposes.

The Arduino framework stuff that TMCStepper uses is primarily the SPI class, used for, obviously, sending commands over the SPI bus.  The Arduino framework's SPI class driver uses some driver stuff from ESP-IDF but also does some stuff by itself.

The SPI bus is shared between the SD card and the TMC devices.  That requires using a different chip select (CS) GPIO for SD vs TMC. That gets complicated because TMCStepper wants to use the Arduino digitalWrite() function to control the TMC CS GPIO, whereas for the SD Card, we use the FluidNC "Pin" framework that lets us use a uniform way to access different "abstract GPIOs" including ones that implemented via I2S.  To make the TMC CS work, we overload a weak function in the TMC Stepper code and redirect it back through the FluidNC Pin framework via a "PinMapper" class.

The old version of ESP-IDF does not have the internal HAL layer because it only supported classic ESP32 chips.  The new version has the internal HAL, and also they changed the behavior of some functions at the driver layer. The new Arduino framework's esp32-hal- layer has changes that reflect that changed behavior at the ESP-IDF driver layer.  There are potential bugs at both places.  There are also potential problems due the use of a newer version of GCC and the associated C++ standard libraries, especially in the area of atomics.

The problem that you are experiencing could be caused by changes or bugs in any of those layers.  Also, we do not entirely understand what the XPro people have done with their hardware.

Now do you see why I am less than thrilled by the prospect of spending a lot of time being your guide through the wilderness?

lagnat, 2022-07-28:
Well that's a mouthful and more than I expected, so thank you for that.  You've outlined the complexity well and it looks quite daunting.  Abstractions on top of abstractions on top of a binary-only library doesn't bode well.

MitchBradley, 2022-07-28:
Not to mention the fact that there are at least 6 major components designed by entirely different people with different architectural styles, goals, and approaches, and you end up needing to comprehend both old and new versions of many of them in order to navigate the change set.

lagnat, 2022-08-07:
As I'm sure you predicted, I didn't get very far.  That being said, I opened an issue on the SparkConcepts GitHub about this and it might have resulted in them sending an XPro to each of you.  They say at least one of the units was already successfully delivered.

bdring, 2022-08-07:
I received the one on Friday. There are a few design issues that make it not 100% compatible with FluidNC at this time. Some of those should be able to be overlooked to allow basic motor and USB functionality, but I still have issues with both.

I am optimistic we can modify FluidNC to make it work, but it could take some time. To get the flexibility we need could force major rewrites of the SPI and Trinamic sections.

Please be patient and remember that FluidNC is free and open source firmware. We are are not affiliated with Spark Concepts. This is the first time we have seen this hardware or the schematics.

lagnat, 2022-08-07:
@bdring Well understood and thank you!

MitchBradley, 2022-08-07:
I received the xpro yesterday and an looking into the SPI issues surrounding daisy-chained SPI for the driver chips.

bdring, 2024-12-11:
Closing very old issues

closed 2024-12-11 by bdring