bdring/FluidNC #486: In M3 laser mode (constant power), G0 moves don't clear laser PWM

Avataar120, 2022-06-11:
### Controller Board

Personnal board
4 DRV8825 driven through I2S_Stream
Laser PWM pin
Enable Laser pin (reversed because of K40 power supply)

### Help From Board Vendor

- [ ] Yes
- [ ] No
- [X] Not Applicable

### Machine Description

Stock K40
And test on table with oscilloscope

### Configuration file

```yaml
name: K40_AvaShield
board: Ava_Shield_7.3

kinematics:
  Cartesian:

stepping:
  engine: I2S_STREAM
  idle_ms: 255
  dir_delay_us: 1
  pulse_us: 4
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN

  x:
    steps_per_mm: 78.74
    max_rate_mm_per_min: 20000.000
    acceleration_mm_per_sec2: 2000.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 5.000
      feed_mm_per_min: 200.000
      seek_mm_per_min: 2500.000
      settle_ms: 250.000
      seek_scaler: 1.500
      feed_scaler: 5.000

    motor0:
      limit_neg_pin: gpio.39
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 5.000
      stepstick:
        ms3_pin: I2SO.3
        step_pin: I2SO.2
        direction_pin: I2SO.1
        disable_pin: I2SO.0

  y:
    steps_per_mm: 78.74
    max_rate_mm_per_min: 20000.000
    acceleration_mm_per_sec2: 2000.000
    max_travel_mm: 200.000
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 5.000
      feed_mm_per_min: 200.000
      seek_mm_per_min: 2500.000
      settle_ms: 250.000
      seek_scaler: 1.500
      feed_scaler: 5.000

    motor0:
      limit_neg_pin: gpio.34
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 5.000
      stepstick:
        ms3_pin: I2SO.7
        step_pin: I2SO.6
        direction_pin: I2SO.5
        disable_pin: I2SO.4

  z:
    steps_per_mm: 1000.000
    max_rate_mm_per_min: 100.000
    acceleration_mm_per_sec2: 500.000
    max_travel_mm: 20.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: true
      mpos_mm: 5.000
      feed_mm_per_min: 200.000
      seek_mm_per_min: 2500.000
      settle_ms: 250.000
      seek_scaler: 1.500
      feed_scaler: 5.000

    motor0:
      limit_neg_pin: gpio.35:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 5.000
      stepstick:
        ms3_pin: I2SO.11
        step_pin: I2SO.10
        direction_pin: I2SO.9
        disable_pin: I2SO.8
    motor1:
      null_motor:

i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5

control:
  safety_door_pin: gpio.27:low:pu
  reset_pin: gpio.12:low:pu
  feed_hold_pin: gpio.14:low:pu
  cycle_start_pin: gpio.13:low:pu

coolant:
  mist_pin: gpio.4
  delay_ms: 50.000

probe:
  pin: gpio.36:low
  check_mode_start: false

macros:
  startup_line0: 
  startup_line1: 
  macro0: 
  macro1: 
  macro2: 
  macro3: 

start:
  must_home: true
  check_limits: true
  deactivate_parking: false

user_outputs:

laser:
  tool_num: 0
  speed_map: 0=0.0% 1000=100.0%
  output_pin: gpio.15
  enable_pin: gpio.16:low
  disable_with_s0: false (tried true)
  s0_with_disable: false (tried true)
  pwm_hz: 15000

arc_tolerance_mm: 0.002
junction_deviation_mm: 0.010
verbose_errors: false
report_inches: false
enable_parking_override_control: false
use_line_numbers: false
```

### Startup Messages

```text
$ss
[MSG:INFO: FluidNC v3.0.x (noGit)]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is SPIFFS]
[MSG:INFO: Configuration file:config.yaml]
[MSG:DBG: Running after-parse tasks]
[MSG:DBG: Checking configuration]
[MSG:INFO: Machine K40_AvaShield]
[MSG:INFO: Board Ava_Shield_7.3]
[MSG:INFO: I2SO BCK:gpio.22 WS:gpio.17 DATA:gpio.21]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN]
[MSG:INFO: Stepping:I2S_stream Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (5.000,305.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.2 Dir:I2SO.1 Disable:I2SO.0]
[MSG:INFO:     Neg Limit gpio.39]
[MSG:INFO: Axis Y (5.000,205.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.6 Dir:I2SO.5:low Disable:I2SO.4]
[MSG:INFO:     Neg Limit gpio.34]
[MSG:INFO: Axis Z (-15.000,5.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.10 Dir:I2SO.9:low Disable:I2SO.8]
[MSG:INFO:     Neg Limit gpio.35:low]
[MSG:INFO:   Motor1]
[MSG:INFO: Door gpio.27:low:pu]
[MSG:INFO: Reset gpio.12:low:pu]
[MSG:ERR: Reset pin is active at startup]
[MSG:INFO: FeedHold gpio.14:low:pu]
[MSG:INFO: CycleStart gpio.13:low:pu]
[MSG:ERR: CycleStart pin is active at startup]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Laser Spindle Ena:gpio.16:low Out:gpio.15 Freq:15000Hz Res:12bits Laser mode:On]
[MSG:INFO: Using spindle Laser]
[MSG:INFO: Mist coolant gpio.4]
[MSG:INFO: Probe Pin: gpio.36:low]
[MSG:INFO: STA SSID is not set]
[MSG:INFO: AP SSID FluidNC IP 192.168.0.1 mask 255.255.255.0 channel 1]
[MSG:INFO: AP started]
[MSG:INFO: WiFi on]
[MSG:INFO: Captive Portal Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
```

### User Interface Software

Lightburn / GCode / Console

### What happened?

GCODE is the following : 

G00 G17 G40 G21 G54
G91
M4
M7
G0 X0Y0                                   ;Move OK - LASER OFF
M3                                            ;Constant power mode
G1 X28S550F720                      ;Move OK - LASER ON - PWM and laser enable pin ok on oscilloscope
                                                 ;At the end of movement, PWM stays at same value and enable is still ON (osscilloscope)
G0 X9Y10                                 ;Move OK - LASER ON -----> I'm expecting LASER OFF

If I suppress M3 command (but I need it), LASER is turned OFF after G1 command

Avataar120, 2022-06-11:
I implemented a dirty solution that is working fro me in GCode.cpp

    //If laser mode and motion is seek (fast movement) -> clear PWM
    if ( spindle->isRateAdjusted() &&  gc_block.modal.motion == Motion::Seek ) {
        pl_data->spindle_speed = 0;
        log_debug ("Clear spindle PWM");
    }

    // [8. Coolant control ]:

MitchBradley, 2022-06-11:
M3 turns on the laser regardless of motion.  That is how it is supposed to work, as defined by [the original GRBL 1.1 spec](https://github.com/gnea/grbl/blob/master/doc/markdown/laser_mode.md) :

[image]

Avataar120, 2022-06-11:
[image]

Avataar120, 2022-06-11:
Note that GRBL ESP32 was working properly, meaning that in laser mode, when issuing G0 command, PWM was put at 0.

Avataar120, 2022-06-11:
[image]

MitchBradley, 2022-06-11:
I will look into this when I have time.

commit 61572869 (2022-06-24): Fix486 (#504)

* Fix #486 laser on in M3 and G0 modes.

* Fix switch to G0 in M3 mode

This version works with both "G0" and "G0 X0".
closed 2022-06-24 by bdring