bdring/FluidNC #491: Problem: Can't connect to firmware on MKS-TinyBee V1.0

wollo-fpv, 2022-06-15:
### Controller Board

MKS-TinyBee V1.0

### Help From Board Vendor

- [ ] Yes
- [X] No
- [ ] Not Applicable

### Machine Description

Busy to setup a 4-axis hot-wire foam cnc

### Configuration file

```yaml
net yet
```

### Startup Messages

```text
'utf-8' codec can't decode byte 0xa0 in position 3: invalid start byte
```

### What happened?

Hello,
Can’t connect to to fluidnc devices over usb

I have 2 devices 
esp32 d1 r32 
and 
MKS-TinyBee V1.0
FluidNc firmaware: fluidnc-v3.4.7
Linux and Windows

For both devices I followed the same steps.

I have executed in the folder those command

fluidnc-v3.4.7-posix /
				   ./erase.sh
				   ./install-fs.sh
				   ./install-wifi.sh
	
MKS-TinyBee V1.0:
I try to connect to this device using
	bCNC (window and linux)
	UGC (windows and linux)
	
I can’t connect
Get this error
'utf-8' codec can't decode byte 0xa0 in position 3: invalid start byte
The software is opening the com port but is not connect to the firmware.
Picture: mks

With serial monitor in Arduino IDE, as well with fluidterm I can connect to the firmware 

Esp32 d1 r32:
I try to connect to this device using
	bCNC (window and linux)
	UGC (windows and linux) 

I have no problems to connect to the firmware 
Picture: esp32_D1_R32

What I need to do….?
What is the different between those 2 devices….?
How to fix the problem on the MKS-TinyBee V1.0

Kr,
Wolf
[image]
[image]

wollo-fpv, 2022-06-15:
Hello,
Here the startup

--- Available ports:
---  1: /dev/ttyS0           'n/a'
---  2: /dev/ttyUSB0         'USB Serial'
--- Enter port index or full name: 2
--- Fluidterm on /dev/ttyUSB0  115200,8,N,1 ---
--- Quit: Ctrl+] or Ctrl+Q | Upload: Ctrl+U | Reset: Ctrl+R ---
Resetting MCU
eׁJ�
   8�&ʶ��:���7C�aH�.���Š        �]�R��%�EU�I��t'�ŢB▒A�_TMT�T�*M�_�ꊕj
��K�g�Zi bZA�W����k�V���A0)}d��x�b�}�r����0��A_������0,
                                                       �d�W���bw�E��i0��C���e'T� ��k"�v�L!��+�i��ff����ɕn��L´C!�d��Q0��°b╹:�&ʲC!�d��Q0°b╹:��ҍ��
                                                                                                                                               �y��Q0꥓H�
[MSG:INFO: FluidNC v3.4.7]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is SPIFFS]
[MSG:INFO: Configuration file:MKS_TinyBee_1_XYZAB.yaml]
[MSG:ERR: Configuration parse error on line 276: Keys must be followed by ':']
[MSG:INFO: Using default configuration]

bdring, 2022-06-15:
You stated "Not Yet" for the config file? Why?

Send **$LocalFS/Show=MKS_TinyBee_1_XYZAB.yaml** to see the contents of that file.

It could have been corrupted.

wollo-fpv, 2022-06-15:
I have uploaded the new config file

--- Available ports:
---  1: /dev/ttyS0           'n/a'
---  2: /dev/ttyUSB1         'USB Serial'
--- Enter port index or full name: 2
--- Fluidterm on /dev/ttyUSB1  115200,8,N,1 ---
--- Quit: Ctrl+] or Ctrl+Q | Upload: Ctrl+U | Reset: Ctrl+R ---
Resetting MCU
eׁ�]
   8 
     &��00'&Ҫ7��HH��i0xLB��WŪ��MET�I��t:��b (��}�AS�e1AS�%=�T)C��nf�֥�: � ���]�:���5
c�5�Ec�v:���)}�rv'��0,�E��:0��0��.����:0�0�
                                           }�rv'�A0,A}��v���0
[��:DɊ▒���k �Z��
늅�:0�&����0,���11�SH�녑:0�:��80�bb���12�LfH���d:��0�0��bb���b30��H����y ��0 �0�YSH�
[MSG:INFO: FluidNC v3.4.7]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is SPIFFS]
[MSG:INFO: Configuration file:config.yaml]
[MSG:INFO: Machine K40 Laser]
[MSG:INFO: Board MKS TinyBee V1.0_001]
[MSG:INFO: I2SO BCK:gpio.25 WS:gpio.26 DATA:gpio.27]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN]
[MSG:INFO: Stepping:I2S_static Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:0ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (0.000,325.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.1 Dir:I2SO.2 Disable:I2SO.0]
[MSG:INFO:     Neg Limit gpio.33]
[MSG:INFO: Axis Y (0.000,220.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.4 Dir:I2SO.5 Disable:I2SO.3]
[MSG:INFO:     Neg Limit gpio.32]
[MSG:INFO: Axis Z (0.000,80.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.7 Dir:I2SO.8 Disable:I2SO.6]
[MSG:INFO:     Neg Limit gpio.22]
[MSG:INFO: Macro 0 gpio.4:low:pu]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Laser Spindle Ena:I2SO.16 Out:gpio.13 Freq:5000Hz Res:13bits Laser mode:On]
[MSG:INFO: Using spindle Laser]
[MSG:INFO: Connecting to STA SSID:CNC_DEV]
[MSG:INFO: Connecting.]
[MSG:INFO: Connected - IP is 192.168.1.121]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl 3.4 [FluidNC v3.4.7 (wifi) '$' for help]

Alle errors are gone... :) !
But I cant connect with bCNC over USB
Still have the same error.

'utf-8' codec can't decode byte 0xd5 in position 4: invalid continuation byte

Kr

Wolf

MitchBradley, 2022-06-15:
I have a guess about what could be happening.  I am going to explain this in more detail than is necessary to just fix your problem, just because I like to explain stuff :-)

Normally when you start an ESP32, you see some messages like this 
```
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:12812
load:0x40080400,len:3032
entry 0x400805e4

[MSG:INFO: FluidNC v3.4.7]
```
The messages before the first `[MSG:INFO: ...` line are from the "bootloader" component of the core ESP32 software.  When the `[MSG:INFO: ...` lines start, that shows that the FluidNC software is now in control.

In your system, instead of the bootload messages, you are seeing garbage characters like```
eׁJ�
8�&ʶ��:���7C�aH�.���Š �]�R��%�EU�I��t'�ŢB▒A�_TMT�T�*M�_�ꊕj
```
That usually means that the bootloader is sending its messages at a different baud rate.  FluidNC uses 115200 baud.  The bootloader is often configured to also use 115200 baud, but that is not always the case.  I suspect that your bootloader is sending at a different baud, so when the receiving program (FluidTerm, bCNC, UGS, or whatever) sees those startup messages, it interprets the characters incorrectly.  That would explain the message
```
'utf-8' codec can't decode byte 0xa0 in position 3: invalid start byte
```
utf-8 is a modern way to express international characters using the restricted set of 256 different character values that can be sent over a serial line.  It expresses the English language alphabet with single characters according to the old "ASCII" standard, and for characters in other languages it uses sequences or two or more characters according to a very clever set of rules.  The garbled (misinterpreted) character values from the bootload baud rate mismatch are triggering the utf-8 decoder software in bCNC. but the sequence does not follow the rules (it is not really utf-8, it is garbage that just happens to look like it might possibly be part of a utf-8 sequence).

So, what do we do about this?  The solution will probably involve changing the bootloader baud rate.  I will have to do some investigation to learn the best way to do so.  I will report back.

bdring, 2022-06-15:
You could try using standard serial terminal and try different baud rates. If you see the standard ESP32 boot text, that will confirm the problem. 

Keep in mind that anything different from 115200 will garble the FluidNC response.

MitchBradley, 2022-06-15:
You might even have to use the non-standard baud rate of 74880.  There is some history going back to ESP8266 of the boot messages coming out at that rate.  It has to do with a bad choice of crystal oscillator frequency that Espressif chose for ESP8266.  You won't find that baud rate in baud rate selection menus, but some terminal programs let you enter the rate manually.  One example of a program that does is Tera Term Pro for Windows.

wollo-fpv, 2022-06-15:
Hello,
@9600 baud:
⸮jB⸮⸮}3G;;&A3⸮⸮⸮⸮⸮.L⸮⸮⸮⸮⸮⸮⸮⸮⸮N⸮⸮⸮sb⸮
19:27:54.373 -> |⸮|⸮r⸮NN⸮⸮1&)⸮,⸮⸮b⸮⸮⸮)$F)g(mN⸮)⸮

@19200 baud;
⸮jB⸮⸮}3G;;&A3⸮⸮⸮⸮⸮.L⸮⸮⸮⸮⸮⸮⸮⸮⸮N⸮⸮⸮sb⸮
19:27:54.373 -> |⸮|⸮r⸮NN⸮⸮1&)⸮,⸮⸮b⸮⸮⸮)$F)g(mN⸮)⸮

@38400
W⸮;⸮⸮Gq⸮/sW⸮⸮⸮w⸮W⸮⸮⸮⸮82⸮⸮đs⸮⸮⸮W⸮Uu:sW⸮E⸮⸮ȳ⸮(⸮u⸮⸮⸮⸮:x⸮⸮⸮!⸮p⸮⸮j⸮⸮⸮⸮#⸮⸮v#]⸮⸮⸮⸮⸮⸮w5⸮⸮⸮*6QQ⸮⸮⸮ɀU⸮⸮0ҁA⸮*⸮⸮⸮⸮*⸮0
⸮\⸮N⸮⸮}⸮⸮⸮؊⸮A⸮⸮A ⸮⸮⸮⸮⸮C⸮⸮⸮
⸮W⸮q;j⸮⸮⸮W⸮2⸮⸮⸮@⸮y\ep⸮q⸮W⸮1⸮⸮⸮ސ⸮S⸮u⸮r⸮⸮d⸮⸮g+⸮u51חwP}sW⸮⸮(s⸮D⸮ssW⸮WAW⸮9⸮⸮\⸮u7⸮"1eA?⸮1QW⸮1sW⸮r⸮r⸮Gp{IGssW⸮⸮ݕ3⸮⸮!::ye⸮(⸮⸮ ;⸮yg⸮Wy⸮1⸮p;⸮ws⸮⸮⸮b⸮⸮W⸮к]w⸮s`@⸮r⸮⸮`1⸮⸮P9⸮⸮us⸮@⸮⸮⸮sW݆⸮⸮⸮⸮⸮⸮0sW⸮ܝ]w⸮s

@57600
W⸮;⸮⸮Gq⸮/sW⸮⸮⸮w⸮W⸮⸮⸮⸮82⸮⸮đs⸮⸮⸮W⸮Uu:sW⸮E⸮⸮ȳ⸮(⸮u⸮⸮⸮⸮:x⸮⸮⸮!⸮p⸮⸮j⸮⸮⸮⸮#⸮⸮v#]⸮⸮⸮⸮⸮⸮w5⸮⸮⸮*6QQ⸮⸮⸮ɀU⸮⸮0ҁA⸮*⸮⸮⸮⸮*⸮0
⸮\⸮N⸮⸮}⸮⸮⸮؊⸮A⸮⸮A ⸮⸮⸮⸮⸮C⸮⸮⸮
⸮W⸮q;j⸮⸮⸮W⸮2⸮⸮⸮@⸮y\ep⸮q⸮W⸮1⸮⸮⸮ސ⸮S⸮u⸮r⸮⸮d⸮⸮g+⸮u51חwP}sW⸮⸮(s⸮D⸮ssW⸮WAW⸮9⸮⸮\⸮u7⸮"1eA?⸮1QW⸮1sW⸮r⸮r⸮Gp{IGssW⸮⸮ݕ3⸮⸮!::ye⸮(⸮⸮ ;⸮yg⸮Wy⸮1⸮p;⸮ws⸮⸮⸮b⸮⸮W⸮к]w⸮s`@⸮r⸮⸮`1⸮⸮P9⸮⸮us⸮@⸮⸮⸮sW݆⸮⸮⸮⸮⸮⸮0sW⸮ܝ]w⸮s

@74880 baud:
⸮
HHHHJ)Ȍ⸮!⸮⸮⸮⸮

@115200 baud:
et⸮⸮R⸮n⸮⸮2⸮L⸮⸮Ҳ&gӓH⸮
19:31:35.253 -> r⸮⸮x⸮B⸮=]ER=⸮}RE⸮U⸮I⸮⸮t⸮⸮Eb⸮iU}F⸮*⸮%1⸮S%⸮%⸮==⸮)⸮⸮⸮nV⸮⸮i⸮bS⸮R⸮⸮⸮
⸮,k⸮E⸮⸮:0⸮⸮)⸮d.W⸮x⸮b⸮}⸮r⸮'⸮⸮⸮⸮i⸮0⸮V⸮:⸮⸮0,}⸮r⸮⸮⸮⸮0,⸮}⸮r⸮'⸮⸮0
⸮[⸮⸮⸮D⸮⸮⸮⸮⸮c-"⸮v⸮j
19:31:35.253 -> ⸮+⸮:⸮⸮ff⸮⸮⸮0⸮⸮⸮:LL⸮
!⸮+⸮i0⸮⸮7⸮⸮⸮,+⸮i1&⸮ҍ!⸮+⸮i0Ѱ4⸮b╹:⸮⸮⸮*⸮Y⸮⸮⸮A0Ѱ5⸮⸮H⸮
19:31:35.717 -> [MSG:INFO: FluidNC v3.4.7]
19:31:35.717 -> [MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
19:31:35.750 -> [MSG:INFO: Local filesystem type is SPIFFS]
19:31:35.750 -> [MSG:INFO: Configuration file:config.yaml]
19:31:35.750 -> [MSG:INFO: Machine K40 Laser]
19:31:35.750 -> [MSG:INFO: Board MKS TinyBee V1.0_001]
19:31:35.783 -> [MSG:INFO: I2SO BCK:gpio.25 WS:gpio.26 DATA:gpio.27]
19:31:35.783 -> [MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
19:31:35.783 -> [MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN]
19:31:35.783 -> [MSG:INFO: Stepping:I2S_static Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:0ms]
19:31:35.783 -> [MSG:INFO: Axis count 3]
19:31:35.783 -> [MSG:INFO: Axis X (0.000,325.000)]
19:31:35.783 -> [MSG:INFO:   Motor0]
19:31:35.783 -> [MSG:INFO:     stepstick Step:I2SO.1 Dir:I2SO.2 Disable:I2SO.0]
19:31:35.816 -> [MSG:INFO:     Neg Limit gpio.33]
19:31:35.816 -> [MSG:INFO: Axis Y (0.000,220.000)]
19:31:35.816 -> [MSG:INFO:   Motor0]
19:31:35.816 -> [MSG:INFO:     stepstick Step:I2SO.4 Dir:I2SO.5 Disable:I2SO.3]
19:31:35.816 -> [MSG:INFO:     Neg Limit gpio.32]
19:31:35.816 -> [MSG:INFO: Axis Z (0.000,80.000)]
19:31:35.816 -> [MSG:INFO:   Motor0]
19:31:35.816 -> [MSG:INFO:     stepstick Step:I2SO.7 Dir:I2SO.8 Disable:I2SO.6]
19:31:35.816 -> [MSG:INFO:     Neg Limit gpio.22]
19:31:35.816 -> [MSG:INFO: Macro 0 gpio.4:low:pu]
19:31:35.850 -> [MSG:INFO: Kinematic system: Cartesian]
19:31:35.850 -> [MSG:INFO: Laser Spindle Ena:I2SO.16 Out:gpio.13 Freq:5000Hz Res:13bits Laser mode:On]
19:31:35.850 -> [MSG:INFO: Using spindle Laser]
19:31:35.949 -> [MSG:INFO: Connecting to STA SSID:CNC_DEV]
19:31:35.949 -> [MSG:INFO: Connecting.]
19:31:37.936 -> [MSG:INFO: Connected - IP is 192.168.1.121]
19:31:37.936 -> [MSG:INFO: WiFi on]
19:31:37.969 -> [MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
19:31:37.969 -> [MSG:INFO: SSDP Started]
19:31:37.969 -> [MSG:INFO: HTTP started on port 80]
19:31:37.969 -> [MSG:INFO: Telnet started on port 23]
19:31:37.969 -> 
19:31:37.969 -> Grbl 3.4 [FluidNC v3.4.7 (wifi) '$' for help]

@230400 baud:
19:32:18.887 -> &>⸮f~⸮x

MitchBradley, 2022-06-15:
The ESP32 uploader defaults to 460800 baud.  Maybe you could try that too.

I am asking for help on the MKS discord.

wollo-fpv, 2022-06-15:
I will , and want to try all...
But how and where I need to change the baud rates?
And next to that, how to re-flash the bootloader ...
With the arduino uno and mega I knwo how to it.
But on esp32 is new for me.
Kr,

Wolf

MitchBradley, 2022-06-15:
I am searching for a solution.  The earliest `ets` messages may come from a component so deep inside the chip that it cannot be reflashed.  We have never encountered this problem before on an ESP32 (it was common on ESP8266), so we do not know the answer yet.

MitchBradley, 2022-06-15:
Please keep in mind that we do not have a Tinybee and we have no relationship with the company that sells it.  It uses a version of the ESP32 module that we do not use in any of the hardware we have developed.

wollo-fpv, 2022-06-15:
Yes,
I know, 
I'm already very happy that you want to help me... !

MitchBradley, 2022-06-15:
Also please note that the problem you are facing is not directly caused by FluidNC, but rather is an interaction between third party hardware, Espressif firmware outside of FluidNC, and third party senders.  None of those are under our control.  We are trying to help, but we may be limited in what we can do, since the problems are in system components that we do not "own".

MitchBradley, 2022-06-15:
I have something you could try.  If we cannot fix the boot message baud rate, maybe we can make them go away entirely.

If you pull down the GPIO15 line on an ESP32, it is supposed to suppress the boot messages.  On the MKS Tinybee board, GPIO15 is the LCD_D6 signal.  It is pin 23 on the ESP32 module and pin 2 on U10.  If you were to pull it down with a 10K resistor to gnd, the problem messages might go away.  Unfortunately the pins are small so careful soldering will be necessary.

Here is a [link to the Tinybee schematic](https://github.com/makerbase-mks/MKS-TinyBee/blob/main/hardware/MKS%20TinyBee%20V1.0_003/MKS%20TinyBee%20V1.0_003%20SCH.pdf), for reference.

wollo-fpv, 2022-06-15:
I will give that a try.....
First have to search resistor.... 

Because the tinybee is a printer-controller 
I have flashed marlin from the mks-side.
https://github.com/makerbase-mks/MKS-TinyBee

And as well, the first boot massage from board is this:

20:37:43.094 -> et.RչA ⸮±⸮0⸮⸮⸮5⸮H⸮⸮⸮⸮:⸮⸮ ⸮⸮⸮E⸮⸮⸮%UM⸮T⸮I⸮o⸮⸮⸮⸮b ⸮*⸮}⸮A⸮⸮e1⸮S⸮%⸮O⸮JH(⸮n⸮Z֥⸮:⸮bS⸮⸮⸮⸮x⸮⸮H⸮⸮d⸮W⸮x⸮b⸮}⸮r⸮⸮⸮⸮0⸮⸮E⸮⸮:0⸮⸮i⸮0⸮v⸮⸮A0⸮⸮_⸮.⸮⸮⸮⸮⸮i⸮_⸮⸮⸮⸮⸮⸮⸮C⸮⸮e⸮⸮⸮c⸮k⸮ V⸮⸮C!⸮d⸮⸮m⸮A0⸮b╹:⸮C!⸮d⸮⸮Mf⸮⸮0⸮,⸮k⸮:⸮⸮⸮
!⸮+⸮i0Ѱ⸮°⸮bl⸮K¸NSȬ+⸮:⸮⸮0⸮⸮⸮⸮bb⸮⸮:⸮⸮⸮
⸮⸮⸮⸮⸮ ⸮⸮0⸮⸮⸮a⸮C⸮start
20:37:43.590 ->  External Reset
20:37:43.590 ->  Brown out Reset
20:37:43.590 ->  Watchdog Reset
20:37:43.590 -> Marlin 2.0.9.2
20:37:43.590 -> echo: Last Updated: 2021-09-03 | Author: (none, default config)
20:37:43.590 -> echo: Compiled: Jun 15 2022
20:37:43.590 -> echo: Free Memory: 251696  PlannerBufferBytes: 1408
20:37:43.789 -> echo:AP started 
20:37:43.888 -> echo:192.168.0.1
20:37:43.988 -> echo:OTA service started
20:37:44.021 -> echo:HTTP server started
20:37:44.352 -> echo:V85 stored settings retrieved (633 bytes; crc 19339)
20:37:48.330 -> echo:192.168.0.1
20:38:13.723 -> et.Rչ ⸮2⸮⸮ 2⸮⸮2&⸮7C⸮⸮rs⸮xLBPjUI⸮N%UM⸮⸮JI⸮⸮⸮'⸮⸮b⸮i⸮%_⸮P*⸮e1A⸮⸮%=O⸮⸮H⸮⸮˥g⸮Z⸮b⸮ӕ⸮⸮'Ս⸮⸮⸮E⸮⸮'⸮⸮0,\E⸮⸮:⸮A⸮0I⸮d.⸮⸮⸮⸮0,⸮A_⸮:⸮A0⸮⸮_⸮⸮:⸮A0i⸮_⸮⸮:⸮⸮0⸮a⸮e⸮⸮ ⸮⸮k⸮A⸮Z⸮⸮j⸮⸮+⸮i⸮⸮f⸮±b⸮⸮n⸮⸮⸮H⸮+⸮⸮0⸮⸮⸮⸮⸮⸮L,⸮⸮⸮:⸮⸮!⸮d⸮⸮Ѱ⸮°⸮⸮⸮YK⸮⸮9⸮H⸮⸮a⸮x⸮8⸮⸮0ɢ⸮n⸮⸮⸮
⸮⸮K⸮y⸮⸮⸮00⸮⸮a⸮H⸮start
20:38:14.186 ->  External Reset
20:38:14.186 ->  Brown out Reset
20:38:14.220 ->  Watchdog Reset
20:38:14.220 -> Marlin 2.0.9.2
20:38:14.220 -> echo: Last Updated: 2021-09-03 | Author: (none, default config)
20:38:14.220 -> echo: Compiled: Jun 15 2022
20:38:14.220 -> echo: Free Memory: 251696  PlannerBufferBytes: 1408
20:38:14.386 -> echo:AP started 
20:38:14.485 -> echo:192.168.0.1
20:38:14.618 -> echo:OTA service started
20:38:14.618 -> echo:HTTP server started
20:38:14.983 -> echo:V85 stored settings retrieved (633 bytes; crc 19339)
20:38:18.960 -> echo:192.168.0.1

MitchBradley, 2022-06-15:
Since you do not know what a resistor is, it is unlikely that you have the skill and soldering tools to connect one to the tiny pins on the board.  Soldering to fine-pitch components requires good tools and a lot of practice.

The options that I see right now are:
1. Find someone with electronics skills to do the mod for you.
2. Buy a different board
3. Convince the authors of bCNC or UGS to add code to ignore garbled startup messages
4. Use a different sender that does not get confused. There are many different senders out there - WebUI, CNCjs, Candle, Lightburn, Commander, and many others.  I expect that some of them would ignore the bad characters.

It is possible that I will get an answer from the MKS Discord, but I am not hopeful.

MitchBradley, 2022-06-15:
[image]
The red arrows show the two possibilities for where you could connect one end of the resistor.  The other end of the resistor should be connected to GND.  The green arrows show some of the many possibilities for nearby GND connections.

DirtyEngineer, 2022-06-15:
I'll have a look at this what I get home later tonight as I have this board running FluidNC.  Also GPIO15 is conveniently next to ground on the EXP1 header so a couple female jumper wires and a resistor would be easy to add.

[image]

MitchBradley, 2022-06-15:
I asked another FluidNC/Tinybee user on our Discord if he has similar problems.  He said:

"nope, startup messages and talking to it via fluidterm, serial port, webUI terminal all worked flawlessly from the first time I flashed it"

Another user said:

"I don't remember if there was any barbled bootloader messages but I also didn't have any problem with fluidterm, the serial port or webUI working. I have not used bCNC"

So I am beginning to wonder if you have a bad Tinybee board.  That said, I cannot think of a hardware flaw that would make the startup messages come out at the wrong baud but the FluidNC messages would be correct.

MitchBradley, 2022-06-15:
"Also GPIO15 is conveniently next to ground on the EXP1 header"

Alas, it is not necessarily that easy.  According to the schematic, GPIO15 from the ESP32 goes through a level translator chip (U10) before being presented to the EXP1 header.  It is conceivable that pulling it down on the EXP1 side would reflect back to the ESP32 side, but it is hard to be sure.  The schematic does not give the part number for the level translator so I can't tell how it works inside.

MitchBradley, 2022-06-15:
I can't find a BOM for the design on the MKS github repo.  I can't read the markings on U10 on the image.  Going by the pin labels on U10, it looks like a tri-stateable buffer chip instead of a bidirectional level translator.  If so, a resistor on the EXP1 side would have not be seen by the ESP32 pin.

DirtyEngineer, 2022-06-15:
Hmm, I seem to remember the level shifter for the display is actually on the display board, not the main board but it's very possible that I may be confusing this with the DLC32 board.  I will verify tonight.

MitchBradley, 2022-06-15:
Look at the schematic (link above).  
[image]

wollo-fpv, 2022-06-15:
OMG,
Thank you so much for all your research and time.
I will order a new board.
I hope that will work...!
Best regards, and one more time thank you

Wolf

Skorpi08, 2022-06-15:
How long is your USB Cable?
Maybe try another USB Port.

DirtyEngineer, 2022-06-15:
My FluidNC TinyBee board has similar garbled text with a board reset.

115200 baud NL & CL

⸮ketsRչ⸮ ⸮016 ⸮:22:⸮SH⸮
r.⸮⸮⸮⸮ (⸮]I=9}RESETJI⸮⸮⸮⸮0x13 (S⸮}MQ}FLASH%==Q⸮
confZ֥⸮⸮ b SPIץ⸮⸮5
clk_⸮⸮⸮0x⸮bq_dr⸮⸮⸮⸮0⸮}⸮⸮⸮⸮0x00,c.z⸮⸮⸮⸮0x00,h⸮E⸮⸮⸮0⸮0,wpߖ⸮⸮0x⸮
modY'T⸮	⸮ock dZ⸮⸮j
l+⸮⸮0x3⸮⸮⸮⸮⸮b╹⸮C!⸮+⸮⸮0x3ff⸮1c,lYK⸮⸮⸮⸮j
load⸮⸮⸮007⸮⸮0,leK⸮⸮⸮⸮⸮j
load'⸮⸮00⸮⸮00,lYK⸮ª⸮j
entr^⸮⸮⸮06a8
⸮
[MSG:INFO: FluidNC v3.4.2]
[MSG:INFO: Compiled with ESP32 SDK:v3.3.5-1-g85c43024c]
[MSG:INFO: Configuration file:config.yaml]
[MSG:INFO: Machine PNP]
[MSG:INFO: Board MKS TinyBee]
[MSG:INFO: I2SO BCK:gpio.25 WS:gpio.26 DATA:gpio.27]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN]
[MSG:INFO: Stepping:I2S_static Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:0ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (0.000,440.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.1 Dir:I2SO.2 Disable:I2SO.0]
[MSG:INFO:     Neg Limit gpio.33]
[MSG:INFO: Axis Y (0.000,440.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.4 Dir:I2SO.5 Disable:I2SO.3]
[MSG:INFO:     Neg Limit gpio.32]
[MSG:INFO: Axis Z (0.000,80.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.7 Dir:I2SO.8 Disable:I2SO.6]
[MSG:INFO:     Neg Limit gpio.22]
[MSG:INFO: Macro 0 gpio.4:low:pu]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Using spindle NoSpindle]

MitchBradley, 2022-06-15:
Do you have an oscilloscope?  If you could get scope traces of the first few characters on the TX line, starting with the "ets", I can figure out exactly what is wrong.

Skorpi08, 2022-06-15:
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5856
entry 0x400806a8

[MSG:INFO: FluidNC v3.0.x (noGit)]
[MSG:INFO: Compiled with ESP32 SDK:v3.3.5-1-g85c43024c]
[MSG:INFO: Configuration file:config.yaml]
[MSG:INFO: Machine K40 Laser]
[MSG:INFO: Board MKS-DLC32 V2.1]

DirtyEngineer, 2022-06-15:
> [image] The red arrows show the two possibilities for where you could connect one end of the resistor. The other end of the resistor should be connected to GND. The green arrows show some of the many possibilities for nearby GND connections.

Can confirm a 10k resistor between the module GPIO15 and GND does bypass the bootloader.

I do have a scope and can possibly get a capture of the output of the TX line later tonight.

Skorpi08, 2022-06-15:
> How long is your USB Cable? Maybe try another USB Port. I have no Problems with Lightburn.
> 
> Arduino Serial:
> 
> ⸮ets Jun 8 2016 00:22:57
> 
> ⸮.⸮⸮⸮⸮ (POWERON_RESET),boot:0x1,B⸮A%}��MT_FLASH_BOOT) configsip: b�⸮A%]P:0xee clk_drv:0x00,q_drv:0x00⸮}⸮⸮⸮⸮0x00,cs0_drv:0x00,hd_drv:0x0b⸮⸮}⸮⸮v:0x00 mode:DIO, clock dZ⸮⸮j load:0x3fff0018,len:4 load:0x3⸮⸮�⸮⸮�⸮len:1044 load:0x40078000,leK⸮⸮⸮⸮⸮j load:0x40080400,len:5856 entry⸮⸮⸮00806a8
> 
> [MSG:INFO: FluidNC v3.0.x (noGit)] [MSG:INFO: Compiled with ESP32 SDK:v3.3.5-1-g85c43024c]

it was power over USB

This is with 12V powersupply:

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5856
entry 0x400806a8

[MSG:INFO: FluidNC v3.0.x (noGit)]
[MSG:INFO: Compiled with ESP32 SDK:v3.3.5-1-g85c43024c]

DirtyEngineer, 2022-06-15:
Can also confirm that a 10k between the EXP1 pin and ground **does not work** but on the first reset I had a near perfect bootloader read out.  I have not been able to replicate it. Very odd. It is a timing issue as parts/characters of the bootloader message are occasionally discernable.

Skorpi08, 2022-06-15:
> MKS DLC32v2.1 001
> 
> ets Jun 8 2016 00:22:57
> 
> rst:0x1 (POWERON_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT) configsip: 0, SPIWP:0xee clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00 mode:DIO, clock div:1 load:0x3fff0018,len:4 load:0x3fff001c,len:1044 load:0x40078000,len:10124 load:0x40080400,len:5856 entry 0x400806a8
> 
> [MSG:INFO: FluidNC v3.0.x (noGit)] [MSG:INFO: Compiled with ESP32 SDK:v3.3.5-1-g85c43024c] [MSG:INFO: Configuration file:config.yaml] [MSG:INFO: Machine K40 Laser] [MSG:INFO: Board MKS-DLC32 V2.1]

MKS DLC32 can only powered with 12/24V

MitchBradley, 2022-06-15:
Let's hold off on the reports about MKS DLC32 for now, as the problem seems to be specific to MKS Tinybee.

MitchBradley, 2022-06-15:
Even if the problem appears to have gone away, I would still like to see a scope trace if you can get one.  It will let me evaluate the bit timing and the data format.  Maybe the stop bit is too short or something like that.

DirtyEngineer, 2022-06-16:
No I was not clear. The garbled bootloader test issue persists. I had one reboot where it was clear but haven't been able to replicate a clean boot since. Let me work on getting the timing capture.

9600, 2022-06-16:
> 

> @9600 baud:

Hello, you called?

wollo-fpv, 2022-06-16:
Hello,
the usb-cable I use is 30cm long. (I have tested with different cables and usb-ports)

Sorry, no I don't have a oscilloscope.

MitchBradley, 2022-06-21:
@DirtyEngineer I would still be interested in a scope trace of the bad messages.  I do not need the whole thing, just a few characters worth.

MitchBradley, 2022-06-22:
There has been no new information over the past few days.

My attempt to reach out to the MKS developers on Discord failed.  The only response I got was from another user who offered some unhelpful (incorrect) guesses.

It is clear to me that this problem is not related to FluidNC itself - there is nothing that we can change in FluidNC that will change the ESP32 ROM startup message corruption.

That leaves some possibilities:
1. Use the hardware method given above to turn off the startup messages entirely.
2. Use a different board, either another Tinybee that does not fail (apparently some do and some don't) or a different kind of board
3. Convince the sender developers to change their code to ignore garbage characters

Since there is nothing more that we can do from the FluidNC side, I am closing this issue.  Personally, I would still be interested in understanding exactly why it is happening, but even with that information, it is unlikely that we would be able to fix it with changes to FluidNC.