bdring/FluidNC #492: Problem: issues running 16 microsteps on FYSETC E4

jamescarruthers, 2022-06-15:
### Controller Board

FYSETC E4

### Help From Board Vendor

- [ ] Yes
- [X] No
- [ ] Not Applicable

### Machine Description

Pen plotter — CoreXY — A1 in size.
https://github.com/jamescarruthers/PlotterXY/tree/main/Hardware

### Configuration file

```yaml
name: "PlotterXY"
board: "FYSETC E4"

stepping:
  engine: RMT
  idle_ms: 255
  dir_delay_us: 0
  pulse_us: 2
  disable_delay_us: 0
  segments: 20

kinematics:
  corexy:

axes:
  shared_stepper_disable_pin: gpio.25:high

  x:
# 16t = 100, 20t = 80 for 16 microsteps – change for y axes also
    steps_per_mm: 20
    max_rate_mm_per_min: 12000
    acceleration_mm_per_sec2: 1200
    max_travel_mm: 860
    soft_limits: true
    homing:
      cycle: 2
      allow_single_axis: true
      mpos_mm: 860
      positive_direction: true
      seek_mm_per_min: 4000
      feed_mm_per_min: 600
      settle_ms: 200
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_pos_pin: gpio.34:high
      hard_limits: true
      pulloff_mm: 10.000
      tmc_2209:
        uart:
          txd_pin: gpio.22
          rxd_pin: gpio.21
          baud: 115200
          mode: 8N1
        addr: 1
        r_sense_ohms: 0.110
        run_amps: 1.200
        hold_amps: 0.800
        microsteps: 4
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: StealthChop
        homing_mode: StealthChop
        use_enable: true
        step_pin: gpio.27
        direction_pin: gpio.26:high
        disable_pin: NO_PIN

  y:
    steps_per_mm: 20
    max_rate_mm_per_min: 12000
    acceleration_mm_per_sec2: 1200
    max_travel_mm: 600
    soft_limits: true
    homing:
      cycle: 1
      allow_single_axis: true
      mpos_mm: 600
      positive_direction: true
      seek_mm_per_min: 4000
      feed_mm_per_min: 600
      settle_ms: 200
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_pos_pin: gpio.35:high
      hard_limits: true
      pulloff_mm: 10.000
      tmc_2209:
        addr: 3
        r_sense_ohms: 0.110
        run_amps: 1.200
        hold_amps: 0.800
        microsteps: 4
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: StealthChop
        homing_mode: StealthChop
        use_enable: true
        step_pin: gpio.33
        direction_pin: gpio.32:high
        disable_pin: NO_PIN

  z:
    steps_per_mm: 80
    max_rate_mm_per_min: 1000
    acceleration_mm_per_sec2: 1000
    max_travel_mm: 5
    
    motor0:
      solenoid:
        output_pin: gpio.2
        pwm_hz: 30000
        off_percent: 0.000
        pull_percent: 100.000
        hold_percent: 15.000
        pull_ms: 25
        direction_invert: false

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5
  card_detect_pin: NO_PIN

macros:
  startup_line0:
  startup_line1:
  macro0:
  macro1:
  macro2:
  macro3:

start:
  must_home: true
```

### Startup Messages

```text
[MSG:INFO: FluidNC v3.4.7]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is SPIFFS]
[MSG:INFO: Configuration file:config.yaml]
[MSG:INFO: Machine PlotterXY]
[MSG:INFO: Board FYSETC E4]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN]
[MSG:INFO: Stepping:RMT Pulse:2us Dsbl Delay:0us Dir Delay:0us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable gpio.25]
[MSG:INFO: Axis X (0.000,860.000)]
[MSG:INFO:   Motor0]
[MSG:INFO: Trinamic Stepper Uart Tx:gpio.22 Rx:gpio.21 RTS:NO_PIN Baud:115200]
[MSG:INFO:     tmc_2209 Step:gpio.27 Dir:gpio.26 Disable:NO_PIN Addr:1 R:0.110]
[MSG:INFO:     Pos Limit gpio.34]
[MSG:INFO: Axis Y (0.000,600.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2209 Step:gpio.33 Dir:gpio.32 Disable:NO_PIN Addr:3 R:0.110]
[MSG:INFO:     Pos Limit gpio.35]
[MSG:INFO: Axis Z (-5.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     solenoid Pin: gpio.2 Off: 0.000 Hold: 15.000 Pull:100.000 Duration:25 pwm hz:30000]
[MSG:INFO: X Axis driver test passed]
[MSG:INFO: Y Axis driver test passed]
[MSG:INFO: Kinematic system: CoreXY]
[MSG:INFO: Using spindle NoSpindle]
[MSG:INFO: Connecting to STA SSID:CatFi]
[MSG:INFO: Connecting.]
[MSG:INFO: Connected - IP is 192.168.0.129]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
```

### User Interface Software

WebUI

### What happened?

If I issue the commands G00 X0 Y0 and then G00 X0 Y500 - instead of the plotter going from 0,0 to 0,500 it goes to something more like 200,400. It moves diagonally. I can "fix" this by changing the microsteps from 16 to 4.

If I use another board, Bart's CNC Dev v3.5 running TMC2209 or DRV8825 — there are no issues using 16 microsteps. The stepper drivers are set to anything from 1A to 1.5A. I originally thought it was a current issue.

If I go back to my original GRBL ESP32 install running on the FYSETC E4 I don't have any issues.

### Other Information

The GRBL ESP32 config I'm using is here; https://github.com/jamescarruthers/PlotterXY/tree/main/Firmware/GRBL-ESP32%20(Depreciated)

jamescarruthers, 2022-06-15:
Microsteps are the same on both axis.

I guess it's true CoreXY. It has the same belt layout as https://corexy.com/

bdring, 2022-06-15:
Check if one motor is not reacting to the change. Do a move and count the revs on each motor. Change the microsteps and repeat.

MitchBradley, 2022-07-03:
Is there any new information on this?  Please try [v3.5.0-pre5](https://github.com/bdring/FluidNC/releases/tag/v3.5.0-pre5) which has several important fixes since v3.4.7