bdring/FluidNC #493: FluidNC 3.4.7 fails to init motors. Hardware configuration works fine in ESP32_Grbl

bDuthieDev, 2022-06-15:
### Controller Board

6-pack V1p5 6/2020

### Help From Board Vendor

- [ ] Yes
- [ ] No
- [X] Not Applicable

### Machine Description

XYYZA configuration with no motor connected to A at this time

### Configuration file

```yaml
name: 6-pack 2130sg XYYZA 4xIn 4x5v 10vCNC (v6)
board: 6-pack

kinematics:
  Cartesian:

stepping:
  engine: I2S_static
  idle_ms: 255
  dir_delay_us: 0
  pulse_us: 4
  disable_delay_us: 0

axes:
  x:
    steps_per_mm: 100.000
    max_rate_mm_per_min: 1000.000
    acceleration_mm_per_sec2: 200.000
    max_travel_mm: 600.000
    soft_limits: true
    homing:
      cycle: 2
      mpos_mm: 0.000
      positive_direction: true
      settle_ms: 250.000
      seek_mm_per_min: 2000.000
      feed_mm_per_min: 200.000
      seek_scaler: 1.100
      feed_scaler: 5.000
    motor0:
      limit_all_pin: gpio.33:low
      hard_limits: false
      tmc_2130:
        direction_pin: i2so.1
        step_pin: i2so.2
        disable_pin: i2so.0
        run_amps: 1.250
        hold_amps: 0.250
        microsteps: 16
        stallguard: 16
        stallguard_debug: false
        run_mode: CoolStep
        homing_mode: Stallguard
        use_enable: false
        r_sense_ohms: 0.110
        cs_pin: i2so.3:low
    motor1:
      null_motor:
  y:
    steps_per_mm: 100.000
    max_rate_mm_per_min: 1000.000
    acceleration_mm_per_sec2: 200.000
    max_travel_mm: 640.000
[... 501 lines cut ...]
$Z/Microsteps=8
$A/Microsteps=16
$B/Microsteps=16
$C/Microsteps=16
$X/StallGuard=4
$Y/StallGuard=4
$Z/StallGuard=4
$A/StallGuard=16
$B/StallGuard=16
$C/StallGuard=16
$Errors/Verbose=Off
```
```

### User Interface Software

Using WebUI and FluidTerm .

### What happened?

Updated to FluidNC from GRBL_Esp32. Created config.yaml use the tool, then edited to conform to machine settings.. The motors init correctly in GRBL_Esp32, and the machine runs correctly. The motors fail to init in FluidNC. 

 Seems as if I must have created some error in config.yaml. I just can't find it myself.

This is with the downloaded version of FluidNC 3.4.7.

MitchBradley, 2022-06-15:
Try with version 3.4.4.  At version 3.4.5 we switched to the latest version of Arduino framework code, which has some changes in the way that GPIOs are handled.  It is possible that those changes are breaking something in the Trinamic setup code.

bdring, 2022-06-15:
I don't think the CS pins for the motors should have the **:low** attribute.

There are a lot of Trinamic SPI examples for the 6 pack here.

https://github.com/bdring/6-Pack_CNC_Controller/tree/main/FluidNC_configs

also see 

https://github.com/bdring/FluidNC/wiki/FluidNC-Motor-Setup#tmc_2130

bDuthieDev, 2022-06-15:
tried 3.4.6 - no luck
reloaded 3.4.7, removed :low on cs pins - motors now init, and are live
some issue with soft limits cause errors on X,Y. Z does not have the issue
I should be able to solve that myself, but it is over 100 in the workshop, so will chase it later

bdring, 2022-06-15:
Closing....subject issue was resolved.

closed 2022-06-15 by bdring