bdring/FluidNC #494: Problem: After soft limit triggered, RESET via USB does not work

Avataar120, 2022-06-16:
### Controller Board

4 pack perso design

### Help From Board Vendor

- [ ] Yes
- [ ] No
- [X] Not Applicable

### Machine Description

4 stepper drivers - DRV8825
Driven through I2S - 74HC595
Board design guidelines followed during personal design - especially I have no problem to download the ESP32 when it's mounted on my board (either for firmware, either for filesystem (with PlatformIO or FLuidterm)

### Configuration file

```yaml
name: K40_AvaShield
board: Ava_Shield_7.3

kinematics:
  Cartesian:

stepping:
  engine: I2S_STREAM
  idle_ms: 255
  dir_delay_us: 1
  pulse_us: 4
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN

  x:
    steps_per_mm: 78.74
    max_rate_mm_per_min: 20000.000
    acceleration_mm_per_sec2: 2000.000
    max_travel_mm: 300.000
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 5.000
      feed_mm_per_min: 200.000
      seek_mm_per_min: 2500.000
      settle_ms: 250.000
      seek_scaler: 1.500
      feed_scaler: 5.000

    motor0:
      limit_neg_pin: gpio.39
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 5.000
      stepstick:
        ms3_pin: I2SO.3
        step_pin: I2SO.2
        direction_pin: I2SO.1
        disable_pin: I2SO.0

  y:
    steps_per_mm: 78.74
    max_rate_mm_per_min: 20000.000
    acceleration_mm_per_sec2: 2000.000
    max_travel_mm: 200.000
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 5.000
      feed_mm_per_min: 200.000
      seek_mm_per_min: 2500.000
      settle_ms: 250.000
      seek_scaler: 1.500
      feed_scaler: 5.000

    motor0:
      limit_neg_pin: gpio.34
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 5.000
      stepstick:
        ms3_pin: I2SO.7
        step_pin: I2SO.6
        direction_pin: I2SO.5:low
        disable_pin: I2SO.4

  z:
    steps_per_mm: 1000.000
    max_rate_mm_per_min: 100.000
    acceleration_mm_per_sec2: 500.000
    max_travel_mm: 20.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: true
      mpos_mm: 5.000
      feed_mm_per_min: 200.000
      seek_mm_per_min: 2500.000
      settle_ms: 250.000
      seek_scaler: 1.500
      feed_scaler: 5.000

    motor0:
      limit_neg_pin: gpio.35:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 5.000
      stepstick:
        ms3_pin: I2SO.11
        step_pin: I2SO.10
        direction_pin: I2SO.9:low
        disable_pin: I2SO.8
    motor1:
      null_motor:

i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5

control:
  safety_door_pin: gpio.27:low:pu
  reset_pin: gpio.12:low:pu
  feed_hold_pin: gpio.14:low:pu
  cycle_start_pin: gpio.13:low:pu

coolant:
  mist_pin: gpio.4
  delay_ms: 50.000

probe:
  pin: gpio.36:low
  check_mode_start: false

macros:
  startup_line0: 
  startup_line1: 
  macro0: 
  macro1: 
  macro2: 
  macro3: 

start:
  must_home: true
  check_limits: true
  deactivate_parking: false

user_outputs:

laser:
  tool_num: 0
  speed_map: 0=0.0% 1000=100.0%
  output_pin: gpio.15
  enable_pin: gpio.16:low
  disable_with_s0: false
  s0_with_disable: false
  pwm_hz: 15000

arc_tolerance_mm: 0.002
junction_deviation_mm: 0.010
verbose_errors: false
report_inches: false
enable_parking_override_control: false
use_line_numbers: false
```

### Startup Messages

```text
[MSG:INFO: FluidNC v3.0.x (noGit)]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is SPIFFS]
[MSG:INFO: Configuration file:config.yaml]
[MSG:INFO: Machine K40_AvaShield]
[MSG:INFO: Board Ava_Shield_7.3]
[MSG:INFO: I2SO BCK:gpio.22 WS:gpio.17 DATA:gpio.21]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN]
[MSG:INFO: Stepping:I2S_stream Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (5.000,305.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.2 Dir:I2SO.1 Disable:I2SO.0]
[MSG:INFO:     Neg Limit gpio.39]
[MSG:INFO: Axis Y (5.000,205.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.6 Dir:I2SO.5:low Disable:I2SO.4]
[MSG:INFO:     Neg Limit gpio.34]
[MSG:INFO: Axis Z (-15.000,5.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.10 Dir:I2SO.9:low Disable:I2SO.8]
[MSG:INFO:     Neg Limit gpio.35:low]
[MSG:INFO:   Motor1]
[MSG:INFO: Door gpio.27:low:pu]
[MSG:INFO: Reset gpio.12:low:pu]
[MSG:ERR: Reset pin is active at startup]
[MSG:INFO: FeedHold gpio.14:low:pu]
[MSG:INFO: CycleStart gpio.13:low:pu]
[MSG:ERR: CycleStart pin is active at startup]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Laser Spindle Ena:gpio.16:low Out:gpio.15 Freq:15000Hz Res:12bits Laser mode:On]
[MSG:INFO: Using spindle Laser]
[MSG:INFO: Mist coolant gpio.4]
[MSG:INFO: Probe Pin: gpio.36:low]
[MSG:INFO: Connecting to STA SSID:Avataar120_Orly]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connecting...]
[MSG:INFO: Connecting....]
[MSG:INFO: Connecting.]
[MSG:INFO: Connected - IP is 192.168.1.79]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
```

### User Interface Software

Lightburn or Fluidterm

### What happened?

X soft limit is [5..305]
I home the machine
And request a manuel movement to X=500
Soft limit is triggered as expected

I get this error message as expected : 
[MSG:INFO: Soft limit on X target:500.000]
[MSG:INFO: Soft limit on Y target:0.000]
ALARM:2
G-code motion target exceeds machine travel. Machine position safely retained. Alarm may be unlocked. (Right-click the 'Devices' button to reset the connection)
On or near line 2:
Stream completed in 0:00
[MSG:INFO: Reset to continue]

But when I trigger SW reset (Right click on DEVICE Button in Lightburn or unselect COM et reselect it again), I get  Waiting for connection... message and computer is not able to connect to ESP32.
I need to trigger an HW reset (button on ESP32 or remove USB cable) to recover the situation

When not in alarm mode, SW reset is working well and ESP32 is rebooting correctly
Message in the console in this use case is the following : 

Waiting for connection...
[VER:3.4 FluidNC v3.4.7 (main-062d169e-avataar120-Release):]
[OPT:MPHS]
[MSG: Machine: K40_AvaShield]
[MSG: No WifiNo Wifi]
ok

bdring, 2022-06-16:
The ESP32 typically does not reset by cycling the COM port. We want to be able to connect to a running machine without forcing a reboot.

Typically alarms are cleared with $X and not a reset. Alarms like soft limits are not critical and can be recovered without a reset.

Avataar120, 2022-06-16:
I would be very happy :)

But it's not working.
When soft limit is triggered, message in console is clearly requesting RESET
And if I try to enter any command, there is no echo or reaction from ESP32

MitchBradley, 2022-06-16:
The GRBL-standard way to reset after a limit - either hard or soft - is to send a CTRL-X reset.  In FluidTerm, you do that by typing CTRL-X directly.  In Lightburn, you hit the Stop button 
[image]

After that, you can send $X to cancel the alarm.

We are considering applying your patch, but it does introduce an incompatibility with standard GRBL so we are not sure whether it will cause problems in the future and with other senders.

Avataar120, 2022-06-16:
Thx, I didn't know that STOP button sends CTRL-X :)

bdring, 2022-06-18:
I tried the patch and to not think it is the ultimate solution. I added an item to our development tracker to take a fresh look at how soft limit alarms are handled. Until then, we should keep what we have. Closing this issue.

- [Discord discussion](https://discord.com/channels/780079161460916227/805613138916802570/987101649510215740)
- [Development tracker](https://github.com/bdring/FluidNC/projects/1#card-83327545)

closed 2022-06-18 by bdring