bdring/FluidNC #498: Problem: Machine position seems not precise with my setup and leads to soft limit error when job is finished

Avataar120, 2022-06-18:
### Controller Board

Perso 4 pack board

### Help From Board Vendor

- [ ] Yes
- [ ] No
- [X] Not Applicable

### Machine Description

K40 with MXL belt & pulleys.
Steps / mm = 78,74 and not 80 like if I were using BT2
Pull offs are 5mm on X & Y axis
mpos are 4mm on X & Y axis

### Configuration file

```yaml
name: K40_AvaShield
board: Ava_Shield_7.3

kinematics:
  Cartesian:

stepping:
  engine: I2S_STREAM
  idle_ms: 255
  dir_delay_us: 1
  pulse_us: 4
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN

  x:
    steps_per_mm: 78.74
    max_rate_mm_per_min: 20000.000
    acceleration_mm_per_sec2: 2000.000
    max_travel_mm: 310.000
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 5.000
      feed_mm_per_min: 200.000
      seek_mm_per_min: 2500.000
      settle_ms: 250.000
      seek_scaler: 1.500
      feed_scaler: 5.000

    motor0:
      limit_neg_pin: gpio.39
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 5.000
      stepstick:
        ms3_pin: I2SO.3
        step_pin: I2SO.2
        direction_pin: I2SO.1
        disable_pin: I2SO.0

  y:
    steps_per_mm: 78.74
    max_rate_mm_per_min: 20000.000
    acceleration_mm_per_sec2: 2000.000
    max_travel_mm: 210.000
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 5.000
      feed_mm_per_min: 200.000
      seek_mm_per_min: 2500.000
      settle_ms: 250.000
      seek_scaler: 1.500
      feed_scaler: 5.000

    motor0:
      limit_neg_pin: gpio.34
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 5.000
      stepstick:
        ms3_pin: I2SO.7
        step_pin: I2SO.6
        direction_pin: I2SO.5:low
        disable_pin: I2SO.4

  z:
    steps_per_mm: 1000.000
    max_rate_mm_per_min: 100.000
    acceleration_mm_per_sec2: 500.000
    max_travel_mm: 20.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: true
      mpos_mm: 9.500
      feed_mm_per_min: 200.000
      seek_mm_per_min: 2500.000
      settle_ms: 250.000
      seek_scaler: 1.500
      feed_scaler: 5.000

    motor0:
      limit_neg_pin: gpio.35:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 9.500
      stepstick:
        ms3_pin: I2SO.11
        step_pin: I2SO.10
        direction_pin: I2SO.9:low
        disable_pin: I2SO.8
    motor1:
      null_motor:

i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5

control:
  safety_door_pin: gpio.27:low:pu
  reset_pin: gpio.12:low:pu
  feed_hold_pin: gpio.14:low:pu
  cycle_start_pin: gpio.13:low:pu

coolant:
  mist_pin: gpio.4
  delay_ms: 50.000

probe:
  pin: gpio.36:low
  check_mode_start: false

macros:
  startup_line0: 
  startup_line1: 
  macro0: 
  macro1: 
  macro2: 
  macro3: 

start:
  must_home: true
  check_limits: true
  deactivate_parking: false

user_outputs:

laser:
  tool_num: 0
  speed_map: 0=0.0% 1000=100.0%
  output_pin: gpio.15
  enable_pin: gpio.16:low
  disable_with_s0: false
  s0_with_disable: false
  pwm_hz: 15000

arc_tolerance_mm: 0.002
junction_deviation_mm: 0.010
verbose_errors: false
report_inches: false
enable_parking_override_control: false
use_line_numbers: false
```

### Startup Messages

```text
No issue
```

### User Interface Software

Lightburn

### What happened?

After homing, I'm expecting to get X=5 & Y=5 when issuing ? command
But I get X=3.988 & Y=3.988 instead

When my job is finishing, lightburn calls G0 X0 Y0 to come back to home position.
Soft limit is triggered with message : 

[MSG:INFO: Soft limit on X target:3.988]
[MSG:INFO: Soft limit on Y target:3.988]

Don't know if it's coming from my weird steps / mm and calculations coming with.

Avataar120, 2022-06-18:
Note : it was the same issue with GRBL ESP32 (but I was not using soft limits)

MitchBradley, 2022-06-18:
It is looking like a rounding error.  78.74 * 5 = 393.7 .  393 / 78.74 = 4.991

MitchBradley, 2022-06-18:
Where does the X=3.998 come from?  On Discord you said 4.991

MitchBradley, 2022-06-18:
The soft limit is cause by the combination of positive_direction: false and mpos_mm: 5.  With positive direction false, 5 is the low limit, so the travel envelope is 5 .. 315 .  0 is outside that range.

Avataar120, 2022-06-18:
> Where does the X=3.998 come from? On Discord you said 4.991

That's true but I tried in the meantime to modify mpos to 4 instead of 5 -> that's why I gave all consistent figures in the issue

Avataar120, 2022-06-18:
> The soft limit is cause by the combination of positive_direction: false and mpos_mm: 5. With positive direction false, 5 is the low limit, so the travel envelope is 5 .. 315 . 0 is outside that range.

G0 X0Y0 is trying to move to 3.988, 3.988. If it was X=4 & Y=4 soft limit won't be triggered no?

Avataar120, 2022-06-18:
I modified info message to give more information in the console.

When issuing G0 X0Y0 I get : 

[MSG:INFO: Soft limit on X target:3.988 Limits:4.000..314.000]
[MSG:INFO: Soft limit on Y target:3.988 Limits:4.000..214.000]

MitchBradley, 2022-06-18:
You keep saying 4, but the config file above says 5.

Avataar120, 2022-06-18:
My mistake ... I'm working on 2 computers ... I update config file. It's 4

MitchBradley, 2022-06-18:
Even after I fix the rounding problem, there will still be a slight offset because internal positions are stored in integer steps.

Assume that mpos_mm = 4 and steps_per_mm = 78.74.  Thus 4 mm is 314.96 steps, but we cannot store 314.96 as an integer, so we round it up to 315.  Then if you convert that back to mm you get 4.001 .

MitchBradley, 2022-06-18:
The easiest way to deal with this problem is not to work right up to the limit.  Set your work zero a few millimeters inside the limit envelope.

Avataar120, 2022-06-18:
I've found a dirty patch that works well for me.
Issue was more to inform you of this rounding problem

Avataar120, 2022-06-18:
Thanks for analysis

MitchBradley, 2022-06-18:
Oh, by the way, G0 X0 Y0 tells us nothing about whether it is inside the limits .  Limits are in machine coordinates, while G0 is in work coordinates unless you say G53 .

Avataar120, 2022-06-18:
I'm not at all an expert in GCODE for machining as you can see :)

What I can say is that my GCODE is like that : 
G00 G17 G40 G21 G54 G90

Then blabla

Then G0 X0Y0

As machine position is X=3.988 Y=3.998 and work position is 0 0 after homing, I assume G0 X0 Y0 will come back to work position 0 0 so meaning machine position 3.988 3.988 which triggers the alarm

Any way, thanks. I took your fix + my workaround and it's working for me.

MitchBradley, 2022-06-19:
Reads about work coordinate systems. You should set the work offset explicitly instead of relying on a default value

MitchBradley, 2022-06-21:
I think that all the topics raised in this issue have been addresses.  File a new issue if not.

closed 2022-06-21 by MitchBradley