bdring/FluidNC #506: Problem: uploading a file to the SD card through FluidTerm on a Win 10 PC

crcayer, 2022-06-23:
### Controller Board

FYSETC E4

### Help From Board Vendor

- [ ] Yes
- [ ] No
- [X] Not Applicable

### Machine Description

Pen plotter with dual steppers with CoreXY config and 3.4.9 release of FluidNC

### Configuration file

```yaml
name: "PlotterXY"
board: "FYSETC E4"

stepping:
  engine: RMT
  idle_ms: 255
  dir_delay_us: 0
  pulse_us: 2
  disable_delay_us: 0
  segments: 20

kinematics:
  corexy:

axes:

  x:
# 16t = 100, 20t = 80 for 16 microsteps – change for y axes also
    steps_per_mm: 20
    max_rate_mm_per_min: 64000
    acceleration_mm_per_sec2: 2400
    max_travel_mm: 280
    soft_limits: true
    homing:
      cycle: 2
      allow_single_axis: true
      mpos_mm: 280
      positive_direction: true
      seek_mm_per_min: 4000
      feed_mm_per_min: 600
      settle_ms: 200
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_pos_pin: gpio.34:high
      hard_limits: true
      pulloff_mm: 10.000
      tmc_2209:
        uart:
          txd_pin: gpio.22
          rxd_pin: gpio.21
          baud: 115200
          mode: 8N1
        addr: 1
        r_sense_ohms: 0.110
        run_amps: 1.200
        hold_amps: 0.800
        microsteps: 4
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: StealthChop
        homing_mode: StealthChop
        use_enable: true
        step_pin: gpio.27
        direction_pin: gpio.26:high
        disable_pin: NO_PIN

  y:
    steps_per_mm: 20
    max_rate_mm_per_min: 64000
    acceleration_mm_per_sec2: 2400
    max_travel_mm: 340
    soft_limits: true
    homing:
      cycle: 1
      allow_single_axis: true
      mpos_mm: 340
      positive_direction: true
      seek_mm_per_min: 4000
      feed_mm_per_min: 600
      settle_ms: 200
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_pos_pin: gpio.35:high
      hard_limits: true
      pulloff_mm: 10.000
      tmc_2209:
        addr: 3
        r_sense_ohms: 0.110
        run_amps: 1.200
        hold_amps: 0.800
        microsteps: 4
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: StealthChop
        homing_mode: StealthChop
        use_enable: true
        step_pin: gpio.33
        direction_pin: gpio.32:high
        disable_pin: NO_PIN

  z:
    steps_per_mm: 80
    max_rate_mm_per_min: 1000
    acceleration_mm_per_sec2: 1000
    max_travel_mm: 5
    
    motor0:
      solenoid:
        output_pin: gpio.2
        pwm_hz: 30000
        off_percent: 0.000
        pull_percent: 100.000
        hold_percent: 20.000
        pull_ms: 25
        direction_invert: false

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5
  card_detect_pin: NO_PIN

coolant:
  mist_pin: gpio.13
  delay: 0

macros:
  startup_line0:
  startup_line1:
  macro0:
  macro1:
  macro2:
  macro3:

start:
  must_home: true
```

### Startup Messages

```text
[MSG:INFO: FluidNC v3.4.9]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is SPIFFS]
[MSG:INFO: Configuration file:config.yaml]
[MSG:WARN: Ignored key delay]
[MSG:INFO: Machine PlotterXY]
[MSG:INFO: Board FYSETC E4]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN]
[MSG:INFO: Stepping:RMT Pulse:2us Dsbl Delay:0us Dir Delay:0us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (0.000,280.000)]
[MSG:INFO:   Motor0]
[MSG:INFO: Trinamic Stepper Uart Tx:gpio.22 Rx:gpio.21 RTS:NO_PIN Baud:115200]
[MSG:INFO:     tmc_2209 Step:gpio.27 Dir:gpio.26 Disable:NO_PIN Addr:1 R:0.110]
[MSG:INFO:     Pos Limit gpio.34]
[MSG:INFO: Axis Y (0.000,340.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2209 Step:gpio.33 Dir:gpio.32 Disable:NO_PIN Addr:3 R:0.110]
[MSG:INFO:     Pos Limit gpio.35]
[MSG:INFO: Axis Z (-5.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     solenoid Pin: gpio.2 Off: 0.000 Hold: 20.000 Pull:100.000 Duration:25 pwm hz:30000]
[MSG:INFO: X Axis driver test passed]
[MSG:INFO: Y Axis driver test passed]
[MSG:INFO: Kinematic system: CoreXY]
[MSG:INFO: Using spindle NoSpindle]
[MSG:INFO: Mist coolant gpio.13]
[MSG:INFO: Connecting to STA SSID:Trilights]
[MSG:INFO: Connecting.]
[MSG:INFO: Connected - IP is 192.168.1.179]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
```

### User Interface Software

FluidTerm

### What happened?

I'm running the 3.49 release and can upload files through the web interface to the SD card.  The wiki shows that a file can also be uploaded via fluidterm by prefixing the filename with /sd

**Adding or Uploading Files to SD Card

From a terminal program such as [fluidterm](https://github.com/bdring/FluidNC/wiki/fluidterm-usage) prefix the local name with /sd and it will upload to the SD card. Sending /sd/Foo.nc will upload Foo.nc to the root directory.**

I'm connected to the controller via a USB cable, and can access the other features of FluidTerm with no issues, but when I tried to upload a local file from my D: drive using the wiki method, I get the following results:

`ok
/sdd:\test.gcode
error:1
/sd/d:\test.gcode
error:1
/sd/d:/test.gcode
error:1`

I don't believe this is an error, just a lack of understanding on my part how to properly format the command.  Yes, the SD card works, I can upload files from the web interface.  Also, the file I referenced also exists on my system.

If I were to use the example from the wiki, does the file foo.nc need to be in the same folder as the FluidTerm executable?
Can I use standard DOS-style formatting for drive and folder names if my file is in a folder on another drive?  Is there some front-slash / back-slash path issue I'm unaware of?  Thanks!

MitchBradley, 2022-06-23:
Did you type CTRL-U to start the upload?  What is supposed to happen is that, after you type CTRL-U, a file selector appears that lets you choose a file on the local Windows system.  After you choose one, a prompt appears in FluidTerm where you can type a different name like for example "/sd/foo.bar".  If you do not add the /sd/ prefix, the name is on the SPIFFS filesystem.  If you just hit enter it uses the same name as you selected, and goes on spiffs.

However, I just discovered a bug in FluidTerm.  In some cases, it does not wait for you to enter a remote name, it just acts like you typed enter, even though you didn't.  It only happens if there is more than one COM port so you have to select one.  In the usual case (on my system at least) where there is only one COM port that is suitable for FluidNC, the bug does not show up.

I issued a PR #507 to fix the FluidNC bug.  You can grab the fluidterm.exe from that PR to see if it solves your problem.

MitchBradley, 2022-07-03:
No updates lately; I think that the problem has been resolved.  Reopen if new information is available.

closed 2022-07-03 by MitchBradley