bdring/FluidNC #510: FluidNC & MKS DLC32 / MPCNC v1.2.2

IgorPack, 2022-06-24:
### Machine Context

Milling machine: X/Y1,Y2/Z, T8, NEMA23 1.2H.M 2.8A, TMS2209 v1.2

### Feature Description

Hi,
Since I can't create messages on Discord, I want to ask here. I want to try to use MKS DLC32 or MPCNC Controller v1.2.2 together with FluidNC for a milling machine (X/Y1,Y2/Z, T8, NEMA23 1.2H.M 2.8A, TMS2209 v1.2). If someone can share an example config.yaml for this or a similar machine, as well as recommendations for setting it up, I would be very grateful for the help. 
Thanks.

### Other Approaches

Haven't tried yet

### How I Can Help

I can help you with testing.

MitchBradley, 2022-06-24:
There are several example config files for both boards in https://github.com/bdring/FluidNC/tree/main/example_configs

There are also quite a lot of auto-translated configs for MPCNC -  https://github.com/bdring/Grbl_Esp32/search?q=MPCNC&type=issues .

Please search before asking questions.  GitHub search in the FluidNC and Grbl_Esp32 repositories turns up many hits for MPCNC and DLC32.  You can search the code, issues, wiki, ...

[image]

closed 2022-06-24 by MitchBradley

IgorPack, 2022-06-24:
Thank you so much for your comment, Mitch. I saw these files, I just wanted to save myself time and get the opinion of those who use them. Good luck.