bdring/FluidNC #513: ESP32 Crashes while homing an axis from ESP32 interface in AP mode

UmerSoftwares, 2022-06-28:
### Controller Board

I designed the board myself. I am pasting the config.yaml file so that developers can see what pins I have used.

### Help From Board Vendor

- [ ] Yes
- [ ] No
- [X] Not Applicable

### Machine Description

It is a 3 axis CNC machine. It has 

- 3 Steppers
- 3 Limit switches
- SDCARD
- 1 pwm output

### Configuration file

```yaml
name: "ESP32 Dev Controller"
board: "ESP32 Dev Controller"

stepping:
  engine: RMT
  idle_ms: 255
  dir_delay_us: 1
  pulse_us: 2
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: gpio.32:high
  
  x:
    steps_per_mm: 1600
    max_rate_mm_per_min: 300
    acceleration_mm_per_sec2: 1000
    max_travel_mm: 300
    homing:
      cycle: 2
      mpos_mm: 0
      positive_direction: false
      seek_mm_per_min: 300
      feed_mm_per_min: 100
    
    motor0:
      pulloff_mm: 3
      limit_all_pin: gpio.13:low:pu
      stepstick:
        direction_pin: gpio.25:high
        step_pin: gpio.33
    motor1:
      null_motor:

  y:
    steps_per_mm: 1600
    max_rate_mm_per_min: 300
    acceleration_mm_per_sec2: 1000
    max_travel_mm: 300
    homing:
      cycle: 2
      mpos_mm: 0
      positive_direction: false
      seek_mm_per_min: 300
      feed_mm_per_min: 100

    motor0:
      pulloff_mm: 3
      limit_all_pin: gpio.14:low:pu
      stepstick:
        direction_pin: gpio.22:low
        step_pin: gpio.21
    motor1:
      null_motor:

  z:
    steps_per_mm: 1600
    max_rate_mm_per_min: 300
    acceleration_mm_per_sec2: 1000
    max_travel_mm: 67
    homing:
      cycle: 1
      mpos_mm: 67
      positive_direction: true
      seek_mm_per_min: 300
      feed_mm_per_min: 100

    motor0:
      pulloff_mm: 3
      limit_all_pin: gpio.27:low:pu
      stepstick:
        direction_pin: gpio.15:low
        step_pin: gpio.4
    motor1:
      null_motor:

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5
  card_detect_pin: NO_PIN

PWM:
  pwm_hz: 5000
  direction_pin: NO_PIN
  output_pin: gpio.26
  enable_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 0
  spindown_ms: 0
  tool_num: 0
  speed_map: 0=0% 10000=100%
```

### Startup Messages

```text
[MSG:INFO: FluidNC v3.4.7]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is SPIFFS]
[MSG:INFO: Configuration file:config.yaml]
[MSG:INFO: Machine ESP32 Dev Controller]
[MSG:INFO: Board ESP32 Dev Controller]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN]
[MSG:INFO: Stepping:RMT Pulse:2us Dsbl Delay:0us Dir Delay:1us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable gpio.32]
[MSG:INFO: Axis X (0.000,300.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.33 Dir:gpio.25 Disable:NO_PIN]
[MSG:INFO:     All Limit gpio.13:low:pu]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Y (0.000,300.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.21 Dir:gpio.22:low Disable:NO_PIN]
[MSG:INFO:     All Limit gpio.14:low:pu]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Z (0.000,67.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.4 Dir:gpio.15:low Disable:NO_PIN]
[MSG:INFO:     All Limit gpio.27:low:pu]
[MSG:INFO:   Motor1]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: PWM Spindle Ena:NO_PIN Out:gpio.26 Dir:NO_PIN Freq:5000Hz Res:13bits]
[MSG:INFO: Using spindle PWM]
[MSG:INFO: Connecting to STA SSID:Tplink 325a]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: No SSID]
[MSG:INFO: AP SSID FluidNC IP 192.168.0.1 mask 255.255.255.0 channel 1]
[MSG:INFO: AP started]
[MSG:INFO: WiFi on]
[MSG:INFO: Captive Portal Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl 3.4 [FluidNC v3.4.7 (wifi) '$' for help]
[MSG:INFO: '$H'|'$X' to unlock]
```

### User Interface Software

ESP3D Web UI

### What happened?

When FLUID NC is in AP mode, I connect to it via hotspot and then home an axis. Sometimes, it will home correctly but in most of the attempts, ESP32 crashes with the following message on the serial.

```
Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed).

Core  1 register dump:
PC      : 0x400817f7  PS      : 0x00060035  A0      : 0x800823a2  A1      : 0x3ffbf1ec
A2      : 0x00000000  A3      : 0x3ffb41cc  A4      : 0x3ffc30e4  A5      : 0x00000003
A6      : 0x00000003  A7      : 0x00000000  A8      : 0xbad00bad  A9      : 0x3ffbf1cc
A10     : 0x3ffb42c8  A11     : 0x003fffff  A12     : 0x3ffb1bf0  A13     : 0x3ffb1bb0
A14     : 0x3ffc0414  A15     : 0x3ffb2e44  SAR     : 0x00000020  EXCCAUSE: 0x00000007
EXCVADDR: 0x00000000  LBEG    : 0x4008aed4  LEND    : 0x4008aedf  LCOUNT  : 0x00000000

Backtrace:0x400817f4:0x3ffbf1ec |<-CORRUPTED

ELF file SHA256: 0000000000000000

Rebooting...
ets Jun  8 2016 00:22:57

rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:12812
load:0x40080400,len:3032
entry 0x400805e4
```

The issue happens only when it is in AP mode. It does not happen when it is in ST mode. 
The issue happens with all x, y and z axis.
After sending the homing command, the axis moves towards the homing direction, hits the limit switch, retracts and then when it tries to move again to the limit switch, ESP32 crashes.

MitchBradley, 2022-06-28:
Try v3.5.0-pre2

MitchBradley, 2022-06-28:
It turns out that v3.5.0-pre2 is identical to v3.5.0-pre and does not have the fixes that I think will change your problem.  There is a mistake in the prerelease build process that caused it to build from the wrong code revision.

UmerSoftwares, 2022-06-28:
> It turns out that v3.5.0-pre2 is identical to v3.5.0-pre and does not have the fixes that I think will change your problem. There is a mistake in the prerelease build process that caused it to build from the wrong code revision.

Can you please guide me what should I do?

MitchBradley, 2022-06-28:
Wait for me to fix it.

MitchBradley, 2022-06-28:
There is definitely a bug that is causing this crash - but even after I fix the bug, we do not recommend using AP mode for actual work, only for setup.  AP mode makes the ESP32 do a lot of extra work that can cause problems.  If you do not have wifi in your shop, it is best to get a router and set up a wifi network there.  It does not have to be connected to the internet, and the router does not have to be a modern fast one.  Any old wifi router that has been taken out of service because it is too slow for modern work will do.

MitchBradley, 2022-06-28:
Just give me time to work through this, okay?

MitchBradley, 2022-06-28:
Try v3.5.0-pre3

UmerSoftwares, 2022-06-29:
I have tried v3.5.0-pre3 now.

Here is what I get at startup
```
[MSG:INFO: FluidNC test1]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is SPIFFS]
[MSG:INFO: Configuration file:config.yaml]
[MSG:INFO: Machine ESP32 Dev Controller]
[MSG:INFO: Board ESP32 Dev Controller]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN]
[MSG:INFO: Stepping:RMT Pulse:2us Dsbl Delay:0us Dir Delay:1us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable gpio.32]
[MSG:INFO: Axis X (0.000,300.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.33 Dir:gpio.25 Disable:NO_PIN]
[MSG:INFO:     All Limit gpio.13:low:pu]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Y (0.000,300.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.21 Dir:gpio.22:low Disable:NO_PIN]
[MSG:INFO:     All Limit gpio.14:low:pu]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Z (0.000,68.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.4 Dir:gpio.15:low Disable:NO_PIN]
[MSG:INFO:     All Limit gpio.27:low:pu]
[MSG:INFO:   Motor1]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: PWM Spindle Ena:NO_PIN Out:gpio.26 Dir:NO_PIN Freq:5000Hz Res:13bits]
[MSG:INFO: Using spindle PWM]
[MSG:INFO: STA SSID is not set]
[MSG:INFO: AP SSID FluidNC IP 192.168.0.1 mask 255.255.255.0 channel 1]
[MSG:INFO: AP started]
[MSG:INFO: WiFi on]
[MSG:INFO: Captive Portal Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl  [FluidNC test1 (wifi) '$' for help]
[MSG:INFO: '$H'|'$X' to unlock]
```

Now, the axis are not homing. I have tried from fluidterm as well as the WebUI. The axis hits the limit switch and stays there. It doesn't retract to the pulloff distance. 

Here is the error in fluidterm
```
$HX
[MSG:ERR: Homing fail]
ALARM:8
ok
```

Here is the error in WebUI
```
$HX
[MSG:ERR: Homing fail]
ALARM:8
Homing fail. Pull off travel failed to clear limit switch. Try increasing pull-off setting or check wiring.
ok
```

MitchBradley, 2022-06-29:
Good, the crashes are gone.  We know about the homing problem.  It is fixed in a recent patch but we have not yet published a release with the homing fix.  I am cleaning up the code and will release it soon.

MitchBradley, 2022-06-29:
v3.5.0-pre4 is homing for me

closed 2022-07-03 by MitchBradley