bdring/FluidNC #514: Un-triggering hard limit behaves the same as triggering hard limit

daxliniere, 2022-07-02:
### Controller Board

MKS DLC32

### Help From Board Vendor

- [ ] Yes
- [ ] No
- [X] Not Applicable

### Machine Description

CNC mill

### Configuration file

```yaml
board: MKS-DLC32 V2.2
name: SWOLE-CNC
meta: (2202.07.01) Dax Liniere, originally by Skorpi

kinematics:
  Cartesian:

stepping:
  engine: I2S_STATIC
  idle_ms: 0
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0
  
axes:
  shared_stepper_disable_pin: I2SO.0
  x:
    steps_per_mm: 802.341
    max_rate_mm_per_min: 1500.000
    acceleration_mm_per_sec2: 80.000
# was 1500.000
    max_travel_mm: 301.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 20.000
      seek_mm_per_min: 200.000
# was 100 in first SWOLE-CNC iteration running MKS firmware
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.36:high
      limit_pos_pin: gpio.34:high
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 0.500
      stepstick:
        step_pin: I2SO.1
        direction_pin: I2SO.2:low
        disable_pin: NO_PIN
        ms1_pin: NO_PIN
        ms2_pin: NO_PIN
        ms3_pin: NO_PIN
        reset_pin: NO_PIN

  y:
    steps_per_mm: 802.322
    max_rate_mm_per_min: 1500.000
    acceleration_mm_per_sec2: 80.000
# was 300.000
    max_travel_mm: 187.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 20.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.35:high
      limit_pos_pin: gpio.23:high
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 0.500
      stepstick:
        step_pin: I2SO.5
        direction_pin: I2SO.6:low
        disable_pin: NO_PIN
        ms1_pin: NO_PIN
        ms2_pin: NO_PIN
        ms3_pin: NO_PIN
        reset_pin: NO_PIN
  z:
    steps_per_mm: 802.497
    max_rate_mm_per_min: 1000.000
    acceleration_mm_per_sec2: 50.000
# was 500.000
    max_travel_mm: 46.000
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 20.000
      seek_mm_per_min: 100.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.18:high
      hard_limits: true
      pulloff_mm: 0.500
      stepstick:
        step_pin: I2SO.3
        direction_pin: I2SO.4:low
        disable_pin: NO_PIN
        ms1_pin: NO_PIN
        ms2_pin: NO_PIN
        ms3_pin: NO_PIN
        reset_pin: NO_PIN

i2so:
  bck_pin: gpio.16
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.12
  mosi_pin: gpio.13
  sck_pin: gpio.14

sdcard:
  cs_pin: gpio.15
  card_detect_pin: NO_PIN
# This could be GPIO.39, but Card Detect has no supported functions in FluidNC

control:
  safety_door_pin: gpio.33:low:pu
  cycle_start_pin: NO_PIN
  feed_hold_pin: NO_PIN
  reset_pin: NO_PIN
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

macros:
  startup_line0:
  startup_line1:
  macro0: $SD/Run=lasertest.gcode
  macro1:
  macro2:
  macro3:

coolant:
  flood_pin: gpio.0
# continuous air
  mist_pin:  gpio.4
# pulsed air 
  delay_ms: 0

probe:
  pin: gpio.22:low
  check_mode_start: true
  

10V:
# Spindle
  direction_pin: NO_PIN
  forward_pin: gpio.5:low
  reverse_pin: NO_PIN
  output_pin: gpio.32
  enable_pin: NO_PIN
  pwm_hz: 5000
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 10000
  spindown_ms: 12000
  tool_num: 0
  speed_map: 0=0% 0=25% 5868=25% 23237=99% 24000=100%

user_outputs:
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 5000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000
  digital0_pin: NO_PIN
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: NO_PIN

start:
  must_home: false
# Pins that could be used: 0(SDA), 4(SCL), 5, 18, 19, 22, 23, 25, 26, 27, 32 (TTL spindle control), 33, 39, I2SO.7
```

### Startup Messages

```text
[MSG:INFO: FluidNC v3.5.0-pre4]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is SPIFFS]
[MSG:INFO: Configuration file:config.yaml]
[MSG:INFO: Machine SWOLE-CNC]
[MSG:INFO: Board MKS-DLC32 V2.2]
[MSG:INFO: I2SO BCK:gpio.16 WS:gpio.17 DATA:gpio.21]
[MSG:INFO: SPI SCK:gpio.14 MOSI:gpio.13 MISO:gpio.12]
[MSG:INFO: SD Card cs_pin:gpio.15 detect:NO_PIN]
[MSG:INFO: Stepping:I2S_static Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:0ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable I2SO.0]
[MSG:INFO: Axis X (0.000,301.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.1 Dir:I2SO.2:low Disable:NO_PIN]
[MSG:INFO:     Neg Limit gpio.36]
[MSG:INFO:     Pos Limit gpio.34]
[MSG:INFO: Axis Y (0.000,187.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.5 Dir:I2SO.6:low Disable:NO_PIN]
[MSG:INFO:     Neg Limit gpio.35]
[MSG:INFO:     Pos Limit gpio.23]
[MSG:INFO: Axis Z (-46.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.3 Dir:I2SO.4:low Disable:NO_PIN]
[MSG:INFO:     Neg Limit gpio.18]
[MSG:INFO: safety_door_pin gpio.33:low:pu]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: 10V Spindle Ena:NO_PIN Out:gpio.32 Dir:NO_PIN Fwd:gpio.5:low Rev:NO_PIN Freq:5000Hz Res:13bits]
[MSG:INFO: Using spindle 10V]
[MSG:INFO: Flood coolant gpio.0]
[MSG:INFO: Mist coolant gpio.4]
[MSG:INFO: Probe Pin: gpio.22:low]
[MSG:INFO: Connecting to STA SSID:Puzzle Factory]
[MSG:INFO: Connecting.]
[MSG:INFO: Connected - IP is 192.168.2.81]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://swolecnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl 3.5 [FluidNC v3.5.0-pre4 (wifi) '$' for help]
```

### User Interface Software

UGS

### What happened?

1) If a hard limit is triggered, you can reset the controller, then unlock and continue. However, the axis that was triggered will not be allowed to move, including away from that limit, while the limit switch is still triggered.

2) However, when you manually move the axis to un-trigger the limit switch, this causes a second hard limit trigger event, despite the fact it should not.

### Other Information

I have just added separate limit switches of each end of X & Y axes, but this was happening before when both ends were chained in series.

daxliniere, 2022-07-02:
Separately, but related, if you are moving away from a limit switch and you trigger it by hand, it still throws a limit error even though it's not possible for it to have been actually triggered.

bdring, 2022-07-02:
Have you tried homing to clear the limit switch? 

Note: switch bounce could trigger a limit. The values of the r/c filter are not shown on the [schematic](https://github.com/makerbase-mks/MKS-TinyBee/blob/main/hardware/MKS%20TinyBee%20V1.0_003/MKS%20TinyBee%20V1.0_003%20SCH.pdf).

daxliniere, 2022-07-02:
Homing takes ages and when I'm just jogging around setting up a job, that's a huge hassle, of course.

Good point about the switch bounce. Are you saying that FNC should't be triggering a limit event when pulling away from a limit?
Also, wasn't there software debouncing available in GRBL_ESP32?

As for r/c circuit, there isn't one as I'm using an optocoupler board with isolated PSU that I made as I was originally getting false triggers from the stepper motor wiring. Maybe I'm better off to just build another one with r/c filters on the inputs to the optocouplers?

MitchBradley, 2022-07-03:
> re you saying that FNC should't be triggering a limit event when pulling away from a limit?

How does FluidNC know that the intention of a move is to pull away from a limit?  I am trying to think about what to do about this but I don't have a good model for the way that states should be managed.  Usually when I have slammed into a hard limit I turn off the machine to release the steppers and manually move it into safe land.

MitchBradley, 2022-07-03:
> Also, wasn't there software debouncing available in GRBL_ESP32?

Yes but we removed it because people were using it as a poor substitute for fixing switch noise sensitivity at the source - i.e. deglitching and noise suppression instead of debouncing.  In typical usage of limit switches, once you have solved the noise problem, you do not have to worry about bouncing.  Even if the switch does bounce and thus send a rapid fire On-Off-On-Off-On sequence, the subsequent pulses do not matter, because the first one triggers a one-way state change.  The Off transitions do not undo that state change, and the repetitions of the On transition are irrelevant since the state has already changed.

There was a problem, though, with Control pins like Macro pins, where the bouncing could cause multiple invocations.  For control pins, we added back a fixed 10 ms interval during which repetitions do not cause additional events.

MitchBradley, 2022-07-03:
One answer to "when are you pulling away from a limit" is "the limit switch is active and the direction of motion is in the opposite direction".  That works to some extent if the limit pin is at a defined end, but not when you have a "both ends" limit pin, i.e. a pair of wire-or'ed switches driving a single input.  I suppose that you could allow any motion in that case, or perhaps any jog, or perhaps a jog with a small distance like say 10 mm.  I also suspect that adding logic like that would take approximately forever to get right and sort out all the edge conditions and interactions with other stuff.

When you have a defined limit end, you can home a single axis to pull off the switch. This feature already exists.  If you are already at the switch location , the homing cycle does not take long, just a couple of short back and forth cycles of the pulloff distance.

daxliniere, 2022-07-03:
> Note: switch bounce could trigger a limit. The values of the r/c filter are not shown on the schematic

I realise this is outside of the scope of FNC, but do you think the values for the r+c elements in this schematic look right?
And am I correct in guessing that R2 is probably not necessary since R1 has a higher value?
[image]

bdring, 2022-07-03:
Without knowing any details, It seems like the resistance is rather high. You need to consider the values the LED wants.

This is the controller side of the circuit.

[image]

MitchBradley, 2022-07-03:
That circuit will only work at all if the optocoupler has a high CTR.  The current through the LED will be 12V/15K = 0.8mA, and most ordinary optocouplers want at least 5mA through the LED in order to turn on solidly.  Furthermore, since you are running a lower current than normal, your noise immunity will be compromised.  In addition to the galvanic isolation effect on noise reduction, another factor is the lower impedance (higher switching current) which make it harder for noise to turn on the LED.  It is R1 that you do not want, not R2.

Also, you don't want a 4.2ms time constant (100 Hz).  That will make the edge stay in the transition region too long.  0.5 to 1.0 ms is better.

daxliniere, 2022-07-04:
> > re you saying that FNC should't be triggering a limit event when pulling away from a limit?
> 
> How does FluidNC know that the intention of a move is to pull away from a limit? I am trying to think about what to do about this but I don't have a good model for the way that states should be managed.

What I had thought about is that if the setup is running with **limit_neg/pos_pin**, then FNC can know what it needs to. For example, if the limit+ is triggered, then negative motion should be allowed for that axis.

>Usually when I have slammed into a hard limit I turn off the machine to release the steppers and manually move it into safe land.

I 3D printed some knobs to fit onto the nut at the non-motor ends of my lead screws, but I'd suggest that most people won't have this and so don't have a way to move away from the triggered limit.

daxliniere, 2022-07-04:
> > Also, wasn't there software debouncing available in GRBL_ESP32?
> 
>Even if the switch does bounce and thus send a rapid fire On-Off-On-Off-On sequence, the subsequent pulses do not matter, because the first one triggers a one-way state change. The Off transitions do not undo that state change, and the repetitions of the On transition are irrelevant since the state has already changed.

Yes, that makes perfect sense (1-0-1-0-1), but the 0-1-0-1-0 sequence when pulling off will cause a re-trigger.

> Yes but we removed it because people were using it to hide instead of fix switch noise sensitivity

Any chance you would reintroduce it via a setting in config.yaml? Seems like penalising people who are trying to solve a legitimate problem in order to save lazy people.

daxliniere, 2022-07-04:
> ..but not when you have a "both ends" limit pin, i.e. a pair of wire-or'ed switches driving a single input. I suppose that you could allow any motion in that case, or perhaps any jog..

I think it would be safer to keep the current behaviour of not allowing any movement when **limit_all_pin** is used.

> When you are limited at the homing end, you can home a single axis to pull off the switch. This feature already exists. If you are already at the switch location , the homing cycle does not take long, just a couple of short back and forth cycles of the pulloff distance.

That's very true. I might set up macro keys for that in UGS. Thanks for the suggestion.

MitchBradley, 2022-09-26:
Is this issue still a problem?

dylanpil, 2022-10-21:
I'm having a similiar issue as i have just tried to upgrade from v3.4.3 to the latest release v3.6.3
Everything seems to work great except when i hit the endstops.
In the webui, in the command space this is the error
"ALARM:1
Hard limit has been triggered. Machine position is likely lost due to sudden halt. Re-homing is highly recommended."
But in the GRBL window no usual Alarm button appears and nothing i do seems to be able to realease that axis, but the other axis's can still move until an endstop is hit and they seem to lock off. I have to turn off my machine and physically move the machine off the endstop.

I've just tested several of the releases, it looks like this issue occurs from v3.4.5.
v3.4.3 and v3.4.4 dont give this issue, but then is seems to occur in every other release above.

any advise?

daxliniere, 2022-10-21:
@dylanpil 
"Hard limit has been triggered."
"Everything seems to work great except when i hit the endstops."

This is exactly what is supposed to happen. I have no idea why you weren't seeing this with earlier releases as this has existed since the beginning of FNC.
I would suggest you measure the travel of each axis and enable soft limits. This will avoid that problem.

dylanpil, 2022-10-21:
@daxliniere 
Its very strange to have the cnc lockup when it hits the endstops and not be able to release it, in all other cnc programs that ive had experience with doesn't seem to do this.
I'll try with soft limits as you suggest, maybe there is something im not understanding, but i've been playing with cncs and 3d printers.... for awhile and this is new news to me :)

dylanpil, 2022-10-21:
OK I see the light!!!
I like it alot, but I still find it strange that the machine cannot be unlocked to move it off the endstop, but I guess I will never hit the endstop again! :)
THANKS!!

closed 2024-12-12 by bdring