bdring/FluidNC #515: Installing FluidNC   Problem:

Ronsayer, 2022-07-03:
### Controller Board

6-pack With XYYZ Relay Setup ( don't use the relay ) works Ok with ' Grbl_Esp32 , 

### Help From Board Vendor

- [ ] Yes
- [ ] No
- [X] Not Applicable

### Machine Description

Gantry router withXYYZ format
[image]
 

### Configuration file

```yaml
oard: 6-pack
name: MPCNC External Steppers Relay
meta: 
stepping:
  engine: I2S_STREAM
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN
  shared_stepper_reset_pin: NO_PIN
  x:
    steps_per_mm: 800.000
    max_rate_mm_per_min: 12000.000
    acceleration_mm_per_sec2: 450.000
    max_travel_mm: 370.000
    soft_limits: true
    homing:
      cycle: 2
      allow_single_axis: true
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 700.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.33:low:pu
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 3.000
      standard_stepper:
        step_pin: I2SO.2
        direction_pin: I2SO.1
        disable_pin: I2SO.0

    motor1:
      limit_neg_pin: gpio.32:low:pu
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 3.000
      standard_stepper:
        step_pin: I2SO.5
        direction_pin: I2SO.4
        disable_pin: I2SO.7

  y:
    steps_per_mm: 800.000
    max_rate_mm_per_min: 12000.000
    acceleration_mm_per_sec2: 450.000
    max_travel_mm: 370.000
    soft_limits: true
    homing:
      cycle: 2
      allow_single_axis: true
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 700.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.35:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 3.000
      standard_stepper:
        step_pin: I2SO.10
        direction_pin: I2SO.9:low
        disable_pin: I2SO.8

    motor1:
      limit_neg_pin: gpio.34:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 3.000
      standard_stepper:
        step_pin: I2SO.18
        direction_pin: I2SO.17:low
        disable_pin: I2SO.16

  z:
    steps_per_mm: 800.000
    max_rate_mm_per_min: 1000.000
    acceleration_mm_per_sec2: 200.000
    max_travel_mm: 126.000
    soft_limits: false
     homing:
      cycle: 2
      allow_single_axis: true
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 700.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100
    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.32
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      stepstick:
        ms1_pin: NO_PIN
        ms2_pin: NO_PIN
        ms3_pin: I2SO.22
        step_pin: I2SO.21
        direction_pin: I2SO.20:low
        disable_pin: I2SO.23

    motor1:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 3.000
      null_motor:

kinematics:
  Cartesian:

i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5
  card_detect_pin: NO_PIN

control:
  safety_door_pin: gpio.2:low   
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

coolant:
  flood_pin: NO_PIN
  mist_pin: NO_PIN
  delay_ms: 1000

probe:
  pin: gpio.25:pu
  check_mode_start: false

macros:
  startup_line0: 
  startup_line1: 
  macro0: 
  macro1: 
  macro2: 
  macro3: 

start:
  must_home: true
  deactivate_parking: false
  check_limits: true
```

### Startup Messages

```text
I have pasted my config file in plain text but have tried it in yaml
```

### User Interface Software

WebUI,  with Grbl_esp32 works with WebUI, and Cnc3d commander

### What happened?

I am tring to load FluidNC because I get Jumping with Grbl_esp on x and other axis I had it on z but it only happen occasionally  @Mitch Bradly said they had found issues with Girbl but did not have the time to look in to it as they were committed to  FluidNc

### Other Information

[image]
[image]
[image]
[image]
[image]
[image]
[image]

closed 2022-07-03 by Ronsayer

MitchBradley, 2022-07-03:
[image]
[image]
Can you spot the spelling difference?

Ronsayer, 2022-07-03:
Thanks for that I have tried both spellings  but don’t get any ware so from that I still don’t know which to use 

I found from another person that Fluidtherm is a command line serial tool  , is hard when you don’t  know what to expect because I haven’t used any of it before 

I have had problems with the github forms that I was to fill in , my config file would not paste in I tried a txt ver and that managed to paste in so I might be getting some comments  from that little exercise.

Thanks for all you have done , it must be frustrating  to try and explain things to non -programmers 

Cheers Ron

 

 <https://user-images.githubusercontent.com/4861133/177050673-c9771745-0e7e-4254-a7ce-8384ec9a02d6.png> image
 <https://user-images.githubusercontent.com/4861133/177050682-b4d279f9-c595-43fe-8e61-68d802416d9d.png> image

MitchBradley, 2022-07-03:
Whatever spelling you use, it must be the same in both places.  In the screenshots you posted, it was .yaml in the settings and the actual file on the filesystem was .yml .  The settings told FluidNC to look for a file named config.yaml but it did not find one because the file that was present was config.yml.

You have to change one or the other so they both match.

It is okay to paste it as .txt.  I will add add a note to the form instructions.

MitchBradley, 2022-07-03:
Also please read the form instructions carefully.  In the "Startup Messages" section, it tells you to issue $SS and paste the result from that, but instead you wrote some text about the config file.  The startup messages are something entirely different from the config file.

MitchBradley, 2022-07-03:
Regarding FluidTerm, it comes up automatically at the end of the install-wifi step in the installer.  You can also run it explicitly by double-clicking on fluidterm.bat in the installer.

Ronsayer, 2022-07-03:
Sorry about that  I  have been on days with this I try to understand it and I must be getting to old for this , I only see the things I use , it might help if I read everything out twice , I will try that instruction $SS and paste the results a.s.a.p. 

Ron