bdring/FluidNC #516: Problem:

deranku, 2022-07-04:
### Controller Board

FluidNC Pen Laser Controller (SPI)

### Help From Board Vendor

- [ ] Yes
- [ ] No
- [X] Not Applicable

### Machine Description

It will be a string art machine with external stepper driver on x, solenoid on y and servo on z.

### Configuration file

```yaml
name: "Stringer"
board: "FluidNC Pen/Laser"

start:
  must_home: false
  deactivate_parking: true

stepping:
  engine: RMT
  # 255: dauerhaft an
  idle_ms: 255
  # dir_delay_us: 5 us
  dir_delay_us: 10
  # pulse_us: not less than 2.5 us
  pulse_us: 3
  # Some motors need a delay from when they are enabled to when they can take the first step.: at least 5 us
  disable_delay_us: 10

axes:
  shared_stepper_disable_pin: gpio.13
  
  # Teller/Stepper
  x: 
    # actually steps per nail
    steps_per_mm: 300
    max_rate_mm_per_min: 2000
    acceleration_mm_per_sec2: 25
    max_travel_mm: 1000
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100
    
    motor0:
      limit_neg_pin: gpio.39:high
      hard_limits: false
      pulloff_mm: 1.0
      standard_stepper:
        step_pin: gpio.14
        direction_pin: gpio.12

  # Fädler/Solenoid
  y:
    steps_per_mm: 100.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 5.000
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 5.000
        
    motor0:
      solenoid:
        output_pin: gpio.27:high
        pwm_hz: 1000
        off_percent: 0.000
        pull_percent: 100.000
        hold_percent: 100.000
        pull_ms: 100

  # Bohrer/Servo
  z:
    steps_per_mm: 100
    max_rate_mm_per_min: 5000
    acceleration_mm_per_sec2: 50
    max_travel_mm: 40
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 0
    
    motor0:
      rc_servo:
        pwm_hz: 50
        # output_pin: gpio.22
        output_pin: gpio.21
        min_pulse_us: 540
        max_pulse_us: 2470

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5
  card_detect_pin: NO_PIN

coolant:
  # M7
  mist_pin:  NO_PIN
  # M8
  flood_pin: NO_PIN
  # M9 turns both off

probe:
  pin: NO_PIN

# macros:
#   startup_line0:
#   startup_line1:
#   macro0: G90&G53G0Z-1&G0X0Y0
#   macro1: $SD/Run=drill.nc
#   macro2:
#   macro3:
```

### Startup Messages

```text
[MSG:INFO: FluidNC v3.5.0-pre4]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is SPIFFS]
[MSG:INFO: Configuration file:stringer_config.yaml]
[MSG:INFO: Machine Stringer]
[MSG:INFO: Board FluidNC Pen/Laser]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN]
[MSG:INFO: Stepping:RMT Pulse:3us Dsbl Delay:10us Dir Delay:10us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable gpio.13]
[MSG:INFO: Axis X (150.000,1150.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:gpio.14 Dir:gpio.12 Disable:NO_PIN]
[MSG:INFO:     Neg Limit gpio.39]
[MSG:INFO: Axis Y (0.000,5.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     solenoid Pin: gpio.21 Off: 0.000 Hold: 100.000 Pull:100.000 Duration:100 pwm hz:1000]
[MSG:INFO: Axis Z (-5.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     rc_servo Pin:gpio.27 Pulse Len(900,2100)]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Using spindle NoSpindle]
[MSG:INFO: Connecting to STA SSID:Hakaniemi]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 192.168.178.63]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://stringer.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl 3.5 [FluidNC v3.5.0-pre4 (wifi) '$' for help]
```

### User Interface Software

WebUI

### What happened?

The z-axis with a servo attached to the servo output pins is not moving at all. The other axis work as expected.

I tried three different ones. The first went up in smoke during an OTA-update of FNC. A little panic came up and I can't rule out that I connected it wrong.

Then I tried two different 5 V hobby servos, which both work fine with the Arduino "Sweep" example. The Arduino Servo lib provides a 50 Hz signal, so I took on to that for the FNC config. The min and max pulse come from one servos datasheet, but I played around with different values (and the default). One of the servos moves a bit when the ESP is powered on and the other one is just dead. Both can not be moved by the Web interface.

The 5 V provided by the board can be measured, but just to rule out that either power or GPIO 22 broke during the death of servo #1, I attached the servos to external power and GPIO 21, with the same result.

### Other Information

[image]
Standard 5 V hobby servos

MitchBradley, 2022-07-04:
What is between the servos and the ESP32?  Is there some sort of H bridge or are you trying to connect the servos directly to the GPIO pins?

deranku, 2022-07-04:
The servos are directly connected to the board.

MitchBradley, 2022-07-04:
That will not work.  ESP32 pins do not supply enough voltage or current to drive a servo.  Even micro servos need 100 mA or more.  An ESP32 GPIO can drive about 20mA max.  You need a DC motor driver circuit between the ESP32 and the servo.

deranku, 2022-07-04:
Aha! Thank you, Mitch. I thought the power part was covered with 5 V coming to the servo separately, so that the signal pin itself would not draw any current. The servo pins on the board implied that for me. Thanks again.

MitchBradley, 2022-07-04:
One chip that is often used for this purpose is L293D.  Inexpensive modules with that chip are readily available on Amazon, eBay, ...

MitchBradley, 2022-07-04:
https://www.adafruit.com/product/4489

closed 2022-07-04 by MitchBradley

MitchBradley, 2022-07-04:
https://www.ebay.com/itm/193471876726

MitchBradley, 2022-07-04:
https://www.amazon.com/KOOBOOK-DRV8833-Module-Bridge-Controller/dp/B07S7Z8CFL

deranku, 2022-07-04:
Thank you, Mitch!

bdring, 2022-07-04:
You should connect to the servo connector and use GPIO.22. If you think you damaged 22, you could use outputs on GPIO.21, GPIO.2 or GPIO.27. I suggest using the 5V and ground from the servo connector.

You can test to see if GPIO.22 is OK with a [simple digital output](https://github.com/bdring/FluidNC/wiki/User-Outputs).

```yaml
user_outputs:
  digital0_pin: gpio.22
```

M62P0 to turn on and M63P0 to turn off. Check the voltage with a meter.

Here is the pinout of the controller.

[image]

Here is the schematic of the area of concern.

[image]

deranku, 2022-07-05:
Thank you, Bart! Just checked, it seems that the GPIO.22 works just fine 👌
Waiting for the H-Bridge to cozy the servo now--

bdring, 2022-07-05:
You should be able to use the on board connector without any other components. 

Does the servo hold position when connected and FluidNC is running?

Take a picture of the connection you used.

MitchBradley, 2022-07-05:
I was mistaken about the H-bridge.  Bart is correct, you just need the 5V output via the dedicated connector.

deranku, 2022-07-05:
This servo turns ~15° clockwise when the board is powered on. When it has already reached it's maximum rotation, it just jerks a little.
[image]

This one does nothing at all.
[image]

Both do not hold their positions. They work when connected to an Arduino with ext. power attached (so general function is ok and the wiring is also ok)

bdring, 2022-07-05:
Try changing the PWM range to 1000 to 2000 us.

What commands Re you using to rotate the servo?

Try G53G0Z0 and G53G0Z-5

Avataar120, 2022-07-05:
If it can help : 

[Link](https://components101.com/motors/mg995-servo-motor)

Now for controlling of servo there are only two important things to remember:

**Frequency of PWM**: MG995 takes in PWM signal of frequency 50Hz and any higher and lower frequency PWM will lead to error. As shown in figure the every single cycle of PWM needs to be 20ms width for 50Hz frequency.

**Duty cycle of PWM**: The duty cycle of PWM (or ratio of ON time to total cycle time) determines the position of servo axis. If we provide a PWM signal of 0.5ms ON time over 20mS complete cycle, the servo axis will move to 0º.

deranku, 2022-07-06:
Changing the PWM range has no effect. I tried both jogging via WebUI and through direct input of G-Code, also the code above.

@Avataar120 
Thank you for the infos. The fequency is already set to 50 Hz. The duty cycle is indirectly set through the pulse length, right? I already played around with different values.

Although I already tried another GPIO pin for the servo PWM, I'll try another board next. Maybe something is broken after all–

deranku, 2022-07-06:
I tried another board, but that made no difference.

Now, for reference, I connected the PWM output pin from an Arduino to an oscilloscope. You could clearly see the pulses getting wider and narrower during the servo sweep demo:

[image]
[image]

I then connected GPIO.21 to the oscilloscope and jogged around. The pulse did not alter its length. Also, it seems a little short overall (see picture: scale factor at 1.25 µs). I also tried GPIO.2 and GPIO.21 (did not forget to change the output pin in the config) but there seemed to be no PWM signal. That's odd.

[image]

bdring, 2022-07-06:
Send **$axes/Z** from a serial or WebUI console and paste the response in a reply. I want to double check your values.

deranku, 2022-07-07:
`/axes/Z:
  steps_per_mm: 100.000
  max_rate_mm_per_min: 1000.000
  acceleration_mm_per_sec2: 50.000
  max_travel_mm: 40.000
  soft_limits: false
  homing:
    cycle: 1
    allow_single_axis: true
    positive_direction: true
    mpos_mm: 0.000
    feed_mm_per_min: 50.000
    seek_mm_per_min: 200.000
    settle_ms: 250
    seek_scaler: 1.100
    feed_scaler: 1.100
  motor0:
    limit_neg_pin: NO_PIN
    limit_pos_pin: NO_PIN
    limit_all_pin: NO_PIN
    hard_limits: false
    pulloff_mm: 1.000
    rc_servo:
      output_pin: gpio.21
      pwm_hz: 50
      min_pulse_us: 1000
      max_pulse_us: 2000`

MitchBradley, 2022-07-07:
I will set up a test for this tomorrow.

deranku, 2022-07-07:
Thank you so much!

MitchBradley, 2022-07-08:
It was yet another bug in the Arduino framework code.  Sigh.

deranku, 2022-07-08:
Thank you for fixing this, Mitch. I really appreciate it. Can't wait for the next release.