bdring/FluidNC #521: Problem: Microservo don't work

xorkrus, 2022-07-09:
### Controller Board

MKS DLC32 V2.01

### Help From Board Vendor

- [ ] Yes
- [ ] No
- [X] Not Applicable

### Machine Description

Simple drawing machine (pen plotter)

### Configuration file

```yaml
board: MKS-DLC32 V2.1
name: DrawBotDLC32
meta: 01.07.2022

kinematics:
  corexy:

stepping:
  engine: I2S_STATIC
  idle_ms: 0
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0
axes:
  shared_stepper_disable_pin: I2SO.0
  x:
    steps_per_mm: 157.750
    max_rate_mm_per_min: 18000.000
    acceleration_mm_per_sec2: 1500.000
    max_travel_mm: 325.000
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 5000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.36
      hard_limits: false
      pulloff_mm: 2.000
      stepstick:
        step_pin: I2SO.1
        direction_pin: I2SO.2

  y:
    steps_per_mm: 157.750
    max_rate_mm_per_min: 12000.000
    acceleration_mm_per_sec2: 300.000
    max_travel_mm: 220.000
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 5000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.35
      hard_limits: false
      pulloff_mm: 2.000
      stepstick:
        step_pin: I2SO.5
        direction_pin: I2SO.6:low

  z:
    steps_per_mm: 100
    max_rate_mm_per_min: 5000
    acceleration_mm_per_sec2: 100
    max_travel_mm: 5
    homing:
      cycle: 1
      mpos_mm: 0
      positive_direction: true

    motor0:
      rc_servo:
        pwm_hz: 50
        output_pin: gpio.22
        min_pulse_us: 1000
        max_pulse_us: 2000

i2so:
  bck_pin: gpio.16
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.12
  mosi_pin: gpio.13
  sck_pin: gpio.14

sdcard:
  cs_pin: gpio.15
  card_detect_pin: NO_PIN

control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  macro0_pin: gpio.33:low:pu
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

macros:
  startup_line0:
  startup_line1:
  macro0: $SD/Run=lasertest.gcode
  macro1: $SD/Run=home.gcode
  macro2:
  macro3:

coolant:
  flood_pin: NO_PIN
  mist_pin: NO_PIN
  delay_ms: 0

user_outputs:
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 5000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000
  digital0_pin: NO_PIN
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: NO_PIN

start:
  must_home: false
```

### Startup Messages

```text
[MSG:INFO: FluidNC v3.5.0]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is SPIFFS]
[MSG:INFO: Configuration file:DLC32XYZ.yaml]
[MSG:INFO: Machine DrawBotDLC32]
[MSG:INFO: Board MKS-DLC32 V2.1]
[MSG:INFO: I2SO BCK:gpio.16 WS:gpio.17 DATA:gpio.21]
[MSG:INFO: SPI SCK:gpio.14 MOSI:gpio.13 MISO:gpio.12]
[MSG:INFO: SD Card cs_pin:gpio.15 detect:NO_PIN]
[MSG:INFO: Stepping:I2S_static Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:0ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable I2SO.0]
[MSG:INFO: Axis X (0.000,325.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.1 Dir:I2SO.2 Disable:NO_PIN]
[MSG:INFO:     Neg Limit gpio.36]
[MSG:INFO: Axis Y (0.000,220.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.5 Dir:I2SO.6:low Disable:NO_PIN]
[MSG:INFO:     Neg Limit gpio.35]
[MSG:INFO: Axis Z (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     rc_servo Pin:gpio.22 Pulse Len(1000,2000)]
[MSG:INFO: macro0_pin gpio.33:low:pu]
[MSG:INFO: Kinematic system: CoreXY]
[MSG:INFO: Using spindle NoSpindle]
[MSG:INFO: Connecting to STA SSID:xopkland]
[MSG:INFO: Connecting.]
[MSG:INFO: Connected - IP is 192.168.1.40]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 8080]
Grbl 3.5 [FluidNC v3.5.0 (wifi) '$' for help]
```

### User Interface Software

WebUI

### What happened?

I connect chinese servo TowerPro SG90 on IO35 (default this a "Probe" pin). Didn't respond to any commands.
I recompile firmware with wifi and change PWM resolution to 12bit instead of 16bit.
`const int SERVO_PWM_RESOLUTION_BITS = 16;`
in file FluidNC/src/Motors/RcServoSettings.h
The servo has started to work and respond to commands, but nothing can make the servo move in the opposite direction. Only setting PWM_HZ to "1" or "2" gives some jerk in the opposite direction

MitchBradley, 2022-07-09:
1. The config file does not agree with the startup messages.  The config file has z/max_travel_mm: 5, z/homing/mpos_mm:0, and z/homing/positive_direction:true.  Those settings would make the Z range be -5mm to 0mm, but the startup messages show a different range: [MSG:INFO: Axis Z (-1000.000,0.000)]
2. You did not tell us which commands that you tried when moving the servo, so we cannot tell if you are doing something wrong with those commands.
3. You did not tell us which version of the source code that you used to perform the recompile, so we do not have enough information to draw any conclusions from that data point,
4. According to the schematic at https://github.com/makerbase-mks/MKS-DLC32/blob/main/hardware/MKS%20DLC32%20V2.1_003/MKS%20DLC32%20V2.1_003%20SCH.pdf , GPIO 35 is Y-, not Probe.
5. Again according to that schematic, the circuit attached to GPIO 35 has resistors and capacitors for input signal conditioning.  That signal conditioning will compromise the quality of the PWM signal that is applied to the servo, possibly causing it to misread the PWM signal.  The Makerbase schematic does not give values for the components, so it is hard to know exactly how much filtering is happening.
6. SG90 servos are specified to use a signal voltage of 5V, not 3.3V as comes from an ESP32 GPIO.  I have a Tower Pro MG90S servo that does work with 5V supply and 3.3V signal (direct from ESP32, not through an RC network).  So some servos do indeed work to some extent with 3.3V signaling, but that is not guaranteed, and the RC network will only make it worse.
Please try with an output pin that does not have an RC filter circuit.  If you have further questions, we need accurate information.  When the information that you provide is inconsistent, we do not know which part of it is correct, so we have to guess, and probably guess wrong.  That wastes our time and yours.

MitchBradley, 2022-07-09:
[image]
[image]

bdring, 2022-07-09:
GPIO.22 (Probe) has a R/C filter of unknown value. R/C filters should generally not be used with PWM signals.

I have had good luck with 3.3V on Analog and Digital RC Servos.

xorkrus, 2022-07-09:
to MitchBradley:
1. Sorry, I've been trying different options all night. Made changes on the fly. Perhaps that is why it disagrees with the configuration. Below output from current configuration.
2. Commands from wiki - G53 G0 Z5 and G53 G0 Z0 and arrows from webui
3. Git version at 08.07.22 (maybe 3.5.0)
4, 5, 6. Yes, i probe any pin. 22/34/35/36 and other. I also tried pins 22,25,26, it seems to be straight from ESP32. The situation is the same.
```

> rst:0xc (SW_CPU_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
> 
> configsip: 0, SPIWP:0xee
> 
> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
> 
> mode:DIO, clock div:1
> 
> load:0x3fff0030,len:1184
> 
> load:0x40078000,len:12812
> 
> load:0x40080400,len:3032
> 
> entry 0x400805e4
> 
> 
> 
> [MSG:INFO: FluidNC v3.5.0]
> 
> [MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
> 
> [MSG:INFO: Local filesystem type is SPIFFS]
> 
> [MSG:INFO: Configuration file:DLC32XYZ.yaml]
> 
> [MSG:INFO: Machine DrawBotDLC32]
> 
> [MSG:INFO: Board MKS-DLC32 V2.1]
> 
> [MSG:INFO: I2SO BCK:gpio.16 WS:gpio.17 DATA:gpio.21]
> 
> [MSG:INFO: SPI SCK:gpio.14 MOSI:gpio.13 MISO:gpio.12]
> 
> [MSG:INFO: SD Card cs_pin:gpio.15 detect:NO_PIN]
> 
> [MSG:INFO: Stepping:I2S_static Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:0ms]
> 
> [MSG:INFO: Axis count 3]
> 
> [MSG:INFO: Shared stepper disable I2SO.0]
> 
> [MSG:INFO: Axis X (0.000,325.000)]
> 
> [MSG:INFO:   Motor0]
> 
> [MSG:INFO:     stepstick Step:I2SO.1 Dir:I2SO.2 Disable:NO_PIN]
> 
> [MSG:INFO:     Neg Limit gpio.36]
> 
> [MSG:INFO: Axis Y (0.000,220.000)]
> 
> [MSG:INFO:   Motor0]
> 
> [MSG:INFO:     stepstick Step:I2SO.5 Dir:I2SO.6:low Disable:NO_PIN]
> 
> [MSG:INFO:     Neg Limit gpio.35]
> 
> [MSG:INFO: Axis Z (-5.000,0.000)]
> 
> [MSG:INFO:   Motor0]
> 
> [MSG:INFO:     rc_servo Pin:gpio.22 Pulse Len(1000,2000)]
> 
> [MSG:INFO: macro0_pin gpio.33:low:pu]
> 
> [MSG:INFO: Kinematic system: CoreXY]
> 
> [MSG:INFO: Using spindle NoSpindle]
> 
> [MSG:INFO: Connecting to STA SSID:xopkland]
> 
> [MSG:INFO: Connecting.]
> 
> [MSG:INFO: Connected - IP is 192.168.1.40]
> 
> [MSG:INFO: WiFi on]
> 
> [MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
> 
> [MSG:INFO: SSDP Started]
> 
> [MSG:INFO: HTTP started on port 80]
> 
> [MSG:INFO: Telnet started on port 8080]
> 
> 
> Grbl 3.5 [FluidNC v3.5.0 (wifi) '$' for help]

```

MitchBradley, 2022-07-09:
Try stepping/idle_ms:255

With idle_ms at 0, the system disables all the motors, thus killing the servo position pulse.

I recommend using GPIO 25 and picking it up from pin 7 of the EXP1 connector (LCD_CS).  That signal is driven via a 5V driver, without any filtering.

xorkrus, 2022-07-09:
I used pin 25. Nothing changes. G53 G0 Z5 and G53 G0 Z5 only move the servomotor in one direction. At the same time, it does not look like a step-by-step movement, but like a jamming-jerk.

bdring, 2022-07-09:
Try **G53G0Z0** and **G53G0Z-5**

Is the servo locked in position or free to manually rotate?

xorkrus, 2022-07-09:
Z-5 the same - only forward not a step back... Position free if idle_ms=0 and still work if idle_ms=255

bdring, 2022-07-09:
Your answer is unclear. Is it locked or free to rotate with idle_ms:255.

Please send a picture of the connections to the controller and make sure we can see the servo.

xorkrus, 2022-07-09:
Answer - as is. On idle_ms:255, servo don't stop until i enter commands $stepping/idle_ms=0 and G53...
[pic](https://fastpic.org/view/120/2022/0710/_4c06566d7b70eeb1d774826535293613.jpg.html)

MitchBradley, 2022-07-10:
> At the same time, it does not look like a step-by-step movement, but like a jamming-jerk.

What do you expect?  A 5 mm movement at z/max_rate_mm_per_min: 5000 takes 1/1000 of a minute, or 1/17 of a second.  At a 50 Hz servo pulse rate, that is less than three servo updates, and certainly much less that human perception of smooth movement.

MitchBradley, 2022-07-10:
Furthermore, with your z settings of max_travel_mm:5, mpos_mm:0, positive direction:true, the range of travel is -5..0, consistent with `[MSG:INFO: Axis Z (-5.000,0.000)]`, so G53 G0 Z5 is outside the range of motion and G53 G0 Z0 (which I assume that you meant when you said `G53 G0 Z5` twice, is also at the very limit of the motion, the same limit as Z5.

No information that you have provided is consistent with motion inside the configured limits.

MitchBradley, 2022-07-10:
If you want the range to be 0..5, set positive_direction:false so that mpos_mm:0 is at the negative end of the travel range.  Set max_rate_mm_per_min to something more reasonable, or use G1 moves with an explicit feedrate that corresponds to the speed that you want, considering the 50 Hz max update rate of a hobby servo.

I duplicated your setup with a MKS DLC32 V1.1, connecting the servo to EXP1 pins 7 (signal), 9 (gnd), 10 (5V).  It worked as expected, using motions like `G1 Z-5 F100` and `G1 Z0 F100`.

closed 2022-07-10 by MitchBradley

xorkrus, 2022-07-10:
OK. Maybe my English is too bad and I didn't explain well. [Here's a video](https://youtu.be/J2Hf5VHUaCw).
With any max_travel and speed settings, the servo behaves like in the video. And it turns under any conditions only in one direction. Turning the servomotor in the other direction is possible only when pvm_hz is set to "1" or "2". if normal 50hz - only one direction.
At the same time, as you can see in the video, the movement of the servomotor is non-uniform and a jerk occurs at the end.

MitchBradley, 2022-07-10:
https://discord.com/channels/780079161460916227/882732763126583326/995800097470304356 shows it working for me.  The config file is as attached; it is the same as what you posted above, except:
1. idle_ms is 255, not 0
2. /axes/z/motor0/rc_servo/output_pin is gpio.25, not gpio.22

The board is MKS DLC32 v1.1 because that is the closest that I have.

The servo is connected to the EXP1 header, pins 7 (signal), 9 (GND), and 10 (5V).

The WCO for G54 is 0,0,0, so the coordinates entered in GCode are the same as MPos.

The operating range is -5 mm to 0 mm as specified in the config file and confirmed by `[MSG:INFO: Axis Z (-5.000,0.000)]`

I do not observer any unexplained or strange behavior.

In your video, it appears that the MPos coordinates differ from what you enter as GCode.  That makes me think that you have a work coordinate offset.
[mks.yaml.txt](https://github.com/bdring/FluidNC/files/9079916/mks.yaml.txt)