bdring/FluidNC #522: Problem: Position not updated

ulegan, 2022-07-09:
### Controller Board

none yet

### Help From Board Vendor

- [ ] Yes
- [ ] No
- [X] Not Applicable

### Machine Description

Not connected to a board yet. Just exploring the WebUI

### Configuration file

```yaml
board: None
name: Default (Test Drive)
stepping:
  engine: RMT
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 0
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 320.000
    max_rate_mm_per_min: 1000.000
    acceleration_mm_per_sec2: 25.000
    max_travel_mm: 200.000
    soft_limits: false

  y:
    steps_per_mm: 320.000
    max_rate_mm_per_min: 1000.000
    acceleration_mm_per_sec2: 25.000
    max_travel_mm: 200.000
    soft_limits: false

  z:
    steps_per_mm: 320.000
    max_rate_mm_per_min: 1000.000
    acceleration_mm_per_sec2: 25.000
    max_travel_mm: 200.000
    soft_limits: false

probe:
  pin: NO_PIN
  check_mode_start: true

macros:
  startup_line0:
  startup_line1:
  macro0:
  macro1:
  macro2:
  macro3:

NoSpindle:
```

### Startup Messages

```text
[MSG:INFO: FluidNC v3.5.0]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is SPIFFS]
[MSG:INFO: Configuration file:config.yaml]
[MSG:INFO: Machine Default (Test Drive)]
[MSG:INFO: Board None]
[MSG:INFO: SPI not defined]
[MSG:INFO: Stepping:RMT Pulse:4us Dsbl Delay:0us Dir Delay:0us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (-200.000,0.000)]
[MSG:INFO: Axis Y (-200.000,0.000)]
[MSG:INFO: Axis Z (-200.000,0.000)]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Using spindle NoSpindle]
[MSG:INFO: Connecting to STA SSID:ulegan]
[MSG:INFO: Connecting.]
E (1858) wifi:Association refused temporarily, comeback time 1024 mSec
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 192.168.1.143]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl 3.5 [FluidNC v3.5.0 (wifi) '$' for help]
[MSG:INFO: '$H'|'$X' to unlock]
```

### User Interface Software

WebUI

### What happened?

Uploaded latest Version to a new ESP32-module without any other hardware.
Connected to WebUI and ran some G-code commands, for example G0 G91 X100
It seems the axis are moving internaly, as it shows a speed (FS:1000.0) but MPos stays 0.000
Same with JOG
Is this a bug or a feature? It worked fine with Grbl-ESP32

### Other Information

Grbl 3.5 [FluidNC v3.5.0 (wifi) '$' for help]
[MSG:INFO: '$H'|'$X' to unlock]
<Alarm|MPos:0.000,0.000,0.000|FS:0,0|WCO:110.000,0.000,0.000>
$X
[MSG:INFO: Caution: Unlocked]
ok
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
g0 g91 x100
ok
<Run|MPos:0.000,0.000,0.000|FS:1000,0>
<Run|MPos:0.000,0.000,0.000|FS:1000,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0|WCO:110.000,0.000,0.000>
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Ov:100,100,100>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>

<Idle|MPos:0.000,0.000,0.000|FS:0,0>
$J=G91 G21 F1000 X10
ok
<Jog|MPos:0.000,0.000,0.000|FS:195,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0|WCO:110.000,0.000,0.000>
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Ov:100,100,100>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>

MitchBradley, 2022-07-09:
With the default configuration there are no motors defined so no motor drivers are active.  The actual position management is done in the motor drivers.  If you add a motor section below one of the axes, that axis's position will update
```
    motor0:
      standard_stepper:
        step_pin: gpio.0
        direction_pin: gpio.2
```

ulegan, 2022-07-09:
Thank you for your help, this explains the behaviour.
It's working fine now, also with my board, which I wanted to update from GRBL-ESP32

closed 2022-07-09 by ulegan