bdring/FluidNC #526: Crashes when reaching the endstop

Skorpi08, 2022-07-13:
### Controller Board

MKS Tinybee

### Help From Board Vendor

- [ ] Yes
- [X] No
- [ ] Not Applicable

### Machine Description

K40 Laser

### Configuration file

```yaml
# MKS_TinyBee_1_XYZAB.yaml
board: MKS TinyBee V1.0_001
name: K40 Laser with Adapter
meta: (10.07.2022) by Skorpi
stepping:
  engine: I2S_STATIC
  idle_ms: 0
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0
  segments: 12

axes:
  shared_stepper_disable_pin: NO_PIN
  shared_stepper_reset_pin: NO_PIN

  y:
    steps_per_mm: 157.750
    max_rate_mm_per_min: 12000.000
    acceleration_mm_per_sec2: 300.000
    max_travel_mm: 220.000
    soft_limits: true
    homing:
      cycle: 1
      allow_single_axis: true
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 5000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.32
      hard_limits: false
      pulloff_mm: 2.000
      stepstick:
        step_pin: I2SO.4
        direction_pin: I2SO.5
        disable_pin: I2SO.3

  z:
    steps_per_mm: 157.750
    max_rate_mm_per_min: 12000.000
    acceleration_mm_per_sec2: 500.000
    max_travel_mm: 80.000
    soft_limits: true
    homing:
      cycle: 0
      allow_single_axis: true
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 1000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.22
      hard_limits: false
      pulloff_mm: 1.000
      stepstick:
        step_pin: I2SO.7
        direction_pin: I2SO.8
        disable_pin: I2SO.6

  x:
    steps_per_mm: 157.750
    max_rate_mm_per_min: 18000.000
    acceleration_mm_per_sec2: 1500.000
    max_travel_mm: 325.000
    soft_limits: true
    homing:
      cycle: 2
      allow_single_axis: true
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 5000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.33
      hard_limits: false
      pulloff_mm: 2.000
      stepstick:
        step_pin: I2SO.10
        direction_pin: I2SO.11:low
        disable_pin: I2SO.9

kinematics:
  Cartesian:

i2so:
  bck_pin: gpio.25
  data_pin: gpio.27
  ws_pin: gpio.26

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5
  card_detect_pin: gpio.39

control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  macro0_pin: gpio.4:low:pu
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

coolant:
  flood_pin: NO_PIN
  mist_pin: NO_PIN
  delay_ms: 0

probe:
  pin: NO_PIN
  check_mode_start: true

macros:
  startup_line0: 
  startup_line1: 
  macro0: $SD/Run=lasertest.gcode
  macro1: $SD/Run=home.gcode
  macro2: 
  macro3: 

start:
  must_home: false
  deactivate_parking: false
  check_limits: false

user_outputs:
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 5000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000
  digital0_pin: NO_PIN
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: NO_PIN

arc_tolerance_mm: 0.002
junction_deviation_mm: 0.010
verbose_errors: false
report_inches: false
enable_parking_override_control: false
use_line_numbers: false
planner_blocks: 16

Laser:
  pwm_hz: 5000
  output_pin: gpio.13
  enable_pin: I2SO.16
  disable_with_s0: false
  s0_with_disable: true
  tool_num: 0
  speed_map: 0=0.000% 0=12.500% 1700=100.000%
```

### Startup Messages

```text
[MSG:INFO: Restarting]
[MSG:INFO: WiFi Disconnected]
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:12812
load:0x40080400,len:3032
entry 0x400805e4

[MSG:INFO: FluidNC v3.5.0]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is SPIFFS]
[MSG:INFO: Configuration file:TinyBee_Adapter.yaml]
[MSG:INFO: Machine K40 Laser with Adapter]
[MSG:INFO: Board MKS TinyBee V1.0_001]
[MSG:INFO: I2SO BCK:gpio.25 WS:gpio.26 DATA:gpio.27]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card Detect gpio.39]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:gpio.39]
[MSG:INFO: Stepping:I2S_static Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:0ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (0.000,325.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.10 Dir:I2SO.11:low Disable:I2SO.9]
[MSG:INFO:     Neg Limit gpio.33]
[MSG:INFO: Axis Y (0.000,220.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.4 Dir:I2SO.5 Disable:I2SO.3]
[MSG:INFO:     Neg Limit gpio.32]
[MSG:INFO: Axis Z (0.000,80.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.7 Dir:I2SO.8 Disable:I2SO.6]
[MSG:INFO:     Neg Limit gpio.22]
[MSG:INFO: macro0_pin gpio.4:low:pu]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Laser Spindle Ena:I2SO.16 Out:gpio.13 Freq:5000Hz Res:13bits Laser mode:On]
[MSG:INFO: Using spindle Laser]
[MSG:INFO: Connecting to STA SSID:Easyfritz]
[MSG:INFO: Connecting.]
[MSG:INFO: Connected - IP is 192.168.168.168]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl 3.5 [FluidNC v3.5.0 (wifi) '$' for help]
```

### User Interface Software

WebUI

### What happened?

After reaching the Endstop on homing, it crashes:

Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1). 

Core  1 register dump:
PC      : 0x40081908  PS      : 0x00060935  A0      : 0x80081960  A1      : 0x3ffbf1bc  
A2      : 0x3ffb62d8  A3      : 0x00000000  A4      : 0x00000001  A5      : 0x3ffb23a0  
A6      : 0x00000000  A7      : 0x3ffb60b8  A8      : 0x80081908  A9      : 0x3ffbf19c  
A10     : 0x00000004  A11     : 0x00000000  A12     : 0x3ffb23a0  A13     : 0x3ffb2380  
A14     : 0x00000000  A15     : 0x3ffcd880  SAR     : 0x00000001  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x400858e1  LEND    : 0x400858e9  LCOUNT  : 0x00000027  
Core  1 was running in ISR context:
EPC1    : 0x40123073  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x00000000

Backtrace:0x40081905:0x3ffbf1bc |<-CORRUPTED

Core  0 register dump:
PC      : 0x401ab782  PS      : 0x00060b35  A0      : 0x80121c20  A1      : 0x3ffd30f0  
A2      : 0x00000000  A3      : 0x80000001  A4      : 0x80091480  A5      : 0x3ffd9f00  
A6      : 0x00000003  A7      : 0x00060023  A8      : 0x80121712  A9      : 0x3ffd30c0  
A10     : 0x00000000  A11     : 0x80000001  A12     : 0x80091480  A13     : 0x3ffd9f00  
A14     : 0x00000003  A15     : 0x00060023  SAR     : 0x00000014  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace:0x401ab77f:0x3ffd30f00x40121c1d:0x3ffd3110 0x4008f550:0x3ffd3130 

ELF file SHA256: 0000000000000000

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:12812
load:0x40080400,len:3032
entry 0x400805e4

[MSG:INFO: FluidNC v3.5.0]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is SPIFFS]
[MSG:INFO: Configuration file:TinyBee_Adapter.yaml]
[MSG:INFO: Machine K40 Laser with Adapter]
[MSG:INFO: Board MKS TinyBee V1.0_001]
[MSG:INFO: I2SO BCK:gpio.25 WS:gpio.26 DATA:gpio.27]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card Detect gpio.39]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:gpio.39]
[MSG:INFO: Stepping:I2S_static Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:0ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (0.000,325.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.10 Dir:I2SO.11:low Disable:I2SO.9]
[MSG:INFO:     Neg Limit gpio.33]

This is all that is displayed in the Terminal

### Other Information

Tested Firmware:
3.4.4 ->ok
3.4.5 ->crash
3.4.9 ->crash
3.5.0-pre4 ->crash
3.5.0  ->crash

MitchBradley, 2022-07-13:
I cannot duplicate the problem.  I adapted your config for use on MKS DLC32 v1.1 - the closest board that I have to your setup.  To do so I had to change the GPIO and I2SO pin assignments.  The config file is attached.  I then installed FluidNC v3.5.0 and, from WebUI, clicked on the Home X button.  I then pressed and released a switch connected to the X limit pins, then pressed/released again during the slow approach phase.  It worked exactly as expected.  I also tried using $HX from FluidTerm, with $message/level=debug so that I could observe the homing phase reports.  Again, it worked correctly.
[issue526.yaml.txt](https://github.com/bdring/FluidNC/files/9106389/issue526.yaml.txt)
[issue526.ss.txt](https://github.com/bdring/FluidNC/files/9106405/issue526.ss.txt)
[issue526.hx.txt](https://github.com/bdring/FluidNC/files/9106408/issue526.hx.txt)

I also tried with the X homing switch set to active high (NC).  It worked.

MitchBradley, 2022-07-14:
I have a guess about what might be happening.  The crash backtrace indicates that Core 0 is in a pin interrupt handler, and the crash is caused by a watchdog timeout.  That could happen if the interrupt handler is firing continuously so that the idle task has no time to feed the watchdog.  The active state of the limit pins - high - suggests that the limit switches are normally closed, which means that the limit inputs are relative immune to noise when the switch is inactive.  Then, when the limit is reached during homing, the switch opens and the impedance of the circuit is much higher and thus more susceptible to noise.  If the limit circuit is picking up noise when the switch is open, the limit interrupt could be firing over and over, saturating that core with interrupts.

One way to check this would be to replace the X limit circuit with a switch on a very short wire.  Test homing by manually operating the switch.  If it works correctly with that test, the real limit circuit is probably picking up noise.

The Tinybee board has RC filters on its limit inputs, but the schematic does not show the component values so I cannot tell how effective those filters might be.  Even if I knew, this ticket gives us no information about what types of limit switches are used, how they are wired, what sort of interfering signals might be present, how possibly-interfering wiring is routed, how the different power supplies are tied to ground or noise filtered, whether or not the laser supply is running when the problem occurs...

Skorpi08, 2022-07-14:
it has optical endstops as shown on this website.
http://donsthings.blogspot.com/2016/06/k40-optical-endstops.html?m=1
I can test it later with DLC32.

Avataar120, 2022-07-14:
I do confirm that those optical endstops are generating a LOT of noise on the ESP32 inputs.
Before designing my own board for ESP32 GRBL and now for FluidNC, I have experienced some difficulties at first with standard shields that were not equipped with RC filters (false positives mainly)
In my personal design I have integrated RC filters and I have no more issues

MitchBradley, 2022-07-14:
Optical endstops can trigger from ambient light.  Some kinds of lighting pulsates at high frequencies that could potentially cause rapid-fire triggering.  When the  interrupter is not in place, you would not experience the effect because the LED is keeping the optocoupler turned on continuously.  When the interrupter is in place, ambient light leaks in and retriggers over and over at the pulsation frequency.
You might be able to shield the limit switches somehow by putting some sort of box over them.

Skorpi08, 2022-07-15:
https://cdn.discordapp.com/attachments/882732763126583326/997572992630542408/IMG_20220715_203808.jpg

https://cdn.discordapp.com/attachments/882732763126583326/997572993150631987/IMG_20220715_203756.jpg

The Problem is not the Endstop.

I have changed on X Axis limit_neg_pin: gpio.33
to
limit_neg_pin: gpio.22
and it works.

On Y Axis it works with limit_neg_pin: gpio.32

MitchBradley, 2022-07-15:
[image]
[image]

MitchBradley, 2022-07-15:
For reference, here is an excerpt from the Tinybee schematic
[image]

Skorpi08, 2022-07-15:
tested with DLC32, no Problem.

board: MKS-DLC32 V2.1
name: K40 Laser with Adapter
meta: (10.07.2022) by Skorpi

kinematics:
  Cartesian:

stepping:
  engine: I2S_STATIC
  idle_ms: 0
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0
axes:
  shared_stepper_disable_pin: I2SO.0

  y:
    steps_per_mm: 473.250
    max_rate_mm_per_min: 12000.000
    acceleration_mm_per_sec2: 300.000
    max_travel_mm: 220.000
    soft_limits: true
    homing:
      cycle: 2
      allow_single_axis: true
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 5000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.34
      hard_limits: false
      pulloff_mm: 2.000
      tmc_2209:
        addr: 1
        r_sense_ohms: 0.110
        run_amps: 1.500
        hold_amps: 0.800
        microsteps: 16
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: CoolStep
        homing_mode: CoolStep
        use_enable: false
        step_pin: I2SO.5
        direction_pin: I2SO.6

  x:
    steps_per_mm: 473.250
    max_rate_mm_per_min: 12000.000
    acceleration_mm_per_sec2: 300.000
    max_travel_mm: 320.000
    soft_limits: true
    homing:
      cycle: 1
      allow_single_axis: true
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 5000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.22
      hard_limits: false
      pulloff_mm: 2.000
      tmc_2209:
        uart:
          txd_pin: gpio.0
          rxd_pin: gpio.4
          baud: 115200
          mode: 8N1
        addr: 2
        r_sense_ohms: 0.110
        run_amps: 1.500
        hold_amps: 0.800
        microsteps: 16
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: CoolStep
        homing_mode: CoolStep
        use_enable: false
        step_pin: I2SO.3
        direction_pin: I2SO.4

i2so:
  bck_pin: gpio.16
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.12
  mosi_pin: gpio.13
  sck_pin: gpio.14

sdcard:
  cs_pin: gpio.15
  card_detect_pin: gpio.39

control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  macro0_pin: gpio.33:low:pu
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

macros:
  startup_line0:
  startup_line1:
  macro0: $SD/Run=lasertest.gcode
  macro1: $SD/Run=home.gcode
  macro2:
  macro3:

coolant:
  flood_pin: NO_PIN
  mist_pin: NO_PIN
  delay_ms: 0

Laser:
  pwm_hz: 5000
  #L on Beeper / IN on TTL
  output_pin: gpio.32
  enable_pin: I2SO.7
  disable_with_s0: false
  s0_with_disable: false
  tool_num: 0
  speed_map: 0=0.000% 0=12.500% 1700=100.000%
# 135=0mA 270=5mA 400=10mA 700=16mA
user_outputs:
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 5000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000
  digital0_pin: NO_PIN
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: NO_PIN

start:
  must_home: false

vadim7272, 2022-07-16:
I also have a tiny bee and the optical pins don't work properly, give out alarm , limit_all_pin: gpio.22 does that help without a macro?

MitchBradley, 2022-07-16:
There is a difference between the MKS DLC32 and MKS Tinybee input circuits.  DLC32 has
[image]
while Tinybee has
[image]

The additional Tinybee components provide protection against spikes, particularly the reverse-biased diodes (D22) that connect to +5 and GND. The Schottky series diode (D31) makes the input 5V tolerant, by disconnecting the input if its voltage goes above about 3.4V.  I wonder if that series diode is contributing to the problem by changing the circuit behavior on a low-to-high transition?  It might also be affecting the bias conditions of the optical sensor under certain conditions.

One way to test that conjecture would be to bypass that diode by connecting a resistor - something like 10K - from 3.3V to the input pin.

Skorpi08, 2022-07-16:
Nothing helps.
With pin 33 it does not work, with pin 22 it works fine.
Something must have changed in the 3.5.0, because with the 3.4.4 it also works with pin 33.

MitchBradley, 2022-07-16:
A lot changed between 3.4.4 and 3.4.5 - we upgraded to the new version of the Arduino framework, which involves a switch to the new version of the ESP-IDK core firmware and a newer version of the C++ compiler.  We are still finding differences and problems in that underlying code.  Somebody sent me a Tinybee for testing but it has not arrived yet.

MitchBradley, 2022-08-09:
I don't know if this will help but it is worth a try

Skorpi08, 2022-08-09:
[MSG:INFO: FluidNC EventQueueTest]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is SPIFFS]
[MSG:INFO: Configuration file:TinyBee_Adapter.yaml]
[MSG:INFO: Machine K40 Laser with Adapter]
[MSG:INFO: Board MKS TinyBee V1.0_001]
[MSG:INFO: I2SO BCK:gpio.25 WS:gpio.26 DATA:gpio.27]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card Detect gpio.34]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:gpio.34]
[MSG:INFO: Stepping:I2S_static Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:0ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (0.000,320.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.10 Dir:I2SO.11:low Disable:I2SO.9]
[MSG:INFO:     Neg Limit gpio.33]
Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1). 

Core  1 register dump:
PC      : 0x400825d0  PS      : 0x00060035  A0      : 0x40085d60  A1      : 0x3ffbf23c  
A2      : 0x00000002  A3      : 0x00000000  A4      : 0x00002180  A5      : 0x4008e058  
A6      : 0x00000002  A7      : 0xffffff00  A8      : 0x3ff44000  A9      : 0xffffff00  
A10     : 0x00000000  A11     : 0x00000020  A12     : 0x00000001  A13     : 0x3ffb2590  
A14     : 0x00000001  A15     : 0x00000000  SAR     : 0x0000001f  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x4008acc0  LEND    : 0x4008acd6  LCOUNT  : 0xffffffff  
Core  1 was running in ISR context:
EPC1    : 0x401231f7  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x00000000

Backtrace:0x400825cd:0x3ffbf23c |<-CORRUPTED

Core  0 register dump:
PC      : 0x401ab8ea  PS      : 0x00060535  A0      : 0x80121da4  A1      : 0x3ffd32a0  
A2      : 0x00000000  A3      : 0x4008a07c  A4      : 0x00060520  A5      : 0x3ffd29f0  
A6      : 0x007bf288  A7      : 0x003fffff  A8      : 0x80121896  A9      : 0x3ffd3270  
A10     : 0x00000000  A11     : 0x80000001  A12     : 0x800913ec  A13     : 0x3ffd31a0  
A14     : 0x00000003  A15     : 0x00060023  SAR     : 0x00000014  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace:0x401ab8e7:0x3ffd32a00x40121da1:0x3ffd32c0 0x4008f4bc:0x3ffd32e0 

ELF file SHA256: 0000000000000000

Rebooting...
ets Jun  8 2016 00:22:57

Skorpi08, 2022-08-10:
$HX
Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1). 

Core  1 register dump:
PC      : 0x400825ee  PS      : 0x00050335  A0      : 0x40085d60  A1      : 0x3ffbf23c  
A2      : 0x00000002  A3      : 0x3ffbdd00  A4      : 0x00000001  A5      : 0x4008e058  
A6      : 0x00000000  A7      : 0x80000001  A8      : 0x00000001  A9      : 0xffffff00  
A10     : 0x00000000  A11     : 0x00000020  A12     : 0x00000001  A13     : 0x3ffb38f4  
A14     : 0x00000001  A15     : 0x3ffb616c  SAR     : 0x0000001f  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  
Core  1 was running in ISR context:
EPC1    : 0x4012324f  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x00000000

Backtrace:0x400825eb:0x3ffbf23c |<-CORRUPTED

Core  0 register dump:
PC      : 0x401ab95a  PS      : 0x00060135  A0      : 0x80121dfc  A1      : 0x3ffd32a0  
A2      : 0x00000000  A3      : 0x80000001  A4      : 0x800913ec  A5      : 0x3ffd31c0  
A6      : 0x00000003  A7      : 0x00060023  A8      : 0x801218ee  A9      : 0x3ffd3270  
A10     : 0x00000000  A11     : 0x80000001  A12     : 0x800913ec  A13     : 0x3ffd31a0  
A14     : 0x00000003  A15     : 0x00060023  SAR     : 0x00000014  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace:0x401ab957:0x3ffd32a00x40121df9:0x3ffd32c0 0x4008f4bc:0x3ffd32e0 

ELF file SHA256: 0000000000000000

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:12812
load:0x40080400,len:3032
entry 0x400805e4

[MSG:INFO: FluidNC EventQueueTest2]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is SPIFFS]
[MSG:INFO: Configuration file:TinyBee_Adapter.yaml]
[MSG:INFO: Machine K40 Laser with Adapter]
[MSG:INFO: Board MKS TinyBee V1.0_001]
[MSG:INFO: I2SO BCK:gpio.25 WS:gpio.26 DATA:gpio.27]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card Detect gpio.34]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:gpio.34]
[MSG:INFO: Stepping:I2S_static Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:0ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (0.000,320.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.10 Dir:I2SO.11:low Disable:I2SO.9]
[MSG:INFO:     Neg Limit gpio.33]
ets Jun  8 2016 00:22:57

rst:0x8 (TG1WDT_SYS_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:12812
load:0x40080400,len:3032
entry 0x400805e4

[MSG:INFO: FluidNC EventQueueTest2]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is SPIFFS]
[MSG:INFO: Configuration file:TinyBee_Adapter.yaml]
[... 105 lines cut ...]
Backtrace:0x4008250e:0x3ffbf21c |<-CORRUPTED

Core  0 register dump:
PC      : 0x401ab95a  PS      : 0x00060535  A0      : 0x80121dfc  A1      : 0x3ffd32a0  
A2      : 0x00000000  A3      : 0x4008a07c  A4      : 0x00060520  A5      : 0x3ffd2910  
A6      : 0x007bf288  A7      : 0x003fffff  A8      : 0x801218ee  A9      : 0x3ffd3270  
A10     : 0x00000000  A11     : 0x3ffbf284  A12     : 0x3ffbf284  A13     : 0x00000000  
A14     : 0x00060520  A15     : 0x00000000  SAR     : 0x00000014  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace:0x401ab957:0x3ffd32a00x40121df9:0x3ffd32c0 0x4008f4bc:0x3ffd32e0 

ELF file SHA256: 0000000000000000

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:12812
load:0x40080400,len:3032
entry 0x400805e4

Skorpi08, 2022-08-28:
https://github.com/bdring/FluidNC/releases/tag/EventQueue2-Test

[MSG:INFO: FluidNC EventQueue2-Test]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is SPIFFS]
[MSG:INFO: Configuration file:TinyBee_Adapter.yaml]
[MSG:INFO: Machine K40 Laser with Adapter]
[MSG:INFO: Board MKS TinyBee V1.0_001]
[MSG:INFO: I2SO BCK:gpio.25 WS:gpio.26 DATA:gpio.27]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card Detect gpio.34]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:gpio.34]
[MSG:INFO: Stepping:I2S_static Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:0ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (0.000,320.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.10 Dir:I2SO.11:low Disable:I2SO.9]
[MSG:INFO:     Neg Limit gpio.33]
[MSG:INFO: Axis Y (0.000,220.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.4 Dir:I2SO.5 Disable:I2SO.3]
[MSG:INFO:     Neg Limit gpio.32]
[MSG:INFO: Axis Z (-1000.000,0.000)]
[MSG:INFO: macro0_pin gpio.4:low:pu]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Laser Ena:I2SO.16 Out:gpio.13 Freq:5000Hz Period:8191]
[MSG:INFO: Using spindle Laser]
[MSG:INFO: Connecting to STA SSID:Router]
[MSG:INFO: Connecting.]
[MSG:INFO: Connected - IP is 192.168.168.168]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl  [FluidNC EventQueue2-Test (wifi) '$' for help]
$X
ok
ok
$HX
ok
ok
$HY
ok
ok

Skorpi08, 2022-08-28:
thx Mitch!

closed 2022-08-28 by Skorpi08