bdring/FluidNC #532: Problem: version from 3.4.7 introduces movement issues on fast move for CoreXY

lollotek, 2022-07-19:
### Controller Board

Custom Esp32 board with TMC_2009 as standard_stepper (no uart)

[image]

### Help From Board Vendor

- [ ] Yes
- [ ] No
- [X] Not Applicable

### Machine Description

custom CoreXY with laser engraver

### Configuration file

```yaml
name: TMC2208 Pen Cutter
board: unknown

kinematics:
  corexy:

stepping:
  engine: RMT
  idle_ms: 255
  dir_delay_us: 0
  pulse_us: 3
  disable_delay_us: 0

laser:
  pwm_hz: 5000
  output_pin: gpio.4
  enable_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: false
  tool_num: 0
  speed_map: 0=0.000% 255=100.000%

axes:
  shared_stepper_disable_pin: gpio.13:high
  x:
    steps_per_mm: 80.000
    max_rate_mm_per_min: 8000.000
    acceleration_mm_per_sec2: 300.000
    max_travel_mm: 550.000
    soft_limits: false
    homing:
      cycle: 2
      mpos_mm: 0.000
      positive_direction: true
      settle_ms: 250.000
      seek_mm_per_min: 1000.000
      feed_mm_per_min: 500.000
      seek_scaler: 1.100
      feed_scaler: 5.000
    motor0:
      limit_all_pin: gpio.17
      hard_limits: false
      standard_stepper:
        direction_pin: gpio.14
        step_pin: gpio.12
    motor1:
      null_motor:
  y:
    steps_per_mm: 80.000
    max_rate_mm_per_min: 8000.000
    acceleration_mm_per_sec2: 300.000
    max_travel_mm: 550.000
    soft_limits: false
    homing:
      cycle: 1
      mpos_mm: 0.000
      positive_direction: true
      settle_ms: 250.000
      seek_mm_per_min: 1000.000
      feed_mm_per_min: 500.000
      seek_scaler: 1.100
      feed_scaler: 5.000
    motor0:
      limit_all_pin: gpio.21
      hard_limits: false
      standard_stepper:
        direction_pin: gpio.15
        step_pin: gpio.26
    motor1:
      null_motor:
  z:
    steps_per_mm: 80.000
    max_rate_mm_per_min: 8000.000
    acceleration_mm_per_sec2: 300.000
    max_travel_mm: 20.000
    soft_limits: false
    motor0:
      limit_all_pin: gpio.16
      standard_stepper:
        direction_pin: gpio.33
        step_pin: gpio.27
    motor1:
      null_motor:

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5

control:

coolant:
  delay_ms: 1000.000

probe:
  check_mode_start: false

macros:
  startup_line0: 
  startup_line1: 
  macro0: 
  macro1: 
  macro2: 
  macro3: 

start:
  must_home: false
  check_limits: false
  deactivate_parking: false

user_outputs:
arc_tolerance_mm: 0.002
junction_deviation_mm: 0.010
verbose_errors: false
report_inches: false
enable_parking_override_control: false
use_line_numbers: false
```

### Startup Messages

```text
[MSG:INFO: FluidNC v3.5.0]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is SPIFFS]
[MSG:INFO: Configuration file:config.yaml]
[MSG:INFO: Machine TMC2208 Pen Cutter]
[MSG:INFO: Board unknown]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN]
[MSG:INFO: Stepping:RMT Pulse:3us Dsbl Delay:0us Dir Delay:0us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable gpio.13]
[MSG:INFO: Axis X (-550.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.12 Dir:gpio.14 Disable:NO_PIN]
[MSG:INFO:     All Limit gpio.17]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Y (-550.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.26 Dir:gpio.15 Disable:NO_PIN]
[MSG:INFO:     All Limit gpio.21]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Z (-20.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.27 Dir:gpio.33 Disable:NO_PIN]
[MSG:INFO:     All Limit gpio.16]
[MSG:INFO:   Motor1]
[MSG:INFO: Kinematic system: CoreXY]
[MSG:INFO: Laser Spindle Ena:NO_PIN Out:gpio.4 Freq:5000Hz Res:13bits Laser mode:On]
[MSG:INFO: Using spindle Laser]
[MSG:INFO: Connecting to STA SSID:Vodafone-34808741]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connecting...]
[MSG:INFO: Connected - IP is 192.168.1.2]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
ok
```

### User Interface Software

CNCjs

### What happened?

After a firmware update, I Started having issues with movements: for example sending only X movement it also change a bit on Y and vice versa.

After some test I found that was introduced starting from version 3.4.7, and is still present on 3.5.0.
I also found that issue is present only during fast movents and not with G1 movements. I edited my test GCODE changing G0 whit G1, and everythings where correctly aligned.

I need to update the firmare because it fixes the issue with laser that don't switch off during fast movement, (so I need fast movement in my gcode).

bdring, 2022-07-20:
I am testing on a corexy and don't immediately see the issue. You need to give me the simplest case to repeat the problem and an exact statement of the problem.

Please quantify "it also change a bit on Y"

lollotek, 2022-07-20:
I recorded a Video: With CNCjs (also tryed with openbuilds control) I send a Jog movement of 100mm on X axes, but it also move on Y.

The complete sequence is +100mm X, +100mm X, -100mm X, -100mm X, +100mm Y, +100mm Y

[https://youtu.be/HgJ-7Kwkc5U](youtube video)

All versions >= 3.4.6 have that issue.

What else I can provide\test to be helpful?

bdring, 2022-07-20:
That is strange, the motion look like an "S" shape.

You should send gcode from a console because I don't know what CNCJS is sending. Are you connected via USB? Your image says no UART.

Try using Fluidterm with these commands. Tell me what works and what does not. Ignore the comments in parentheses.

$H (Home All sequences)
G91 G0 X-100 (Rapid move in X)
G91 G0 Y-100
G91 G1 F1000 X-100 (slow feed move in X)
G91 Y-100 
$J=G91 F8000 X-100 (fast jog in X)
$J=G91 F8000 Y-100
$J=G91 F8000 X100 Y100
G90 (return to g90 mode)

Also send a picture of the controller.

lollotek, 2022-07-21:
This is the video with those commands sent from fluid interface. I skipped the homing because i didn't have it.
As you can see only with fast moments it do the 'S shape'. 
[https://www.youtube.com/watch?v=S8VqbhyBhng](url)
the controller board is DIY on breadboard
[https://imgur.com/a/LVAvg6Q](url)

SeamusCheng, 2023-10-24:
I also encountered the same problem, which I discovered during the homing process. The corexy structure first home X,  X displays 0, and then home Y, Y displays 0 and X displays 0.25. Have you solved the problem?

lollotek, 2023-11-14:
For that I remember, I found an older version without that issue.
unfortunately, I'm not using my cnc recently because of life changes.
I'm sorry that I cannot help more..