bdring/FluidNC #535: Feature: Separate homing on dual axis motors

lagnat, 2022-07-24:
### Machine Context

I'll first say that I don't know if this is a new idea of something that's already proven to be dumb but couldn't hurt to ask.

Many machines have an axis with dual motors.  That creates the scenario where the CNC can get out of square.  Independent homing would enable the machine to be calibrated once and then auto-square every time.

### Feature Description

The idea would be that $HY would run the homing algorithm independently for each motor.  As one limit switch is hit, that particular motor would stop but the other would keep going until that motor's switch was hit.  I could also see the algorithm slightly tweaked such that when the first switch is hit and the motor stops, it waits for the other motor to hit the switch before doing a simultaneous backoff and slow approach, however still independently.  This idea would require a limit switch per motor and likely a bunch of trial and error before getting it dialed in.

I look forward to your thoughts.

### Other Approaches

Right now I use a framing square clamped to a long level, and then manually turn one of the motors until it's square.  This is not so important most of the time but on occasion I need something as close to square as possible and the manual method is cumbersome.  

### How I Can Help

I would love to contribute but my c++ is entirely too rusty so I'm not entirely confident I could go it alone.  For this request, or otherwise, I'm open to helping out.

MitchBradley, 2022-07-24:
This feature already exists - https://github.com/bdring/FluidNC/wiki/FluidNC-Homing#axis-squaring .

lagnat, 2022-07-24:
Well shucks.. I guess it was a good idea after all.  I checked the documentation for $H but didn't look further.  Sorry for the noise.

closed 2022-07-24 by lagnat