bdring/FluidNC #536: Feature:

PhGoffin, 2022-07-24:
### Machine Context

Hi, I am new with ESP. I built a sculpture to draw on sand (like sisyphus kinetctic sculpture). The sculpture is based on the arduino + raspberry pi architecture. I am thinking to move to ESP but I would like to know if I can customize the webUI to propose to my customers a very simplify UI (connection via smartphone)

### Feature Description

I would like to use the webUI to create my own user interface. To give you an idea, the user interface is composed of 10 buttons to select a pattern, an image to preview the drawing pattern and a button "execute" to load the gcode and send to the cnc.
Before spending time on this project, I would like to have your opinion on the feasibility.
Can I create a new folder on the web server and set my pages? what is the language to create the pages (currently, I use Apache + PHP).
I am an IT guy so programming pages in another language shouldn't be a problem!

### Other Approaches

Using Arduino (to host grbl) + Raspberry pi (to host the web server)

### How I Can Help

do you have already experience with this feature.
Of course, I will be happy to share my experience with the community if I can implement a new user interface

MitchBradley, 2022-07-24:
WebUI is a separate project by a different author - [ESP3D-WebUI](https://github.com/luc-github/ESP3D-WEBUI).  It is written in client-side JavaScript. We are currently using a [fork of version 2](https://github.com/MitchBradley/ESP3D-WEBUI/tree/TabletStuff) but version 3 is now out and we are planning to migrate to the new version as time permits.  It is compiled down (using gulp for version 2 and npm for version 3) to a single "index.html.gz" file that is placed on the FLASH filesystem of the ESP32.  All of the server-side logic is written in C++ and compiled into the FluidNC binary.  You would probably not need to change any server logic.

Our fork has some tweaks for FluidNC and, more significantly, extra code to implement an additional UI tab that is optimized for use with a touchscreen tablet computer.  That additional UI has some features in common with your needs, in that

1. It shows the feasibility of and provides an example of adding an additional UI to ESP3D-WEBUI.
2. It is intended for use on a smaller screen than the normal UI, with larger buttons relative to the screen size and attempts to avoid scrolling
3. I has a gcode previewer panel.

[image]

Implementing extensions like this extra tablet UI differs between version 2 and version 3.  In version 2, there was no facility for such additions so I had to change the code directly, adding includes of extra .js, .html, and .css files and references to them in menus.  Version 3 has an ExtraPages facility whereby you can add your own file in an ExtraContent directory.  In principle that provides an obvious place to hook in custom UI stuff.  I have not yet ported the Tablet UI to version 3 so I do not know what changes will be required from the standpoint of its interaction with the core UI and messaging code.  I will say that, from what I have seen so far, version 3 has much better internal structure than version 2.

MitchBradley, 2022-07-24:
Since the issue is really a question about feasibility (yes it is feasible), the author proposes to do the work himself for his own use, and the implied questions about how to do have been answered, I am now closing this issue.

closed 2022-07-24 by MitchBradley