bdring/FluidNC #537: problemi di scansione utilizzando mks dlc32 e lightburn

distebia, 2022-07-24:
### Controller Board

mks dlc32 v.2.1

### Help From Board Vendor

- [ ] Yes
- [ ] No
- [X] Not Applicable

### Machine Description

Laser co2 50w su telaio autocostruito con piano di 1300x900.

### Configuration file

```yaml
board: MKS DLC32 2.1
name: MKS DLC32 XYZ
meta: (08/07/2022) Biagio

arc_tolerance_mm: 0.002
junction_deviation_mm: 0.010
verbose_errors: false
report_inches: false
enable_parking_override_control: false
use_line_numbers: false
planner_blocks: 16

stepping:
  engine: I2S_STREAM
  idle_ms: 255
  pulse_us: 10
  dir_delay_us: 1
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: i2so.0
  x:
    steps_per_mm: 79.973
    max_rate_mm_per_min: 20000.000
    acceleration_mm_per_sec2: 3000.000
    max_travel_mm: 1226.000
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 1000.000
      seek_mm_per_min: 5000.000
      settle_ms: 250.000
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.36:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm:1.000
      stepstick:
        step_pin:  i2so.1
        direction_pin: I2SO.2

  y:
    steps_per_mm: 80.194
    max_rate_mm_per_min: 10000.000
    acceleration_mm_per_sec2: 350.000
    max_travel_mm: 898.000
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 1000.000
      seek_mm_per_min: 5000.000
      settle_ms: 250.000
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.35:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm:1.000
      stepstick:
        step_pin: I2SO.5
        direction_pin: I2SO.6

  z:
    steps_per_mm: 640.000
    max_rate_mm_per_min: 200.000
    acceleration_mm_per_sec2: 50.000
    max_travel_mm: 170.000
    soft_limits: true
    homing:
      cycle: 0
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 200.000
      seek_mm_per_min: 200.000
      settle_ms: 250.000
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.34:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm:1.000
      stepstick:
        step_pin: I2SO.3
        direction_pin: I2SO.4:low

i2so:
  bck_pin: gpio.16
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.12
  mosi_pin: gpio.13
  sck_pin: gpio.14

sdcard:
  card_detect_pin: NO_PIN
  cs_pin: gpio.15

probe:
  pin: gpio.2
  check_mode_start: false

start:
  must_home: false

Laser:
  pwm_hz: 5000
  output_pin: gpio.32
  enable_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  speed_map: 0=0.000% 0=9.000% 1000=100.000%
```

### Startup Messages

```text
nessun errore
```

### User Interface Software

lightburn

### What happened?

Salve, ho la configurazione come sopra e ho notato problemi di scansione di testo riempito con lightburn. in pratica il mio testo, anche impostando una sovrascansione del 5% viene come sdoppiato. sono riuscito ad accendere il laser al 1% grazie a speed_map, anche variandolo non ottengo cambiamenti. ho trovato in rete di impostare I2S_static al posto di I2S_stream. ottengo benefici senza perdere niente?

Skorpi08, 2022-07-24:
Du kannst auch einfach meine yaml Datei nehmen, die funktioniert in einem K40 Laser mit Lightburn.

distebia, 2022-07-24:
ho anche problemi con i finecorsa...se carico il firmware originale tutto funziona perfettamente, ripetibile e preciso. se carico fluidnc l'homing certe volte parte verso la direzione sbagliata, riavvio e funziona. altre volte è come se non legge i finecorsa e va a sbattere, altre volte mi richiede una distanza di pull-up maggiore e, anche aumentandola, mi segnala l'errore. riavvio e tutto torna alla normalità ma non mi sento sicuro in quanto i problemi si ripresentano ad ogni accensione. se rimetto il firmware originale tutto funziona a meraviglia, tranne il fatto che sono costretto ad accendere il laser al 10%

MitchBradley, 2022-07-24:
Use IS2_STATIC mode with lasers.  IS2_STREAM mode has extra latency that prevents the laser power from synchronizing properly with the motion.  Use FluidNC version 3.4.4 instead of later versions.  In later versions, we switched over to the new core code from Espressif and we are still debugging problems with that code.

distebia, 2022-07-24:
Grazie per la dritta. Attualmente uso la 3.4.9. ho un dubbio però. Ho scaricato la 3.4.9 e ho messo i dentro il repository di makerbase/fluidnc, decomprimendo il file zip nella robot di fluidnc. Poi ho lanciato installa wifi. Forse sbaglio questo passaggio?

MitchBradley, 2022-07-24:
Do not use 3.4.9.

I do not now what you mean by "il repository di makerbase/fluidnc".  I did not know that makerbase had such a repository.

Skorpi08, 2022-07-24:
https://github.com/makerbase-mks/FluidNC

distebia, 2022-07-24:
https://github.com/makerbase-mks?tab=repositories&q=Fluid&type=&language=&sort=

distebia, 2022-07-24:
Ma ho installato correttamente il repository di makerbase dentro la root di fluidnc?

MitchBradley, 2022-07-24:
Ignore the makerbase repository. The only thing of value in the Makerbase repo is just the example file MKS_DLC32_BASE.yaml, and that file will have to be edited for different machine setups, as you have already done. You should install FluidNC from a [release installer](https://github.com/bdring/FluidNC/releases).

After you have installed FluidNC, you can upload your YAML config file either from WebUI or with FluidTerm via the CTRL-U upload command.

distebia, 2022-07-24:
Quindi non devo caricare il loro repository? Pensavo potesse essere di aiuto a definire i Pin ecc.... Domani provo con la 3.4.4 e vediamo che succede. Un'altra cosa, devo cancellare prima tutto con installfs? Ho ricaricato il firmware makerbase

MitchBradley, 2022-07-24:
The Makerbase FluidNC repository is useless.  They did not make any changes to FluidNC.  They just took a snapshot of a very old version of FluidNC and added an example configuration file.  The current bdring/FluidNC repo has two DLC32 examples, and there are many more DLC32 configurations that have been auto-generated by the automatic translation program.

You do not need to fork the entire source code just to define pins.

You should do an installfs when migrating from Makerbase/Grbl_ESP32 to FluidNC.  There is nothing magic about installfs; it just deletes everything in the SPIFFS filesystem and replaces index.html.gz with an up to date version. You could do that manually by using WebUI or FluidTerm to delete individual files and upload new ones.  The only important files are index.html.gz, which is necessary for the full WebUI, and your own custom config.yaml that describes your machine.

If there is no index.html.gz, you can still use the browser to upload files, because FluidNC includes a built-in very simple WebUI subset that only has file management features.

distebia, 2022-07-25:
sembra che tutto funzioni a meraviglia, grazie per l'aiuto. non so quali sono i pin sulla mks per fare un sistema autoquadrante (2 motori e 2 finecorsa per l'asse y) ma, eventualmente, come dovrei configurare il file yaml?

MitchBradley, 2022-07-25:
Can somebody else help with this?  I have already spent more time on it than I can spare.  I think he wants to make an auto-squaring XYY system (DLC32 has only 3 stepper sockets).

daxliniere, 2022-07-25:
Hi @distebia,

> sembra che tutto funzioni a meraviglia, grazie per l'aiuto. non so quali sono i pin sulla mks per fare un sistema autoquadrante (2 motori e 2 finecorsa per l'asse y) ma, eventualmente, come dovrei configurare il file yaml?

Okay, sounds like this thread covers multiple issues. Assuming it's just the auto-squaring Y-axis you're trying to get working now, your config.yaml file should look something like this. I have highlighted the important lines:

```
  x:
    steps_per_mm: 157.750
    max_rate_mm_per_min: 18000.000
    acceleration_mm_per_sec2: 1500.000
    max_travel_mm: 325.000
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 5000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.36
      hard_limits: false
      pulloff_mm: 2.000
      stepstick:
        step_pin: I2SO.1
        direction_pin: I2SO.2

  y:
    steps_per_mm: 157.750
    max_rate_mm_per_min: 12000.000
    acceleration_mm_per_sec2: 300.000
    max_travel_mm: 220.000
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 5000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    **motor0:**
      limit_neg_pin: gpio.35
      hard_limits: false
      pulloff_mm: 2.000
      stepstick:
        **step_pin: I2SO.5
        direction_pin: I2SO.6**
        
    **motor1:**
      limit_neg_pin: gpio.34
      hard_limits: false
      pulloff_mm: 1.000
      stepstick:
        **step_pin: I2SO.3
        direction_pin: I2SO.4**
```

distebia, 2022-07-25:
quindi collego sia il y2 che il finecorsa y2 all'attacco del motore Z e finecorsa Z?

daxliniere, 2022-07-25:
Correct.
You will need up change the pin assignments for Y2, of course. (Z settings -> Y2)

distebia, 2022-07-25:
ma non esiste un altro modo per non perdere la Z? con arduino uno il sistema c'era. ma alla fine la limitazione è data dalla mks. nel pcb sono presenti X Y1 Y2 e Z. per i driver esterni solo 3 prese. e se metto un populo attaccato sopra Z?

daxliniere, 2022-07-25:
You have bought a 3-axis control board. You are asking for 4-axes... do you see a problem here?

daxliniere, 2022-07-25:
Yes, it is possible to add an extra axis, but you will need to do some soldering, or at the very least, crimp some cables. If you are using the MKS LCD display, then you really only have 3 pins available (SDA, SCL and buzzer), but buzzer will still require a modification.
Are you using the MKS LCD?

distebia, 2022-07-25:
No, non ho il display

daxliniere, 2022-07-25:
I jsut found out that, even though you would have more pins available, they are not all coming from the same 'source' (GPIO vs I2S). It would be possible, but would require a lot of work cutting pins and soldering.
Another user suggested looking at the MKS TinyBee. It's not very expensive and has sockets for 5 stepper motor drivers.

One example: https://www.aliexpress.com/item/1005004268076123.html

bdring, 2022-07-25:
There are many other hardware options.

<https://github.com/bdring/FluidNC/wiki/Hardware-that-Runs-FluidNC>

distebia, 2022-07-26:
> I jsut found out that, even though you would have more pins available, they are not all coming from the same 'source' (GPIO vs I2S). It would be possible, but would require a lot of work cutting [...]

Lo so che sono su una 3 assi ma pensavo questo....non utilizzo il display e lì ho un pò di pin da usare, quello che mi manca è l'attacco motore. nel pcb posso mettere su Y un dvr8825 e ho 2 uscite per il motore. in questo modo posso avere u motori su Y. quello che non so è se i motori sono semplicemente sdoppiati o si riferiscono a pin differenti circa step e dir. per il finecorsa posso usare qualcosa sul connettore display

MitchBradley, 2022-07-26:
It is possible to drive two motors from the same stepper driver by wiring them in parallel.  It is not ideal but it can work. Squaring will not be possible with that setup.  The DLC32 board cannot be used to drive 4 independent motors.  That board uses a shift register connected to the I2S bus to drive motors, and there are no spare shift register outputs.  The extra GPIOs cannot be used to drive a fourth motor, because FluidNC requires that all stepping must be done with the same method, either I2S or GPIO.  Mixing them is not possible because the steps would not synchronize correctly.  If you want to use 4 motors with 4 stepper driver modules, the only options are to buy a different board that supports 4 stepper drivers, or rewire the DLC32 board.  Based on the questions that you have asked so far, I believe that rewiring the board is beyond your skill level and we are unwilling to spend the large amount of time it would take to teach you how to rewire it.  4-way boards are inexpensive, and we have already spent time that is much more valuable that the cost of a different board.

I am closing this ticket because we have already helped you as much as we can.

closed 2022-07-26 by MitchBradley

distebia, 2022-07-26:
Tanto per curiosità...orientativamente...come dovrei ricablare la scheda? dissaldare componenti? oppure cablaggi esterni? non sono un mago di elettronica ma è la terza macchina che realizzo partendo da arduino uno, opto a forcella come finecorsa, tb6600, sensori di prossimità, 4n25, relè 5v. asse z con fuoco automatico, w1209 e sensore di flusso ecc.. L'ultima macchina che ho fatto è una 1300x1000 di piano con 1600x1200 di misure esterne, co2 50w+diodo 445nm. ho anche aggiunto l'asse rotante sdoppiando la Z. insomma diciamo che qualcosina so fare.

MitchBradley, 2022-07-26:
Look up the schematic for the MKS DLC32 board.  It is on the web and can be found with a little searching.  Disconnect the XYZ_EN signal from the stepper sockets, rewire it to the step pin of a new socket, wire the BEEPER signal to the dir pin of a new socket, and make a configuration file that uses i2so.0 as step, i2so.7 as dir for the new motor.

distebia, 2022-07-26:
forse ho capito...scollego i pin EN da tutte le prese per driver esterni (dove vado a tb6600), ne prendo uno qualsiasi ed ho ottenuto STEP. prendo il segnale BEEPER dal display ed ho ottenuto DIR. dopodichè STEP e DIR e GND vanno ad un nuovo tb6600 ed ho un altro motore attaccato. e per i finecorsa?

MitchBradley, 2022-07-26:
What you said about STEP and DIR is correct.  For limit switches, read the schematic and find some signal that is not used.

distebia, 2022-07-26:
ne ho molti liberi, nella sonda (gpio.22, nel display (gpio.25, gpio.26 ecc.). uno vale l'altro o ne devo prendere uno con qualche requisito? naturalmente modifico il file yaml come indicato sopra assegnando esempio:
y:
    steps_per_mm: 80.194
    max_rate_mm_per_min: 10000.000
    acceleration_mm_per_sec2: 350.000
    max_travel_mm: 898.000
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 1000.000
      seek_mm_per_min: 5000.000
      settle_ms: 250.000
      seek_scaler: 1.100
      feed_scaler: 1.100
motor0:
    limit_neg_pin: gpio.35:low
    limit_pos_pin: NO_PIN
    limit_all_pin: NO_PIN
    hard_limits: false
    pulloff_mm:1.000
    stepstick:
    step_pin: I2SO.5
    direction_pin: I2SO.6

**motor1:**
    limit_neg_pin: **gpio.25:low**       <-- preso da EXP1
    limit_pos_pin: NO_PIN
    limit_all_pin: NO_PIN
    hard_limits: false
    pulloff_mm:1.000
    stepstick:
    step_pin: **i2so.0**                   <-- preso da XYZ_EN
    direction_pin: **i2so.7**            <-- preso da BEEPER EXP1 

fin qua tutto corretto? o sbaglio?
inoltre come dovrei commentare la riga
**axes:
  shared_stepper_disable_pin: i2so.0**
visto che il i2so.0 l'ho usato per step?

MitchBradley, 2022-07-26:
Delete the shared_stepper_disable line; disabling the drivers will not be possible with this setup.

The probe pin is probably a good one to use as a limit because it is suitable for input use.

Please look at the schematic.

distebia, 2022-07-26:
ok, elimino la riga shared_stepper_disable, tanto utilizzo idle_ms: 255  per tenere i motori sempre alimentati. ma devo invertire la logica in yaml? tipo $4 di grbl? ok per il pin della sonda gpio.22. ultima cosa, con il tester ho fatto dei test di continuità fra i pin gnd (e avevo continuità fra di loro) ma non avevo continuità fra i pin EN. il mio i2so.0 lo devo prendere dal connettore Y? o uno vale l'altro?

MitchBradley, 2022-07-26:
Read the schematic and figure it out.

distebia, 2022-07-26:
ok, lo prendo da Y, mi sembra logico. e per quanto riguarda di invertire la logica di EN tipo $4?