bdring/FluidNC #543: Problem:  Steppers Motors Won't Run - Fail Test At Startup

Andycart, 2022-07-26:
### Controller Board

6-pack

### Help From Board Vendor

- [ ] Yes
- [ ] No
- [X] Not Applicable

### Machine Description

Cartesian CO2 laser cutter. 2 steppers on the Z axis, 2 steppers on the Y axis, one stepper on the X axis. Automatic air assist using the relay board on the 6 pack. Laser enable and PWM via the 4 Ch 5 volt O/P board. Microswitch limit switches (NC). One on X, one on Z and 2 on Y. (for auto squaring). Original GRBLESP32 machine file attached. This is the file I ran through the GitHub converter to create the .yaml file. The machine works with GRBLESP32 and this machine file.

[Machine.txt](https://github.com/bdring/FluidNC/files/9190117/Machine.txt)
.

### Configuration file

```yaml
name: 6 Pack 2130 XYYZZ 5v Laser Mist
board: 6-pack

kinematics:
  Cartesian:

stepping:
  engine: I2S_stream
  idle_ms: 250
  dir_delay_us: 0
  pulse_us: 4
  disable_delay_us: 0

axes:
  x:
    steps_per_mm: 100.000
    max_rate_mm_per_min: 1000.000
    acceleration_mm_per_sec2: 200.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 2
      mpos_mm: 0.000
      positive_direction: true
      settle_ms: 250.000
      seek_mm_per_min: 2000.000
      feed_mm_per_min: 200.000
      seek_scaler: 1.100
      feed_scaler: 5.000
    motor0:
      limit_all_pin: gpio.33
      hard_limits: false
      tmc_2130:
        direction_pin: i2so.1
        step_pin: i2so.2
        disable_pin: i2so.0
        run_amps: 0.250
        hold_amps: 0.125
        microsteps: 16
        stallguard: 16
        stallguard_debug: false
        run_mode: CoolStep
        homing_mode: CoolStep
        use_enable: false
        r_sense_ohms: 0.110
        cs_pin: i2so.3:low
    motor1:
      null_motor:
  y:
    steps_per_mm: 100.000
    max_rate_mm_per_min: 1000.000
    acceleration_mm_per_sec2: 200.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 2
      mpos_mm: 0.000
      positive_direction: true
      settle_ms: 250.000
      seek_mm_per_min: 2000.000
      feed_mm_per_min: 200.000
      seek_scaler: 1.100
      feed_scaler: 5.000
    motor0:
      limit_all_pin: gpio.32
      hard_limits: false
      tmc_2130:
        direction_pin: i2so.4
        step_pin: i2so.5
        disable_pin: i2so.7
        run_amps: 0.250
        hold_amps: 0.125
        microsteps: 16
        stallguard: 16
        stallguard_debug: false
        run_mode: CoolStep
        homing_mode: CoolStep
        use_enable: false
        r_sense_ohms: 0.110
        cs_pin: i2so.6:low
    motor1:
      limit_all_pin: gpio.35
      hard_limits: false
      tmc_2130:
        direction_pin: i2so.9
        step_pin: i2so.10
        disable_pin: i2so.8
        run_amps: 0.250
        hold_amps: 0.125
        microsteps: 16
        stallguard: 16
        stallguard_debug: false
        run_mode: CoolStep
        homing_mode: CoolStep
        use_enable: false
        r_sense_ohms: 0.110
        cs_pin: i2so.11:low
  z:
    steps_per_mm: 100.000
    max_rate_mm_per_min: 1000.000
    acceleration_mm_per_sec2: 200.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 1
      mpos_mm: 0.000
      positive_direction: false
      settle_ms: 250.000
      seek_mm_per_min: 2000.000
      feed_mm_per_min: 200.000
      seek_scaler: 1.100
      feed_scaler: 5.000
    motor0:
      limit_all_pin: gpio.34
      hard_limits: false
      tmc_2130:
        direction_pin: i2so.12
        step_pin: i2so.13
        disable_pin: i2so.15
        run_amps: 0.250
        hold_amps: 0.125
        microsteps: 16
        stallguard: 16
        stallguard_debug: false
        run_mode: CoolStep
        homing_mode: CoolStep
        use_enable: false
        r_sense_ohms: 0.110
        cs_pin: i2so.14:low
    motor1:
      tmc_2130:
        direction_pin: i2so.17
        step_pin: i2so.18
        disable_pin: i2so.16
        run_amps: 0.250
        hold_amps: 0.125
        microsteps: 16
        stallguard: 16
        stallguard_debug: false
        run_mode: CoolStep
        homing_mode: CoolStep
        use_enable: false
        r_sense_ohms: 0.110
        cs_pin: i2so.19:low

i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5

control:

coolant:
  mist_pin: gpio.14
  delay_ms: 1000.000

probe:
  check_mode_start: false

macros:
  startup_line0: 
  startup_line1: 
  macro0: 
  macro1: 
  macro2: 
  macro3: 

start:
  must_home: true
  check_limits: true
  deactivate_parking: false

user_outputs:

laser:
  tool_num: 0
  speed_map: 0=0.0% 1000=100.0%
  output_pin: gpio.26
  enable_pin: gpio.4
  disable_with_s0: false
  s0_with_disable: false
  pwm_hz: 5000
arc_tolerance_mm: 0.002
junction_deviation_mm: 0.010
verbose_errors: false
report_inches: false
enable_parking_override_control: false
use_line_numbers: false
```

### Startup Messages

```text
Resetting MCU
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:12812
load:0x40080400,len:3032
entry 0x400805e4

[MSG:INFO: FluidNC v3.5.0]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is SPIFFS]
[MSG:INFO: Configuration file:config.yaml]
[MSG:INFO: Machine 6 Pack 2130 XYYZZ 5v Laser Mist]
[MSG:INFO: Board 6-pack]
[MSG:INFO: I2SO BCK:gpio.22 WS:gpio.17 DATA:gpio.21]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN]
[MSG:INFO: Stepping:I2S_stream Pulse:4us Dsbl Delay:0us Dir Delay:0us Idle Delay:250ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (-300.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2130 Step:I2SO.2 Dir:I2SO.1 CS:I2SO.3:low Disable:I2SO.0 Index:-1 R:0.110]
[MSG:INFO:     All Limit gpio.33]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Y (-300.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2130 Step:I2SO.5 Dir:I2SO.4 CS:I2SO.6:low Disable:I2SO.7 Index:-1 R:0.110]
[MSG:INFO:     All Limit gpio.32]
[MSG:INFO:   Motor1]
[MSG:INFO:     tmc_2130 Step:I2SO.10 Dir:I2SO.9 CS:I2SO.11:low Disable:I2SO.8 Index:-1 R:0.110]
[MSG:INFO:     All Limit gpio.35]
[MSG:INFO: Axis Z (0.000,300.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2130 Step:I2SO.13 Dir:I2SO.12 CS:I2SO.14:low Disable:I2SO.15 Index:-1 R:0.110]
[MSG:INFO:     All Limit gpio.34]
[MSG:INFO:   Motor1]
[MSG:INFO:     tmc_2130 Step:I2SO.18 Dir:I2SO.17 CS:I2SO.19:low Disable:I2SO.16 Index:-1 R:0.110]
[MSG:ERR: X Axis driver test failed. Check motor power]
[MSG:ERR: Y Axis driver test failed. Check motor power]
[MSG:ERR: Y2 Axis driver test failed. Check motor power]
[MSG:ERR: Z Axis driver test failed. Check motor power]
[MSG:ERR: Z2 Axis driver test failed. Check motor power]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Laser Spindle Ena:gpio.4 Out:gpio.26 Freq:5000Hz Res:13bits Laser mode:On]
[MSG:INFO: Using spindle Laser]
[MSG:INFO: Mist coolant gpio.14]
[MSG:INFO: Connecting to STA SSID:SKY3DA89]
[MSG:INFO: Connecting.]
[MSG:INFO: Connected - IP is 192.168.0.76]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl 3.5 [FluidNC v3.5.0 (wifi) '$' for help]
[MSG:INFO: '$H'|'$X' to unlock]
[MSG:INFO: Caution: Unlocked]
```

### User Interface Software

WebUI, Fluidterm

### What happened?

The motors don't appear to have any power. I get the start up error for driver power (which is verified OK) and if I command a jog nothing happens.

bdring, 2022-07-26:
How did you make the config file? Did you use the translator tool or create it yourself?

I think your cs_pin definitions should not have the ":low" attribute.

[This is a similar config file for reference](https://github.com/bdring/6-Pack_CNC_Controller/blob/main/FluidNC_configs/6P_2130_XXYYZ_Rly.yaml).

Andycart, 2022-07-26:
Hi Bart

I ran it through your conversion tool on GitHub

Regards

Andy

This email has been checked for viruses by AVG.
https://www.avg.com

bdring, 2022-07-26:
Try fixing the cs_pin definitions.

Andycart, 2022-07-26:
Hi Bart

 

Tried that:-

 

motor0:

      limit_all_pin: gpio.33

      hard_limits: false

      tmc_2130:

        direction_pin: i2so.1

        step_pin: i2so.2

        disable_pin: i2so.0

        run_amps: 0.250

        hold_amps: 0.125

        microsteps: 16

        stallguard: 0

        stallguard_debug: false

        run_mode: CoolStep

        homing_mode: CoolStep

        use_enable: false

        r_sense_ohms: 0.110

        cs_pin: i2so.3

 

No Joy

 

Andy

 

 

 <http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=emailclient> 

Virus-free.  <http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=emailclient> www.avg.com

MitchBradley, 2022-07-26:
Try it with  version 3.4.4 in case the problem is with the new Arduino framework.  We are having lots of problems with the new toolchain and vendor code.

bdring, 2022-07-26:
Are you 100% sure the main power is on? Is the green LED on?

Andycart, 2022-07-26:
Yes 😊

https://www.avg.com

Andycart, 2022-07-26:
I’m not using the Arduino IDE. Just the install-fs.bat and the install-wifi.bat?

https://www.avg.com

MitchBradley, 2022-07-26:
We build out code on top of the Arduino framework and other libraries from Espressif.  The Arduino framework is not the same thing as the Arduino IDE.  There was a major update to all that stuff.  We switched to the new versions starting with 3.4.5, in order to pick up some features that we will need soon.  There were some deep changes that have caused problems.  We are still sorting them out.  Please, grab the 3.4.4 release and try it.  If that works, it means that the problem you are having is related to those changes.

Andycart, 2022-07-26:
Will do and I’ll report back

https://www.avg.com

Andycart, 2022-07-27:
Hi

 

I reverted to 3.4.4 and got exactly the same result i.e no motor current. I programmed another ESP32 with GRBLESP32 using the machine file Bart sent me and everything works Ok. It would seem that the issue is with the .yaml file from the converter

 

Andy

 

 

 <http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=emailclient> 

Virus-free.  <http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=emailclient> www.avg.com

bdring, 2022-07-27:
I am not sure what your problem is. You did not post the revised file, so I cannot comment on that.

I manually created a file based on the initial post with the tweaks I mentioned. It worked for me.

You may need to tweak a few values, like motor current, etc.

```
[MSG:INFO: Axis X (-300.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2130 Step:I2SO.2 Dir:I2SO.1 CS:I2SO.3 Disable:I2SO.0 Index:-1 R:0.110]
[MSG:INFO:     All Limit gpio.33]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Y (-300.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2130 Step:I2SO.5 Dir:I2SO.4 CS:I2SO.6 Disable:I2SO.7 Index:-1 R:0.110]
[MSG:INFO:     All Limit gpio.32]
[MSG:INFO:   Motor1]
[MSG:INFO:     tmc_2130 Step:I2SO.10 Dir:I2SO.9 CS:I2SO.11 Disable:I2SO.8 Index:-1 R:0.110]
[MSG:INFO:     All Limit gpio.35]
[MSG:INFO: Axis Z (0.000,300.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2130 Step:I2SO.13 Dir:I2SO.12 CS:I2SO.14 Disable:I2SO.15 Index:-1 R:0.110]
[MSG:INFO:     All Limit gpio.34]
[MSG:INFO:   Motor1]
[MSG:INFO:     tmc_2130 Step:I2SO.18 Dir:I2SO.17 CS:I2SO.19 Disable:I2SO.16 Index:-1 R:0.110]
[MSG:INFO: X Axis driver test passed]
[MSG:INFO: Y Axis driver test passed]
[MSG:INFO: Y2 Axis driver test passed]
[MSG:INFO: Z Axis driver test passed]
[MSG:INFO: Z2 Axis driver test passed]
```

I have added a ticket to look at the cs_pin in the conversion program.

```yaml
board: 6 Pack
name: TMC2130 XXYYZ 5v Laser Mist
meta: 07/27/2022 B. Dring for Andy Cart

kinematics:
  Cartesian:

stepping:
  engine: I2S_Static
  idle_ms: 250
  dir_delay_us: 0
  pulse_us: 4
  disable_delay_us: 0

axes:
  x:
    steps_per_mm: 100.000
    max_rate_mm_per_min: 1000.000
    acceleration_mm_per_sec2: 200.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 2
      mpos_mm: 0.000
      positive_direction: true
      settle_ms: 250.000
      seek_mm_per_min: 2000.000
      feed_mm_per_min: 200.000
      seek_scaler: 1.100
      feed_scaler: 5.000
    motor0:
      limit_all_pin: gpio.33
      hard_limits: false
      #socket 1
      tmc_2130:
[... 163 lines cut ...]
  macro2: 
  macro3: 

start:
  must_home: true
  check_limits: true
  deactivate_parking: false

user_outputs:

laser:
  tool_num: 0
  speed_map: 0=0.0% 1000=100.0%
  output_pin: gpio.26
  enable_pin: gpio.4
  disable_with_s0: false
  s0_with_disable: false
  pwm_hz: 5000
arc_tolerance_mm: 0.002
junction_deviation_mm: 0.010
verbose_errors: false
report_inches: false
enable_parking_override_control: false
use_line_numbers: false
```

Andycart, 2022-07-28:
Thanks Bart

I’ll give that config a go.

Regards

Andy

https://www.avg.com

closed 2022-08-17 by bdring