bdring/FluidNC #545: Problem: Steppers no longer move after Firmware update to v3.5.0

kaptaink-cg, 2022-07-29:
### Controller Board

FluidNC Pen Laser Controller TMC2209

### Help From Board Vendor

- [ ] Yes
- [ ] No
- [X] Not Applicable

### Machine Description

[ZenXY 2 ](https://docs.v1engineering.com/zenxy/)  CoreXY Belting system

### Configuration file

```yaml
Currently not using a config file.
```

### Startup Messages

```text
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:12812
load:0x40080400,len:3032
entry 0x400805e4

[MSG:INFO: FluidNC v3.5.0]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is SPIFFS]
[MSG:WARN: Cannot open configuration file:Xconfig.yaml]
[MSG:INFO: Using default configuration]
[MSG:INFO: Axes: using defaults]
[MSG:INFO: Machine Default (Test Drive)]
[MSG:INFO: Board None]
[MSG:INFO: SPI not defined]
[MSG:INFO: Stepping:RMT Pulse:4us Dsbl Delay:0us Dir Delay:0us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (-1000.000,0.000)]
[MSG:INFO: Axis Y (-1000.000,0.000)]
[MSG:INFO: Axis Z (-1000.000,0.000)]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Using spindle NoSpindle]
[MSG:INFO: STA SSID is not set]
[MSG:INFO: AP SSID FluidNC IP 192.168.0.1 mask 255.255.255.0 channel 1]
[MSG:INFO: AP started]
[MSG:INFO: WiFi on]
[MSG:INFO: Captive Portal Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl 3.5 [FluidNC v3.5.0 (wifi) '$' for help]
[MSG:INFO: '$H'|'$X' to unlock]
```

### User Interface Software

WebUI

### What happened?

I tested the board when it first arrived and both stepper motors worked (although in cartesian mode)
I updated the firmware, loaded a config file, and the stepper motors no longer move.  I deleted the config file to see if that was causing an issue, but the same thing happens.  
I clear the startup alarm in the WebUI, tell it to move X or Y, but nothing happens.
The steppers "lock" when I boot up the board, so I don't think the drivers are damaged.

### Other Information

System Status Info:

$CD
board: None
name: Default (Test Drive)
meta:
stepping:
  engine: RMT
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 0
  disable_delay_us: 0
  segments: 12

axes:
  shared_stepper_disable_pin: NO_PIN
  shared_stepper_reset_pin: NO_PIN
  x:
    steps_per_mm: 80.000
    max_rate_mm_per_min: 1000.000
    acceleration_mm_per_sec2: 25.000
    max_travel_mm: 1000.000
    soft_limits: false
    homing:
      cycle: 0
      allow_single_axis: true
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 50.000
      seek_mm_per_min: 200.000
      settle_ms: 250
      seek_scaler: 1.100
      feed_scaler: 1.100

  y:
    steps_per_mm: 80.000
    max_rate_mm_per_min: 1000.000
    acceleration_mm_per_sec2: 25.000
    max_travel_mm: 1000.000
    soft_limits: false
    homing:
      cycle: 0
      allow_single_axis: true
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 50.000
      seek_mm_per_min: 200.000
      settle_ms: 250
      seek_scaler: 1.100
      feed_scaler: 1.100

  z:
    steps_per_mm: 80.000
    max_rate_mm_per_min: 1000.000
    acceleration_mm_per_sec2: 25.000
    max_travel_mm: 1000.000
    soft_limits: false
    homing:
      cycle: 0
      allow_single_axis: true
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 50.000
      seek_mm_per_min: 200.000
      settle_ms: 250
      seek_scaler: 1.100
      feed_scaler: 1.100

kinematics:
  Cartesian:

spi:
  miso_pin: NO_PIN
  mosi_pin: NO_PIN
  sck_pin: NO_PIN

sdcard:
  cs_pin: NO_PIN
  card_detect_pin: NO_PIN

control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

coolant:
  flood_pin: NO_PIN
  mist_pin: NO_PIN
  delay_ms: 0

probe:
  pin: NO_PIN
  check_mode_start: true

macros:
  startup_line0:
  startup_line1:
  macro0:
  macro1:
  macro2:
  macro3:

start:
  must_home: true
  deactivate_parking: false
  check_limits: false

user_outputs:
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 5000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000
  digital0_pin: NO_PIN
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: NO_PIN

arc_tolerance_mm: 0.002
junction_deviation_mm: 0.010
verbose_errors: false
report_inches: false
enable_parking_override_control: false
use_line_numbers: false
planner_blocks: 16
NoSpindle:

ok

Avataar120, 2022-07-29:
Without a proper config file, no chance to get your board working.
Config file purpose is to tell the firmware which PIN of the ESP32 shall be used for stepper control, inputs, ...
If you have updated a vendor board, you'll have to find out what were the good parameters and fill them in a config.yaml

bdring, 2022-07-29:
Thanks for uploading the info. There are no motors defined in your config. Try this one.

I would like to create the perfect Zen Sand config file. If this works and you do any tweaking or tuning, please let me know and I will update it and make it accessible for future sand people.

```yaml
name: "TMC2209 ZenXY"
board: "FluidNC Pen/Laser 2209"

stepping:
  engine: RMT
  idle_ms: 255
  dir_delay_us: 1
  pulse_us: 2
  disable_delay_us: 0
  
kinematics:
  corexy:

axes:
  shared_stepper_disable_pin: gpio.13:high
  
  x:
    steps_per_mm: 80
    max_rate_mm_per_min: 1000
    acceleration_mm_per_sec2: 25
    max_travel_mm: 1000
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100
    
    motor0:
      limit_neg_pin: gpio.36:low
      #Diag pin is gpio.34
      tmc_2209:
        uart:
          txd_pin: gpio.17
          rxd_pin: gpio.16
          rts_pin: NO_PIN
          cts_pin: NO_PIN
          baud: 115200
          mode: 8N1

        addr: 0
        r_sense_ohms: 0.110
        run_amps: 0.800
        hold_amps: 0.500
        microsteps: 16
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: StealthChop
        homing_mode: StealthChop
        use_enable: false
        direction_pin: gpio.12
        step_pin: gpio.14
        disable_pin: NO_PIN
    
    
    motor1:
      null_motor:

  y:
    steps_per_mm: 80
    max_rate_mm_per_min: 1000
    acceleration_mm_per_sec2: 25
    max_travel_mm: 1000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.39:low
      #diag pin is gpio.35
      tmc_2209:
        addr: 1
        r_sense_ohms: 0.110
        run_amps: 0.500
        hold_amps: 0.500
        microsteps: 16
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: StealthChop
        homing_mode: StealthChop
        use_enable: false
        direction_pin: gpio.26
        step_pin: gpio.25
        disable_pin: NO_PIN
    motor1:
      null_motor:

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5
  card_detect_pin: NO_PIN
```

bdring, 2022-07-29:
Actually, the file you uploaded was the Test Drive (no I/O).

kaptaink-cg, 2022-07-29:
They MOVE!! :)
This seems to work pretty well.  I'm getting a weird intermittent clicking
noise when the X-Axis moves in either direction, like the belt is skipping
teeth.  No issues when the Y-Axis moves either direction.
Not sure what could be causing that, but I'll troubleshoot on my end first.

A few more questions before you publish this for other ZenXY2 Sand Table
users.

   - The hardware kit sold for this build includes two optical end stops
   (for home position) that I have wired to the board.
      1. Do these need to be enabled in the config file as well?
      2. Is it possible to have the machine automatically home when it is
      powered on, or is that better handled in the G-Code startup lines?
      3. If it is possible, can we set the machine to always home Y before
      X?  (This is due to the locations of the endstops.  If X is
homed first it
      will fail.)
      4. Is it possible to disable the alarm that needs to be cleared on
      board power up?

Ideally, I would like to power on the board and have the machine
automatically home and be ready to go.

Thanks for your help!  I've built a CNC before but FluidNC and CoreXY are
very new to me.

bdring, 2022-07-30:
1. I think the optical end stops need a 3.3v voltage source when used with the ESP32. There is no easy way to access that.
2. Homing on startup is not generally done on CNC machines. I don't think out startup macros allow that.
3. Yes. [See the wiki](http://wiki.fluidnc.com/en/config/axes)
4. Yes. [See the wiki](must_home)

kaptaink-cg, 2022-07-30:
These endstops are rated from 2.7V to 5V according to the[ V1 Forums.](https://forum.v1engineering.com/t/zenxy-control-board-recommendation/27789/13)
I'll dig back into the WIki for the rest. Thanks.
Once I get a fully working Config File I can send it back to you if you'd like.

bdring, 2022-07-30:
You must use 3.3V, because that is all the ESP32 can handle unless you level shift one your own.

The V1 ZenXY page appears to show normal switches as an option.

https://www.v1engineering.com/zenxy/

MitchBradley, 2022-09-26:
No updates in a long time; closing

closed 2022-09-26 by MitchBradley