bdring/FluidNC #549: Feature: Negative feed rate override

ipsod, 2022-08-02:
### Machine Context

LinuxCNC has this feature with their "adaptive feed" functionality.  I need it because I'm trying to build "pseudo-manual" machines, where I can define a path that the machine should be able to take using g-code, and then manually jog that path, forward and backward, by controlling feed rate.

I primarily want this for a camera rail/gimbal, but I believe it will be useful for automating many processes that would otherwise be difficult.

### Feature Description

Instantaneous feed rate changes, not just slowing down or speeding up feed, but walking the g-code in reverse.  This would be controlled either manually, or through pressure sensors of sorts (like LinuxCNC uses them, for adaptive feed, where the feed rate is proportional to the forces actively sensed in the system).

### Other Approaches

LinuxCNC can do it, but FluidNC runs on smaller, less expensive, less complicated, easier to source hardware, and seems to be a less complicated system overall, since it all runs on an MCU.

### How I Can Help

I will do what I can.  I would be happy to write documentation, do testing, or anything.  If I make money from it, and I do hope to, I'll share some.

MitchBradley, 2022-08-02:
Based on what I know about the current design of the planner and the way gcode is streamed, buffered and pipelined through various phases, I think this would be very difficult to implement.

ipsod, 2022-08-03:
@MitchBradley  Ah, that's too bad. Do you know if (not-negative) instantaneous feed rate change is a current feature?

bdring, 2022-08-03:
Real time feed and speed override is a feature.

ipsod, 2022-08-03:
@bdring Cool.  Real time pause is a feature as well, right?  Is there a way to cancel the current running program, after pausing?

bdring, 2022-08-03:
Realtime feed hold is a feature. You can resume or cancel. It is the same interface as regular Grbl.

The wiki is a good source for some of this information 

<https://github.com/bdring/FluidNC/wiki/Overrides>

ipsod, 2022-08-03:
Ok, thank you.  I searched the wiki for the word "cancel", but didn't find it.  https://github.com/bdring/FluidNC/search?q=cancel&type=wikis

MitchBradley, 2022-08-03:
To stop a job, whether paused or not, you use the reset realtime character CTRL-X.

closed 2022-08-17 by bdring