bdring/FluidNC #550: Feature: Where would be best to insert my own code for jog control?

ipsod, 2022-08-04:
### Machine Context

I'm making a machine that is initially "taught" by manual jogging and creating waypoints, then running through those waypoints.

### Feature Description

Because I'll be doing so much manual jogging, I want to have it work really well.  I've already got code that does that (though I plan to rewrite it using FastAccelStepper).

### Other Approaches

Where would be best to put my code?  It looks like it would be easy enough just to insert it in the main loop, before the call to protocol_main_loop(), maybe with its own little loop so that reset_variables() isn't getting called every time I switch.  Would some other way be better?

### How I Can Help

n/a (I think?)

bdring, 2022-08-04:
FastAccelStepper and FluidNC are different animals. There is no way to mix them. 

Either use FluindNC as a motion processor with a specialized app to control it or write the whole thing in FastAccelStepper.

FluidNC is designed for CNC with tight multiaxes coordination and motion planning for good cornering.

ipsod, 2022-08-04:
I've already tried writing my own system, and I know that it's a task that requires more time/ability than I currently have.

I don't understand why they can't be mixed.  I would only be using one or the other at a given time.  I understand that they're entirely different.  Why could I not enter one state to jog with FastAccelStepper, get all of the coordinates that I need to move to, then enter FluidNC's `protocol_main_loop` and feed those coordinates as g-code commands, then exit protocol_main_loop and go back into my machine-jogging loop?  I can, at a higher level, do any coordinate management I need to (ie: if FastAccelStepper moves motors, then FluidNC would need to be made aware of the machine's new position, and vice-versa).

That said, would you say that jogging with FluidNC can be made to be excellent?  I wouldn't mind rewriting to use strictly FluidNC, if so.  With my current jogging program, I can get real time control of speed/accel - as good of joystick-driven motion control as I could hope for - and I really don't want to take a step backward.

bdring, 2022-08-04:
Create a fork, give it a try. It is not something we would consider working on.

We support the enhanced jogging as described here. 

https://github.com/gnea/grbl/wiki/Grbl-v1.1-Jogging

ipsod, 2022-08-04:
Ok, thank you.

ipsod, 2022-08-04:
Might there be a better way to improve jogging, that you would support?  The GRBL document on jogging describes the problem with it [here](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Jogging#joystick-implementation)- input lag that averages somewhere between 0.5 and 1.0 seconds (aside from the cancel command, which triggers the stop instantly).  GRBL accepts their half-solution because they don't have room for better on their platform, but the ESP32 has plenty of room, surely?

bdring, 2022-08-04:
If you send a small jog the machine will move there with no lag. If you send another jog before the first one ends, it will join the jogs together and go to the end of the second without stopping between them (assuming the jogs do not create a sharp corner). There is no lag.

With this system, you can use a continuous stream of tiny jogs. There should be no lag assuming you are not exceeding the communication rate. You can cancel the entire jog buffer at any time.

At no time can you alter the acceleration while moving. Our #1 priority is coordinated multiaxis motion control.

It is very unlikely we will ever be interested what you are proposing. Please take our free and open source code and alter it however you want. Please don't ask us to do it for you.

ipsod, 2022-08-04:
Ok, thank you.

closed 2022-08-04 by ipsod

MitchBradley, 2022-08-04:
If you really wanted to do something cool, hook up a camera and make a system where you can point at waypoints and have it go there.  Jogging is so 20th century.

ipsod, 2022-08-04:
@MitchBradley 

That's a neat idea, but wouldn't help with my current project.

I'm trying to create a platform for shooting product videos.  I use jogging to choose the shots I want to shoot.  My setup has a platform for an item that's on a  2-axis rotational platform, which rides on a 2-axis X-Z platform, and the camera has a 2-axis gimbal, a 1-axis slider, a motor on the focus ring, and then I'm also sending codes to the camera to control ISO.  I really don't need all of the tightly-coupled precision that FluidNC offers, but I've had a very hard time trying to create my own motion planner.  I'm shooting videos of art that I make and sell, and I make a lot of it, and I photograph each item individually, so the better jogging works, the faster I can shoot the videos, the more time I can save.

I plan to use a similar system in creating some art.  It's one of those things where manual control is just superior - you don't know how far to go, until you're there.  But, once it's done once, you can repeat it - that's the idea.

MitchBradley, 2022-08-04:
The 0.5 - 1 s latency that is mentioned in the reference that you cite is not primarily caused by the software, but rather by the momentum of the mechanical system.  The software can probably mostly keep up with human reaction time.  The trick will be tuning the motion speeds so the system is moving at a speed where it is possible to slow down in a controlled fashion.  If a little overshoot and then back up is okay, then things get easier.  Overshoot is often completely unacceptable in a CNC setup because it can break bits, mar workpieces, etc.  Good luck.

ipsod, 2022-08-04:
@MitchBradley 

Are you certain?  I read it that way on first reading, but then came to think that the planner had to plan deceleration at the end of all of the queued motions, which meant that faster motions required a longer duration to be queued to prevent it from planning the deceleration, and this would push the response to a requested deceleration back by 0.5 to 1.0 seconds, unless, instead of decelerating, the jog was cancelled.

MitchBradley, 2022-08-04:
You are on your own at this point.  Your use case is very far from the design center of FluidNC so many of FluidNC's constraints, optimizations, and considerations do not apply.  We are very busy with stuff that is applicable to our intended use case and do not have any more time to help with something that is outside that scope.

ipsod, 2022-08-04:
Ok, thank you for your time.