bdring/FluidNC #555: Feature: IP Address on OLED or whatever display.

brakthehun, 2022-08-05:
### Machine Context

I build lasers and it would be nice to not hook them up to USB to find the IP

### Feature Description

Shows the IP,, maybe scrolling. Shows the Telnet port. Shows the web server address??

### Other Approaches

I currently have to open a serial monitor and connect the machine to a computer and ask it it's address.

### How I Can Help

Release a binary with OLED support and have an $OLED fullof things like  $oled/Mode=IP or POSition or SPINdle speed and set OLED/Size=32x128 or 6x126 or LCD backpack. Maybe $OLED/Font_size??

MitchBradley, 2022-08-05:
OLED is sort of a troublesome thing to release as a binary because the set of things that someone might want to see is pretty open-ended.  I anticipate that there would be a never-ending set of requests for something different sort of like this one.

Ideally we would have some sort of template language that lets you configure what is on the display, but that is a big design problem.

MitchBradley, 2022-08-05:
I do agree that having the IP address is very helpful.  The OLED_TINY config shows that and very little else.

On the other hand, it is possible to configure the DHCP server in most Wifi routers to supply a fixed IP address for certain MAC addresses, and you can also configure a static IP address directly in FluidNC.

If you sell the lasers that you build, it might make sense to provide customized releases to your customers, with support for the specific OLED that you supply.

closed 2024-12-12 by bdring