bdring/FluidNC #561: Feature: I2C pin extender

sebosfato, 2022-08-08:
### Machine Context

Hello im from Brail my name is Fabio and im building my own cnc milling almost 4 axis machine... 

my idea is to build this machine because its cost prohibited here developing molds for developing small or simple products here..  and to buy a cnc machine here is also very expensive... so im using an xy table a big drilling machine to make a sort of cnc machine that can make some precision jobs for me... 

### Feature Description

Does Fluid nc support I2C pin extenders?

Here in Brazil we find this arduino modules with 16 outputs, would be very nice to use for not critical buttons or outputs like enable of the motors spindle cooling etc also it would be nice to be able to implement some kind of buttons without consuming the pins of the esp32.. 

in the example i noticed there is a I2SO protocol being used but im not sure what module i could use with it... 

i tried to find this info in the wiki and in the forum issues but coudnt find clear info about it... 

would you please tell me if i2c is available and if its not, a suggestion of module that use this I2CO compatible with fluid and perhaps how to wire it...  

i tried to initialize the i2c like this 

i2c:
  sda_pin: gpio.21
  scl_pin: gpio.22

Can it be compiled in arduino ide? 

however in the fluidterminal it says its being skiped... 

compliments for the great job....

a question about the negative and positive limit switches... why is not possible to use together in the same line all the switches from different axis if the axis tests each direction and axis independently and can be pulled off from the switches?

it would make a big difference in pin requirement... specially for more axis machine.. 

Thanks for all building it! 

### Other Approaches

I think there not many ways around this problem of pin limitation... 

### How I Can Help

i can test it extensively and provide good feedback about! promise!

sebosfato, 2022-08-09:
Hey thanks for answering. I tried the following but had no success yet.. .

i2c:
  sda: gpio.21
  scl: gpio.22
  bus: 0
  frequency: 100000

extenders:
  pinextender0:
    i2c_extender:
      device: pca9685
      device_id: 0
      interrupt: gpio.16

However it seem to me its not even being loaded because on fluiterm it get me this errors 

[MSG:INFO: FluidNC v3.5.1]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is SPIFFS]
[MSG:INFO: Configuration file:config.yaml]
[MSG:WARN: Ignored key i2c]
[MSG:ERR: Skipping key sda indent 2 thisIndent 0]
[MSG:ERR: Skipping key scl indent 2 thisIndent 0]
[MSG:ERR: Skipping key bus indent 2 thisIndent 0]
[MSG:ERR: Skipping key frequency indent 2 thisIndent 0]
[MSG:WARN: Ignored key extenders]
[MSG:ERR: Skipping key pinextender0 indent 2 thisIndent 0]
[MSG:ERR: Skipping key i2c_extender indent 4 thisIndent 0]
[MSG:ERR: Skipping key device indent 6 thisIndent 0]
[MSG:ERR: Skipping key device_id indent 6 thisIndent 0]
[MSG:ERR: Skipping key interrupt indent 6 thisIndent 0]

should it be placed inside the motor for example? 
i tried everything so far no good... aways show skiping key

bdring, 2022-08-09:
We do not offer support for it at this time. It is not 100% working yet. Read the wiki page fully to fix your errors.

closed 2022-08-09 by bdring