bdring/FluidNC #563: Problem: No stepper movement

codingishard404, 2022-08-11:
### Controller Board

Hello. I'm using the FluidNC Pen Laser Controller (SPI) with TMC2130 drivers. The board is V1.2, which I'm assuming has the same pinout as v1.0, running FluidNC v3.4.5

### Help From Board Vendor

- [ ] Yes
- [ ] No
- [X] Not Applicable

### Machine Description

The machine is a corexy sand table (ZENXYv2). 

### Configuration file

```yaml
Config file pulled from memory via the $CD command in the web interface

board: FluidNC Pen/Laser 2130
name: TMC2130 ZenXY
meta: 
stepping:
  engine: RMT
  idle_ms: 255
  pulse_us: 2
  dir_delay_us: 1
  disable_delay_us: 0
  segments: 6
axes:
  shared_stepper_disable_pin: NO_PIN
  shared_stepper_reset_pin: NO_PIN
  x:
    steps_per_mm: 80.000
    max_rate_mm_per_min: 1000.000
    acceleration_mm_per_sec2: 25.000
    max_travel_mm: 450.000
    soft_limits: true
    homing:
      cycle: 2
      allow_single_axis: true
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100
    motor0:
      limit_neg_pin: gpio.39
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      tmc_2130:
        step_pin: gpio.14
        direction_pin: gpio.12
        disable_pin: NO_PIN
        r_sense_ohms: 0.110
        run_amps: 0.250
        hold_amps: 0.250
        microsteps: 32
        toff_disable: 0
        toff_stealthchop: 5
        use_enable: false
        cs_pin: gpio.16
        spi_index: -1
        run_mode: StealthChop
        homing_mode: StealthChop
        stallguard: 0
        stallguard_debug: false
        toff_coolstep: 3
    motor1:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      null_motor:
  y:
    steps_per_mm: 80.000
    max_rate_mm_per_min: 1000.000
    acceleration_mm_per_sec2: 25.000
    max_travel_mm: 885.000
    soft_limits: true
    homing:
      cycle: 1
      allow_single_axis: true
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100
    motor0:
      limit_neg_pin: gpio.36
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      tmc_2130:
        step_pin: gpio.25
        direction_pin: gpio.26
        disable_pin: NO_PIN
        r_sense_ohms: 0.110
        run_amps: 0.250
        hold_amps: 0.250
        microsteps: 32
        toff_disable: 0
        toff_stealthchop: 5
        use_enable: false
        cs_pin: gpio.17
        spi_index: -1
        run_mode: StealthChop
        homing_mode: StealthChop
        stallguard: 0
        stallguard_debug: false
        toff_coolstep: 3
    motor1:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      null_motor:
  z:
    steps_per_mm: 80.000
    max_rate_mm_per_min: 1000.000
    acceleration_mm_per_sec2: 25.000
    max_travel_mm: 1000.000
    soft_limits: false
    homing:
      cycle: 0
      allow_single_axis: true
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 50.000
      seek_mm_per_min: 200.000
      settle_ms: 250
      seek_scaler: 1.100
      feed_scaler: 1.100
kinematics:
  CoreXY:
spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18
sdcard:
  cs_pin: gpio.5
  card_detect_pin: NO_PIN
control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN
coolant:
  flood_pin: NO_PIN
  mist_pin: NO_PIN
  delay_ms: 0
probe:
  pin: NO_PIN
  check_mode_start: true
macros:
  startup_line0: 
  startup_line1: 
  macro0: 
  macro1: 
  macro2: 
  macro3: 
start:
  must_home: true
  deactivate_parking: false
  check_limits: false
user_outputs:
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 5000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000
  digital0_pin: NO_PIN
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: NO_PIN
arc_tolerance_mm: 0.002
junction_deviation_mm: 0.010
verbose_errors: false
report_inches: false
enable_parking_override_control: false
use_line_numbers: false
planner_blocks: 16
NoSpindle:
```

### Startup Messages

```text
[MSG:INFO: FluidNC v3.4.5]
[MSG:INFO: Compiled with ESP32 SDK:v4.4-beta1-189-ga79dc75f0a]
[MSG:INFO: Local filesystem type is SPIFFS]
[MSG:INFO: Configuration file:config_01.yaml]
[MSG:INFO: Machine TMC2130 ZenXY]
[MSG:INFO: Board FluidNC Pen/Laser 2130]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN]
[MSG:INFO: Stepping:RMT Pulse:2us Dsbl Delay:0us Dir Delay:1us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (0.000,450.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2130 Step:gpio.14 Dir:gpio.12 CS:gpio.16 Disable:NO_PIN Index:-1 R:0.110]
[MSG:INFO:     Neg Limit gpio.39]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Y (0.000,885.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2130 Step:gpio.25 Dir:gpio.26 CS:gpio.17 Disable:NO_PIN Index:-1 R:0.110]
[MSG:INFO:     Neg Limit gpio.36]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Z (-1000.000,0.000)]
[MSG:INFO: X Axis driver test passed]
[MSG:INFO: Y Axis driver test passed]
[MSG:INFO: Kinematic system: CoreXY]
[MSG:INFO: Using spindle NoSpindle]
[MSG:INFO: STA SSID is not set]
[MSG:INFO: AP SSID FluidNC IP 192.168.0.1 mask 255.255.255.0 channel 1]
[MSG:INFO: AP started]
[MSG:INFO: WiFi on]
[MSG:INFO: Captive Portal Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
```

### User Interface Software

webUI

### What happened?

I can't get any movement out of the steppers at all. I'm fairly certain I've gone wrong somewhere in my config file. I've attempted to create one by merging the example code for a coreXY plotter using 2209 drivers (https://github.com/bdring/FluidNC/blob/main/example_configs/TMC2209_corexy.yaml) and the example config for a cartesian plotter using TMC2130 drivers (https://github.com/bdring/FluidNC/blob/main/example_configs/TMC2130_pen.yaml).

bdring, 2022-08-11:
I think you need shared disable pin defined.

```yaml
  shared_stepper_disable_pin: gpio.13:high
```

I put a pinout image for the controller here.
http://wiki.fluidnc.com/en/hardware/official/FluidNC_Pen_Laser_CNC_Controller_SPI

codingishard404, 2022-08-11:
We've got movement! Thank you. Now to work on the homing.

codingishard404, 2022-08-12:
New problem: if I over-travel while jogging it refuses to move in any direction until I restart FluidNC. Moving from 0 to 100 and back to 0 is fine, but -100 breaks all movement in both axes.

Is this how the soft-limit function is supposed to function?

bdring, 2022-08-12:
What is the error?

codingishard404, 2022-08-12:
It doesn't error, it just doesn't move.

bdring, 2022-08-12:
Check your startup messages for the range on that axis.

codingishard404, 2022-08-12:
Well that would explain a lot, it's now "skipping configuration file due to panic." Not sure why the X and Y are working at all since it looks like the default is cartesian not corexy. That aside, if I'm reading it correctly X and Y are -1000, 0

Is there a list of the various error messages? I tried searching the wiki without much success.

bdring, 2022-08-12:
Paste the startup messages before the panic.

codingishard404, 2022-08-12:
Feels like I'm chasing my tail here.  I *think* movement commands stopped working at the first connection error message.

`ESP3D for
FluidNC
 Dashboard FluidNCTablet
Controls
X
Y
Z
0.1
1
10
100
+Y
-Y
-X
+X
+Z
-Z
10
1
0.1
Manage macros
 

Xw:0.000	
Xm:0.000
Yw:0.000	
Ym:0.000
Zw:0.000
Zm:0.000

XY:	
1000
mm/min
 
Z:	
100
mm/min
 
GRBL
  	auto-check every:
3
sec
Idle
 		 	

Override
 	
 	
Reset the firmware
SD Files	  Make a directory
  		  	
Show runnable files or all files
 
       
No SD Card
Commands
$/axes/x/max_travel_mm
$/axes/x/homing/mpos_mm
$/axes/x/homing/positive_direction
$/axes/y/max_travel_mm
$/axes/y/homing/mpos_mm
$/axes/y/homing/positive_direction
$G
$/axes/x/max_travel_mm=450.000
ok
$/axes/x/homing/mpos_mm=0.000
ok
$/axes/x/homing/positive_direction=false
ok
$/axes/y/max_travel_mm=885.000
ok
$/axes/y/homing/mpos_mm=0.000
ok
$/axes/y/homing/positive_direction=false
ok
[GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S0]
ok
<Alarm|MPos:0.000,0.000,0.000|FS:0,0|Pn:Y|WCO:0.000,0.000,0.000>
<Alarm|MPos:0.000,0.000,0.000|FS:0,0|Pn:Y|Ov:100,100,100>
<Alarm|MPos:0.000,0.000,0.000|FS:0,0|Pn:Y>
$X
<Alarm|MPos:0.000,0.000,0.000|FS:0,0|Pn:Y>
[MSG:INFO: Caution: Unlocked]
ok
$X
ok
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Pn:Y>
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Pn:Y>
$SS
[MSG:INFO: FluidNC v3.4.5]
[MSG:INFO: Compiled with ESP32 SDK:v4.4-beta1-189-ga79dc75f0a]
[MSG:INFO: Local filesystem type is SPIFFS]
[MSG:INFO: Configuration file:config_02.yaml]
[MSG:INFO: Machine TMC2130 ZenXY]
[MSG:INFO: Board FluidNC Pen/Laser 2130]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN]
[MSG:INFO: Stepping:RMT Pulse:2us Dsbl Delay:0us Dir Delay:1us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable gpio.13]
[MSG:INFO: Axis X (0.000,450.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2130 Step:gpio.14 Dir:gpio.12 CS:gpio.16 Disable:NO_PIN Index:-1 R:0.110]
[MSG:INFO:     Neg Limit gpio.39]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Y (0.000,885.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2130 Step:gpio.25 Dir:gpio.26 CS:gpio.17 Disable:NO_PIN Index:-1 R:0.110]
[MSG:INFO:     Neg Limit gpio.36]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Z (-1000.000,0.000)]
[MSG:INFO: X Axis driver test passed]
[MSG:INFO: Y Axis driver test passed]
[MSG:INFO: Kinematic system: CoreXY]
[MSG:INFO: Using spindle NoSpindle]
[MSG:INFO: STA SSID is not set]
[MSG:INFO: AP SSID FluidNC IP 192.168.0.1 mask 255.255.255.0 channel 1]
[MSG:INFO: AP started]
[MSG:INFO: WiFi on]
[MSG:INFO: Captive Portal Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
ok
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Pn:Y>
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Pn:Y>
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Pn:Y>
$J=G91 G21 F1000 X-100
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Pn:Y>
ok
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Pn:Y|WCO:0.000,0.000,0.000>
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Pn:Y|Ov:100,100,100>
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Pn:Y>
$J=G91 G21 F1000 X100
ok
<Jog|MPos:0.013,0.000,0.000|FS:191,0|Pn:Y>
<Jog|MPos:14.712,0.000,0.000|FS:1414,0|Pn:Y>
<Jog|MPos:64.688,0.000,0.000|FS:1414,0|Pn:Y>
<Idle|MPos:100.000,0.000,0.000|FS:0,0|Pn:Y>
$J=G91 G21 F1000 X-100
<Idle|MPos:100.000,0.000,0.000|FS:0,0|Pn:Y>
ok
<Jog|MPos:100.000,0.000,0.000|FS:191,0|Pn:Y>
<Jog|MPos:62.537,0.000,0.000|FS:1414,0|Pn:Y>
<Jog|MPos:12.637,0.000,0.000|FS:1414,0|Pn:Y|WCO:0.000,0.000,0.000>
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Pn:Y|Ov:100,100,100>
$J=G91 G21 F1000 X-100
ok
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Pn:Y>
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Pn:Y>
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Pn:Y>
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Pn:Y>
$J=G91 G21 F1000 X100
ok
<Jog|MPos:0.013,0.000,0.000|FS:191,0|Pn:Y>
<Jog|MPos:4.537,0.000,0.000|FS:1379,0|Pn:Y>
<Jog|MPos:54.787,0.000,0.000|FS:1414,0|Pn:Y>
<Jog|MPos:99.950,0.000,0.000|FS:0,0|Pn:Y>
<Idle|MPos:100.000,0.000,0.000|FS:0,0|Pn:Y|WCO:0.000,0.000,0.000>
$J=G91 G21 F1000 Y100
ok
<Jog|MPos:100.000,0.000,0.000|FS:191,0|Pn:Y|Ov:100,100,100>
<Jog|MPos:100.000,31.250,0.000|FS:1414,0|Pn:Y>
<Jog|MPos:100.000,81.825,0.000|FS:1414,0|Pn:Y>
<Idle|MPos:100.000,100.000,0.000|FS:0,0|Pn:Y>
$J=G91 G21 F1000 Y-100
$J=G91 G21 F1000 Y-100
Connection error
$J=G91 G21 F1000 X100
<Alarm|MPos:0.000,0.000,0.000|FS:0,0>
<Alarm|MPos:0.000,0.000,0.000|FS:0,0>
<Alarm|MPos:0.000,0.000,0.000|FS:0,0>
<Alarm|MPos:0.000,0.000,0.000|FS:0,0>
<Alarm|MPos:0.000,0.000,0.000|FS:0,0>
<Alarm|MPos:0.000,0.000,0.000|FS:0,0>
$J=G91 G21 F1000 X100
ERROR:8
Grbl '$' command cannot be used unless Grbl is IDLE. Ensures smooth operation during a job.
<Alarm|MPos:0.000,0.000,0.000|FS:0,0|WCO:0.000,0.000,0.000>
<Alarm|MPos:0.000,0.000,0.000|FS:0,0|Ov:100,100,100>
<Alarm|MPos:0.000,0.000,0.000|FS:0,0>
$X
[MSG:INFO: Caution: Unlocked]
ok
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
$J=G91 G21 F1000 Y100
ok
<Jog|MPos:0.000,0.000,0.000|FS:135,0>
<Jog|MPos:0.000,0.000,0.000|FS:330,0>
<Jog|MPos:0.000,0.000,0.000|FS:1000,0>
<Jog|MPos:0.000,0.000,0.000|FS:655,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
$SS
[MSG:INFO: FluidNC v3.4.5]
[MSG:INFO: Compiled with ESP32 SDK:v4.4-beta1-189-ga79dc75f0a]
[MSG:INFO: Local filesystem type is SPIFFS]
[MSG:ERR: Skipping configuration file due to panic]
[MSG:INFO: Using default configuration]
[MSG:INFO: Axes: using defaults]
[MSG:INFO: Machine Default (Test Drive)]
[MSG:INFO: Board None]
[MSG:INFO: SPI not defined]
[MSG:INFO: Stepping:RMT Pulse:4us Dsbl Delay:0us Dir Delay:0us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (-1000.000,0.000)]
[MSG:INFO: Axis Y (-1000.000,0.000)]
[MSG:INFO: Axis Z (-1000.000,0.000)]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Using spindle NoSpindle]
[MSG:INFO: STA SSID is not set]
[MSG:INFO: AP SSID FluidNC IP 192.168.0.1 mask 255.255.255.0 channel 1]
[MSG:INFO: AP started]
[MSG:INFO: WiFi on]
[MSG:INFO: Captive Portal Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
`

bdring, 2022-08-12:
I need to see the startup messages or I cannot help you.

Connect with Fluidterm only. Cut and paste the startup messages here.

http://wiki.fluidnc.com/en/support/requesting_help

codingishard404, 2022-08-12:
Roger. Let me get that installed, I've been connecting via the web interface.

codingishard404, 2022-08-12:
Resetting MCU
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:12804
ho 0 tail 12 room 4
load:0x40080400,len:3032
entry 0x400805e4

[MSG:INFO: FluidNC v3.4.5]
[MSG:INFO: Compiled with ESP32 SDK:v4.4-beta1-189-ga79dc75f0a]
[MSG:INFO: Local filesystem type is SPIFFS]
[MSG:INFO: Configuration file:config_02.yaml]
[MSG:INFO: Machine TMC2130 ZenXY]
[MSG:INFO: Board FluidNC Pen/Laser 2130]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN]
[MSG:INFO: Stepping:RMT Pulse:2us Dsbl Delay:0us Dir Delay:1us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable gpio.13]
[MSG:INFO: Axis X (0.000,450.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2130 Step:gpio.14 Dir:gpio.12 CS:gpio.16 Disable:NO_PIN Index:-1 R:0.110]
[MSG:INFO:     Neg Limit gpio.39]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Y (0.000,885.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2130 Step:gpio.25 Dir:gpio.26 CS:gpio.17 Disable:NO_PIN Index:-1 R:0.110]
[MSG:INFO:     Neg Limit gpio.36]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Z (-1000.000,0.000)]
[MSG:INFO: X Axis driver test passed]
[MSG:INFO: Y Axis driver test passed]
[MSG:INFO: Kinematic system: CoreXY]
[MSG:INFO: Using spindle NoSpindle]
[MSG:INFO: STA SSID is not set]
[MSG:INFO: AP SSID FluidNC IP 192.168.0.1 mask 255.255.255.0 channel 1]
[MSG:INFO: AP started]
[MSG:INFO: WiFi on]
[MSG:INFO: Captive Portal Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl 3.4 [FluidNC v3.4.5 (wifi) '$' for help]
[MSG:INFO: '$H'|'$X' to unlock]

bdring, 2022-08-12:
I don't see a panic. Can you cause the panic in Fluidterm?

codingishard404, 2022-08-13:
Working on learning the proper commands here.

```Resetting MCU
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:12804
ho 0 tail 12 room 4
load:0x40080400,len:3032
entry 0x400805e4

[MSG:INFO: FluidNC v3.4.5]
[MSG:INFO: Compiled with ESP32 SDK:v4.4-beta1-189-ga79dc75f0a]
[MSG:INFO: Local filesystem type is SPIFFS]
[MSG:INFO: Configuration file:config_02.yaml]
[MSG:INFO: Machine TMC2130 ZenXY]
[MSG:INFO: Board FluidNC Pen/Laser 2130]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN]
[MSG:INFO: Stepping:RMT Pulse:2us Dsbl Delay:0us Dir Delay:1us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable gpio.13]
[MSG:INFO: Axis X (0.000,450.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2130 Step:gpio.14 Dir:gpio.12 CS:gpio.16 Disable:NO_PIN Index:-1 R:0.110]
[MSG:INFO:     Neg Limit gpio.39]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Y (0.000,885.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2130 Step:gpio.25 Dir:gpio.26 CS:gpio.17 Disable:NO_PIN Index:-1 R:0.110]
[MSG:INFO:     Neg Limit gpio.36]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Z (-1000.000,0.000)]
[MSG:INFO: X Axis driver test passed]
[MSG:INFO: Y Axis driver test passed]
[MSG:INFO: Kinematic system: CoreXY]
[MSG:INFO: Using spindle NoSpindle]
[MSG:INFO: STA SSID is not set]
[MSG:INFO: AP SSID FluidNC IP 192.168.0.1 mask 255.255.255.0 channel 1]
[MSG:INFO: AP started]
[MSG:INFO: WiFi on]
[MSG:INFO: Captive Portal Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl 3.4 [FluidNC v3.4.5 (wifi) '$' for help]
[MSG:INFO: '$H'|'$X' to unlock]
$X
[MSG:INFO: Caution: Unlocked]
ok
$J=G91 X100 F100
ok
$J=G91 Y100 F100
ok
```

If I'm understanding things correctly, this should've moved 100mm in X, then in Y. I got no movement at all in either axis.

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:12804
ho 0 tail 12 room 4
load:0x40080400,len:3032
entry 0x400805e4

[MSG:INFO: FluidNC v3.4.5]
[MSG:INFO: Compiled with ESP32 SDK:v4.4-beta1-189-ga79dc75f0a]
[MSG:INFO: Local filesystem type is SPIFFS]
[MSG:INFO: Configuration file:config_03.yaml]
[MSG:INFO: Machine TMC2130 ZenXY]
[MSG:INFO: Board FluidNC Pen/Laser 2130]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN]
[MSG:INFO: Stepping:RMT Pulse:2us Dsbl Delay:0us Dir Delay:1us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable gpio.13]
[MSG:INFO: Axis X (0.000,450.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2130 Step:gpio.14 Dir:gpio.12 CS:gpio.16 Disable:NO_PIN Index:-1 R:0.110]
[MSG:INFO:     Neg Limit gpio.39]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Y (0.000,885.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2130 Step:gpio.25 Dir:gpio.26 CS:gpio.17 Disable:NO_PIN Index:-1 R:0.110]
[MSG:INFO:     Neg Limit gpio.36]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Z (-1000.000,0.000)]
[MSG:INFO: X Axis driver test passed]
[MSG:INFO: Y Axis driver test passed]
[MSG:INFO: Kinematic system: CoreXY]
[MSG:INFO: Using spindle NoSpindle]
[MSG:INFO: STA SSID is not set]
[MSG:INFO: AP SSID FluidNC IP 192.168.0.1 mask 255.255.255.0 channel 1]
[MSG:INFO: AP started]
[MSG:INFO: WiFi on]
[MSG:INFO: Captive Portal Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl 3.4 [FluidNC v3.4.5 (wifi) '$' for help]
[MSG:INFO: '$H'|'$X' to unlock]
[MSG:INFO: Caution: Unlocked]
Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed).

Core  1 register dump:
PC      : 0x400817e3  PS      : 0x00060035  A0      : 0x80082382  A1      : 0x3ffbf1ec
A2      : 0x00000000  A3      : 0x3ffb4ad0  A4      : 0x3ffc324c  A5      : 0x00000003
A6      : 0x00000003  A7      : 0x00000002  A8      : 0xbad00bad  A9      : 0x3ffbf1cc
A10     : 0x3ffb4c14  A11     : 0x003fffff  A12     : 0x3ffb1de0  A13     : 0x3ffb1da0
A14     : 0x3ffc04a8  A15     : 0x3ffb28e0  SAR     : 0x00000018  EXCCAUSE: 0x00000007
EXCVADDR: 0x00000000  LBEG    : 0x4008ae1c  LEND    : 0x4008ae27  LCOUNT  : 0x00000000

Backtrace:0x400817e0:0x3ffbf1ec |<-CORRUPTED

ELF file SHA256: 0000000000000000

Rebooting...
ets Jun  8 2016 00:22:57

rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:12804
ho 0 tail 12 room 4
load:0x40080400,len:3032
entry 0x400805e4

[MSG:INFO: FluidNC v3.4.5]
[MSG:INFO: Compiled with ESP32 SDK:v4.4-beta1-189-ga79dc75f0a]
[MSG:INFO: Local filesystem type is SPIFFS]
[MSG:ERR: Skipping configuration file due to panic]
[MSG:INFO: Using default configuration]
[MSG:INFO: Axes: using defaults]
[MSG:INFO: Machine Default (Test Drive)]
[MSG:INFO: Board None]
[MSG:INFO: SPI not defined]
[MSG:INFO: Stepping:RMT Pulse:4us Dsbl Delay:0us Dir Delay:0us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (-1000.000,0.000)]
[MSG:INFO: Axis Y (-1000.000,0.000)]
[MSG:INFO: Axis Z (-1000.000,0.000)]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Using spindle NoSpindle]
[MSG:INFO: STA SSID is not set]
[MSG:INFO: AP SSID FluidNC IP 192.168.0.1 mask 255.255.255.0 channel 1]
[MSG:INFO: AP started]
[MSG:INFO: WiFi on]
[MSG:INFO: Captive Portal Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl 3.4 [FluidNC v3.4.5 (wifi) '$' for help]
[MSG:INFO: '$H'|'$X' to unlock]
```

Okay. I got it to break by connecting over the webUI and jogging via that interface. Not quite sure why I can't get it to move via the terminal.

MitchBradley, 2022-08-13:
Please revert to 3.4.4.  3.4.5 is bad

bdring, 2022-08-13:
Try using [version 3.4.4](https://github.com/bdring/FluidNC/releases/tag/v3.4.4). That is a last release before the Espressif update that is still causing some mysterious issues.

Try using [Wifi STA mode rather than AP mode](http://wiki.fluidnc.com/en/features/wifi_bt#connect-to-an-external-network)

codingishard404, 2022-08-13:
Okay, leaving aside the wifi for now, I've loaded 3.4.4.

```
FluidNC v1.2.0 using COM4
Exit: Ctrl-C, Ctrl-Q or Ctrl-], Clear screen: CTRL-W
Upload: Ctrl-U, Reset ESP32: Ctrl-R, Send Override: Ctrl-O
$Startup/Show
[MSG:INFO: FluidNC v3.4.4]
[MSG:INFO: Compiled with ESP32 SDK:v3.3.5-1-g85c43024c]
[MSG:INFO: Configuration file:config_03.yaml]
[MSG:INFO: Machine TMC2130 ZenXY]
[MSG:INFO: Board FluidNC Pen/Laser 2130]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN]
[MSG:INFO: Stepping:RMT Pulse:2us Dsbl Delay:0us Dir Delay:1us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable gpio.13]
[MSG:INFO: Axis X (0.000,450.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2130 Step:gpio.14 Dir:gpio.12 CS:gpio.16 Disable:NO_PIN Index:-1 R:0.110]
[MSG:INFO:     Neg Limit gpio.39]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Y (0.000,885.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2130 Step:gpio.25 Dir:gpio.26 CS:gpio.17 Disable:NO_PIN Index:-1 R:0.110]
[MSG:INFO:     Neg Limit gpio.36]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Z (-1000.000,0.000)]
[MSG:INFO: X Axis driver test passed]
[MSG:INFO: Y Axis driver test passed]
[MSG:INFO: Kinematic system: CoreXY]
[MSG:INFO: Using spindle NoSpindle]
[MSG:INFO: STA SSID is not set]
[MSG:INFO: AP SSID FluidNC IP 192.168.0.1 mask 255.255.255.0 channel 1]
[MSG:INFO: AP started]
[MSG:INFO: WiFi on]
[MSG:INFO: Captive Portal Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
ok
<Alarm|MPos:0.000,0.000,0.000|FS:0,0|Pn:Y|Ov:100,100,100>
$X
[MSG:INFO: Caution: Unlocked]
ok
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Pn:Y>
$J=G91 X100 F100
ok
Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)
Core 1 register dump:
PC      : 0x40081517  PS      : 0x00060034  A0      : 0x80082283  A1      : 0x3ffc0010
A2      : 0x00000000  A3      : 0x3ffc64ec  A4      : 0x3ffb9920  A5      : 0x00000000
A6      : 0x00000003  A7      : 0x3ffb9920  A8      : 0xbad00bad  A9      : 0x3ffbfff0
A10     : 0x3ffb3c20  A11     : 0xb0000000  A12     : 0x00060021  A13     : 0x00000000
A14     : 0x00000001  A15     : 0x3ffb8058  SAR     : 0x0000001e  EXCCAUSE: 0x00000007
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000
Core 1 was running in ISR context:
EPC1    : 0x4009149b  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x40081517

ELF file SHA256: 0000000000000000

Backtrace: 0x40081517:0x3ffc0010 0x40082280:0x3ffc0030 0x400822dc:0x3ffc0050 0x40082891:0x3ffc0070 0x400887f1:0x3ffc0090 0x40091498:0x3ffb1490 0x40091c91:0x3ffb14b0 0x40089748:0x3ffb14d0 0x401ab2b2:0x3ffb1540 0x401525e6:0x3ffb1570 0x40152802:0x3ffb15a0 0x40153c9f:0x3ffb15d0 0x40155681:0x3ffb1610 0x401519b3:0x3ffb1650 0x40150ef6:0x3ffb1680 0x4019b046:0x3ffb16a0 0x4000bd67:0x3ffb16c0 0x40128bd2:0x3ffb16e0 0x40128c61:0x3ffb1710 0x400d668a:0x3ffb1730 0x400f4382:0x3ffb1790 0x401b45bd:0x3ffb1cd0 0x40101b66:0x3ffb1cf0 0x4010628a:0x3ffb1d10 0x4010635d:0x3ffb1d60 0x400f3661:0x3ffb1db0 0x400fc6c6:0x3ffb1df0 0x400e999f:0x3ffb1e10 0x400e85e9:0x3ffb1e30 0x400ddc9e:0x3ffb1f60 0x4010d75c:0x3ffb1fb0 0x4008c152:0x3ffb1fd0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5856
entry 0x400806a8

[MSG:INFO: FluidNC v3.4.4]
[MSG:INFO: Compiled with ESP32 SDK:v3.3.5-1-g85c43024c]
[MSG:INFO: Configuration file:config_03.yaml]
[MSG:INFO: Machine TMC2130 ZenXY]
[MSG:INFO: Board FluidNC Pen/Laser 2130]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN]
[MSG:INFO: Stepping:RMT Pulse:2us Dsbl Delay:0us Dir Delay:1us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable gpio.13]
[MSG:INFO: Axis X (0.000,450.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2130 Step:gpio.14 Dir:gpio.12 CS:gpio.16 Disable:NO_PIN Index:-1 R:0.110]
[MSG:INFO:     Neg Limit gpio.39]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Y (0.000,885.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2130 Step:gpio.25 Dir:gpio.26 CS:gpio.17 Disable:NO_PIN Index:-1 R:0.110]
[MSG:INFO:     Neg Limit gpio.36]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Z (-1000.000,0.000)]
[MSG:INFO: X Axis driver test passed]
[MSG:INFO: Y Axis driver test passed]
[MSG:INFO: Kinematic system: CoreXY]
[MSG:INFO: Using spindle NoSpindle]
[MSG:INFO: STA SSID is not set]
[MSG:INFO: AP SSID FluidNC IP 192.168.0.1 mask 255.255.255.0 channel 1]
[MSG:INFO: AP started]
[MSG:INFO: WiFi on]
[MSG:INFO: Captive Portal Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl 3.4 [FluidNC v3.4.4 (wifi) '$' for help]
[MSG:INFO: '$H'|'$X' to unlock]

```

MitchBradley, 2022-08-13:
I wonder if the z axis without a motor definition is causing the problem.

MitchBradley, 2022-08-13:
I don't think the z axis is the problem.  This part of the crash dump
```
Core 1 was running in ISR context:
EPC1    : 0x4009149b ...
```
 tells me that the ESP32 is trying to read FLASH memory from inside an interrupt routine.  The ESP32 hardware does not allow that.  I do not not see anything else in the crash dump to explain why that FLASH access is happening, so I suspect that it might be related to AP mode.  We know that AP mode has a lot of problems.  We recommend using AP mode only for initial configuration.

codingishard404, 2022-08-13:
Tried defining motor0 and motor1 for z, I think it's having the same issue. $CD at the end

```
Grbl 3.4 [FluidNC v3.4.4 (wifi) '$' for help]
[MSG:INFO: '$H'|'$X' to unlock]
$Startup/Show
[MSG:INFO: FluidNC v3.4.4]
[MSG:INFO: Compiled with ESP32 SDK:v3.3.5-1-g85c43024c]
[MSG:INFO: Configuration file:config_04.yaml]
[MSG:INFO: Machine TMC2130 ZenXY]
[MSG:INFO: Board FluidNC Pen/Laser 2130]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN]
[MSG:INFO: Stepping:RMT Pulse:2us Dsbl Delay:0us Dir Delay:1us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable gpio.13]
[MSG:INFO: Axis X (0.000,450.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2130 Step:gpio.14 Dir:gpio.12 CS:gpio.16 Disable:NO_PIN Index:-1 R:0.110]
[MSG:INFO:     Neg Limit gpio.39]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Y (0.000,885.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2130 Step:gpio.25 Dir:gpio.26 CS:gpio.17 Disable:NO_PIN Index:-1 R:0.110]
[MSG:INFO:     Neg Limit gpio.36]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Z (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:   Motor1]
[MSG:INFO: X Axis driver test passed]
[MSG:INFO: Y Axis driver test passed]
[MSG:INFO: Kinematic system: CoreXY]
[MSG:INFO: Using spindle NoSpindle]
[MSG:INFO: STA SSID is not set]
[MSG:INFO: AP SSID FluidNC IP 192.168.0.1 mask 255.255.255.0 channel 1]
[MSG:INFO: AP started]
[MSG:INFO: WiFi on]
[MSG:INFO: Captive Portal Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
ok
$X
[MSG:INFO: Caution: Unlocked]
ok
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Pn:Y>
$J=G91 X100 F100
ok
Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)
Core 1 register dump:
PC      : 0x40081517  PS      : 0x00060034  A0      : 0x80082283  A1      : 0x3ffc0010
A2      : 0x00000000  A3      : 0x3ffc64ec  A4      : 0x3ffb9920  A5      : 0x00000000
A6      : 0x00000003  A7      : 0x3ffb9920  A8      : 0xbad00bad  A9      : 0x3ffbfff0
A10     : 0x3ffb3c6c  A11     : 0xb0000000  A12     : 0x00060021  A13     : 0x00000000
A14     : 0x00000001  A15     : 0x3ffb8058  SAR     : 0x00000012  EXCCAUSE: 0x00000007
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000
Core 1 was running in ISR context:
EPC1    : 0x400914be  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x40081517

ELF file SHA256: 0000000000000000

Backtrace: 0x40081517:0x3ffc0010 0x40082280:0x3ffc0030 0x400822dc:0x3ffc0050 0x40082891:0x3ffc0070 0x400887f1:0x3ffc0090 0x400914bb:0x3ffb1490 0x40091c91:0x3ffb14b0 0x40089748:0x3ffb14d0 0x401ab2b2:0x3ffb1540 0x401525e6:0x3ffb1570 0x40152802:0x3ffb15a0 0x40153c9f:0x3ffb15d0 0x40155681:0x3ffb1610 0x401519b3:0x3ffb1650 0x40150ef6:0x3ffb1680 0x4019b046:0x3ffb16a0 0x4000bd67:0x3ffb16c0 0x40128bd2:0x3ffb16e0 0x40128c61:0x3ffb1710 0x400d668a:0x3ffb1730 0x400f4382:0x3ffb1790 0x401b45bd:0x3ffb1cd0 0x40101b66:0x3ffb1cf0 0x4010628a:0x3ffb1d10 0x4010635d:0x3ffb1d60 0x400f3661:0x3ffb1db0 0x400fc6c6:0x3ffb1df0 0x400e999f:0x3ffb1e10 0x400e85e9:0x3ffb1e30 0x400ddc9e:0x3ffb1f60 0x4010d75c:0x3ffb1fb0 0x4008c152:0x3ffb1fd0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
[... 234 lines cut ...]
user_outputs:
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 5000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000
  digital0_pin: NO_PIN
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: NO_PIN

arc_tolerance_mm: 0.002
junction_deviation_mm: 0.010
verbose_errors: false
report_inches: false
enable_parking_override_control: false
use_line_numbers: false
planner_blocks: 16
NoSpindle:

ok
```

bdring, 2022-08-13:
$Wifi/Mode=Off

codingishard404, 2022-08-13:
That seems to have gotten rid of the movement issues. Going to play with it a bit more. Thank you both.

bdring, 2022-08-13:
What do you think did it? The version change or the Wifi change.

codingishard404, 2022-08-13:
I'm fairly certain disabling the wifi fixed it, I was still getting the reset with 3.4.4.

MitchBradley, 2022-08-14:
Both are important

MitchBradley, 2022-08-14:
3.4.5 is unstable even with STA mode

MitchBradley, 2022-08-14:
With 3.4.5 you would likely still have crashes, but they would be much less frequent, typically happening sporadically in the middle of long jobs.  Unless you did something that explicitly accesses FLASH, such as reloading WebUI in the middle of a GCode job run - that will crash 3.4.5 pretty quickly.

codingishard404, 2022-08-14:
Hello again. I thought I had the homing figured out, but while trying to make it home faster I came across another error. Below is the logs from two config files, the only change between the two is I increased seek_mm_per_min: (line 27) from 200 to 600. 

If I'm understanding correctly, I should be able to have any value between 1 and 100,000?

Output below, split into two for readability.

```
FluidNC v1.2.0 using COM4
Exit: Ctrl-C, Ctrl-Q or Ctrl-], Clear screen: CTRL-W
Upload: Ctrl-U, Reset ESP32: Ctrl-R, Send Override: Ctrl-O
$SS
[MSG:INFO: FluidNC v3.4.4]
[MSG:INFO: Compiled with ESP32 SDK:v3.3.5-1-g85c43024c]
[MSG:INFO: Configuration file:config_01.yaml]
[MSG:INFO: Machine TMC2130 ZenXY]
[MSG:INFO: Board FluidNC Pen/Laser 2130]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN]
[MSG:INFO: Stepping:RMT Pulse:2us Dsbl Delay:0us Dir Delay:1us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable gpio.13]
[MSG:INFO: Axis X (0.000,450.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2130 Step:gpio.14 Dir:gpio.12 CS:gpio.16 Disable:NO_PIN Index:-1 R:0.110]
[MSG:INFO:     Neg Limit gpio.39]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Y (0.000,885.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2130 Step:gpio.25 Dir:gpio.26 CS:gpio.17 Disable:NO_PIN Index:-1 R:0.110]
[MSG:INFO:     Neg Limit gpio.36]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Z (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:   Motor1]
[MSG:INFO: X Axis driver test passed]
[MSG:INFO: Y Axis driver test passed]
[MSG:INFO: Kinematic system: CoreXY]
[MSG:INFO: Using spindle NoSpindle]
[MSG:INFO: WiFi is disabled]
ok
$CD
board: FluidNC Pen/Laser 2130
name: TMC2130 ZenXY
meta:
stepping:
  engine: RMT
  idle_ms: 255
  pulse_us: 2
  dir_delay_us: 1
  disable_delay_us: 0
  segments: 6

axes:
  shared_stepper_disable_pin: gpio.13
  shared_stepper_reset_pin: NO_PIN
  x:
    steps_per_mm: 80.000
    max_rate_mm_per_min: 1000.000
    acceleration_mm_per_sec2: 25.000
    max_travel_mm: 450.000
    soft_limits: false
    homing:
      cycle: 2
      allow_single_axis: true
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
[... 331 lines cut ...]
  check_limits: false

user_outputs:
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 5000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000
  digital0_pin: NO_PIN
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: NO_PIN

arc_tolerance_mm: 0.002
junction_deviation_mm: 0.010
verbose_errors: false
report_inches: false
enable_parking_override_control: false
use_line_numbers: false
planner_blocks: 16
NoSpindle:
```

bdring, 2022-08-14:
Maybe line 45 has an error?

MitchBradley, 2022-08-15:
We cannot see syntax errors in the config file from the $CD output.  $CD reconstructs the configuration from the in-memory copy of the config tree, supplying default values for items that are not present in the actual file.  If there is a syntax error in the file, the information on that line will not be incorporated into the in-memory tree.

The error message says that there is a problem on line 45 of the config file.  That is not the same as line 45 of the $CD output.  The line numbers in $CD do not track the line numbers in the file.

You should look at line 45 of your file and figure out what is wrong - probably either a missing colon or a misspelling or some extraneous junk.  If you want us to look at it, you will need to attach the file.

codingishard404, 2022-08-16:
Line 45 should be microsteps: 32.  The only change I made was the value for line 27.  I've tried a few different values (I think 400, 600 and 800) and they all gave an error. I also tried adding an additional ":", but that upload failed (160, followed by an error 2 on each line).

Maybe this is a dumb question. Is the microsteps value dependent on the seek/feed value? 

Changed it to a text file for uploading.

[config_01a.txt](https://github.com/bdring/FluidNC/files/9353051/config_01a.txt)

MitchBradley, 2022-08-16:
I tried your config_01a.txt on a test machine.  There were no error message from the parsing phase.

```

[MSG:INFO: FluidNC v3.4.6 (EventQueue2-c1fb102-dirty)]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is SPIFFS]
[MSG:INFO: Configuration file:config_01a.txt]
[MSG:INFO: Machine TMC2130 ZenXY]
[MSG:INFO: Board FluidNC Pen/Laser 2130]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN]
[MSG:INFO: Stepping:RMT Pulse:2us Dsbl Delay:0us Dir Delay:1us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable gpio.13]
[MSG:INFO: Axis X (0.000,450.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2130 Step:gpio.14 Dir:gpio.12 CS:gpio.16 Disable:NO_PIN Index:-1 R:0.110]
[MSG:INFO:     Neg Limit gpio.39]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Y (0.000,885.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2130 Step:gpio.25 Dir:gpio.26 CS:gpio.17 Disable:NO_PIN Index:-1 R:0.110]
[MSG:INFO:     Neg Limit gpio.36]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Z (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:   Motor1]
[MSG:ERR: X Axis driver test failed. Check motor power]
[MSG:ERR: Y Axis driver test failed. Check motor power]
[MSG:INFO: Kinematic system: CoreXY]
[MSG:INFO: Using spindle NoSpindle]
[MSG:INFO: Connecting to STA SSID:CampOlinda]
[MSG:INFO: Connecting.]
[MSG:INFO: Connected - IP is 192.168.68.125]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl 3.4 [FluidNC v3.4.6 (EventQueue2-c1fb102-dirty) (wifi) '$' for help]
[MSG:INFO: '$H'|'$X' to unlock]
```
The driver test errors were because this machine does not have TMCs.  I did not expect that to work, I was just looking for config file syntax errors; there were none.

If you are still getting an error message, try re-uploading the file.  Perhaps it got corrupted.  We have seen cases where the SPIFFS filesystem has errors.

Regarding your question about "the seek/feed value", I cannot answer that because there are several config options related to seek and feed, so "the seek/feed value" does not uniquely identify what you are asking about.

codingishard404, 2022-08-16:
Thanks for giving that a try Mitch. Can't even get it to upload now... I'm sure I'm missing something obvious. I tried re-uploading the current config file (which doesn't give errors), and got the below

```
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Ov:100,100,100>
FluidNC filename [config_01aa.yaml]:
XModem Upload C:\Users\Christian\Documents\Projects\ZenXY\FluidNC\config_01aa.yaml config_01aa.yaml
$Xmodem/Receive=config_01aa.yaml
annot open config_01aa.yaml]
error:160
board: FluidNC Pen/Laser 2130
error:2

ok
name: TMC2130 ZenXY
error:2

ok
meta:
error:2

ok
stepping:
error:2

ok
  engine: RMT
error:2

ok
  idle_ms: 255
error:2

ok
  pulse_us: 2
error:2

ok
  dir_delay_us: 1
error:2

ok
  disable_delay_us: 0
error:2

ok
  segments: 6
error:2

ok

ok

ok
axes:
error:2

ok
  shared_stepper_disable_pin: gpio.13
error:2

ok
  shared_stepper_reset_pin: NO_PIN
error:2

ok
  x:
error:2

ok
    steps_per_mm: 80.000
error:2

ok
    max_rate_mm_per_min: 1000.000
error:2

ok
    acceleration_mm_per_sec2: 25.000
error:2

ok
    max_travel_mm: 450.000
error:2

ok
    soft_limits: false
error:2

ok
    homing:
error:2

ok
      cycle: 2
error:2

ok
      allow_single_axis: true
error:2

ok
      positive_direction: false
error:2

ok
      mpos_mm: 0.000
error:2

ok
      feed_mm_per_min: 100.000
error:2

ok
      seek_mm_per_min: 200.000
error:2

ok
      settle_ms: 500
error:2

ok
      seek_scaler: 1.100
error:2

ok
      feed_scaler: 1.100
error:2

ok

ok

ok
    motor0:
error:2

ok
      limit_neg_pin: gpio.39
error:2

ok
     250
error:1

ok
        hold_amps: 0.250
error:2

ok
        mboard: FluidNC Pen/Laser 2130        microstepsc
error:2

ok
name: TMC2130 ZenXY
error:2

ok
meta:
error:2

ok
stepping:
error:2

ok
  engine: RMT
error:2

ok
  idle_ms: 255
error:2

ok
  pulse_us: 2
error:2

ok
  dir_delay_us: 1
error:2

ok
  disable_delay_us: 0
error:2

ok
  segments: 6
error:2

ok

ok

ok
axes:
error:2

ok
  shared_stepper_disable_pin: gpio.13
error:2

ok
  shared_stepper_reset_pin: NO_PIN
error:2

ok
  x:
error:2

ok
    steps_per_mm: 80.000
error:2

ok
    max_rate_mm_per_min: 1000.000
error:2

ok
    acceleration_mm_per_sec2: 25.50
error:2

ok
        mboard: FluidNC Pen/Laser 2130        microstepsc
error:2

ok
name: TMC2130 ZenXY
error:2

ok
meta:
error:2

ok
stepping:
error:2

ok
  engine: RMT
error:2

ok
  idle_ms: 255
error:2

ok
  pulse_us: 2
error:2

ok
  dir_delay_us: 1
error:2

ok
  disable_delay_us: 0
error:2

ok
  segments: 6
error:2

ok

ok

ok
axes:
error:2

ok
  shared_stepper_disable_pin: gpio.13
error:2

ok
  shared_stepper_reset_pin: NO_PIN
error:2

ok
  x:
error:2

ok
    steps_per_mm: 80.000
error:2

ok
    max_rate_mm_per_min: 1000.000
error:2

ok
    acceleration_mm_per_sec2: 25.50
error:2

ok
        mboard: FluidNC Pen/Laser 2130        microstepsc
error:2

ok
name: TMC2130 ZenXY
error:2

ok
meta:
error:2

ok
stepping:
error:2

ok
  engine: RMT
error:2

ok
  idle_ms: 255
error:2

ok
  pulse_us: 2
error:2

ok
  dir_delay_us: 1
error:2

ok
  disable_delay_us: 0
error:2

ok
  segments: 6
error:2

ok

ok

ok
axes:
error:2

ok
  shared_stepper_disable_pin: gpio.13
error:2

ok
  shared_stepper_reset_pin: NO_PIN
error:2

ok
  x:
error:2

ok
    steps_per_mm: 80.000
error:2

ok
    max_rate_mm_per_min: 1000.000
error:2

ok
    acceleration_mm_per_sec2: 25.50
error:2

ok
        mboard: FluidNC Pen/Laser 2130        microstepsc
error:2

ok
name: TMC2130 ZenXY
error:2

ok
meta:
error:2

ok
stepping:
error:2

ok
  engine: RMT
error:2

ok
  idle_ms: 255
error:2

ok
  pulse_us: 2
error:2

ok
  dir_delay_us: 1
error:2

ok
  disable_delay_us: 0
error:2

ok
  segments: 6
error:2

ok

ok

ok
axes:
error:2

ok
  shared_stepper_disable_pin: gpio.13
error:2

ok
  shared_stepper_reset_pin: NO_PIN
error:2

ok
  x:
error:2

ok
    steps_per_mm: 80.000
error:2

ok
    max_rate_mm_per_min: 1000.000
error:2

ok
    acceleration_mm_per_sec2: 25.50
error:2

ok
        microstepsc Timeout
Giving up
Returned -4
```

bdring, 2022-08-16:
I just loaded your file on a Pen  Laser 2130. I used Version 3.4.4

```
[MSG:INFO: FluidNC v3.4.4]
[MSG:INFO: Compiled with ESP32 SDK:v3.3.5-1-g85c43024c]
[MSG:INFO: Configuration file:config_01aa.yaml]
[MSG:INFO: Machine TMC2130 ZenXY]
[MSG:INFO: Board FluidNC Pen/Laser 2130]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN]
[MSG:INFO: Stepping:RMT Pulse:2us Dsbl Delay:0us Dir Delay:1us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable gpio.13]
[MSG:INFO: Axis X (0.000,450.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2130 Step:gpio.14 Dir:gpio.12 CS:gpio.16 Disable:NO_PIN Index:-1 R:0.110]
[MSG:INFO:     Neg Limit gpio.39]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Y (0.000,885.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2130 Step:gpio.25 Dir:gpio.26 CS:gpio.17 Disable:NO_PIN Index:-1 R:0.110]
[MSG:INFO:     Neg Limit gpio.36]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Z (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:   Motor1]
[MSG:INFO: X Axis driver test passed]
[MSG:INFO: Y Axis driver test passed]
[MSG:INFO: Kinematic system: CoreXY]
[MSG:INFO: Using spindle NoSpindle]
[MSG:INFO: STA SSID is not set]
[MSG:INFO: AP SSID FluidNC IP 192.168.0.1 mask 255.255.255.0 channel 1]
[MSG:INFO: AP started]
[MSG:INFO: WiFi on]
[MSG:INFO: Captive Portal Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl 3.4 [FluidNC v3.4.4 (wifi) '$' for help]
[MSG:INFO: '$H'|'$X' to unlock]
```

codingishard404, 2022-08-16:
Okay. I'll close this out and start a new one.

closed 2022-08-16 by codingishard404

bdring, 2022-08-16:
Thanks, my phone is so slow with long Github issues.