bdring/FluidNC #564: Problem: Homing drives 2nd motor until other axes reqach limit switch

HoverClub, 2022-08-11:
### Controller Board

TinyBee

### Help From Board Vendor

- [ ] Yes
- [X] No
- [ ] Not Applicable

### Machine Description

Simple gantry router with XYZ steppers - dual steppers on X axis. 

### Configuration file

```yaml
board: MKS TinyBee V1.0
name: HovTek_CNC

kinematics:
  Cartesian:

i2so:
  bck_pin: gpio.25
  data_pin: gpio.27
  ws_pin: gpio.26

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5
  # uses TH2 IO34 - MAKE SURE jumper J2 is set to SDDET!!!
  card_detect_pin: gpio.34

stepping:
  # I2S_STATIC to make sure laser is sync'ed with steppers!
  engine: I2S_STATIC
  idle_ms: 255
  # 4usec min. - I2S bit rate 
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 2

axes:
  x:
    # X = 1/8 uStep
    steps_per_mm: 26.66
    max_rate_mm_per_min: 8000.000
    acceleration_mm_per_sec2: 80.000
    max_travel_mm: 2500.000
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 1500.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.33:low:pu
      hard_limits: true
      pulloff_mm: 4.000
      stepstick:
        step_pin: I2SO.1
        direction_pin: I2SO.2
        disable_pin: I2SO.0

    # use E0 driver for 2nd X axis motor!
    motor1:
      hard_limits: true
      pulloff_mm: 4.000
      stepstick:
        step_pin: I2SO.10
        direction_pin: I2SO.11
        disable_pin: I2SO.9

  y:
    # Y = 1/8 uStep
    steps_per_mm: 26.66
    max_rate_mm_per_min: 8000.000
    acceleration_mm_per_sec2: 70.000
    max_travel_mm: 1250.000
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 2000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.32:low:pu
      hard_limits: true
      pulloff_mm: 4.000
      stepstick:
        step_pin: I2SO.4
        # invert DIR signal from default high
        direction_pin: I2SO.5:low
        disable_pin: I2SO.3

  z:
    # Z = 1/8 uStep
    steps_per_mm: 200.000
    max_rate_mm_per_min: 8000.000
    acceleration_mm_per_sec2: 60.000
    max_travel_mm: 100.000
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 500.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_pos_pin: gpio.22:low:pu
      hard_limits: true
      pulloff_mm: 3.000
      stepstick:
        step_pin: I2SO.7
        direction_pin: I2SO.8
        disable_pin: I2SO.6

control:
  safety_door_pin: NO_PIN
  # on MT_DET connector
  reset_pin: gpio.35:low
  # on TH1 connector
  feed_hold_pin: gpio.36:low
  # on TB connector
  cycle_start_pin: gpio.39:low
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

# spindle PWM signal
PWM:
  pwm_hz: 2500
  # on EXP1 IO15 connector
  output_pin: gpio.15:high
  s0_with_disable: true
  tool_num: 0
  spinup_ms: 4000
  spindown_ms: 4000
  speed_map: 0=0.000% 12000=100.000%
 
Laser:
  pwm_hz: 5000
  # on 3D Touch connector (has pdwn + PWM)
  output_pin: gpio.2:high:pd
  s0_with_disable: true
  tool_num: 1
  speed_map: 0=0.000% 100=100.000%

start:
  must_home: false
```

### Startup Messages

```text
[MSG:INFO: FluidNC v3.5.1-pre]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is SPIFFS]
[MSG:INFO: Configuration file:config.yaml]
[MSG:INFO: Machine HovTek_CNC]
[MSG:INFO: Board MKS TinyBee V1.0]
[MSG:INFO: I2SO BCK:gpio.25 WS:gpio.26 DATA:gpio.27]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card Detect gpio.34]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:gpio.34]
[MSG:INFO: Stepping:I2S_static Pulse:4us Dsbl Delay:2us Dir Delay:1us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (0.000,2500.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.1 Dir:I2SO.2 Disable:I2SO.0]
[MSG:INFO:     Neg Limit gpio.33:low:pu]
[MSG:INFO:   Motor1]
[MSG:INFO:     stepstick Step:I2SO.10 Dir:I2SO.11 Disable:I2SO.9]
[MSG:INFO: Axis Y (0.000,1250.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.4 Dir:I2SO.5:low Disable:I2SO.3]
[MSG:INFO:     Neg Limit gpio.32:low:pu]
[MSG:INFO: Axis Z (-100.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.7 Dir:I2SO.8 Disable:I2SO.6]
[MSG:INFO:     Pos Limit gpio.22:low:pu]
[MSG:INFO: reset_pin gpio.35:low]
[MSG:INFO: feed_hold_pin gpio.36:low]
[MSG:INFO: cycle_start_pin gpio.39:low]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: PWM Spindle Ena:NO_PIN Out:gpio.15 Dir:NO_PIN Freq:2500Hz Res:14bits]
[MSG:INFO: Laser Spindle Ena:NO_PIN Out:gpio.2:pd Freq:5000Hz Res:13bits Laser mode:On]
[MSG:INFO: Using spindle PWM]
[MSG:INFO: STA SSID is not set]
[MSG:INFO: AP SSID FluidNC IP 192.168.0.1 mask 255.255.255.0 channel 1]
[MSG:INFO: AP started]
[MSG:INFO: WiFi on]
[MSG:INFO: Captive Portal Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl 3.5 [FluidNC v3.5.1-pre (wifi) '$' for help]
```

### User Interface Software

Anything - Fluidterm, UGS, Web interface, etc.

### What happened?

On simultaneous homing ($H) with X & Y on same homing cycle #, if the Y axis is further from home than the X axis, the second X axis motor (not the first X motor) continues to drive (against the end stop after it has triggered the Xlimit sw) until the Y axis has reached the Y limit switch.   If I set X & Y to different homing cycle #'s then it all works OK (Z is on separate # already).  Individual axis homing all works regardless of cycle #.

bdring, 2022-08-11:
I do not know why it is behaving that way, but we do not support 2 motors on one switch.

http://wiki.fluidnc.com/en/config/homing_and_limit_switches#ganged-motors-with-one-switch-input

HoverClub, 2022-08-11:
Yep - I got that but understood it meant that squaring wouldn't be attempted if only one switch?  Most, if not all, commercial hardware I've come across only has one limit switch on ganged motors. 
[https://github.com/bdring/FluidNC/issues/170#issuecomment-991253619](url)

sjonholle, 2022-08-11:
I ran into the same problem. With 2 endswitches my problem was gone. pulloff-mm 4mm wil do. motor0: limit_all_pin: gpio.21:low, motor1: limit_all_pin: gpio.4:low

bdring, 2022-08-11:
I have made hundreds of ganged motor configurations and they have all been dual switch.

It probably a lot easier to add the second switch than for us to change the firmware. I recall we put that limitation for a good reason. I can research that, but we are still dealing with things the Espressif broke when they updated their framework.

HoverClub, 2022-08-12:
... and just to add to the odd behaviour the system drives only the 2nd motor if the X axis on end stop and you try to jog away from it.  Same config.yaml as above except Xaxis cycle is 3.

bdring, 2022-08-16:
Closing this issue. We require a limit switch per axis at this time.

closed 2022-08-16 by bdring