bdring/FluidNC #566: Problem when enabled hard_limits

jemsengine, 2022-08-13:
### Controller Board

Makerbase MKS DLC32

### Help From Board Vendor

- [ ] Yes
- [X] No
- [ ] Not Applicable

### Machine Description

[Proxxon MF70](http://proxxonspain.es/micromot/fresado/micro-fresadora-mf-70.htm)

### Configuration file

```yaml
board: MKS-DLC32 V2.1
name: Proxxon MF70
meta: JEMS 220813

kinematics:
  Cartesian:

stepping:
  engine: I2S_STATIC
  idle_ms: 0
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0
axes:
  shared_stepper_disable_pin: I2SO.0
  x:
    steps_per_mm: 400.000
    max_rate_mm_per_min: 100.000
    acceleration_mm_per_sec2: 10.000
    max_travel_mm: 100.000
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 50.000
      seek_mm_per_min: 50.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.36
      hard_limits: false
      pulloff_mm: 2.000
      stepstick:
        step_pin: I2SO.1
        direction_pin: I2SO.2

  y:
    steps_per_mm: 400.000
    max_rate_mm_per_min: 100.000
    acceleration_mm_per_sec2: 10.000
    max_travel_mm: 40.000
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 50.000
      seek_mm_per_min: 50.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.35
      hard_limits: false
      pulloff_mm: 2.000
      stepstick:
        step_pin: I2SO.5
        direction_pin: I2SO.6:low

  z:
    steps_per_mm: 400.000
    max_rate_mm_per_min: 100.000
    acceleration_mm_per_sec2: 10.000
    max_travel_mm: 80.000
    soft_limits: false
    homing:
      cycle: 0
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 50.000
      seek_mm_per_min: 50.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.34
      hard_limits: false
      pulloff_mm: 1.000
      stepstick:
        step_pin: I2SO.3
        direction_pin: I2SO.4

i2so:
  bck_pin: gpio.16
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.12
  mosi_pin: gpio.13
  sck_pin: gpio.14

sdcard:
  cs_pin: gpio.15
  card_detect_pin: NO_PIN

control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  macro0_pin: gpio.33:low:pu
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

macros:
  startup_line0:
  startup_line1:
  macro0: $SD/Run=lasertest.gcode
  macro1: $SD/Run=home.gcode
  macro2:
  macro3:

coolant:
  flood_pin: NO_PIN
  mist_pin: NO_PIN
  delay_ms: 0

probe:
  pin: gpio.22
  check_mode_start: true

Laser:
  pwm_hz: 5000
  #L on Beeper / IN on TTL
  output_pin: gpio.32
  enable_pin: I2SO.7
  disable_with_s0: false
  s0_with_disable: false
  tool_num: 0
  speed_map: 0=0.000% 0=12.500% 1700=100.000%
# 135=0mA 270=5mA 400=10mA 700=16mA
user_outputs:
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 5000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000
  digital0_pin: NO_PIN
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: NO_PIN

start:
  must_home: false

# 5,18,19,22,23,25,26,27,32,33,39,I2SO.7
# SDA 0 / SCL 4
```

### Startup Messages

```text
$/axes/x/max_travel_mm
$/axes/x/homing/mpos_mm
$/axes/x/homing/positive_direction
$/axes/y/max_travel_mm
$/axes/y/homing/mpos_mm
$/axes/y/homing/positive_direction
$G
<Alarm|MPos:0.000,0.000,0.000|FS:0,0|Pn:PXYZ|WCO:0.000,0.000,0.000>
<Alarm|MPos:0.000,0.000,0.000|FS:0,0|Pn:PXYZ|Ov:100,100,100>
<Alarm|MPos:0.000,0.000,0.000|FS:0,0|Pn:PXYZ>
```

### User Interface Software

WebUI

### What happened?

When you try to set the field axes/x/motor0/hard_limits=true from the configuration file, an error appears that blocks the program.

If the field axes/s/motor0/hard_limits=false all the configuration works propertly.
If the field is set to true when the file is loaded an alarm appears and cannot be cleared.

### Other Information

If the axis is configured by command line with hard_limits=true, it does not move.
Although the axis does not move, when the limit switch is manually pushed, the correct limit switch reached message appears and the system requests a reset for safety.

MitchBradley, 2022-08-14:
The startup messages are not what you posted - to get the startup messages, send $ss or type CTRL-R in FluidTerm.  We need the startup messages to determine what is going wrong.

bdring, 2022-08-17:
I am having trouble following the problem. Can you get the problem to occur using FluidTerm? If so, paste the session text here.

jemsengine, 2022-08-21:
Sorry for my absence. During the week I do not have access to the machine to test.
### First test. Configuration file upload
When I use the `$ss` command from the command line using a configuration file that has a line with `hard_limits: true` nothing comes up and the system goes into alarm.

```
$ss
Not apear any info. 
```

If I execute the same file with `hard_limits: false` the `$ss` command responds correctly and this appears:
```
$ss
[MSG:INFO: FluidNC v3.5.1]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is SPIFFS]
[MSG:INFO: Configuration file:MKS_DLC32_21_MF70.yaml]
[MSG:INFO: Machine Proxxon MF70]
[MSG:INFO: Board MKS-DLC32 V2.1]
[MSG:INFO: I2SO BCK:gpio.16 WS:gpio.17 DATA:gpio.21]
[MSG:INFO: SPI SCK:gpio.14 MOSI:gpio.13 MISO:gpio.12]
[MSG:INFO: SD Card cs_pin:gpio.15 detect:NO_PIN]
[MSG:INFO: Stepping:I2S_static Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:0ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable I2SO.0]
[MSG:INFO: Axis X (0.000,100.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.1 Dir:I2SO.2 Disable:NO_PIN]
[MSG:INFO:     Neg Limit gpio.36]
[MSG:INFO: Axis Y (0.000,40.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.5 Dir:I2SO.6:low Disable:NO_PIN]
[MSG:INFO:     Neg Limit gpio.35]
[MSG:INFO: Axis Z (-80.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.3 Dir:I2SO.4 Disable:NO_PIN]
[MSG:INFO:     Neg Limit gpio.34]
[MSG:INFO: macro0_pin gpio.33:low:pu]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Laser Ena:I2SO.7 Out:gpio.32 Freq:5000Hz Period:8191]
[MSG:INFO: Using spindle Laser]
[MSG:INFO: Probe Pin: gpio.22]
[MSG:INFO: Connecting to STA SSID:MOVISTAR_CD89]
[MSG:INFO: Connecting.]
[MSG:INFO: Connected - IP is 192.168.8.2]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://ProxxonMF70.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
```

### Second test. Modify the `hard_limits` via command line
In this case the system does not fail but the x axis does not move. The Y and Z axes move propertly.
When pressed to move the axis there seems to be a slight clicking noise from the motor. I don't know if this detail can help you.
When I change the field to false (`hard_limits: false`) the X axis works again without problem.
```
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Pn:PXYZ|Ov:100,100,100>
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Pn:PXYZ>
$axes/x/motor0/hard_limits
$/axes/x/motor0/hard_limits=false
ok
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Pn:PXYZ>
$ss
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Pn:PXYZ|WCO:0.000,0.000,0.000>
[MSG:INFO: FluidNC v3.5.1]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is SPIFFS]
[MSG:INFO: Configuration file:MKS_DLC32_21_MF70.yaml]
[MSG:INFO: Machine Proxxon MF70]
[MSG:INFO: Board MKS-DLC32 V2.1]
[... 55 lines cut ...]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.5 Dir:I2SO.6:low Disable:NO_PIN]
[MSG:INFO:     Neg Limit gpio.35]
[MSG:INFO: Axis Z (-80.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.3 Dir:I2SO.4 Disable:NO_PIN]
[MSG:INFO:     Neg Limit gpio.34]
[MSG:INFO: macro0_pin gpio.33:low:pu]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Laser Ena:I2SO.7 Out:gpio.32 Freq:5000Hz Period:8191]
[MSG:INFO: Using spindle Laser]
[MSG:INFO: Probe Pin: gpio.22]
[MSG:INFO: Connecting to STA SSID:MOVISTAR_CD89]
[MSG:INFO: Connecting.]
[MSG:INFO: Connected - IP is 192.168.8.2]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://ProxxonMF70.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
ok
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Pn:PXYZ|WCO:0.000,0.000,0.000>
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Pn:PXYZ|Ov:100,100,100>
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Pn:PXYZ>
```

MitchBradley, 2022-08-21:
The problem is that your limit switches are reporting as active:

<Idle|MPos:0.000,0.000,0.000|FS:0,0|Pn:PXYZ|Ov:100,100,100>

Pn:PXYZ means that all of the switches - Probe , X Limit, Y Limit, Z Limit - are active.  Your machine description does not tell us anything about how your limit switches are wired, so we can only guess.

Your config file says, for example:

```
  x:
     ...
    motor0:
      limit_neg_pin: gpio.36

```

That would be correct for a limit switch connected in the Normally Closed (NC) configuration.  If your switches are Normally Open, you would need to say "gpio.36:low" instead of "gpio.36".

If adding :low to all the limit and probe pin specifications does not work, you will need to tell us all the details about your switches and how they are wired.

MitchBradley, 2022-08-21:
Please note that the link to the Proxxon machine is for the manual version of the machine.  It gives us no information about steppers, limit switches, and their wiring.

bdring, 2022-08-21:
We have a wiki page to help with switch problems.

http://wiki.fluidnc.com/en/support/help_with_switch_problems

jemsengine, 2022-08-22:
> Please note that the link to the Proxxon machine is for the manual version of the machine. It gives us no information about steppers, limit switches, and their wiring.

I have updated the machine information.
During the week I will update the configuration of the limit switches and I will answer you.

MitchBradley, 2022-09-26:
Closed as stale; no updates for a long time

closed 2022-10-09 by MitchBradley