bdring/FluidNC #567: Problem

mac7988, 2022-08-14:
### Controller Board

Custom board.

### Help From Board Vendor

- [ ] Yes
- [ ] No
- [X] Not Applicable

### Machine Description

Testing on bench. No machine.

### Configuration file

```yaml
board: benchtop
name: 6axsrig
stepping:
  engine: I2S_STREAM
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 3

axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 4.444
    max_rate_mm_per_min: 350000.000
    acceleration_mm_per_sec2: 30000.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 2
      allow_single_axis: true
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 10000.000
      seek_mm_per_min: 500.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.26
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 10.000
      standard_stepper:
        step_pin: I2SO.18
        direction_pin: I2SO.17:low
        disable_pin: I2SO.16

    motor1:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 10.000
      standard_stepper:
        step_pin: I2SO.6
        direction_pin: I2SO.5:high
        disable_pin: I2SO.4
  y:
    steps_per_mm: 4.444
    max_rate_mm_per_min: 350000.000
    acceleration_mm_per_sec2: 30000.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 2
      allow_single_axis: true
      positive_direction: false
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: gpio.15:pu
      hard_limits: true
      pulloff_mm: 10.000
      standard_stepper:
        step_pin: I2SO.22
        direction_pin: I2SO.21
        disable_pin: I2SO.20
  z:
    steps_per_mm: 4.444
    max_rate_mm_per_min: 350000.000
    acceleration_mm_per_sec2: 30000.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 0
      allow_single_axis: true
      positive_direction: false
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: I2SO.26
        direction_pin: I2SO.25
        disable_pin: I2SO.24      

  b:
    steps_per_mm: 4.444
    max_rate_mm_per_min: 400000.000
    acceleration_mm_per_sec2: 10000.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 2
      allow_single_axis: true
      positive_direction: false
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: I2SO.14
        direction_pin: I2SO.13
        disable_pin: I2SO.12
i2so:
  bck_pin: gpio.22
  data_pin: gpio.12
  ws_pin: gpio.21

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  card_detect_pin: NO_PIN
  cs_pin: gpio.5

control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: gpio.36
  cycle_start_pin: gpio.39
  macro0_pin: gpio.27
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

coolant:
  flood_pin: gpio.33 
  mist_pin: gpio.13
  delay_ms: 0

probe:
  pin: NO_PIN
  check_mode_start: false

macros:
  startup_line0:
  startup_line1:
  macro0: $SD/Run=TESTCODE.txt
  macro1:
  macro2:
  macro3:

user_outputs:
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 5000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000
  digital0_pin: I2SO.31
  digital1_pin: I2SO.29
  digital2_pin: I2SO.30
  digital3_pin: NO_PIN

start:
  must_home: false
```

### Startup Messages

```text
[MSG:INFO: FluidNC v3.4.4]
[MSG:INFO: Compiled with ESP32 SDK:v3.3.5-1-g85c43024c]
[MSG:INFO: Configuration file:config.yaml]
[MSG:WARN: max_rate_mm_per_min value 350000.000 constrained to range (0.001,100000.000)]
[MSG:WARN: max_rate_mm_per_min value 350000.000 constrained to range (0.001,100000.000)]
[MSG:WARN: max_rate_mm_per_min value 350000.000 constrained to range (0.001,100000.000)]
[MSG:WARN: max_rate_mm_per_min value 400000.000 constrained to range (0.001,100000.000)]
[MSG:INFO: Machine 6xsrig]
[MSG:INFO: Board benchtop]
[MSG:INFO: I2SO BCK:gpio.22 WS:gpio.21 DATA:gpio.12]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN]
[MSG:INFO: Stepping:I2S_stream Pulse:4us Dsbl Delay:3us Dir Delay:1us Idle Delay:255ms]
[MSG:INFO: User Digital Output:0 on Pin:I2SO.31]
[MSG:INFO: User Digital Output:1 on Pin:I2SO.29]
[MSG:INFO: User Digital Output:2 on Pin:I2SO.30]
[MSG:INFO: Axis count 5]
[MSG:INFO: Axis X (0.000,300.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.18 Dir:I2SO.17:low Disable:I2SO.16]
[MSG:INFO:     Neg Limit gpio.26]
[MSG:INFO:   Motor1]
[MSG:INFO:     standard_stepper Step:I2SO.6 Dir:I2SO.5 Disable:I2SO.4]
[MSG:INFO: Axis Y (150.000,450.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.22 Dir:I2SO.21 Disable:I2SO.20]
[MSG:INFO:     All Limit gpio.15:pu]
[MSG:INFO: Axis Z (150.000,450.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.26 Dir:I2SO.25 Disable:I2SO.24]
[MSG:INFO: Axis A (-1000.000,0.000)]
[MSG:INFO: Axis B (150.000,450.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.14 Dir:I2SO.13 Disable:I2SO.12]
[MSG:INFO: FeedHold gpio.36]
[MSG:INFO: CycleStart gpio.39]
[MSG:INFO: Macro 0 gpio.27]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Using spindle NoSpindle]
[MSG:INFO: Flood coolant gpio.33]
[MSG:INFO: Mist coolant gpio.13]
[MSG:INFO: Connecting to STA SSID:test]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connecting...]
[MSG:INFO: Connecting....]
[MSG:INFO: Connected - IP is 192.168.0.101]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
```

### User Interface Software

webui

### What happened?

I have a macro button assigned to a Gcode "testcode.txt". When I press it, It runs the program and at the end of the program it
will disable the X axis.  Below is the code that I run. The problem I am seeing is that there is a delay on when the axis are reenabled. the causes the axis to not return to the same position.  Please note, I am not homing after reenable.

g90g1F5000x360
g90g1F5000x0
g4 p.2
m64p1
g4 p.5
m65p1
m64p0
g4 p.5
m65p0
m64p2
g4 p.5
m65p2
m30
(MSG NICE)
$MD=X

bdring, 2022-08-14:
How far off is it?

Generally you can't expect accuracy if you disable the motors and do not rehome.

bdring, 2022-08-14:
BTW. You have some unusually high config values that are being constrained.

mac7988, 2022-08-14:
> 

ok, I was under the impression something is wrong on my end. It off by 20 to 30 degree. 
<

mac7988, 2022-08-14:
> BTW. You have some unusually high config values that are being constrained.

I am using linear movement as angular plus micro stepping. It's accurate but slow so I cranked up the value. I know about the max velocity limit. I will dial it in later 

thank you Bart!

bdring, 2022-08-16:
It is probably OK if you stay under the max step rate. You are below that.  We could raise the speed limit. 

BTW: I usually use radians for rotational axes.

Can this issue be closed. The title of the issue is fixed, right?

Start a new one if needed

mac7988, 2022-08-16:
Yes thank you

closed 2022-08-17 by bdring