bdring/FluidNC #568: Problem:

Parker3123, 2022-08-16:
### Controller Board

6-pack

### Help From Board Vendor

- [ ] Yes
- [ ] No
- [X] Not Applicable

### Machine Description

gantry router with dual y motors, xyyz setup dm542t stepper drivers and 6pack control board with RS485 module 

### Configuration file

```yaml
board: 6 Pack
name: 6 Pack External XYYZ Huany
meta: 2022-06-28 B. Dring for Forest Darling Forest Darling
stepping:
  engine: I2S_STREAM
  idle_ms: 250
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 100.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: I2SO.2
        direction_pin: I2SO.1
        disable_pin: I2SO.0

  y:
    steps_per_mm: 100.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 3
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: I2SO.5
        direction_pin: I2SO.4
        disable_pin: I2SO.7
    
    motor1:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: I2SO.10
        direction_pin: I2SO.9
        disable_pin: I2SO.8

  z:
    steps_per_mm: 100.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 800.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: I2SO.13
        direction_pin: I2SO.12
        disable_pin: I2SO.15
        
probe:
  pin: NO_PIN
  check_mode_start: true
  
        
i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  card_detect_pin: NO_PIN
  cs_pin: gpio.5

control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

start:
  must_home: false
  
coolant:
  flood_pin: gpio.15
  mist_pin: gpio.12
  delay_ms: 0
  
YL620:
  uart:
    txd_pin: gpio.26
    rxd_pin: gpio.16
    rts_pin: gpio.4
    baud: 9600
    mode: 8N1
  modbus_id: 1
  tool_num: 0
  speed_map: 0=0% 0=25% 6000=25% 24000=100%
  
  
Laser:
  pwm_hz: 5000
  output_pin: gpio.14
  enable_pin: gpio.13
  disable_with_s0: false
  s0_with_disable: true
  tool_num: 1
  speed_map: 0=0.000% 255=100.000%
```

### Startup Messages

```text
[MSG:INFO: FluidNC v3.3.0 (pr/easytarget/272-c8cdb9b-dirty)]
[MSG:INFO: Compiled with ESP32 SDK:v3.3.5-1-g85c43024c]
[MSG:INFO: Configuration file:config.yaml]
[MSG:INFO: Machine 6 Pack External XYYZ Huany]
[MSG:INFO: Board 6 Pack]
[MSG:INFO: I2SO BCK:gpio.22 WS:gpio.17 DATA:gpio.21]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 dectect:NO_PIN]
[MSG:INFO: Stepping:I2S_stream Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:250ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (0.000,300.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.2 Dir:I2SO.1 Disable:I2SO.0]
[MSG:INFO: Axis Y (-300.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.5 Dir:I2SO.4 Disable:I2SO.7]
[MSG:INFO:   Motor1]
[MSG:INFO:     standard_stepper Step:I2SO.10 Dir:I2SO.9 Disable:I2SO.8]
[MSG:INFO: Axis Z (-300.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.13 Dir:I2SO.12 Disable:I2SO.15]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: YL620 Spindle Uart Tx:gpio.26 Rx:gpio.16 RTS:gpio.4 Baud:9600]
[MSG:INFO: Laser Spindle Ena:gpio.13 Out:gpio.14 Freq:5000Hz Res:13bits Laser mode:On]
[MSG:INFO: Using spindle YL620]
[MSG:INFO: Flood coolant gpio.15]
[MSG:INFO: Mist coolant gpio.12]
[MSG:INFO: STA SSID is not set]
[MSG:INFO: AP SSID FluidNC IP 192.168.0.1 mask 255.255.255.0 channel 1]
[MSG:INFO: AP started]
[MSG:INFO: WiFi on]
[MSG:INFO: Captive Portal Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl 3.3 [FluidNC v3.3.0 (pr/easytarget/272-c8cdb9b-dirty) (wifi) '$' for help]
[MSG:INFO: VFD RS485 Unresponsive]
```

### User Interface Software

Terminal mostly and WebUI

### What happened?

cant get the spindle (YL620) to move at all says the RS485 is unresponsive. I have spent quite a bit of time digging around and feel like I have things right. The wiring is A-A,B-B, COM-G, the first of those being from the vdf to the rs485 module. I am just not sure if Im dumb or missing somthing the code seems correct for the yaml file. Here I am though desperate for any guidance!

MitchBradley, 2022-08-16:
You haven't mentioned what you have tried for settings on the VFD.

Parker3123, 2022-08-16:
I apologize the setting I have made for the VFD are 

P00.00 | 400.0 
P00.01 = 3 
P03.00 = 3 
P03.01  = 1 
P03.02 = 2
P03.08  = 100.0 
I found these in this link under the YL620 section

https://github.com/bdring/FluidNC/wiki/Fluid-Spindle-Setup#yl620-rs485

<br class="Apple-interchange-newline">

MitchBradley, 2022-08-16:
Please attach some photos.  Is the RS485 module in slot number 3?

Parker3123, 2022-08-16:
[image]!

[image](https://user-images.githubusercontent.com/111254083/184992586-0862abcf-2e11-4c8f-9360-ef4565925c83.png)

The RS458 is in socket number 3

bdring, 2022-08-16:
Try swapping A and B at the controller end.

Parker3123, 2022-08-16:
[MSG:INFO: FluidNC v3.3.0 (pr/easytarget/272-c8cdb9b-dirty)]
[MSG:INFO: Compiled with ESP32 SDK:v3.3.5-1-g85c43024c]
[MSG:INFO: Configuration file:config.yaml]
[MSG:INFO: Machine 6 Pack External XYYZ Huany]
[MSG:INFO: Board 6 Pack]
[MSG:INFO: I2SO BCK:gpio.22 WS:gpio.17 DATA:gpio.21]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 dectect:NO_PIN]
[MSG:INFO: Stepping:I2S_stream Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:250ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (0.000,300.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.2 Dir:I2SO.1 Disable:I2SO.0]
[MSG:INFO: Axis Y (-300.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.5 Dir:I2SO.4 Disable:I2SO.7]
[MSG:INFO:   Motor1]
[MSG:INFO:     standard_stepper Step:I2SO.10 Dir:I2SO.9 Disable:I2SO.8]
[MSG:INFO: Axis Z (-300.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.13 Dir:I2SO.12 Disable:I2SO.15]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: YL620 Spindle Uart Tx:gpio.26 Rx:gpio.16 RTS:gpio.4 Baud:9600]
[MSG:INFO: Laser Spindle Ena:gpio.13 Out:gpio.14 Freq:5000Hz Res:13bits Laser mode:On]
[MSG:INFO: Using spindle YL620]
[MSG:INFO: Flood coolant gpio.15]
[MSG:INFO: Mist coolant gpio.12]
[MSG:INFO: STA SSID is not set]
[MSG:INFO: AP SSID FluidNC IP 192.168.0.1 mask 255.255.255.0 channel 1]
[MSG:INFO: AP started]
[MSG:INFO: WiFi on]
[MSG:INFO: Captive Portal Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl 3.3 [FluidNC v3.3.0 (pr/easytarget/272-c8cdb9b-dirty) (wifi) '$' for help]
[MSG:INFO: VFD RS485 Unresponsive]

the [MSG:INFO: VFD RS485 Unresponsive] is still appearing when I swap the wires.

MitchBradley, 2022-08-16:
You appear to be running a customized firmware build.

Parker3123, 2022-08-16:
I should have mentioned that I believe you posted it related to a similar issue way back. I should also mention that I get the same results with 3.4.4 as well. Do you recommend a version to use?

MitchBradley, 2022-08-16:
3.4.4 is as good as any with respect to VFDs.  At this point, if I were having the problem, I would get out the oscilloscope.

I used to have a YL620 but I gave it to a friend who lives in another town, so I no longer have easy access to it.

Parker3123, 2022-08-16:
Okay I have one but really only use it for synth making. Any leads onto what types of things i should be looking for or just probe the output of the module? Thanks again for helping me through this

MitchBradley, 2022-08-16:
I would connect it to one of the A or B leads.  Check to see which one is at the highest voltage of the two and connect to that one.  Set it to trigger on a falling edge with a threshold value midway between the idle value of A and B.  Look for a sequence of pulses that happens just before the error message.  That will tell you if a transmission is going out.  If you have a second channel, look at the RTS line, which should change state just after the transmission to invert the direction from ESP>VFD to VFD>ESP.  Then, if things are working right, there should be some more pulses as the VFD sends back a response.

MitchBradley, 2022-08-16:
https://sbc-support.com/uploads/tx_srcfaq/RS%20485%20Oscilloscope%20Screenshots_100179.pdf

MitchBradley, 2022-08-16:
https://www.youtube.com/watch?v=yNtuVDKkOnk

Parker3123, 2022-08-17:
thank you will look at this and get back later with any updates!

bdring, 2022-08-17:
Is the main power on and not just the USB power?

Parker3123, 2022-08-25:
Its been a long week so finally trying to get this sorted. I have both USB and main power on. Have been messing with the scope but have to be honest when I say Im a novice at viewing the RS485. I appear to get pulses at the start but dont trust my skills enough to know where to go next. I have noticed through searching that the JP1 part of the YL620A changes the input voltage. Not totally sure what the 4 switches should be set to or if that would cause a problem. Forgive me for such a long winded response with so many problems. I really appreciate any feedback/time you guys have given me. Im not here to give up just have lots of learning curves with this aspect!

Parker3123, 2022-08-25:
my scope is a rigol DS1054z if that helps btw just learning how to set up what mitch advised and being humbled quickly

Parker3123, 2022-10-16:
Figured I'd updated. I bought a new VFD a huanyang. Followed the instructions and even bought a new RS485 module nothing worked. Fluidterm still says vfd unresponsive. I have gotten 0-10v module to work but was hoping for the RS485. I have tried different module sockets and no help. I thought maybe my main board was bad but all the other modules work fine. Not sure where to go from here but feeling pretty discouraged and like I have spent countless hours researching for nothing.

bdring, 2022-10-16:
I can take a picture of my Huanyang setup tomorrow. I think on my setup I did not use the ground wire. Also try swapping the A & B.

MitchBradley, 2022-10-16:
Have you managed to look at the RS485 signals with your scope?

Parker3123, 2022-10-16:
Trying Barts recommendations first, if no luck I will try and upload what I see on my scope

bdring, 2022-10-16:
[image]

Parker3123, 2022-10-16:
[image]

Parker3123, 2022-10-16:
I tried my pest to set up my scope the right way. It appears I am getting pulses, they look kind of noisy to me but Im not sure if thats due to my novice scope skills

MitchBradley, 2022-10-16:
Do you have the scope probe ground connected? It should be connected to G on the RS485 adapter module?

Is one of the channels inverted?  They should be going in opposite direction - one going up while the other goes down.

Parker3123, 2022-10-16:
[image]

Parker3123, 2022-10-16:
that is with ground connected I inverted channel 2 as it seems to have a lower voltage.

MitchBradley, 2022-10-16:
Set both channels to non-inverting, slow down the timebase so we can see more of the waveform horizontally, and zoom out on the camera so I can see all of the setup info on the screen.

MitchBradley, 2022-10-16:
And move the cable that is obscuring part of the screen

Parker3123, 2022-10-16:
https://user-images.githubusercontent.com/111254083/196062394-229099d0-b2b9-4976-bac6-cf92a6b1d216.mov

Parker3123, 2022-10-16:
[image]
[image]
[image]
[image]

MitchBradley, 2022-10-16:
That looks like a reasonable transmit waveform at 9600 baud.  The first three transmitted bytes are 0x01, 0x01, 0x03, which is correct for Huanyang initialization.  The first 0x01 is the address, the second means "read setting", and the 0x03 means there are 3 more bytes in the packet.

MitchBradley, 2022-10-16:
Zoom out (slower timebase) to see if the VFD responds sometime later, by sending its own response packet.  The initial transmit packet from FluidNC should be roughly twice the length of what is shown on the screen.

Parker3123, 2022-10-16:
[image]

Parker3123, 2022-10-16:
I am pretty bad at this thank you for your patience, learning lots though

MitchBradley, 2022-10-16:
So it would appear that the VFD is responding.  Can you probe the RX and TX pins on the adapter module with the other two channels?

MitchBradley, 2022-10-17:
That looks good, assuming that ch 3 is tx and ch 4 is rx.  I need to add some debug stuff to the driver to see what is going on there.  Unfortunately I have to leave now.  I will try to put something together later.

Parker3123, 2022-10-17:
No worries again thank you for the help that is how the channels are set up

bdring, 2022-10-17:
I just tested huanyang again. Here is some info from that. ($Message/Level=Debug)

```yaml
[MSG:INFO: FluidNC v3.6.3]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:6P_Huany.yaml]
[MSG:DBG: Running after-parse tasks]
[MSG:INFO: Axes: using defaults]
[MSG:DBG: Checking configuration]
[MSG:INFO: Machine 6 Pack Huany]
[MSG:INFO: Board 6 Pack]
[MSG:INFO: I2SO BCK:gpio.22 WS:gpio.17 DATA:gpio.21]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN]
[MSG:INFO: Stepping:I2S_stream Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:250ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (-1000.000,0.000)]
[MSG:INFO: Axis Y (-1000.000,0.000)]
[MSG:INFO: Axis Z (-1000.000,0.000)]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Huanyang Spindle Uart Tx:gpio.26 Rx:gpio.16 RTS:gpio.4 Baud:9600]
[MSG:INFO: Using spindle Huanyang]
[MSG:INFO: STA SSID is not set]
[MSG:INFO: AP SSID FluidNC IP 192.168.0.1 mask 255.255.255.0 channel 1]
[MSG:INFO: Huanyang PD005,PD011 Freq range (100,400) Hz (6000,24000) RPM]
[MSG:INFO: Huanyang PD144 Rated RPM @ 50Hz:3000]
[MSG:INFO: AP started]
[MSG:INFO: WiFi on]
[MSG:INFO: Captive Portal Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
[MSG:DBG: VFD setState:0 SpindleSpeed:0]
[MSG:DBG: RPM:0 mapped to device units:0]

Grbl 3.6 [FluidNC v3.6.3 (wifi) '$' for help]
[MSG:INFO: Huanyang PD143 Poles:2]
[MSG:INFO: Huanyang PD014 Accel:6.000]
[MSG:INFO: Huanyang PD015 Decel:6.000]
[MSG:DBG: vfd_cmd_task mode:1]
M3S12000
[MSG:DBG: VFD setState:1 SpindleSpeed:12000]
[MSG:DBG: RPM:12000 mapped to device units:19999]
[MSG:DBG: vfd_cmd_task mode:1]
[MSG:DBG: prep speed 19999 curr -1]
ok
M5
[MSG:DBG: VFD setState:0 SpindleSpeed:12000]
[MSG:DBG: RPM:12000 mapped to device units:19999]
ok
[MSG:DBG: vfd_cmd_task mode:1]
```

I verified all programmed values on the VFD are per this page.

http://wiki.fluidnc.com/en/config/config_spindles#huanyang-rs485

If you want to verify any other settings, I can.

bdring, 2022-10-17:
Can you post your config with the Huanyang?

bdring, 2022-10-17:
I tried changing PD001 and PD002 away from RS485 control to see if that created any problems.

It responded to all the reads properly with no time outs.

Strangely, I was not able to read PD001 or PD002.

Parker3123, 2022-10-17:
board: 6 Pack
name: 6 Pack External XYYZ 10V
meta: 2022-04-28 B. Dring for Leif Beaton
stepping:
  engine: I2S_STREAM
  idle_ms: 250
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 100.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.33
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: I2SO.2
        direction_pin: I2SO.1
        disable_pin: I2SO.0

  y:
    steps_per_mm: 100.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.32
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: I2SO.5
        direction_pin: I2SO.4
        disable_pin: I2SO.7
    
    motor1:
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.35
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: I2SO.10
        direction_pin: I2SO.9
        disable_pin: I2SO.8

  z:
    steps_per_mm: 100.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 800.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.34
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: I2SO.13
        direction_pin: I2SO.12
        disable_pin: I2SO.15
        

i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  card_detect_pin: NO_PIN
  cs_pin: gpio.5

control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN
  
probe:
  pin: gpio.2
  check_mode_start: true

start:
  must_home: false
  
coolant:
  flood_pin: NO_PIN
  mist_pin: NO_PIN
  delay_ms: 0
  
Laser:
  pwm_hz: 5000
  output_pin: gpio.26
  enable_pin: gpio.4
  disable_with_s0: false
  s0_with_disable: true
  tool_num: 0
  speed_map: 0=0.000% 255=100.000%

Huanyang:
  uart:
    txd_pin: gpio.14
    rxd_pin: gpio.15
    rts_pin: gpio.3
    baud: 9600
    mode: 8N1
  modbus_id: 1
  tool_num: 1
  speed_map: 0=0% 0=25% 6000=25% 24000=100%
  off_on_alarm: false

bdring, 2022-10-18:
I see in your picture that the controller is a 6 Pack with an RS485 module in socket #3.  Is this true? The pins you used are way off.

[image]

___

Try using the config I used. It used the same module socket. It only configures the Huanyang, so no other features  will work. This is only a test.

```yaml
board: 6 Pack
name: 6 Pack Huany
meta: 2022-10-17 B. Dring
stepping:
  engine: I2S_STREAM
  idle_ms: 250
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0

i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  card_detect_pin: NO_PIN
  cs_pin: gpio.5

start:
  must_home: false

# Module Socket #3
Huanyang:
  uart:
    txd_pin: gpio.26
    rxd_pin: gpio.16
    rts_pin: gpio.4
    baud: 9600
    mode: 8N1
  modbus_id: 1
  tool_num: 0
  speed_map: 0=0% 0=25% 6000=25% 24000=100%
  ```

bdring, 2022-10-18:
The pins were wrong for any socket.

Parker3123, 2022-10-18:
well that is embarrassing. I swear I had triple checked but yours worked immediately. To confirm if I change my config now for my laser in socket number four it should look like this right?

Laser:
  pwm_hz: 5000
  output_pin: gpio.14
  enable_pin: gpio.13
  disable_with_s0: false
  s0_with_disable: true
  tool_num: 1
  speed_map: 0=0.000% 255=100.000%

bdring, 2022-10-18:
That will work for the laser.

Where does that leave this issue? Can we close it? I am pretty sure the YL620 problems are not firmware or hardware related.

Parker3123, 2022-10-18:
Yes you may close it It is 100% user error on my part

closed 2022-10-18 by bdring