bdring/FluidNC #569: Problem: Laser stays on when it shouldn't

JHarding86, 2022-08-16:
### Controller Board

4 Axis TMC2209 CNC Controller

https://www.tindie.com/products/33366583/4-axis-tmc2209-cnc-controller/

### Help From Board Vendor

- [ ] Yes
- [ ] No
- [X] Not Applicable

### Machine Description

It is a custom Co2 laser.

Single stepper x axis - [StepperOnline](https://www.amazon.com/gp/product/B00PNEQKC0/ref=ppx_yo_dt_b_search_asin_image?ie=UTF8&psc=1)
Single stepper y axis (dual ends - rod connects each end) -> [Amazon Link](https://www.amazon.com/gp/product/B071S7RMSX/ref=ppx_yo_dt_b_search_asin_image?ie=UTF8&psc=1)
Single stepper z axis - [StepperOnline](https://www.amazon.com/gp/product/B00PNEQKC0/ref=ppx_yo_dt_b_search_asin_image?ie=UTF8&psc=1)

GPIO used to control external High Voltage relay for Exhaust Fan, Air Assist, and Water Pump. This is the "coolant" gpio in the config file.

### Configuration file

```yaml
name: "TMC2209 XYZ  Laser"
board: "FluidNC 4X 2209"
meta: 2022-08-09 B. Dring for James Harding 340638

stepping:
  engine: I2S_STATIC
  idle_ms: 255
  dir_delay_us: 1
  pulse_us: 4
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN
  
  x:
    steps_per_mm: 53.33
    max_rate_mm_per_min: 18000
    acceleration_mm_per_sec2: 600
    max_travel_mm: 750
    homing:
      cycle: 1
      mpos_mm: 0
      positive_direction: false
      seek_mm_per_min: 3000
      feed_mm_per_min: 240
    
    motor0:
      limit_neg_pin: gpio.36:high:pu
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      pulloff_mm: 5
      tmc_2209:
        uart:
          txd_pin: gpio.16
          rxd_pin: gpio.4
          rts_pin: NO_PIN
          cts_pin: NO_PIN
          baud: 115200
          mode: 8N1

        addr: 0
        r_sense_ohms: 0.110
        run_amps: 2.0
        hold_amps: 0.500
        microsteps: 16
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: CoolStep
        homing_mode: CoolStep
        use_enable: false
        direction_pin: i2so.1
        step_pin: i2so.2
        disable_pin: i2so.0

  y:
    steps_per_mm: 53.33
    max_rate_mm_per_min: 18000
    acceleration_mm_per_sec2: 600
    max_travel_mm: 410
    homing:
      cycle: 1
      mpos_mm: 0
      positive_direction: false
      seek_mm_per_min: 3000
      feed_mm_per_min: 240

    motor0:
      limit_neg_pin: gpio.39:high:pu
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      pulloff_mm: 5
      tmc_2209:
        addr: 1
        r_sense_ohms: 0.110
        run_amps: 1.800
        hold_amps: 0.500
        microsteps: 16
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: StealthChop
        homing_mode: StealthChop
        use_enable: false
        direction_pin: i2so.4
        step_pin: i2so.5
        disable_pin: i2so.7

  z:
    steps_per_mm: 400
    max_rate_mm_per_min: 200
    acceleration_mm_per_sec2: 25
    max_travel_mm: 200
    homing:
      cycle: 2
      mpos_mm: 0
      positive_direction: false
      seek_mm_per_min: 200
      feed_mm_per_min: 100
    
    motor0:
      limit_neg_pin: gpio.34:high:pu
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      pulloff_mm: 6
      tmc_2209:
        addr: 2
        r_sense_ohms: 0.110
        run_amps: 2.000
        hold_amps: 0.500
        microsteps: 16
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: StealthChop
        homing_mode: StealthChop
        use_enable: false
        direction_pin: i2so.9
        step_pin: i2so.10
        disable_pin: i2so.8

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18
  
i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17

sdcard:
  cs_pin: gpio.5
  card_detect_pin: NO_PIN

coolant:
  flood_pin: gpio.25
  delay_ms: 1000

Laser:
  pwm_hz: 5000
  output_pin: gpio.26
  enable_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  tool_num: 0
  speed_map: 0=5% 255=100.000%
```

### Startup Messages

```text
[MSG:INFO: FluidNC v3.4.4]
[MSG:INFO: Compiled with ESP32 SDK:v3.3.5-1-g85c43024c]
[MSG:INFO: Configuration file:4x_2209_laser.yaml]
[MSG:WARN: gpio.36 does not support :pu attribute]
[MSG:WARN: gpio.39 does not support :pu attribute]
[MSG:WARN: gpio.34 does not support :pu attribute]
[MSG:INFO: Machine TMC2209 XYZ  Laser]
[MSG:INFO: Board FluidNC 4X 2209]
[MSG:INFO: I2SO BCK:gpio.22 WS:gpio.17 DATA:gpio.21]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN]
[MSG:INFO: Stepping:I2S_static Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (0.000,750.000)]
[MSG:INFO:   Motor0]
[MSG:INFO: Trinamic Stepper Uart Tx:gpio.16 Rx:gpio.4 RTS:NO_PIN Baud:115200]
[MSG:INFO:     tmc_2209 Step:I2SO.2 Dir:I2SO.1 Disable:I2SO.0 Addr:0 R:0.110]
[MSG:INFO:     Neg Limit gpio.36]
[MSG:INFO: Axis Y (0.000,410.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2209 Step:I2SO.5 Dir:I2SO.4 Disable:I2SO.7 Addr:1 R:0.110]
[MSG:INFO:     Neg Limit gpio.39]
[MSG:INFO: Axis Z (0.000,200.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2209 Step:I2SO.10 Dir:I2SO.9 Disable:I2SO.8 Addr:2 R:0.110]
[MSG:INFO:     Neg Limit gpio.34]
[MSG:INFO: X Axis driver test passed]
[MSG:INFO: Y Axis driver test passed]
[MSG:INFO: Z Axis driver test passed]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Laser Spindle Ena:NO_PIN Out:gpio.26 Freq:5000Hz Res:13bits Laser mode:On]
[MSG:INFO: Using spindle Laser]
[MSG:INFO: Flood coolant gpio.25]
[MSG:INFO: WiFi is disabled]
Grbl 3.4 [FluidNC v3.4.4 (wifi) '$' for help]
[MSG:INFO: '$H'|'$X' to unlock]
```

### User Interface Software

LightBurn

### What happened?

It appears as though sometimes the laser does not turn off sometimes when it should. 

[image]

MitchBradley, 2022-08-16:
What is the circuitry between the controller and the laser?  Is it possible that the ESP32 is trying to say "laser off" but the signal is not getting through?

MitchBradley, 2022-08-16:
Are you sending directly from Lightburn or running from an SD card?

JHarding86, 2022-08-16:
I burned at least 20 different items on 3.5.1 and didn't have this issue. I do not think the message is getting lost, there is always a chance of that or interference. 

I am using LightBurn to send the Goode. SD card is not installed, it is in the config file just in case I ever needed it (sounds like I might need it).

MitchBradley, 2022-08-16:
Remind me what problem you were having with 3.5.1.  Does that problem go away with wifi off?

JHarding86, 2022-08-16:
It started out with I2S_STREAM vs I2S_STATIC. Progressed to acceleration and speed settings. 
[image]

I thought I had got it figured out with one of my designs, so I started lasering different designs and the same problems started rearing their heads. It got to the point that my acceleration and speed values were so low that I determined they couldn't possibly be the issue. 3.5.1 was generally acting strange but I kept at it. Then I started seeing the reports come through on discord and ultimately the tag on the most recent release to use 3.4.4.

I tried 3.4.4 and most everything seems to be great except I have to have $Wifi/Mode=Off to get further than a few Gcode commands into a burn. Solved that to find out that now the laser isn't turning off sometimes, seemingly randomly as I've done a few of the exact same gcode files and this problem crops up in different locations.

JHarding86, 2022-08-16:
Interestingly I did just change what LightBurn calls [S Value Max](https://docs.lightburnsoftware.com/UI/DeviceSettings.html). It was set to 1000, I changed it to 255 because the FluidNC config file has a "speed map" which I do believe is laser power in FluidNC laser mode. 

This feels like it might be the issue... I guess I should change it back to 1000 and see if this issue continues to happen.

bdring, 2022-08-16:
Traditionally, 255 was the number used by Grbl, but the ESP32 PWM resolution can go much higher.

I have used 1000 of LaserGrbl with no issue.

JHarding86, 2022-08-16:
How exactly does the speed map work then? Can I map values up to 1000?

bdring, 2022-08-16:
It maps a value to a percentage. The percentage is the full scale of the spindle. In you case ...

`[MSG:INFO: Laser Spindle Ena:NO_PIN Out:gpio.26 Freq:5000Hz Res:13bits Laser mode:On]`

...the full scale is 13bits, but it can map any value to that.

[MSG:INFO: Laser Spindle Ena:NO_PIN Out:gpio.26 Freq:5000Hz Res:13bits Laser mode:On]

bdring, 2022-08-16:
http://wiki.fluidnc.com/en/config/spindle_speed_maps

MitchBradley, 2022-08-16:
https://github.com/bdring/FluidNC/wiki/FluidNC-Spindle-Speed-Mapping

bdring, 2022-08-16:
Mapping anything less than 8,191 is wasting resolution at 13bits, but in reality can the laser really engrave differently at fractions of a percent?

Extra digits in the streaming gcode is probably more of a waste.

JHarding86, 2022-08-17:
So I set my speed map to:
0=0% 1000=100%

When I fire my laser at 7% (svalue = 70) it turns the laser on. When I fire my laser at 6% (svalue 60) it doesnt turn on.

What should my new speed map be?

bdring, 2022-08-17:
I'll assume the PWM is correct, but your laser has a threshold before it turns on. A DC spindle can act the same way. It takes a little power to get it spinning.

You will have to study [this wiki page](http://wiki.fluidnc.com/en/config/spindle_speed_maps) and adjust, but this might work for you.
```yaml
 speed_map: 0=0% 0=7% 1000=100%
 ```
 
 basically S values of 0 to 1000 will be linearized to 7% to 100%
 
 
[image]

bdring, 2022-08-17:
Do you still see a little bit of light from the LED on the controller at 6%?

JHarding86, 2022-08-17:
Yes, at 6% the led on the output module does still come on. The laser does not leave a mark on wood though.

bdring, 2022-08-17:
That is the whole intent. 0%-100% does not always match what you get.

JHarding86, 2022-08-17:
Right, that's the point of the speed map correct? It's like a calibration for your particular setup.

JHarding86, 2022-08-17:
Working now. Thank you.

closed 2022-08-17 by JHarding86